chan-sccp/chan-sccp #435: Unable to build against Asterisk 16.0.0

CRCinAU, 2018-11-19:
When trying to build against asterisk 16.0.0, I get the following errors:

```
In file included from ast114.c:37:
ast114.c:3665:175: error: ‘struct ast_module_info’ has no member named ‘nonoptreq’
AST_MODULE_INFO(ASTERISK_GPL_KEY, AST_MODFLAG_LOAD_ORDER, SCCP_VERSIONSTR,.load = load_module,.unload = unload_module,.reload = module_reload,.load_pri = AST_MODPRI_DEFAULT,.nonoptreq = "chan_local");
^~~~~~~~~
```

The compile summary is as follows:
```
Operating System : Linux
Kernel : 4.19.2-300.fc29.x86_64
Architecture : x86_64
Asterisk Version : 11400 / 114
Asterisk Repos : TRUNK
Build user : root
Version : 4.3.1_develop_r69fdf958
Compiler Version : 8
CC : gcc
CPP : gcc -E
HOST_CC : gcc
CFLAGS : -Og -Wall -I. -D_LARGEFILE_SOURCE -D_FILE_OFFSET_BITS=64 -std=gnu11 -fgnu89-inline  -isystem /usr/include -DHAVE_ASTERISK -g3 -ggdb3 -pthread
CPPFLAGS : -U_FORTIFY_SOURCE  -isystem /usr/include -DHAVE_ASTERISK  -g3 -ggdb3
LDFLAGS :      -L/usr/lib64
PBX_TYPE : Asterisk
PBX_PREFIX : /usr
PBX_ETC : /etc/asterisk
PBX_LIB : /usr/lib64
PBX_VARLIB : /usr/share/asterisk
PBX_INCLUDE : /usr/include/asterisk
Module Directory : /usr/lib64/asterisk/modules
```

CRCinAU, 2018-11-19:
This basic patch fixes the issue - however I figure this will need to become some kind of conditional to support multiple asterisk versions:

```
diff --git a/src/pbx_impl/ast114/ast114.c b/src/pbx_impl/ast114/ast114.c
index 82cb4629..b9199dab 100644
--- a/src/pbx_impl/ast114/ast114.c
+++ b/src/pbx_impl/ast114/ast114.c
@@ -3662,7 +3662,7 @@ static const __attribute__ ((unused))
struct ast_module_info *ast_module_info = &__mod_info;
#else
-AST_MODULE_INFO(ASTERISK_GPL_KEY, AST_MODFLAG_LOAD_ORDER, SCCP_VERSIONSTR,.load = load_module,.unload = unload_module,.reload = module_reload,.load_pri = AST_MODPRI_DEFAULT,.nonoptreq = "chan_local");
+AST_MODULE_INFO(ASTERISK_GPL_KEY, AST_MODFLAG_LOAD_ORDER, SCCP_VERSIONSTR,.load = load_module,.unload = unload_module,.reload = module_reload,.load_pri = AST_MODPRI_DEFAULT);
#endif
PBX_CHANNEL_TYPE *sccp_wrapper_asterisk114_findPickupChannelByExtenLocked(PBX_CHANNEL_TYPE * chan, const char *exten, const char *context)
```

dkgroot, 2018-11-26:
@CRCinAU 
Thanks for reporting the issue. The main issue was that ast-16 was not yet detected by the configure script and did not have it's own ast116 files. This has been fixed now.

commit 597b2181 (2018-11-26): Add configure/compile support for asterisk-16

(only partially tested)

issue reported by Steven Haigh
fixes #435
closed 2018-11-26 by dkgroot