chan-sccp/chan-sccp #436: Drop support for old (<13) asterisk versions

mpaland, 2018-11-23:
The [version table](https://wiki.asterisk.org/wiki/display/AST/Asterisk+Versions) shows the actual supported asterisk versions.

IMHO the code can be simplified by dropping support for anything below asterisk version 13.

It is not adviseable at all to use any asterisk version smaller than 13 by now, cause they won't get any security fixes anymore.

dkgroot, 2018-12-31:
Closing this 'issue' until further notice !!

dkgroot, 2019-02-17:
@mpaland Just sent a notice to the mailinglist that older asterisk version support will be deprecated and removed beginning next month. Knowing users they will only start reposponding after the fact though :-)

dkgroot, 2019-02-18:
Deprecation of Asterisk-15 was reverted (because it's still actively used by FreePBX), it should be deprecated by 2019-10-03 which is it's EOL date.

mpaland, 2019-02-18:
What about 11?
It should be deprecated as well cause it's not getting any security fixes for over 2 years now - and there had been many security related issues...

dkgroot, 2019-02-18:
Hi Marco,

I do fully agree with you. Some people might need it as a stepping stone from 1.6.2, 1.8 or 10 for example. Also because it was an LTS version, there might be more people stranded on one of these LTS versions. Chan-sccp should not be help responsible for the choices people make. We could/should plan a deprecation for 11 as well though, maybe it should coincide with the next round in october ? asterisk-11 is not holding back development of chan-sccp as much as the older version, so i am ok with it from that perspective. If there are 0 complaints in the coming days we could even check if deprecation of 11 would form a problem for anyone.

Regards,
Diederik

beta2k1, 2019-02-20:
Biggest group of stranded users I think will be those on Elastix, they topped out at 11.21 before switching the platform to 3CX.  Not sure if there's an easier way for them than taking a FreePBX backup, rebuild and restore or not.  

Also not sure how many of us there are running FreePBX derivative PBXs :)

Either way, don't count me as an objection, I've already been wanting to rebuild the ones I watch over, way too many other aspects of the systems are way too far out of date for my liking.

driods, 2019-02-20:
I agree with beta2k1. We have been using Elastix  - asterisk 11.2x for over four years. Rock stable. 
I am sure there are many more users out there who would be reluctant to migrate/upgrade.

dkgroot, 2019-02-20:
@driods  @beta2k1 Appreciate your input guys. Not saying 11 will be dropped right away.

I am guessing/assuming that when they don't update their asterisk-11 instances, chances are high that they are not touching their chan-sccp revisions/versions either (right). So us deprecating support for ast11, would not hinder them at all, in their continued use.

I do worry for them (a little) though, as their linux versions, ssh and ssl implementation are also a long way behind (with known attack vectors). Brrrrr.....
But i guess that's not my problem :-)

driods, 2019-02-20:
You are right D, however after rethinking , we have decided it's wise to move forward to newer versions of asterisk. 
Will keep you updated ? Thanks for all your support. 

ho

dkgroot, 2019-02-20:
@driods I would say, something good came out of the the deprecation threat :-) Good luck with the upgrade. Please use an intermediate test/staging machine, so you can switch over slowly + after thorough testing/checking.

dkgroot, 2020-01-17:
Closing this issue report for now.
Note the only two version we are going to be carrying forward are asterisk-13, asterisk-16 and asterisk-17. We will currently also maintain asterisk-14/15 if they don't hinder progress, even though they are deprecated.

closed 2020-01-17 by dkgroot