chan-sccp/chan-sccp #437: Record feature works in one direction only

sdstorts, 2018-12-03:
When placing or receiving calls on a 7945/7965 Cisco phone with sccpfeature (monitor) enabled on the second line, the recording only works for the channel the call was placed from.

Reproduction Steps:

    [Place call from phone with Record(monitor) sccpfeature as second button on phone, then press the record button.]
    [Place call to phone with Record(monitor) sccpfeature as second button on phone, then press the record button.]
   

Expected behavior:

[Two files are created for audio in each direction.  Both files have the recorded audio from the respective channel when the recording button is pressed a second time or when the call is completed]

Observed behavior:

[Two files are created, however, only the file recording the Caller is storing audio.  The phone being called is not storing audio in the recorded file.  This happens in calls established in either direction
and only records the audio from the phone that made the phone call.  This is also happening on multiple servers that are setup with a similar config.  ]

chan-sccp version: Release: 4.3.1 (HEAD - dfe82abM (built by 'root' on '2018-12-03 17:47:11 UTC')

asterisk version: GIT-13-13.15.0-rc1-1500-gbb33daf

Addition / Optional Information:

    Problem started happening recently, didn't happen in an older version of Chan-SCCP: [No] I noticed this after upgrading to Asterisk 13 from Asterisk 11.18 running sccp version Release: 4.3.1 issues/420_1 - de1fa29M (built by 'root' on '2018-08-20 19:29:11 UTC')
    Problem can be reliably reproduced, doesn't happen randomly: [Yes]
    Platform:
        [CPU - 4 core - Intel Xeon E5-2670 V2(esxi host)]
        [64 Bit]
        [Big/Small Endian]
        [Ubuntu]
        [Server 16.04 LTS	]
    Direct Contact information: [You can contact me here]

sdstorts, 2018-12-31:
Just an update on this: I tried the update c95a6fc1a868f932f5ceffd3baa78b3b68314b73 and can confirm that this issue is not happening anymore.  Thanks so much for your help on this!

dkgroot, 2018-12-31:
@sdstorts thank you very much for reporting this back. Seems we are slowly moving in the right direction !

closed 2018-12-31 by dkgroot