chan-sccp/chan-sccp #438: Some outgoing calls are quiet.

rolakron, 2018-12-23:
Some outgoing calls are quiet.

If I call i.e. time info *60, the call is quiet, phone however shows that the call is answered. If I make a outgoing call to an external number (i.e. mobilephone) I don't hear any call progress tones (ringing tone) but when the call is answered I can talk normally.

Version: 4.3.1
Asterisk verson: Asterisk 13.23.0, running on Raspberry as RasPBX with FreePBX

rolakron, 2018-12-24:
My settings in sccp.conf are:
disallow=all
allow=alaw
allow=ulaw
allow=g729

earlyrtp=progress

dkgroot, 2018-12-24:
Ok that sounds ok, do the codecs match up to your other channel driver (same order) ?
What happens if you set earlyrtp to dialing or ringout (is earlier) ? Does that change the behaviour in a way you expect. Did you try the latest develop git revision ?

dkgroot, 2018-12-24:
You can set "sccp debug core, channel, indicate, rtp, pbx" and just see if you notice anything in the debug stream that might show the issue with codecs not matching for example.
You can also add "rtp set debug on" to see when rtp is starting to flow on your system. This could help to show if it is an earlyrtp issue or a codec matching issue.

rolakron, 2018-12-24:
I enabled debugging and this is what I got when i dialed the number to listen to a radio channel on the phone. The radio channels worked before but are quiet now.
" SEP0023331C1276: (sccp_pbx_softswitch) channel SCCP/1474-00000009 is dialing number 2002
    -- SEP0023331C1276: Set asterisk state Ring (4) for call 9
    -- SEP0023331C1276: Asterisk prodded channel SCCP/1474-00000009.
    -- SCCP/1474-00000009: (pbx_pbx_start) autoloop has started, set requestHangup = requestQueueHangup
 SEP0023331C1276: (sccp_pbx_softswitch) pbx started
    -- SCCP: sccp_pbx_answer
    -- SEP0023331C1276: sccp_pbx_answer checking parent channel
    -- SEP0023331C1276: (sccp_pbx_answer) Outgoing call SCCP/1474-00000009 has been answered by remote party
 SEP0023331C1276: (sccp_pbx_softswitch) finished
    -- SEP0023331C1276: Starting Phone RTP/UDP Transmission (State: DIALING[14])
 SCCP: (sccp_pbx_softswitch) exit
    -- SCCP/1474-00000009: Using dtmf rtp_code : 101
    -- SEP0023331C1276: Tell Phone to send RTP/UDP media from  to 10.20.30.13:18598 (NAT: (Auto)Off)
    -- SEP0023331C1276: Using codec: No codec(0), TOS 184, Silence Suppression: OFF for call with PassThruId: 4294967286 and CallID: 9
    -- SEP0023331C1276: Indicate SCCP new state:PROCEED, current channel state:DIALING on call:SCCP/1474-00000009, lineInstance:1 (previous channelstate:DIGITSFOLL)
    -- SEP0023331C1276: Finish to indicate channel state:PROCEED on call:SCCP/1474-00000009, lineInstance:1. New channel state:PROCEED
    -- SEP0023331C1276: Indicate SCCP new state:CONNECTED, current channel state:PROCEED on call:SCCP/1474-00000009, lineInstance:1 (previous channelstate:DIALING)
    -- SEP0023331C1276 - SCCP/1474-00000009: (recalculateCodecformat) 
	capabilities: [slin16, ulaw/64k, alaw/64k, ilbc, g729b, g729ab, g729, g729a]
	preferences: [alaw/64k, ulaw/64k, g729, g729a, g729b, g729ab]
	remote caps: []
	Format:
    -- SCCP/1474-00000009, OpenReceiveChannel with format No codec, payload 0, echocancel: YES, passthrupartyid: 4294967286, pbx_channel_name: SCCP/1474-00000009
    -- SEP0023331C1276: Finish to indicate channel state:CONNECTED on call:SCCP/1474-00000009, lineInstance:1. New channel state:CONNECTED
    -- SCCP: (pbx_indicate) start indicate 'AST_CONTROL_PROD: Kick remote channel'
    -- SEP0023331C1276: (pbx_indicate) start indicate 'AST_CONTROL_PROD: Kick remote channel' (-1) condition on channel SCCP/1474-00000009 (readStat:1, writeState:1, rtp:yes)
  == SEP0023331C1276: (pbx_indicate) finish: send indication (res:-1)
"

dkgroot, 2018-12-24:
The interesting thing about the first example is "Using codec: No codec(0)" during Proceed. And "Using codec: No codec(0)" when the call is being CONNECTED. This is very wrong and we certainly need to fix that. I will have to do some tests to see how to reproduce this behaviour over here (Little hard in my lab setup).

The github pull is recent you say, does that mean today, or a couple of days ago ? I did merge a big change today (just to give me heads up if this is potentially a recent regression on my part). Can you tell me the revision number you are on at the moment.

rolakron, 2018-12-24:
I did also notice the "No codec"-thing. I didn't test to call the *60 time info, but I assume that it will cause the same issue as well. You can also set up one radio stream reachable via an extension number for testing purposes.

The last one (that is installed now) gives me 95868a58, but I cannot see the older ones (that did work) at all, as I did delete the directory after I compiled the code.
But I have an another system also where I compiled the code for about one week ago, I checked that and it gave me a7bca36e. That one is working correctly.

dkgroot, 2018-12-24:
I don't live in the US (guess), so *60 doesn't mean anything to me :-) I am guessing you mean the time and weather services you can reach over the phone (right), my development/lab server is not connected to pstn/sip connection though. I will figure something out though.

Ok that means that 9586a5 does entail a recent regression that i will have to figure out. I will look into it and see what i can do later today.

rolakron, 2018-12-24:
Ah, sorry for the confusion. :) No, I don't live in US either, the *60 time info is a part of the system. But I'm unsure if it is a part of Asterisk or FreePBX, it's hovever not on the PSTN. I have not configured that part. I can provide you the dial plan for the radio stations?

However, with earlyrtp = ringout, then the radio stations do work (but no ringing tone on outgoing external calls, as earlier.)
"SEP0023339D632D: (sccp_pbx_softswitch) channel SCCP/35222-00000008 is dialing number 2002
    -- SEP0023339D632D: Set asterisk state Ring (4) for call 8
    -- SEP0023339D632D: Asterisk prodded channel SCCP/35222-00000008.
    -- SCCP/35222-00000008: (pbx_pbx_start) autoloop has started, set requestHangup = requestQueueHangup
 SEP0023339D632D: (sccp_pbx_softswitch) pbx started
    -- SCCP: sccp_pbx_answer
    -- SEP0023339D632D: sccp_pbx_answer checking parent channel
    -- SEP0023339D632D: (sccp_pbx_answer) Outgoing call SCCP/35222-00000008 has been answered by remote party
    -- SEP0023339D632D: Starting Phone RTP/UDP Transmission (State: DIALING[14])
    -- SCCP/35222-00000008: Using dtmf rtp_code : 101
    -- SEP0023339D632D: Tell Phone to send RTP/UDP media from  to 10.20.30.13:11926 (NAT: (Auto)Off)
    -- SEP0023339D632D: Using codec: G.711 A-law 64k(2), TOS 184, Silence Suppression: OFF for call with PassThruId: 4294967287 and CallID: 8
    -- SEP0023339D632D: Indicate SCCP new state:PROCEED, current channel state:DIALING on call:SCCP/35222-00000008, lineInstance:1 (previous channelstate:OFFHOOK)
    -- SEP0023339D632D: Finish to indicate channel state:PROCEED on call:SCCP/35222-00000008, lineInstance:1. New channel state:PROCEED
    -- SEP0023339D632D: Indicate SCCP new state:CONNECTED, current channel state:PROCEED on call:SCCP/35222-00000008, lineInstance:1 (previous channelstate:DIALING)
    -- SEP0023339D632D: Did not reopen an RTP stream as old SCCP state was (PROCEED)
    -- SEP0023339D632D: Finish to indicate channel state:CONNECTED on call:SCCP/35222-00000008, lineInstance:1. New channel state:CONNECTED
 SEP0023339D632D: (sccp_pbx_softswitch) finished
 SCCP: (sccp_pbx_softswitch) exit
    -- SCCP: (pbx_indicate) start indicate 'AST_CONTROL_PROD: Kick remote channel'
    -- SEP0023339D632D: (pbx_indicate) start indicate 'AST_CONTROL_PROD: Kick remote channel' (-1) condition on channel SCCP/35222-00000008 (readStat:1, writeState:1, rtp:yes)
  == SEP0023339D632D: (pbx_indicate) finish: send indication (res:-1)
    -- SEP0023339D632D: Got OpenChannel ACK. Status:'Media Status: OK' (0), Remote RTP/UDP:'10.20.30.52:31598', Type:Indirect RTP, PassThruPartyId:4294967287, CallID:8
    -- SEP0023339D632D: Getting the active channel on device.
    -- SCCP: (asterisk113_setPhoneRTPAddress) Update PBX to send RTP/UDP media to '10.20.30.52:31598' (new remote) (NAT: no)
    -- SEP0023339D632D: Tell PBX   to send RTP/UDP media from 10.20.30.13:11926 to 10.20.30.52:31598 (NoNat)
    -- SEP0023339D632D: Opened Receive Channel (State: CONNECTED[30])
    -- SEP0023339D632D: Set the active channel 8 on device
    -- SEP0023339D632D: send callInfo on SCCP/35222-00000008 with lineInstance: 1
    -- SEP0023339D632D: Got startMediaTransmission ACK. Status:'Media Status: OK' (0), Remote RTP/UDP:'10.20.30.52:31598', Type:Indirect RTP, PassThruPartyId:4294967287, CallID:8, CallID1:8
    -- SEP0023339D632D: Getting the active channel on device.
    -- SEP0023339D632D: Media Transmission Started (State: CONNECTED[30])
    -- SEP0023339D632D: Set asterisk state Up (6) for call 8
"
And here is the dial plan for a web radio stream:

[from-internal-xfer-custom]
include => from-internal-custom
include => from-internal-additional ; auto-generated
exten => h,1,Macro(hangupcall)

exten => 2002,1,Answer
exten => 2002,n,Set(CALLERID(name)="DR P8 Jazz")
exten => 2002,n,MusicOnHold(streaming1)
exten => 2002,n,Hangup

Then in FreePBX, go to Music on hold, add a new category. Give the category name "streaming1" (or whaterver as long as it is the same as in "MusicOnHold(streaming1)"). Type must be "Custom application". Press submit. Then edit the newly created category and add this into the Application-field:  "/usr/bin/mpg123 -q -s --mono -r 8000 -f 8192 -b 0 http://live-icy.gss.dr.dk/A/A22L.mp3" (without quotes)

dkgroot, 2018-12-24:
Thanks for the example. Aah you are just listening to MusicOnHold, that's easy to reproduce. Funny that you are answering the call, so it's not like the "This call costs xxx per minute" message you would get from a pstn/sip connection during Progress/Proceed.

What does a call to an outside line look like in debug ? Maybe i will spot the difference. I may have to roll back issue #420 (again) which i was not looking forward to. I would rather fix the issue.

dkgroot, 2018-12-24:
Ok so earlyrtp=progress works just fine when using it during normal calls. But only reaching the "radio" station gives a problem... right?

Just as a little test. Can you change your dialplan example to:
```
exten => 2002,1,Wait(2)
exten => 2002,n,Progress()
exten => 2002,n,Wait(2)
exten => 2002,n,Answer()
exten => 2002,n,Set(CALLERID(name)="DR P8 Jazz")
exten => 2002,n,MusicOnHold(streaming1)
exten => 2002,n,Hangup()
```
Just curious if that would change matters and would work with "earlyrtp=progress". If it does it will give me a headstart fixing this later today.

I am also a little perplexed by your debug log. What is producing:
```
...........
    -- SCCP: destroying PBX rtp server on channel SCCP/35222-00000001
    -- QualityStats: MLQK=0.0000;MLQKav=0.0000;MLQKmn=0.0000;MLQKmx=0.0000;ICR=0.0066;CCR=0.0051;ICRmx=0.0066;CS=2;SCS=0;MLQKvr=0.95
    -- SEP0023339D632D: Call Statistics:
       [
         Last Call        : CallID: 1 Packets sent: 364 rcvd: 70 lost: 0 jitter: 0 latency: 0
         Last Quality     : MLQK=0.0000;MLQKav=0.0000;MLQKmn=0.0000;MLQKmx=0.0000;MLQKvr=0.95|ICR=0.0066;CCR=0.0051;ICRmx=0.0066|CS=2;SCS=0
         Mean Statistics  : #Calls: 1 Packets sent: 364 rcvd: 70 lost: 0 jitter: 0 latency: 0
         Mean Quality     : MLQK=0.0000;MLQKav=0.0000;MLQKmn=0.0000;MLQKmx=0.0000;MLQKvr=0.95|ICR=0.0066;CCR=0.0051;ICRmx=0.0066|CS=2;SCS=0
       ]
```
Normal Call hangup ending : So far so good:
But then i see this after it...
```
    -- SEP0023339D632D: Getting the active channel on device.
    -- SEP0023339D632D: No active channel on device.
    -- SEP0023339D632D: Taken Offhook
    -- SCCP: Looking for channel by state '22' on device: SEP0023339D632D
    -- SEP0023339D632D: line: '35222'
    -- SEP0023339D632D: Could not find active channel with state RINGING(22) on device
SEP0023339D632D: (getEmptyChannel) on line:35222, maybe_c:
SEP0023339D632D: (getEmptyChannel) getActiveChannel
    -- SEP0023339D632D: Getting the active channel on device.
    -- SEP0023339D632D: No active channel on device.
    -- SEP0023339D632D: Set the active channel 2 on device
    -- 35222: New channel number: 2 on line 35222
    -- SCCP: (pbx_channel_allocate) try to allocate Outbound channel on line: 35222, 
    -- SCCP:              cid_num: "35222"
    -- SCCP:             cid_name: "ExtraColor"
    -- SCCP:          accountcode: "79021"
    -- SCCP:                exten: ""
    -- SCCP:              context: "from-internal-xfer-custom"
    -- SCCP:             amaflags: "0"
    -- SCCP:            chan/call: "SCCP/35222-00000002"
    -- SCCP:combined capabilities: "[slin16, ulaw/64k, alaw/64k, ilbc, g729b, g729ab, g729, g729a]"
    -- SCCP:  reduced preferences: "[alaw/64k, ulaw/64k, g729, g729a, g729b, g729ab]"
    -- SCCP: (allocPBXChannel) Create New Channel with name: SCCP/35222-00000002
    -- SCCP/35222-00000002: (recalculateReadformat) 
	capabilities: [slin16, ulaw/64k, alaw/64k, ilbc, g729b, g729ab, g729, g729a]
	preferences: [alaw/64k, ulaw/64k, g729, g729a, g729b, g729ab]
	remote caps: []
	READ:[alaw/64k]
    -- SCCP/35222-00000002: (recalculateWriteformat) 
	capabilities: [slin16, ulaw/64k, alaw/64k, ilbc, g729b, g729ab, g729, g729a]
	preferences: [alaw/64k, ulaw/64k, g729, g729a, g729b, g729ab]
	remote caps: []
	Write:[alaw/64k]
    -- 35222: Allocated asterisk channel SCCP/35222-00000002
    -- SEP0023339D632D: Set asterisk state OffHook (2) for call 2
    -- SEP0023339D632D: Indicate SCCP new state:OFFHOOK, current channel state:DOWN on call:SCCP/35222-00000002, lineInstance:1 (previous channelstate:DOWN)
    -- SEP0023339D632D: Finish to indicate channel state:OFFHOOK on call:SCCP/35222-00000002, lineInstance:1. New channel state:OFFHOOK
    -- SCCP/35222-00000002: schedule digittimeout 16
    -- SEP0023339D632D: is Onhook (buttonIndex: 0, callid: 0)
    -- SEP0023339D632D: Getting the active channel on device.
    -- SCCP/35222-00000002: Disabling scheduler / Removing Scheduled tasks (digittimeout_id:14) (hangup_id:-1)
  == SEP0023339D632D: Ending call SCCP/35222-00000002 (state:OFFHOOK)
    -- SEP0023339D632D: Sending hangupRequest to Call SCCP/35222-00000002 (state: OFFHOOK)
  == SEP0023339D632D: (sccp_channel_closeAllMediaTransmitAndReceive) Stop All Media Reception and Transmission on channel 2
    -- SEP0023339D632D: Current channel SCCP/35222-00000002 state OFFHOOK(10)
    -- SEP0023339D632D: Indicate SCCP new state:ONHOOK, current channel state:OFFHOOK on call:SCCP/35222-00000002, lineInstance:1 (previous channelstate:DOWN)
    -- SEP0023339D632D: Finish to indicate channel state:ONHOOK on call:SCCP/35222-00000002, lineInstance:1. New channel state:DOWN
    -- SEP0023339D632D: Device is Requesting CallStatisticsAndClear
    -- SCCP: Cleaning channel SCCP/35222-00000002
  == SEP0023339D632D: (sccp_channel_closeAllMediaTransmitAndReceive) Stop All Media Reception and Transmission on channel 2
    -- SEP0023339D632D: Set the active channel 0 on device
    -- SEP0023339D632D: Looking for selected channel (2)
    -- Destroying channel SCCP/35222-00000002
    -- QualityStats: MLQK=0.0000;MLQKav=0.0000;MLQKmn=0.0000;MLQKmx=0.0000;ICR=0.0066;CCR=0.0051;ICRmx=0.0066;CS=2;SCS=0;MLQKvr=0.95
    -- SEP0023339D632D: Call Statistics:
       [
         Last Call        : CallID: 2 Packets sent: 364 rcvd: 70 lost: 0 jitter: 0 latency: 0
         Last Quality     : MLQK=0.0000;MLQKav=0.0000;MLQKmn=0.0000;MLQKmx=0.0000;MLQKvr=0.95|ICR=0.0066;CCR=0.0051;ICRmx=0.0066|CS=2;SCS=0
         Mean Statistics  : #Calls: 2 Packets sent: 364 rcvd: 70 lost: 0 jitter: 0 latency: 0
         Mean Quality     : MLQK=0.0000;MLQKav=0.0000;MLQKmn=0.0000;MLQKmx=0.0000;MLQKvr=0.95|ICR=0.0066;CCR=0.0051;ICRmx=0.0066|CS=2;SCS=0
       ]"
```
After you hangup. That is not normal behaviour.

rolakron, 2018-12-24:
With the latest chan_sccp the radio channels and the speaking clock are quiet with earlyrtp = progress.
And I don't either get ringing tone on a external call.

Oh, the last part is added here by accident, I struggled with the handset and isn't really relevant for this. Sorry for that. :P

I feel I have to sleep now, I can try to take a look at it tomorrow.

dkgroot, 2018-12-24:
did you try my little change to the dialplan, to see if adding progress() in front of answer() makes a difference ? Would be belpfull to know if your checked.
If you are already in bed / asleep, then i will recheck myself :-)  I should have been asleep 7 hours ago :-)

dkgroot, 2018-12-24:
@kbstechnologies 
- Is that in anyway related to the rtp issue mentioned above ?
- If this is a seperate issue, then please create/open a new issue report. And include a example of sccp.conf with this "sccp user id" in it. I am not really sure what you mean by "sccp user id"

dkgroot, 2018-12-24:
@rolakron 
I created an issue branch called "issues/438" for you to test. Please do:
```
git pull
git fetch -a
git checkout issues/438
make && make reload
```
And run your test again (using earlyrtp=progress).

FYI: I was able to reproduce the issue using your little dialplan excerpt. And just for your info, inserting Progress() before Answer() resolved the issue on my system, which gave me a headstart in finding the cause and fixing it.
Please report back if the issue is resolved using this issue branch, so i can merge it.

rolakron, 2018-12-24:
Ok. I can test that. But will be with a little delay from my side, I'll possible have time to test it out tomorrow.
BTW, night time is the best time to get some work done. ;)

dkgroot, 2018-12-24:
I already had Alexey test it for me. So you can skip my previous instructions. Patch will be merged in the develop branch in a couple of minutes... That also closes this issue automatically.
if you find it does not work correctly tomorrow (working on christmas ?? ) Then just reopen this issue and let me know.

dkgroot, 2018-12-24:
Thanks for reporting the issue and helping me out, to figure out the cause !!

@kbstechnologies please report your issue in a seperate issue report

dkgroot, 2018-12-24:
:-) Hopefully the issue is fixed for you, and there are no further nasty surprises. That would be a nice ending of the year :-)

rolakron, 2018-12-24:
Ok. I have tested on another system now, the speaking clock seems to work with earlyrtp = progress, but when I place an external call (in my case to my mobilephone) I still don't get any call progress tone, but the mobilephone rings. Sound is coming thru when I answer the call and works as intended in that way. I may have to point out that the call progress tone (ringingtone) when i call my mobilephone is not generated either by Asterisk or by my SCCP-phone, and have never been. The call progress tone itself comes from the receiving phoneswitch for the mobile network. But that is not happening now, it's just quiet.
I have not been able to test the radio stations (they are not configured yet on my other system), but I assume they work as the speaking clock also did. But I will check that later.

Here is the log from when I called my mobilephone (i declined to answer the call).

"SCCP new debug status: (1 -> 12357) core,rtp,channel,indicate,pbx
    -- SEP00070E15D0C4: Getting the active channel on device.
    -- SEP00070E15D0C4: No active channel on device.
    -- SEP00070E15D0C4: Taken Offhook
    -- SCCP: Looking for channel by state '22' on device: SEP00070E15D0C4
    -- SEP00070E15D0C4: line: '3001'
    -- SEP00070E15D0C4: Could not find active channel with state RINGING(22) on device
SEP00070E15D0C4: (getEmptyChannel) on line:3001, maybe_c:
SEP00070E15D0C4: (getEmptyChannel) getActiveChannel
    -- SEP00070E15D0C4: Getting the active channel on device.
    -- SEP00070E15D0C4: No active channel on device.
    -- SEP00070E15D0C4: Set the active channel 5 on device
    -- 3001: New channel number: 5 on line 3001
    -- SCCP: (pbx_channel_allocate) try to allocate Outbound channel on line: 3001, 
    -- SCCP:              cid_num: "3001"
    -- SCCP:             cid_name: "3001"
    -- SCCP:          accountcode: "79021"
    -- SCCP:                exten: ""
    -- SCCP:              context: "from-internal-xfer"
    -- SCCP:             amaflags: "0"
    -- SCCP:            chan/call: "SCCP/3001-00000005"
    -- SCCP:combined capabilities: "[slin16, ulaw/64k, alaw/64k, g729b, g729ab, g729, g729a]"
    -- SCCP:  reduced preferences: "[alaw/64k, ulaw/64k, g729, g729a, g729b, g729ab]"
    -- SCCP: (allocPBXChannel) Create New Channel with name: SCCP/3001-00000005
    -- SCCP/3001-00000005: rtp server instance created at [::]:12478
    -- SCCP/3001-00000005: (createAudioServer) updated phone Audio RTP destination to : [::ffff:172.31.1.30]:12478, family:IPv4, mapped: True
    -- 3001: Allocated asterisk channel SCCP/3001-00000005
    -- SEP00070E15D0C4: Set asterisk state OffHook (2) for call 5
    -- SEP00070E15D0C4: Indicate SCCP new state:OFFHOOK, current channel state:DOWN on call:SCCP/3001-00000005, lineInstance:1 (previous channelstate:DOWN)
    -- SEP00070E15D0C4: Finish to indicate channel state:OFFHOOK on call:SCCP/3001-00000005, lineInstance:1. New channel state:OFFHOOK
    -- SCCP/3001-00000005: schedule digittimeout 16
    -- SCCP/3001-00000005: stop schedule digittimeout 4
    -- SEP00070E15D0C4: (sccp_pbx_softswitch) New call on line 3001
    -- SEP00070E15D0C4: (sccp_pbx_softswitch) Dial Extension 00505123456
    -- SEP00070E15D0C4: Indicate SCCP new state:DIALING, current channel state:OFFHOOK on call:SCCP/3001-00000005, lineInstance:1 (previous channelstate:DOWN)
    -- SEP00070E15D0C4: Send the dialed number:00505478848, callid:5, lineInstance:1
    -- SEP00070E15D0C4: Finish to indicate channel state:DIALING on call:SCCP/3001-00000005, lineInstance:1. New channel state:DIALING
    -- SCCP: (sccp_pbx_softswitch) set variable SKINNY_PRIVATE to: 0
    -- SCCP: (sccp_pbx_softswitch) set variable SKINNY_PRIVATE to: 0
    -- +- pbx extension matcher (00505478848    ): ---+
    -- |ignore     |exists     |can match  |match more|
    -- | no        |yes        |yes        | no       |
    -- +----------------------------------------------+
 SEP00070E15D0C4: (sccp_pbx_softswitch) channel SCCP/3001-00000005 is dialing number 00505478848
    -- SEP00070E15D0C4: Set asterisk state Ring (4) for call 5
    -- SEP00070E15D0C4: Asterisk prodded channel SCCP/3001-00000005.
    -- SCCP/3001-00000005: (pbx_pbx_start) autoloop has started, set requestHangup = requestQueueHangup
 SEP00070E15D0C4: (sccp_pbx_softswitch) pbx started
 SEP00070E15D0C4: (sccp_pbx_softswitch) finished
 SCCP: (sccp_pbx_softswitch) exit
 SEP00070E15D0C4: (asterisk113_get_rtp_info) Asterisk requested EarlyRTP peer for channel SCCP/3001-00000005
 SEP00070E15D0C4: (asterisk113_get_rtp_info) Direct RTP disabled ->  Using AST_RTP_TRY_PARTIAL for channel SCCP/3001-00000005
 SEP00070E15D0C4: (asterisk113_get_vrtp_info) Asterisk requested EarlyRTP peer for channel SCCP/3001-00000005
    -- SCCP/3001-00000005: (pbx_indicate) start indicate 'AST_CONTROL_PVT_CAUSE_CODE: Contains an update to the protocol-specific cause-code stored for branching dials'
    -- SEP00070E15D0C4: (pbx_indicate) start indicate 'AST_CONTROL_PVT_CAUSE_CODE: Contains an update to the protocol-specific cause-code stored for branching dials' (33) condition on channel SCCP/3001-00000005 (readStat:0, writeState:0, rtp:yes)
    -- SCCP/3001-00000005: hangup cause set: 0
  == SEP00070E15D0C4: (pbx_indicate) finish: send indication (res:-1)
    -- SCCP/3001-00000005: (pbx_indicate) start indicate 'AST_CONTROL_PVT_CAUSE_CODE: Contains an update to the protocol-specific cause-code stored for branching dials'
[... cut ...]
    -- SEP00070E15D0C4: Call Statistics:
       [
         Last Call        : CallID: 5 Packets sent: 0 rcvd: 580 lost: 0 jitter: 0 latency: 0
         Last Quality     : MLQK=4.5000;MLQKav=4.5000;MLQKmn=4.5000;MLQKmx=4.5000;MLQKvr=0.95|ICR=0.0000;CCR=0.0000;ICRmx=0.0000|CS=0;SCS=0
         Mean Statistics  : #Calls: 5 Packets sent: 125 rcvd: 234 lost: 0 jitter: 4 latency: 0
         Mean Quality     : MLQK=1.3000;MLQKav=1.3000;MLQKmn=1.3000;MLQKmx=4.5000;MLQKvr=0.95|ICR=0.0653;CCR=0.0557;ICRmx=0.5174|CS=0;SCS=0
       ]
"

kbstechnologies, 2018-12-24:
I was only bringing it up because the behavior was similar by the description.  If you are interested in it then I will open it as a new issue I just wrote it up as a limitation adjusted my code to compensate and moved on.

dkgroot, 2018-12-24:
@kbstechnologies 
I like to keep issues seperate, so if there is something, open a new one :-) 

I still don't fully understand the issue you were trying to report, nor what "sccp user id" you mean. 

If you that some of the sccp.conf parameter have a max length, then i fully agree that is the case. Not all strings are unlimited in length. If you want to check how long a string can be, you can use the SCCPConfigMetaData AMI command, which will generate a report with every sccp.conf parameter etc. So if this is what you wanted to report, i am fully aware :-)

rolakron, 2018-12-25:
Thanks for the comments in the log.
I can confirm that the phones and the server is on the same network/subnet , so Nonat is a correct assumption. I can check tomorrow and compare the logs from the working configuration and the one that is not working correctly.
I can test your sample configuration and see what results I get.
The busy-tone is played by Asterisk (also in the working configuration) and that's also what's expected. The ringing-tone is played by the remote system, not by Asterisk or the phone.

dkgroot, 2018-12-25:
This raises an interesting point. The patch merged a couple of days, was related to fixing some of the issues regarding directrtp handling, and this might be a negative side effect of it. The TRY Partial part is related to one side of the call using directrtp (or trying to) and the other side not supporting it.

Question1: Is this from a sccp->sip-trunk call or from the radio example ?
Question2: do you have direcrtrtp (or directmedia in sip.conf lingo) enabled on either side ? Looking at previous debug logs i don't think so.

I created a new branch "issues/438_1" for you to checkout. It adds a little (more) bit of logging to the get_rtp_info function so we can see how we got to a different decision between the two calls. I think your might have figured out the potential solution, or at least closing onto it.
```
git pull
git fetch -a
git checkout issues/483_1
make && make install && make reload
```
Please set "sccp debug core, channel, pbx, rtp, codec, indicate" during the test calls.

rolakron, 2018-12-25:
It's a SCCP-phone placing an external call over a SIP-trunk.
I haven't tried setting directrtpsetup=yes yet on the trunk, I'm going to test that.
I can pull and test that to see what we get.

dkgroot, 2018-12-25:
That's strange i don't see any of the debug lines i added in "issues/438_1". At least one extra line should have appeared after ```(asterisk113_get_rtp_info) Asterisk requested EarlyRTP peer for channel SCCP/3001-00000005```. Did you checkout the correct branch (438_1) ? Did you set "sccp debug core, channel, pbx, rtp, codec, indicate." (the rtp flag is the one that is required to show the extra added lines of debug, the others just give some channel context and timing) ? Sorry to ask.

I do have one confession to make. I have been testing using asterisk-16 instead of asterisk-13. I will switch to 13 later today, to make sure my observations make sense in comparison.

rolakron, 2018-12-25:
I copy-pasted your commands, but I will check again later today and make shure that I have the correct version/branch.

dkgroot, 2018-12-25:
The debug samples you gave have different ip-ranges, so these are completely different machines and phones (right) ?

I have figured out a way to reproduce the issue, so that's good.
I also found a location in the source which i can revert to a previous situation which makes everything work. Which means startMediaTransmission as soon as the openReceiveChannel has been opened, but the whole point of #420 was to not do exactly that. They should be seperate.
So scratching my head here.

Going for Christmas Party now, so will have to get back to this tomorrow.

dkgroot, 2018-12-25:
No matter, i was finally able to reproduce the issue (Had to set up a sip trunk etc etc...). This is always the biggest hurdle for me to cross. Once i am able to reproduce the issue, i don't have to stare at endless pieces of supplied debug (with strange ip-addresses which it is hard for me to reason about ) :-).

rolakron, 2018-12-25:
I know what you mean. It's always easier (and way faster) to have the faulty system available when there is need to troubleshoot something.

dkgroot, 2018-12-26:
@rolakron 
Current Analysis:
 - The current develop branch (and 438_1 for that matter) removed the starting of rtp sending by the outbound phone during earlyRTP. This makes sense, because it only has to listen and it does allow us to more easily adapt to changes in later stages.
 - This does has a negative side effect though, if there are any firewalls an route they could block incoming RTP. We are not punching the port open from the inside by sending anymore.
 - On my simple SIP trunk setup (without STUN / ICE etc), this seems to be the cause of some problems. I do see the remote SIP (ast2) end transmitting RTP to the other side (So the signalling that earlyrtp should start is correct).
 - The sent packets never arrive to the outbound asterisk server (ast1) though.
 - If i add "startMediaTransmission" immediatly after "openReceiveChannelAck", making the phone start sending rtp packets, everything start working immediatly.
 - However this is not something i really want to do.

Possible solutions:
1. Add startMediaTransmission in openReceiveAck, immediatly followed by stopMediaTransmission. That would punch open the firewalls from both sides and we are back on track.
2. Send a single audio frame from the asterisk server (ast1) to punch open the firewall from the inside. (Have not figured out yet how to send an empty audioframe yet, looks harder that i had hoped).
3. Send a sing udp packet coming from the same ip-address/port as the phone to the remote end, doing the punching without having the phone or asterisk-frames involved. This might be the simplest and fastest. We could even have a scheduled event to do an RTP-keepalive every couple of min as long as media is only flowing in one way (earlyrtp).
    Note: I am hoping this does not confuse asterisk rtp handling to much, and it will just discard the packet.

I am opting for variant 3 at the moment, as it should be the easiest to implement without negative side effects.

dkgroot, 2018-12-26:
@rolakron
You can either store the sample in a file name "patch-file" in the chan-sccp root directory (not in the src dir). and use
```git apply --reject <patch-file```
Or you can simply got into src/sccp_channel.c and make the small change yourself.
I have not made a commit out of it, to keep the history clean.

dkgroot, 2018-12-26:
@rolakron It might help to also sign in to out [gitter chat](https://gitter.im/chan-sccp/chan-sccp) to help speeding up the issue handling a little :-)

rolakron, 2018-12-27:
Ok, nice! I will test it at some point when I have time. :)

dkgroot, 2018-12-27:
I figured out a nice way to do it.
Briefly startmediatransmission to punch open bi-directional rtp flowing during early rtp
Once the first packet has been receive from the phone (and been transferred to the remote side),
the mediatransmission can be closed again until further notice. We makde sure the rtp port is
not closed though, so that it will remain the same.
You can forget about the patch i send earlier (already removed it above) and just pull the latest issues/438_1 revision. As far as i can confirm, it seems to be working quite nicely.

dkgroot, 2018-12-27:
Debug:
```
    -- +- pbx extension matcher (4447           ): ---+
    -- |ignore     |exists     |can match  |match more|
    -- | no        |yes        |yes        |yes       |
    -- +----------------------------------------------+
 SEP0023043403F9: (sccp_pbx_softswitch) channel SCCP/98041-00000005 is dialing number 4447
    -- SEP0023043403F9: Set asterisk state Ring (4) for call 5
    -- SCCP/98041-00000005: (pbx_pbx_start) autoloop has started, set requestHangup = requestQueueHangup
 SEP0023043403F9: (sccp_pbx_softswitch) pbx started
 SEP0023043403F9: (sccp_pbx_softswitch) finished
 SCCP: (sccp_pbx_softswitch) exit
    -- Executing [4447@internal:1] Set("SCCP/98041-00000005", "CALLERID(name)="Dialing SIP on asterisk.chan-sccp.net"") in new stack
```
Dialing a remote sip channel
```
    -- Executing [4447@internal:2] Dial("SCCP/98041-00000005", "SIP/908@ast-chan-sccp, 60") in new stack
[Dec 27 17:42:44] NOTICE[10177][C-000001b9]: app_dial.c:2512 dial_exec_full: SCCP/98041-00000005: (app_dial) NativeAudioFormats:(g722|alaw|ulaw|g729)
  == Using SIP RTP CoS mark 5
 SEP0023043403F9: (asterisk113_get_rtp_info) Asterisk requested EarlyRTP peer for channel SCCP/98041-00000005
  == SCCP/98041-00000005: (rtp_getAudioPeerInfo) returning 1
 SEP0023043403F9: (asterisk113_get_rtp_info) Direct RTP disabled ->  Using AST_RTP_TRY_PARTIAL for channel SCCP/98041-00000005
 SEP0023043403F9: (asterisk113_get_vrtp_info) Asterisk requested EarlyRTP peer for channel SCCP/98041-00000005
    -- Called SIP/908@ast-chan-sccp
    -- SCCP/98041-00000005: (pbx_indicate) start indicate 'AST_CONTROL_PVT_CAUSE_CODE: Contains an update to the protocol-specific cause-code stored for branching dials'
    -- SEP0023043403F9: (pbx_indicate) start indicate 'AST_CONTROL_PVT_CAUSE_CODE: Contains an update to the protocol-specific cause-code stored for branching dials' (33) condition on channel SCCP/98041-00000005 (readStat:0, writeState:0, rtp:yes)
    -- SCCP/98041-00000005: hangup cause set: 0
  == SEP0023043403F9: (pbx_indicate) finish: send indication (res:-1)
    -- SCCP/98041-00000005: (pbx_indicate) start indicate 'AST_CONTROL_PVT_CAUSE_CODE: Contains an update to the protocol-specific cause-code stored for branching dials'
    -- SEP0023043403F9: (pbx_indicate) start indicate 'AST_CONTROL_PVT_CAUSE_CODE: Contains an update to the protocol-specific cause-code stored for branching dials' (33) condition on channel SCCP/98041-00000005 (readStat:0, writeState:0, rtp:yes)
    -- SCCP/98041-00000005: hangup cause set: 0
  == SEP0023043403F9: (pbx_indicate) finish: send indication (res:-1)
    -- SIP/ast-chan-sccp-00000003 is making progress passing it to SCCP/98041-00000005
[Dec 27 17:42:45] NOTICE[10177][C-000001b9]: rtp_engine.c:1451 ast_rtp_instance_early_bridge: EarlyBridge: channel c0:SCCP/98041-00000005, channel c1:SIP/ast-chan-sccp-00000003
[Dec 27 17:42:45] NOTICE[10177][C-000001b9]: rtp_engine.c:1463 ast_rtp_instance_early_bridge: EarlyBridge: Locked channel c0:SCCP/98041-00000005 && channel c1:SIP/ast-chan-sccp-00000003
 SEP0023043403F9: (asterisk113_get_rtp_info) Asterisk requested EarlyRTP peer for channel SCCP/98041-00000005
  == SCCP/98041-00000005: (rtp_getAudioPeerInfo) returning 1
```
Notive still using TRY_PARIAL
```
 SEP0023043403F9: (asterisk113_get_rtp_info) Direct RTP disabled ->  Using AST_RTP_TRY_PARTIAL for channel SCCP/98041-00000005
 SEP0023043403F9: (asterisk113_get_vrtp_info) Asterisk requested EarlyRTP peer for channel SCCP/98041-00000005
    -- SCCP/98041-00000005: (pbx_indicate) start indicate 'AST_CONTROL_PROGRESS: Indicate PROGRESS'
    -- SEP0023043403F9: (pbx_indicate) start indicate 'AST_CONTROL_PROGRESS: Indicate PROGRESS' (14) condition on channel SCCP/98041-00000005 (readStat:0, writeState:0, rtp:yes)
    -- SEP0023043403F9: Indicate SCCP new state:PROGRESS, current channel state:DIALING on call:SCCP/98041-00000005, lineInstance:1 (previous channelstate:OFFHOOK)
  == SEP0023043403F9: SCCP_CHANNELSTATE_PROGRESS
    -- SCCP: Asterisk requests to change state to (Progress) from (DIALING)
    -- SCCP/98041-00000005, Could not find a common prefered codec (yet), using g722/64k (6)
    -- SCCP/98041-00000005: (findBestJoint) returning prefered codec g722/64k (6)
    -- SCCP/98041-00000005: (setNativeAudioFormats) (g722)
    -- SEP0023043403F9 - SCCP/98041-00000005: (recalculateCodecformat) 
	capabilities: [slin16, g722/64k, ulaw/64k, alaw/64k, g722/56k, g722/48k, g729b, g729ab, g729, g729a]
	preferences: [g722/64k, g722/56k, g722/48k, alaw/64k, ulaw/64k, g729, g729a, g729b, g729ab]
	remote caps: []
	Format:g722/64k
    -- SCCP/98041-00000005, OpenReceiveChannel with format G.722 64k, payload 6, echocancel: YES, passthrupartyid: 4294967290, pbx_channel_name: SCCP/98041-00000005
    -- SEP0023043403F9: Finish to indicate channel state:PROGRESS on call:SCCP/98041-00000005, lineInstance:1. New channel state:PROGRESS
  == SEP0023043403F9: (pbx_indicate) finish: send indication (res:-1)
    -- SEP0023043403F9: Got OpenChannel ACK. Status:'Media Status: OK' (0), Remote RTP/UDP:'192.168.178.5:18562', Type:Indirect RTP, PassThruPartyId:4294967290, CallID:5
    -- SEP0023043403F9: Getting the active channel on device.
    -- SCCP: (asterisk113_setPhoneRTPAddress) Update PBX to send RTP/UDP media to '192.168.178.5:18562' (new remote) (NAT: no)
    -- SEP0023043403F9: Tell PBX   to send RTP/UDP media from 192.168.178.3:20680 to 192.168.178.5:18562 (NoNat)
[... cut ...]
Got  RTP packet from    192.168.178.5:18562 (type 09, seq 042907, ts 7243944, len 000160)
Got  RTP packet from    192.168.178.5:18562 (type 09, seq 042908, ts 7244104, len 000160)
Sent RTP packet to      185.21.103.35:17330 (type 09, seq 011885, ts 7242504, len 000160)
```
RTP flows in both directions again.
```
Got  RTP packet from    192.168.178.5:18562 (type 09, seq 042909, ts 7244264, len 000160)
Sent RTP packet to      185.21.103.35:17330 (type 09, seq 011886, ts 7242664, len 000160)
```
```

rolakron, 2018-12-28:
I will test that tomorrow! :) Thank you!

dkgroot, 2018-12-28:
(FYI: these commits with the repeated same name, are mainly because of rebasing on top of changes in the development branch (making merging later on a little easier), so don't worry about that).

rolakron, 2018-12-29:
I pulled the latest of 438_1 now and it seems to work as it should. I hear the call progress tones now when I call my mobilephone. Thank you! :)

dkgroot, 2018-12-29:
@rolakron 
Happy to hear it works !

Note: I am still not 100% sure if what we're doing is the correct/right thing to do.

Old behaviour:
- On earlyrtp request
- Send OpenReceiveChannel to the phone
- On OpenReceiveChannelAck:
  - Transmit the new RTP ip-address to the remote end
  - Send StartMediaTransmission to the phone
  - On StartMediaTransmissionAck, asterisk would see the incoming RTP Packets and start forwaring in both directions.

New behaviour:
- On earlyrtp request
- Send OpenReceiveChannel to the phone
- On OpenReceiveChannelAck:
  - Transmit the new RTP ip-address to the remote end
  - Send StartMediaTransmission to the phone
  - On StartMediaTransmissionAck:
    - We schedule a StopMediaTransmission to happen 500 uSecs later
    - asterisk would see the incoming RTP Packets and start forwaring in both directions

Pro:
- We can choose the sending rtp codec, later in the call process (note: skinny phones cannot adapt to the incoming rtp stream automatically, they have to be told to stop and restart rtp (which takes considerble time).
Con:
- If Progress is followed  by Proceed or Connected because of the remote end Answering the call, we might have already started to stopMediaTransmission, even though the codec has not (even) changed:
  - either waisting more time than before (instead of less)
  - or worse even, not re-opening the outgoing stream (please test this does not happen) because the two events overlap.

Maybe the old (simpler) method was better. Please do some extensive testing (maybe even include extra Wait() / Progress() / Proceed() / Answer() steps in your test dialplan). I would like to prevent some regression because of this.

Note to self: Thing about earlyrtp=immediate.

rolakron, 2018-12-30:
First test. I encountered som errors, the call disconnects immediatly after I have dialed the speaking clock (*60):
``` SEP00070E15D0C4: (sccp_pbx_softswitch) channel SCCP/3001-0000000B is dialing number *60
    -- SEP00070E15D0C4: Set asterisk state Ring (4) for call 11
    -- SEP00070E15D0C4: Asterisk prodded channel SCCP/3001-0000000B.
    -- SCCP/3001-0000000B: (pbx_pbx_start) autoloop has started, set requestHangup = requestQueueHangup
 SEP00070E15D0C4: (sccp_pbx_softswitch) pbx started
 SEP00070E15D0C4: (sccp_pbx_softswitch) finished
 SCCP: (sccp_pbx_softswitch) exit
    -- SCCP/3001-0000000B: (pbx_indicate) start indicate 'AST_CONTROL_PROGRESS: Indicate PROGRESS'
    -- SEP00070E15D0C4: (pbx_indicate) start indicate 'AST_CONTROL_PROGRESS: Indicate PROGRESS' (14) condition on channel SCCP/3001-0000000B (readStat:0, writeState:0, rtp:yes)
    -- SEP00070E15D0C4: Indicate SCCP new state:PROGRESS, current channel state:DIALING on call:SCCP/3001-0000000B, lineInstance:1 (previous channelstate:DIGITSFOLL)
    -- SCCP: Asterisk requests to change state from (DIALING) to (PROGRESS)
    -- SCCP/3001-0000000B, Could not find a common prefered codec (yet), using alaw/64k (2)
    -- SCCP/3001-0000000B: (findBestJoint) returning prefered codec alaw/64k (2)
    -- SCCP/3001-0000000B: (setNativeAudioFormats) (alaw)
    -- SEP00070E15D0C4 - SCCP/3001-0000000B: (recalculateCodecformat) 
	capabilities: [slin16, ulaw/64k, alaw/64k, g729b, g729ab, g729, g729a]
	preferences: [alaw/64k, ulaw/64k, g729, g729a, g729b, g729ab]
	remote caps: []
	Format:alaw/64k
    -- SCCP/3001-0000000B, OpenReceiveChannel with format G.711 A-law 64k, payload 2, echocancel: YES, passthrupartyid: 4294967284, pbx_channel_name: SCCP/3001-0000000B
    -- SCCP/3001-0000000B: Using dtmf rtp_code : 101
    -- SEP00070E15D0C4: Finish to indicate channel state:PROGRESS on call:SCCP/3001-0000000B, lineInstance:1. New channel state:PROGRESS
  == SEP00070E15D0C4: (pbx_indicate) finish: send indication (res:-1)
    -- SCCP: sccp_pbx_answer
    -- SEP00070E15D0C4: sccp_pbx_answer checking parent channel
    -- SEP00070E15D0C4: (sccp_pbx_answer) Outgoing call SCCP/3001-0000000B has been answered by remote party
    -- SEP00070E15D0C4: Indicate SCCP new state:PROCEED, current channel state:PROGRESS on call:SCCP/3001-0000000B, lineInstance:1 (previous channelstate:DIALING)
    -- SEP00070E15D0C4: Finish to indicate channel state:PROCEED on call:SCCP/3001-0000000B, lineInstance:1. New channel state:PROCEED
    -- SEP00070E15D0C4: Starting Phone RTP/UDP Transmission (State: PROCEED[23])
    -- SCCP/3001-0000000B: Using dtmf rtp_code : 101
    -- SEP00070E15D0C4: Tell Phone to send RTP/UDP media from  to 172.31.1.30:17898 (NAT: (Auto)Off)
    -- SEP00070E15D0C4: Using codec: G.711 A-law 64k(2), TOS 184, Silence Suppression: OFF for call with PassThruId: 4294967284 and CallID: 11
    -- SEP00070E15D0C4: Indicate SCCP new state:CONNECTED, current channel state:PROCEED on call:SCCP/3001-0000000B, lineInstance:1 (previous channelstate:PROGRESS)
    -- SEP00070E15D0C4: Did not reopen an RTP stream as old SCCP state was (PROCEED)
    -- SEP00070E15D0C4: Finish to indicate channel state:CONNECTED on call:SCCP/3001-0000000B, lineInstance:1. New channel state:CONNECTED
    -- SCCP/3001-0000000B: (pbx_indicate) start indicate 'AST_CONTROL_PROD: Kick remote channel'
    -- SEP00070E15D0C4: (pbx_indicate) start indicate 'AST_CONTROL_PROD: Kick remote channel' (-1) condition on channel SCCP/3001-0000000B (readStat:1, writeState:1, rtp:yes)
  == SEP00070E15D0C4: (pbx_indicate) finish: send indication (res:-1)
    -- SEP00070E15D0C4: Got OpenChannel ACK. Status:'Media Error: Unknown' (1), Remote RTP/UDP:'172.31.1.223:0', Type:Indirect RTP, PassThruPartyId:4294967284, CallID:11
    -- SEP00070E15D0C4: Getting the active channel on device.
[2018-12-30 02:40:30] ERROR[3633][C-00000000]: sccp_actions.c:3414 handle_openReceiveChannelAck: SEP00070E15D0C4: Device returned: 'Media Error: Unknown' (1) !. Giving up.
  == SEP00070E15D0C4: (sccp_channel_closeAllMediaTransmitAndReceive) Stop All Media Reception and Transmission on channel 11
    -- SEP00070E15D0C4: Stop mediatransmission on channel 11 (KeepPortOpen: NO)
    -- SEP00070E15D0C4: Close receivechannel on channel 11 (KeepPortOpen: NO)
       > SEP00070E15D0C4: Stopping PBX audio rtp transmission on channel SCCP/3001-0000000B
    -- SCCP/3001-0000000B: Disabling scheduler / Removing Scheduled tasks (digittimeout_id:-1) (hangup_id:-1)
  == SEP00070E15D0C4: Ending call SCCP/3001-0000000B (state:CONNECTED)
    -- SEP00070E15D0C4: Sending hangupRequest to Call SCCP/3001-0000000B (state: CONNECTED)
    -- SEP00070E15D0C4: Got startMediaTransmission ACK. Status:'Media Status: OK' (0), Remote RTP/UDP:'172.31.1.223:0', Type:Indirect RTP, PassThruPartyId:4294967284, CallID:11, CallID1:11
    -- SEP00070E15D0C4: Getting the active channel on device.
  == SEP00070E15D0C4: (sccp_channel_closeAllMediaTransmitAndReceive) Stop All Media Reception and Transmission on channel 11
       > SEP00070E15D0C4: Stopping PBX audio rtp transmission on channel SCCP/3001-0000000B
    -- SEP00070E15D0C4: Current channel SCCP/3001-0000000B state CONNECTED(30)
    -- SEP00070E15D0C4: Indicate SCCP new state:ONHOOK, current channel state:CONNECTED on call:SCCP/3001-0000000B, lineInstance:1 (previous channelstate:PROCEED)
    -- SEP00070E15D0C4: Finish to indicate channel state:ONHOOK on call:SCCP/3001-0000000B, lineInstance:1. New channel state:DOWN
[... cut ...]
       > SEP00070E15D0C4: Stopping PBX audio rtp transmission on channel SCCP/3001-0000000B
    -- SCCP/3001-0000000B: Disabling scheduler / Removing Scheduled tasks (digittimeout_id:-1) (hangup_id:-1)
  == SEP00070E15D0C4: Ending call SCCP/3001-0000000B (state:CONNECTED)
    -- SEP00070E15D0C4: Sending hangupRequest to Call SCCP/3001-0000000B (state: CONNECTED)
    -- SEP00070E15D0C4: Got startMediaTransmission ACK. Status:'Media Status: OK' (0), Remote RTP/UDP:'172.31.1.223:0', Type:Indirect RTP, PassThruPartyId:4294967284, CallID:11, CallID1:11
```

dkgroot, 2018-12-30:
Ooh my this is the wrong order of events:
It should be:
Srv->Phone: OpenReceiveChannel (Request)
Phone->Srv: OpenReceiveChannelAck
Srv->Phone: StartMediaTransmission (Request)
Phone->Srv: StartMediaTransmissionAck

Under normal circumstances, OpenReceiveChannelAck handles the calling of StartMediaTransmission.
In an addition last week i accidentally changed this behaviour slightly during call_answer.
Pushed a fixed just now.
This is what i was a little affraid of, and it requires quite a bit of testing on both sides to make sure it is right.

dkgroot, 2019-01-01:
I have done some more tests... And come to the conclusion that 9fa377e57 was not such a good idea. I have reverted that change and committed it.
I think issues/438_1 should now be working for you without the phones locking up after a number of calls.

dkgroot, 2019-01-03:
Rebased on "develop" branch (which already contains a partial fix to the earlyrtp issue)

dkgroot, 2019-01-05:
@rolakron Reported back that "develop" seems to be ok now. Calling the issue closed. The issue will be closed later on, when i have checked if anything can be salvaged from the issue/438_1 branch.

anutator, 2019-10-07:
I also have problems and don't hear ringing tone. When I set earlyrtp=dial I can hear ringing tone while calling to sccp phones but I don'hear tone while calling to Queue with sccp phones an calling mobile numbers. When I set earlyrtp=progress I don't hear ringing tone while calling to sccp phone (wired or wifi, doesn't matterr).

dkgroot, 2019-10-07:
earlyrtp and what it does:
- It will start the incoming rtp flow at the first indication set by earlyrtp=xxxx. So in the cause of the default (ie: progress) that will happen between after starting the dial out and before the call is connected. This way (if everything is working correctly) we should be able to hear any dialtone, ringtone and or announcements produced by asterisk or the remote side.
When the remote side does not produce any sound (simulated or real), then chan-sccp should be producing similated sounds instead (this might be the part that is currently broken, if it is at all).

1. earlyrtp should always be set to the default value (ie: removed from your sccp.conf), the current value would be progress, unless you really know what you are doing.
2. If you are using freepbx, it's dialplan might be interfering with our assumptions.

Analyzing earlyrtp can be a little bit difficult, but a good start would be to set "rtp set debug on" for a single call, to see when the first rtp packages are arriving.

I will run a couple of tests tomorrow to see if dialing a queue on my (vanila) asterisk system, does what it is supposed to to.

anutator, 2019-10-08:
Thank you.
Queue is not a problem, I add Answer().
For other calls I changed earlyrtp.

dkgroot, 2019-10-08:
@anutator When i call i Queue, i normally set Progress() before it, which i think is a better indication to use in this case, but Answer() should also work.

I don't know which chan-driver you use to call mobile. I retested calling chan_sip / chan_pjsip and get earlyrtp + ringing in both cases. If you are using some other channel driver and it does not support earlyrtp, you can set Progress() before the Dial, to indicate we are making progress towards the final destination. (Normally the channel_driver we are calling, should indicate 'progress' or 'proceed' to chan-sccp, so that we can initiate earlyrtp).

dkgroot, 2019-10-08:
@anutator 
https://wiki.asterisk.org/wiki/display/AST/Early+Media+and+the+Progress+Application
This clarifies that Queue() should have Progress() before it, as different phones will behave differently without it.

dkgroot, 2019-10-08:
Can you show me the call to your mobile device with 'sccp debug core, channel, pbx, codec, indicate' set, maybe there is something in there.

dkgroot, 2019-10-10:
@anutator 
Can you show me the call to your mobile device with 'sccp debug core, channel, pbx, codec, indicate' set, maybe there is something in there.

dkgroot, 2019-10-16:
@anutator

Can you please set:
- core set verbose 4
- core set debug 4
- pjsip set logger on
- sccp debug core, channel, pbx, codec, indicate'

maybe there is something in there that will show the reason for missing ringing sounds. I would like to figure out why there is no inband ringing sound.

anutator, 2019-10-16:
Hi!
I made tests with earlyrtp=progress.
If I delete option **r** from Dial application, everything is working:

> r( tone ) - Default: Indicate ringing to the calling party, even if the called party isn't actually ringing. Pass no audio to the calling party until the called channel has answered.
>     **tone - Indicate progress to calling party. Send audio 'tone' from the indications.conf tonezone currently in use**.

The option R has no influence, I hear progress tone with this option and without this option:

> R - Default: Indicate ringing to the calling party, even if the called party isn't actually ringing. Allow interruption of the ringback if early media is received on the channel.

I attached file with option r when I don't hear progress
`Dial("SCCP/4416-00000038", "PJSIP/89160239103@Beeline,,TtrU(record^4416^989160239103)")`

So I will remove option **r** and use earlyrtp=progress. Thank you.

[call-with-option-r.txt](https://github.com/chan-sccp/chan-sccp/files/3732974/call-with-option-r.txt)

dkgroot, 2019-10-16:
@anutator 
Thanks for including the Flag Descriptions. Re-reading the flag description, it looks like 'R' is exactly what we want in this case. We want simulated ringing as long as there is no inband (earlyrtp) yet. If/when earlyrtp is coming in, we want to stop simulated ringing and get the actual audio. 

'R' is what you want. And when that works, we are all ok.

I will look at the debug log output, to make sure the events are coming in the right order.

Please do check the 'Tt' options carefull, and make sure it makes sense to use them. For example on chan-sccp they are not really used a lot, because most people will just use the softkey on the phone instead of the feature *3 keys.
Using 'Tt' options on inbound calls is mostly a bad idea, as it could allow inbound callers to the IVR, to jump out and tranfer themselves to an international/outbound call.

dkgroot, 2019-10-16:
In line 269 the log says: 
>    -- PJSIP/Beeline-00000024 is making progress passing it to SCCP/4416-00000038
But chan-sccp never seems to get that message. I am going to look at the pjsip source to see what is going on there. Without a notification of 'Progress' we are never going to be starting actual earlyrtp at 'progress'.

anutator, 2019-10-16:
> 
> 
> While reading the code, 'r' is prevent the Progress indication from being delivered. Can you send me the same debug log with only 'r' removed, so keeping the 'R'.

I removed R option also - it influences calls to other sccp phones but doesn't influence calls to pjsip trunk.
In log my call from sccp 4416 to sccp 1127 phone.
[call-with-option-R-to-SCCP.txt](https://github.com/chan-sccp/chan-sccp/files/3733085/call-with-option-R-to-SCCP.txt)

dkgroot, 2019-10-16:
You can and maybe should keep the 'R' Flag. I am interested in a call from SCCP to PJSIP in the debug log, so i can see if 'Progress' is being delived this time :-)

For inbound calls to sccp phones the 'R' Flag should not make much of a difference (I think).

anutator, 2019-10-16:
Call with option R to PJSIP trunk (hear progress tone).
`Dial("SCCP/4416-0000005A", "PJSIP/89160239103@Beeline,,TtR
[call-with-option-R-to-PJSIP-trunk.txt](https://github.com/chan-sccp/chan-sccp/files/3733101/call-with-option-R-to-PJSIP-trunk.txt)
U(record^4416^989160239103)")`

dkgroot, 2019-10-16:
Nice: With the 'R' option in place. We see:
```
    -- PJSIP/Beeline-00000043 is making progress passing it to SCCP/4416-0000005A
    -- SCCP: (pbx_indicate) start indicate 'AST_CONTROL_PROGRESS: Indicate PROGRESS'
    -- SCCP/4416-0000005A: (pbx_indicate) start indicate 'AST_CONTROL_PROGRESS: Indicate PROGRESS'
 SEP04C5A4B08D1B: (pbx_indicate) start indicate 'AST_CONTROL_PROGRESS: Indicate PROGRESS' (14) condition on channel SCCP/4416-0000005A (rtp_instance:yes, reception.state:2/G.711 A-law 64k, transmission.state:2/G.711 A-law 64k)
    -- SEP04C5A4B08D1B: Indicate SCCP new state:PROGRESS, current channel state:RINGOUT_ALERTING on call:SCCP/4416-0000005A, lineInstance:1 (previous channelstate:RINGOUT)
    -- SCCP: Asterisk requests to change state from (RINGOUT_ALERTING) to (PROGRESS)
```
So Dial( ) passes the 'Progress' indication to chan-sccp. We check the rtp receptionstate to make sure it is open. And switch from simulated ringing to actual .. Perfect.

anutator, 2019-10-16:
New test - call to PJSIP trunk when phone is busy (earlyrtp=progress, R and r are deleted). We don't hear any busy tone, call is disconnected and we don't understand if it was the error or the phone was busy:
```
SEP04C5A4B08D1B: (sccp_pbx_softswitch) channel SCCP/4416-00000061 is dialing number 8405333
    -- SCCP/4416-00000061: (pbx_pbx_start) autoloop has started, set requestHangup = requestQueueHangup
 SEP04C5A4B08D1B: (sccp_pbx_softswitch) pbx started
    -- Executing [8405333@operators:1] Set("SCCP/4416-00000061", "CALLERID(num)=8914416") in new stack
 SEP04C5A4B08D1B: (sccp_pbx_softswitch) finished
 SCCP: (sccp_pbx_softswitch) exit
    -- Executing [8405333@operators:2] Dial("SCCP/4416-00000061", "PJSIP/8405333@smolensk,,TtU(record^8914416^8405333)") in new stack
    -- Called PJSIP/8405333@smolensk
    -- SCCP: (pbx_indicate) start indicate 'AST_CONTROL_CONNECTED_LINE: Indicate connected line has changed'
    -- SCCP/4416-00000061: (pbx_indicate) start indicate 'AST_CONTROL_CONNECTED_LINE: Indicate connected line has changed'
 SEP04C5A4B08D1B: (pbx_indicate) start indicate 'AST_CONTROL_CONNECTED_LINE: Indicate connected line has changed' (22) condition on channel SCCP/4416-00000061 (rtp_instance:yes, reception.state:0/No codec, transmission.state:0/No codec)
    -- SCCP: Connected Line
    -- OUTBOUND: SCCP/4416-00000061 call Got connected line update, connected.id.number=8405333, connected.id.name=(nil), source=answer
    -- SEP04C5A4B08D1B: send callInfo on SCCP/4416-00000061 with lineInstance: 1
  == SEP04C5A4B08D1B: (pbx_indicate) finish: send indication (res:0)
    -- SCCP: (pbx_indicate) start indicate 'AST_CONTROL_PVT_CAUSE_CODE: Contains an update to the protocol-specific cause-code stored for branching dials'
    -- SCCP/4416-00000061: (pbx_indicate) start indicate 'AST_CONTROL_PVT_CAUSE_CODE: Contains an update to the protocol-specific cause-code stored for branching dials'
 SEP04C5A4B08D1B: (pbx_indicate) start indicate 'AST_CONTROL_PVT_CAUSE_CODE: Contains an update to the protocol-specific cause-code stored for branching dials' (33) condition on channel SCCP/4416-00000061 (rtp_instance:yes, reception.state:0/No codec, transmission.state:0/No codec)
    -- SCCP/4416-00000061: hangup cause set: 0
  == SEP04C5A4B08D1B: (pbx_indicate) finish: send indication (res:0)
    -- SCCP: (pbx_indicate) start indicate 'AST_CONTROL_PVT_CAUSE_CODE: Contains an update to the protocol-specific cause-code stored for branching dials'
    -- SCCP/4416-00000061: (pbx_indicate) start indicate 'AST_CONTROL_PVT_CAUSE_CODE: Contains an update to the protocol-specific cause-code stored for branching dials'
 SEP04C5A4B08D1B: (pbx_indicate) start indicate 'AST_CONTROL_PVT_CAUSE_CODE: Contains an update to the protocol-specific cause-code stored for branching dials' (33) condition on channel SCCP/4416-00000061 (rtp_instance:yes, reception.state:0/No codec, transmission.state:0/No codec)
    -- SCCP/4416-00000061: hangup cause set: 0
  == SEP04C5A4B08D1B: (pbx_indicate) finish: send indication (res:0)
  == Everyone is busy/congested at this time (1:1/0/0)
    -- Executing [8405333@operators:3] Hangup("SCCP/4416-00000061", "") in new stack
```

[11 more replies]

commit 91314821 (2018-12-24): Issue: No audio when dialplan answers the call immediately using Answer();
- Run updateChannelCapability if writeFormat is still CODEC_NONE when running StartMediaTransmission
- Make sure OpenReceiveChannel is run before StartMediaTransmission when the call is immediately answered
through the dialplan (ie: without Progress/Proceed preceeding it).

Partially related improvements:
- Issue Warning if StartMediaTransmission is requested before OpenReceiveChannel
- Renamed sccp_pbx_answered to sccp_pbx_answer
- Issue Progress Indication to remote side just before handling sccp_pbx_answer

fixes #43 [... cut]
commit fd0ec30d (2019-01-03): Fix sccp_pbx_answer to make sure StartMediaTransmission does not overtake OpenReceiveChannel

fixes #438
commit b72d0509 (2019-10-23): Fix inband ringing on outbound call
Move stoptone to sccp_channel_openReceiveChannel

Fixes #438
closed 2019-10-23 by dkgroot