chan-sccp/chan-sccp #440: error messages for each incoming call

hhornung, 2018-12-31:
with the newest trunk I receive the following messages for each incoming call:
2018-12-31 13:24:11] ERROR[11660][C-0000000b] translate.c: Cannot determine best translation path since one capability supports no formats
[2018-12-31 13:24:11] WARNING[11660][C-0000000b] ast113.c: SCCP/sccp-00000003: Zero Native Codecs[2018-12-31 13:24:11] ERROR[11660][C-0000000b] astobj2.c: FRACK!, Failed assertion user_data is NULL (0)
[2018-12-31 13:24:11] ERROR[11660][C-0000000b] : Got -3 backtrace records

Is it a bug or how can I avoid this?
asterisk-version is 13.24.1

dkgroot, 2018-12-31:
- What was the previous revision you were using, before this one?
- I need a little bit more debug (context) to see where/when this is happening, can you collect some debug log (for one of these calls) with "sccp debug core, channel, pbx, codec, rtp" set.
- Can you try the "issues/438_1" branch to see if that resolves the issue (it is close to being merged). 
```
git pull
git fetch -a
git checkout issues/438_1
make && make reload
```

hhornung, 2018-12-31:
I have installed 438_1 but the problem is the same:
[2018-12-31 16:07:28] ERROR[27085][C-00000012]: translate.c:1384 ast_translator_best_choice: Cannot determine best translation path since one capability supports no formats
[2018-12-31 16:07:28] WARNING[27085][C-00000012]: ast113.c:1193 sccp_wrapper_asterisk113_allocPBXChannel: SCCP/sccp-00000001: Zero Native Codecs[2018-12-31 16:07:28] ERROR[27085][C-00000012]: astobj2.c:126 INTERNAL_OBJ: FRACK!, Failed assertion user_data is NULL (0)
[2018-12-31 16:07:28] ERROR[27085][C-00000012]:   Got -3 backtrace records

last working tree (for me without this error) was a7bca36 (Dec. 14th)

dkgroot, 2018-12-31:
OOh boy, that was just before #420 got merge (which i was a little afraid of). Please give me some extra info:
- is this a SCCP->SCCP call or does it involve another channel driver (if so, which one).
- What are your allow/disallow settings for the two sides ?
- I need the debug log for a single call, with "sccp debug core, channel, pbx, codec, rtp" to figure out the flow of events
Looking at the error location, i should be able to figure it out.

dkgroot, 2018-12-31:
[hhornung debug log](https://github.com/chan-sccp/chan-sccp/commit/05a12e6b324445a50d482b435b5eed12665ec8fc#commitcomment-31809373)
I see where it is going wrong, not 100% sure yet how. But it looks quite intriguing.

dkgroot, 2019-01-01:
- Found the comma (",") issue, which was just a typo.
- Fixed the issue where multiple lines are dialed at once, including an SCCP line that is currently not connected to a device (We should give up early on this one, continuing the probe).
- Fixed the issue reported in #441 where codec selection failed because of reduced preferences: returned "[]"
This should hopefully fix the fallout from merging 420_2

Created a new branch called "issues/440", please verify and report back. 
```
git pull
git fetch -a
git checkout issues/440
make && make reload
```

hhornung, 2019-01-01:
I can confirm that the problem (#440) is fixed now :)
I wish you a happy new year and all the best!
ciao
Holger

zipiju, 2019-01-01:
I can confirm this fixed the one-way audio as described in issues/441.
Calling both ways with or without transcoding is now working, I've tried:
- SIP alaw to SCCP (C7965) g722 and the other way
- SCCP g722 to SCCP g722
- SIP g722 to SCCP g722 and the other way
- SIP alaw trunk to SCCP g722 and the other way

Thank you!

dkgroot, 2019-01-01:
Thank you guys, for testing this. I will merge it into develop (Which will dissolve the issues/440 branch) !

closed 2019-01-01 by dkgroot

dkgroot, 2019-01-01:
BTW: If you require earlyrtp=progress to work, you should have a look at issues/438_1, this will land in the "develop" branch shortly, when it has been tested / reported to work.

commit 58a26fb5 (2019-01-01): Fix calling multiple lines at the same time (DIAL(SIP/123&SCCP/432)), whereby the SCCP line is currently not bound to a device
Make sure caps is always allocated in allocPbx, even in the event that
c->preferences.audio could not be determined.
Fixes:
 - Correct codec Selection
 - Fallout from merging #420_2
 - Continuing without a valid nativeformats (returning FALSE if all codec selection failed)
 - AstObj2Error: INTERNAL OBJ: Frack

fixes #440
fixes #441
commit 747df2d7 (2019-01-01): Fix calling multiple lines at the same time (DIAL(SIP/123&SCCP/432)), whereby the SCCP line is currently not bound to a device
Make sure caps is always allocated in allocPbx, even in the event that
c->preferences.audio could not be determined.
Fixes:
 - Correct codec Selection
 - Fallout from merging #420_2
 - Continuing without a valid nativeformats (returning FALSE if all codec selection failed)
 - AstObj2Error: INTERNAL OBJ: Frack

fixes #440
fixes #441
commit d0708a63 (2019-01-01): Fix calling multiple lines at the same time (DIAL(SIP/123&SCCP/432)), whereby the SCCP line is currently not bound to a device
Make sure caps is always allocated in allocPbx, even in the event that
c->preferences.audio could not be determined.
Fixes:
 - Correct codec Selection
 - Fallout from merging #420_2
 - Continuing without a valid nativeformats (returning FALSE if all codec selection failed)
 - AstObj2Error: INTERNAL OBJ: Frack

fixes #440
fixes #441