chan-sccp/chan-sccp #441: One way audio when call originates at SIP end (similiar to #420)

zipiju, 2018-12-31:
Hello Diederik,

With the latest commit 05a12e6 there is one way audio when the call originates on alaw SIP end (alaw SIP end can hear SCCP g722 end but SCCP g722 end cannot hear alaw SIP end).
Calling from SCCP g722 to alaw SIP works ok.
Asterisk is showing two different codecs on one end and partial (one-way) translation on the other end, please see below.

I was on issues/420_1 branch for a while where audio between SIP and SCCP worked okay.

If there is something else you need please let me know.
Thank you.

One way audio when calling SIP->SCCP:
```
serenity*CLI> core show channels
Channel              Location             State   Application(Data)             
SIP/209-00000002     201@dial-ext:2       Up      Dial(SCCP/201,60,htwk)        
SCCP/201-00000003    (None)               Up      AppDial((Outgoing Line))      
2 active channels
1 active call
3 calls processed

serenity*CLI> core show channel SCCP/201-00000003
 -- General --
           Name: SCCP/201-00000003
           Type: SCCP
       UniqueID: 1546273729.7
       LinkedID: 1546273729.6
      Caller ID: 201
 Caller ID Name: Jojko
Connected Line ID: 209
Connected Line ID Name: Jojko mobil
Eff. Connected Line ID: 209
Eff. Connected Line ID Name: Jojko mobil
    DNID Digits: (N/A)
       Language: en
          State: Up (6)
  NativeFormats: (g723|ulaw|alaw|gsm|g726|g726aal2|adpcm|slin|slin|slin|slin|slin|slin|slin|slin|slin|lpc10|g729|speex|speex|speex|ilbc|g722|siren7|siren14|testlaw|g719|opus|silk|silk|silk|silk|h261|h263|h263p|h264|mpeg4|vp8|vp9)
    WriteFormat: alaw
     ReadFormat: g722
 WriteTranscode: No 
  ReadTranscode: No 
 Time to Hangup: 0
   Elapsed Time: 0h0m17s
      Bridge ID: 7761bfd0-1524-432f-9518-01badeb9e11f
 --   PBX   --
        Context: sip
      Extension: 
       Priority: 1
     Call Group: 26
   Pickup Group: 58
    Application: AppDial
           Data: (Outgoing Line)
 Call Identifer: [C-00000002]
      Variables:
BRIDGEPVTCALLID=5db127a6435ebe6430c29936e14101f8@192.168.0.50
BRIDGEPEER=SIP/209-00000002
DIALEDPEERNUMBER=201
SCCP_DEVICE_TYPE=Cisco 7965
SCCP_DEVICE_IP=192.168.0.135
SCCP_DEVICE_MAC=SEP002414B3D5F7
  CDR Variables:
level 1: clid="Jojko" <201>
level 1: src=201
level 1: dcontext=sip
level 1: channel=SCCP/201-00000003
level 1: lastapp=AppDial
level 1: lastdata=(Outgoing Line)
level 1: start=1546273729.807541
level 1: answer=1546273731.447608
level 1: end=1546273731.448471
[... 327 lines cut ...]
+--- SpeeddialButtons -------------------------------------------------------------------------------------------------+
| Id   Name                    Number                 Hint                                                             |
+ ==== ======================= ====================== ================================================================ +
| 4    Bridge 750              750                    1@blf                                                            |
| 5    Parking 701             701                    701@parkedcalls                                                  |
| 6    Parking 702             702                    702@parkedcalls                                                  |
+----------------------------------------------------------------------------------------------------------------------+

+--- FeatureButtons ---------------------------------------------------------------------------------------------------+
| Id   Name                    Status                 Options                                                          |
+ ==== ======================= ====================== ================================================================ +
+----------------------------------------------------------------------------------------------------------------------+

+--- ServiceURLButtons ------------------------------------------------------------------------------------------------+
| Id   Name                   URL                                                                                      |
+ ==== ====================== ======================================================================================== +
+----------------------------------------------------------------------------------------------------------------------+

+--- CallStatistics ---------------------------------------------------------------------------------------------------+
| Type     Calls    PcktSnt  PcktRcvd Lost     Jitter   Latency  Quality  avgQual  meanQual maxQual  rConceal sConceal |
+ ======== ======== ======== ======== ======== ======== ======== ======== ======== ======== ======== ======== ======== +
| LAST     7        397      188      0        1        0        0.000000 0.000000 0.000000 0.000000 0.018300 2        |
| AVG      7        5144     2725     0        1        0        1.813943 1.864014 1.675714 4.500000 0.012043 7        |
+----------------------------------------------------------------------------------------------------------------------+
```

dkgroot, 2018-12-31:
@zipiju 
Does issues/420_1 still work in this scenario ? If so i will check the difference between 420_1 and 420_2. Maybe i made a mistake merging the 420_2 branch instead. Hopefully the difference between branches are not to big.
I will try to check it out in the coming 1-2 days.

dkgroot, 2018-12-31:
Maybe you can also include the logging for a single call setup that fails (sccp debug core, channel, pbx, rtp, codec) i think that would help, once i get around to looking at it (1-2 days)

dkgroot, 2018-12-31:
Looking at the first bit of the debug log, this is the same problem reported in #440
I will focus on this first bit, to see how to resolve the issue.

If need be you can go back to revision:22e7f18fca, just before the 420_2 tree was merged in.

commit 58a26fb5 (2019-01-01): Fix calling multiple lines at the same time (DIAL(SIP/123&SCCP/432)), whereby the SCCP line is currently not bound to a device
Make sure caps is always allocated in allocPbx, even in the event that
c->preferences.audio could not be determined.
Fixes:
 - Correct codec Selection
 - Fallout from merging #420_2
 - Continuing without a valid nativeformats (returning FALSE if all codec selection failed)
 - AstObj2Error: INTERNAL OBJ: Frack

fixes #440
fixes #441
commit 747df2d7 (2019-01-01): Fix calling multiple lines at the same time (DIAL(SIP/123&SCCP/432)), whereby the SCCP line is currently not bound to a device
Make sure caps is always allocated in allocPbx, even in the event that
c->preferences.audio could not be determined.
Fixes:
 - Correct codec Selection
 - Fallout from merging #420_2
 - Continuing without a valid nativeformats (returning FALSE if all codec selection failed)
 - AstObj2Error: INTERNAL OBJ: Frack

fixes #440
fixes #441
commit d0708a63 (2019-01-01): Fix calling multiple lines at the same time (DIAL(SIP/123&SCCP/432)), whereby the SCCP line is currently not bound to a device
Make sure caps is always allocated in allocPbx, even in the event that
c->preferences.audio could not be determined.
Fixes:
 - Correct codec Selection
 - Fallout from merging #420_2
 - Continuing without a valid nativeformats (returning FALSE if all codec selection failed)
 - AstObj2Error: INTERNAL OBJ: Frack

fixes #440
fixes #441
closed 2018-12-31 by dkgroot