chan-sccp/chan-sccp #442: Missing voicemail updates from stasis

mscdex, 2019-01-03:
I'm not sure if there is something that can be done at the chan-sccp level for this or not, but we recently upgraded from Asterisk 11.23.0 to 16.1.1 and we noticed that MWI was no longer being updated. After some debugging, it seems that the stasis subscriptions for mailboxes are only sometimes notified. For now, I've had to explicitly disable stasis support in config.h and use polling with a short interval as a temporary workaround.

I also tried executing the VoicemailRefresh action via AMI but that did not seem to help any.

**Reproduction Steps:**

1. Call extension handled by chan-sccp and leave a new voicemail; or dial voicemail from an extension handled by chan-sccp and delete all new voicemails from the ivr

**chan-sccp version:** develop@05a12e6b324445a50d482b435b5eed12665ec8fc

**asterisk version:** 16.1.1

**Expected behavior:**

MWI updated on affected devices

**Observed behavior:**

No MWI updates or only occasionally

**Screenshots**

**Addition / Optional Information:**
- Problem started happening recently, didn't happen in an older version of Chan-SCCP: Can not test with older chan-sccp versions since Asterisk 16 support was only added very recently
- Problem can be reliably reproduced, doesn't happen randomly: Not Sure
- Platform:
  - Intel
  - 64 Bit
  - Small Endian
  - Ubuntu
  - 18.04.1
- Direct Contact information: [You can contact me here]
- Log Files (attach .txt / [pastbin](http://pastebin.com/) / [gist](http://gist.github.com/)):
- Comments: I was testing with a couple of 7970 phones, but I don't think it matters for this particular issue

mscdex, 2019-01-04:
I just compiled 353be6d7cb0cd7e6c63b9472a8c6a07b84afc054 and loaded it and I'm still seeing the same behavior unfortunately. Another extension calls, leaves a voicemail, MWI light comes on, dial VM, delete message, hang up, MWI light stays on.

mscdex, 2019-01-04:
After more digging it's interesting that AMI connections see the MessageWaiting event when the voicemail is deleted.

dkgroot, 2019-01-04:
To be honest before "fixing" the issue, i did not even check if i could reproduce the issue on my system.
I however was able to leave a voicemail, see the mwi light come on, then pressed the voicemail button, and listenend to comedian mail, pressed 7 to delete. hungup and the light went off. I only did two tests, but it looked like it worked and i was pleased with myself :-) Apparently to early.
Can you set "sccp debug core, device, line, channel, mwi" and repeat your actions please. Maybe there is something in there that will help me figure this out.
BTW: We do not hook into the AMI message stream to get this event notifications, they come over a statis callback event instead.
I have a feelling (based on the work you did when reporting the issue) you are quite capable of helping out with the debugging, with a couple of well place debug print statements in sccp_mwi (correct ?).

dkgroot, 2019-01-04:
Can you perform a "sccp show mwi subscriptions" before leaving a voicemail, after leaving a voicemail and after deleting the voicemail. That might show that we do get the notification and handle it, but somehow skip switching off the light. Or that we don't get the deletion event at all (fyi there is no deletion event perse, we just get an update for the new/old vm count, and should adjust accordingly).

mscdex, 2019-01-04:
> Can you set "sccp debug core, device, line, channel, mwi" and repeat your actions please. Maybe there is something in there that will help me figure this out.

I will get it sometime tomorrow.

> BTW: We do not hook actually into the AMI message stream to get this event notifications, they come over a statis callback event instead.

Right, I wasn't sure where the problem lies, but acknowledging that AMI sees the event (it subscribes to the same `ast_mwi_state_type()`, although in a different way) lets us know that the event is indeed being sent out.

> I have a feelling (based on the work you did when reporting the issue) you are quite capable of helping out with the debugging, with a couple of well place debug print statements in sccp_mwi (correct ?).

I've checked and `sccp_mwi_event()` only gets called the one time, after the voicemail is left. After that it never gets called again. I tried various things, like checking that the subscription was still active inside the callback (it is) as well as forcibly re-subscribing within the callback (doesn't help).

This is all just guesswork on my end as I'm not familiar at all with Asterisk internals/C API.

mscdex, 2019-01-04:
> You said you only "receive" the first event. It could be that you mean during device registration

The device is already registered a fair amount of time before I leave the voicemail message, so I am pretty sure the event emitted after the voicemail is left is not related to the sccp device registration event.

> This might indicate that there is a mismatch between your voicemail.conf and the settings in sccp.conf (regarding mailbox and mailbox context).

When we upgraded from 11 to 16, we did not touch our /etc/asterisk/*.conf files at all, although we did also update FreePBX (from 13 to 14), which may have made some changes on its own (to its own separate config files). As far as the mailbox contexts go, those are all correct and use the '@default' suffix. Voicemail itself works just fine, it's just the MWI that's not working. Additionally `voicemail show users` in the console shows the correct count after deleting a message.

dkgroot, 2019-01-04:
I guess we will have to wait, until you have time to gather the sccp show output and the debug log and hope we will figure it out. Updating to things at the same time, always has the potential to bring surprises. I just copied the voicemail.conf & sccp.conf from asterisk-13 over, so zero change there :-)

mscdex, 2019-01-05:
Here is the information from develop@c1ff984 and Asterisk 16.1.1:

BEFORE VOICEMAIL LEFT
======================

```
asterisk*CLI> sccp show mwi subscriptions

+--- MWISubscriptions --------------------------------------------------+
| Mailbox    LineName                       Context         New Old Sub |
+ ========== ============================== =============== === === === +
| 944        944                            default         000 000 YES |
+-----------------------------------------------------------------------+

asterisk*CLI> voicemail show users
Context    Mbox  User                      Zone       NewMsg
default    944   Testing                                   0
```

VOICEMAIL BEING LEFT
====================

```
    -- SCCP: Asterisk asked us to create a channel with type=SCCP, format=(g722), lineName=944, options=
 SCCP: Adding channel 1 to line 944
    -- 1005: New channel number: 1 on line 944
    -- SCCP: (pbx_channel_allocate) try to allocate Inbound channel on line: 944
    -- SCCP:              cid_num: "944"
    -- SCCP:             cid_name: "Testing"
    -- SCCP:          accountcode: "testing"
    -- SCCP:                exten: ""
    -- SCCP:              context: "from-internal-xfer"
    -- SCCP:             amaflags: "0"
    -- SCCP:            chan/call: "SCCP/944-00000001"
    -- SCCP:combined capabilities: "[slin16, g722/64k, ulaw/64k, alaw/64k, g722/56k, g722/48k, g729b, g729ab, g729, g729a]"
    -- SCCP:  reduced preferences: "[g722/64k, g722/56k, g722/48k, ulaw/64k, alaw/64k, g729, g729a, g729b, g729ab]"
    -- SCCP: (allocPBXChannel) Create New Channel with name: SCCP/944-00000001
    -- 1005: Allocated asterisk channel SCCP/944-00000001
 SCCP: (asterisk116_get_rtp_info) Asterisk requested EarlyRTP peer for channel SCCP/944-00000001
 SCCP: (asterisk116_get_vrtp_info) Asterisk requested EarlyRTP peer for channel SCCP/944-00000001
    -- SCCP: Asterisk request to call SCCP/944-00000001 (dest:944, timeout: 0)
    -- 944: Asterisk request to call SCCP/944-00000001
 SCCP/944-00000001: 0x7fcdd0050030: (getCallInfoStr):
 - calledParty: Testing <944>, valid
 - callingParty:  <2802>, valid
 - presentation: CallerId Presentation Allowed

    -- SEP000102030405: Getting the active channel on device.
    -- SEP000102030405: No active channel on device.
    -- SEP000102030405: Indicate SCCP new state:RINGING, current channel state:DOWN on call:SCCP/944-00000001, lineInstance:1 (previous channelstate:DOWN)
    -- SEP000102030405: Clear the display notify message
    -- SEP000102030405: Clear the status prompt on line 1 and callid 0
    -- SEP000102030405: Send and Set the call state ring-in(4) on call 1 (visibility:default)
    -- SEP000102030405: Found matching linedevice. Aux parameter = 
    -- SEP000102030405: Send ringer mode Outside(3) on device
    -- SEP000102030405: Normal ring occurred.
    -- SEP000102030405: Set softkeyset to RINGIN(3) on line 1  and call 1
    -- SEP000102030405: validKeyMask 4294967295
    -- SEP000102030405: ( sccp_indicate.c:252:__sccp_indicate ) sccp_dev_displayprompt '�2802' for line 1 (4)
    -- SEP000102030405: Display prompt on line 1, callid 1, timeout 4
  == SCCP: (mwi_lineStatusChangedEvent) Get lineStatusChangedEvent
    -- SEP000102030405: (mwi_setMWILineStatus) instance: 1, mwilight:0, mask:0000000000000000000000000000001 (2)
    -- SEP000102030405: (mwi_setMWILineStatus) state: 0. status:0000000000000000000000000000000 (0) 
    -- SEP000102030405: (mwi_setMWILineStatus) Device already knows this state OFF on line 944 (1). skipping update
    -- SEP000102030405: Checking Line Instance: 1 = 944, mwioncall: Off
    -- SEP000102030405: (mwi_check) line 944 voicemail count 0 new/0 old (instance: 1)
    -- SEP000102030405: (mwi_check) Remove Have Voicemail from Display
       > SEP000102030405: clear message stack 1
    -- SEP000102030405: Finish to indicate channel state:RINGING on call:SCCP/944-00000001, lineInstance:1. New channel state:RINGING
    -- SCCP: Set asterisk state Ringing (5) for call 1
[... 256 lines cut ...]
    -- SEP000102030405: Call Statistics:
       [
         Last Call        : CallID: 2 Packets sent: 657 rcvd: 584 lost: 0 jitter: 0 latency: 0
         Last Quality     : MLQK=4.2597;MLQKav=4.2443;MLQKmn=4.2290;MLQKmx=4.2597;MLQKvr=0.95|ICR=0.0000;CCR=0.0051;ICRmx=0.0100|CS=4;SCS=0
         Mean Statistics  : #Calls: 1 Packets sent: 657 rcvd: 584 lost: 0 jitter: 0 latency: 0
         Mean Quality     : MLQK=4.2597;MLQKav=4.2443;MLQKmn=4.2290;MLQKmx=4.2597;MLQKvr=0.95|ICR=0.0000;CCR=0.0051;ICRmx=0.0100|CS=4;SCS=0
       ]
```

AFTER VOICEMAIL DELETE
=======================

```
asterisk*CLI> sccp show mwi subscriptions

+--- MWISubscriptions --------------------------------------------------+
| Mailbox    LineName                       Context         New Old Sub |
+ ========== ============================== =============== === === === +
| 944        944                            default         001 000 YES |
+-----------------------------------------------------------------------+

asterisk*CLI> voicemail show users
Context    Mbox  User                      Zone       NewMsg
default    944   Testing                                   0
```

mscdex, 2019-01-05:
> FYI: If this system is running in production and does require mwi to work right now, we should be able to to create a simple patch that does a forced voicemail poll during device hangup if necessary (This is reminicent of the the way we did this in asterisk-1.6 :-))

It is a production installation. I noticed in one of the other channel drivers they do check the stasis cache first before calling out to the `inboxcount()` API. I'm assuming because `inboxcount()` hits the filesystem and is less efficient? Perhaps it would be a good idea to implement this in chan-sccp as well for when `inboxcount()` is called?

dkgroot, 2019-01-05:
@mscdex
We currently support the asterisk-1.6/asterisk-1.8 method using inboxcount (ie: polling), asterisk-11 HAS_EVENT method and the >asterisk-13 HAS_STASIS methods. We only do inboxcount if HAS_EVENT and HAS_STASIS are not available.

Can you do "manager set debug on" and leave + remove another voicemail. The AMI event should be inserted into the same debug stream so we can see that they are being handled (and when). I think we might be filtering out some of the mailbox/mwi events by accident.

For your production system a temporary fix would be to insert an pbx_app_inboxcount in sccp_mwi_lineStatusChangedEvent(), that would mean we check a little bit often, but it would work. Let me know if you require a temporary patch for this, until we figure the other part out.

dkgroot, 2019-01-09:
@mscdex Maybe you can just attach an excerpt of your sccp.conf & voicemail.conf so i can recreate the issue on my local development system, that might speed things up a little.

mscdex, 2019-01-09:
@dkgroot Here are the "full" files (only other extensions/devices removed):

[sccp.conf.txt](https://github.com/chan-sccp/chan-sccp/files/2742023/sccp.conf.txt)
[voicemail.conf.txt](https://github.com/chan-sccp/chan-sccp/files/2742024/voicemail.conf.txt)

dkgroot, 2019-01-09:
Using your voicemail.conf i can reproduce the issue. Looking into it now

Note: when loading your config a couple of small errors/warnings are displayed (like maxsilence > minsecs for example), but this should not be the cause.

dkgroot, 2019-01-10:
When the line is first created we subscribe to the stasis_event for mwi. And we even get the first event, but all subsequent events never seem to reach us. I still have not been able to figure out what is wrong. Several guesses:
 - Do we forget to flush the queue of event (doesn't look like it)
 - Are we subscribing to early, before the endpoint is actually registered (sounds reasonable, difficult to test with the current source).
 - Are we filtering events out, down the line. Tried several options, an this does not seem to be the case.

I think the best way to continue, is digging a little deeper inside asterisk and set some gdb watchpoints to see what is going on.

BTW: This change has only been in the latest asterisk-13/16 heads since the last 4 months, which seems to also be the time when this stopped working. So if the gdb watchpoints is not going to help, i might have to go through the hassle of doing a git bisect on asterisk... juck.

mscdex, 2019-01-11:
I just tried commenting out everything in [general] in voicemail.conf with develop@b553664, followed by a `voicemail reload`, and the MWI still doesn't go off after deleting a message. Very strange....

dkgroot, 2019-01-11:
@mscdex 
I finally got it back into a working condition (tested under asterisk-13 for now, will run asterisk-16 tests later). Quite a bit of work was done (some functional, some cleanup). Not 100% sure why it is working now, but it does.
One addition the "sccp show mwi subscriptions" out put was extension to show the asterisk subscription uuid. Which you can use for "stasis statistics show subscription [uuid]" to see if/when/how events are/were processed.
Please checkout "issues/442" and give it a go:
```
git pull
git fetch -a
git checkout issues/442
```

Question: why does asterisk add new required stuff in the middle of a minor release... grmbl...

dkgroot, 2019-01-12:
These latter commits are just some cleanup

mscdex, 2019-01-12:
I just tried issues/442@9e960fc and that seems to work with my limited testing. Thanks for working on this! I will chime in if I hear of anyone having any issues.

mscdex, 2019-01-12:
Oops, I spoke too soon. I noticed that while the handset light goes off, the indicator on the screen still shows a flashing envelope next to the line extension.

mscdex, 2019-01-12:
Also when I do a `sccp show mwi subscriptions` it still shows 1 new voicemail.

dkgroot, 2019-01-12:
I am already pretty happy that the event is arriving again...
The calculation of the number of # messages, taking shared line into account is somewhat messy (to say the least), and i might have messed one of them up while working on it.
Will have another look, thanks for the heads up !

dkgroot, 2019-01-13:
Most of the sccp_mwi.c is pretty ancient (ast-1.6.2) and shared line is complicating things quite a bit. I am thinking about an overhaul, to see if that:
- fixes the issue
- improves code readability/quality
- simplifies the implementation
It will take 1-2 days to finish though, hope you can bare with me.

mscdex, 2019-01-13:
No problem, polling with a short interval is working good enough for now.

dkgroot, 2019-01-16:
@mscdex Just noticed a small memory corruption issue still left behind, so maybe skip the testing until i have had a look and fixed it.

dkgroot, 2019-01-16:
@mscdex Found and corrected the memory issue :-). Running some more tests, to check the mwi total count is correct.

dkgroot, 2019-01-16:
@mscdex As far as i can tell the new implementation is working. Can you give it a test run ? (FYI: still in the "issues/442" branch).

darkponay, 2019-01-17:
Hi,
I have experienced the same problems with MWI. I have installed the latest version of Asterisk (16.1.1) and tried issue/442 as mentioned earlier. 

`voice14*CLI> sccp show version
Skinny Client Control Protocol (SCCP). Release: 4.3.1 issues/442 - 733492fM (built by 'root' on '2019-01-17 21:50:13 UTC')`

Here are the steps of my test :
1. Check Asterisk and SCCP MWI count => 0 in each case.
2. Call someone and leave a message.
3. The recipient phone shows an enveloppe but no headset light and "you have voicemail" until i got the phone off and on hook.
4. Call and leave a message again.
5. Recipient phone changed the count to 2/0.
6. Call the voicemail and delete the messages.
7. The phone still has the enveloppe and headset light but shows 1/0.
8. SCCP still shows 1 message when Asterisk show no message in console.

This is the console output of the test, hoping it could help you.
Thank you very much for all your incredible work !!

[https://pastebin.com/5FC55YRq](url)

darkponay, 2019-01-18:
Oh sorry, it's one of my first time with the command line. I come from XiVo/Wazo and then FreePBX (with your insanely great module). i will redo the same operations after issuing the debug command. I'll be back in 15 or 20 minutes, the time for me to set up everything.

dkgroot, 2019-01-18:
No problem. 
BTW if you are referring to the [freepbx plugin](https://github.com/PhantomVl/sccp_manager), that was created by @PhantomVI :-)
I will be asleep by then, but will checkout you report in the morning !
A broken virtual machine is not as bad as real hardware i guess. Advice: when you have a VM, make snapshots often, so you can roll back easily.
Would love to hear more about your Xivo/Wazo experience in our [chan-sccp chat room](https://gitter.im/chan-sccp/chan-sccp).

darkponay, 2019-01-18:
I'm doing the test now. It runs on Hyper-V so i have to be veeeery careful with snapshots and Linux... :D 
I do the config manually, i've never been able to make it work with the realtime db, and dont really want FreePBX, i find it more complicated than editing some config files by myself as a beginner !

darkponay, 2019-01-18:
It's done !

https://pastebin.com/6cmnug71

dkgroot, 2019-01-18:
And here is the problem:
```
    -- Destroying channel SCCP/9914-00000001
 9912: (mwi::pbx_mwi_event) uniqueid:9912@default, msgtype:ast_mwi_state_type
  == 9912: (mwi::NotifyLine) Notify newmsgs:1 oldmsgs:0
    -- 9912: (sccp_line_indicateMWI) Set voicemail lamp:on on device:SEP00082FB71729
    -- SEP00082FB71729: (sccp_device_setMWI), newmsgs:1, oldmsgs:0
    -- 9912: (sccp_line_indicateMWI) Set voicemail lamp:on on device:SEP00230433A9B9
    -- SEP00230433A9B9: (sccp_device_setMWI), newmsgs:1, oldmsgs:0
```
We update the device mwi calculations (sccp_device_setMWI), which is good, But we "forget" to call sccp_device_indicateMWI, so the result of the calculation is never shown. Should be pretty easy to fix, and i will give it a go tomorrow.
Already happy to see that the asterisk event is received correctly, and that the calculation is correct so far.

BTW: I have looked at the wazo sccp implementation, and it is very clean, (but also a little limited) and i don't think there is any further development on it (AFAIK)

darkponay, 2019-01-18:
No they told me they won't touch it anymore and they focus on their new REST APIs and web interface. They also explained me that is not required by their customers so it has a very low priority. I came to your program because i had a UC540 last year, and when you try the Shared lines, SLA and BLF, you can't go back ! I will go to sleep (i'm in France, it's almost 5am now). I came to Asterisk 16 after seeing this issue here. I was originally on the 13.24.x.
Have a good night and sorry if i made stay awake late!

dkgroot, 2019-01-18:
@mscdex / @darkponay 
There were some more recent changes in asterisk-16 in the last 7 days, involving stasis and app_voicemail. I updated / Recompiled / Recompiled chan_sccp and Ran the following Test:

- Made sure all mailboxes were empty
- Restarted Asterisk
- Left two Voicemails:
  - Phone Lamp is On
  - Line Icon shows Voicemail Icon blinking
  - StatusMessage says "You have 2/0 voicemails"
- Dialing Voicemail
  - PhoneLamp switches off during call (sccp.conf: mwioncall = off)
  - Line Icon shows Voicemail Icon blinking
- Action: Deleting one of them + Hangup
  - Line Icon shows Voicemail Icon blinking
  - StatusMessage updated to 1/0
  - After hangup the PhoneLamp is turned back on
- Dialing Voicemail
  - PhoneLamp switches off during call (sccp.conf: mwioncall = off)
- Action: Deleting second voicemail + Hangup
  - Line Icon  removed
  - StatusMessage is removed
  - After hangup the PhoneLamp is turned off
So on my system, it looks like everything is working as expected.

darkponay, 2019-01-18:
Hi,
Just tested on my yesterday's machine : https://pastebin.com/BfTKr0yE
I can't test with shared line for the moment, i only have 2 phones available.

dkgroot, 2019-01-19:
I always use [asterisk git repo](http://github.com/asterisk/asterisk directly), i guess the tar.gz may not have been updated yet. Which would mean you are still on some older revision i guess. And you are only receiving "(mwi::pbx_mwi_event)" once, but you should be getting it for each and every voicemail change.

If you could pull the latest asterisk revision from the repo directly and check if the issue goes away/remain, we know for sure. If it remains it must be an issue on the chan_sccp side. If it goes away it was a regression in asterisk. Thanks for helping out !

darkponay, 2019-01-19:
It wooooorks!
https://pastebin.com/xVCcBSzm

Two messages leaved : Lamp+enveloppe+"2/0"
one deleted : Lamp+enveloppe+"1/0"
Last deleted : no lamp, no enveloppe, and bottom line cleared.

One caveat, i tested with mwioncall=off for the test. lamp stays on while calling voicemail, but it switches off when calling a phone.

dkgroot, 2019-01-19:
@darkponay
I guess you tested with the latest asterisk repo ast-16 revision (right) ? So that means there was a "little" bug in asterisk-16 instead of chan_sccp (and all the redesigning wasn't even necessary). Still like the new code a lot better, so we will be keeping it i think.

Pushed a fix for the mwioncall issue.

mscdex, 2019-01-19:
I don't see any commits since 16.1.1 that would affect this though. For stasis it was just printf and ABI fixes and for voicemail it was simply adding a missing channel variable for urgent messages. I haven't had time lately to try asterisk 16 head or the changes in here unfortunately.

dkgroot, 2019-01-19:
As soon as @darkponay updated to latest, everything started working again. Asterisk-16 started failing when the stasis_filter method was added (about 4 months ago if i am not mistaken). Not sure if we should investigate any further(i did try to find the cause, by looking at the asterisk diffs) to find out what is really behind the issue. Doing a git bisect on asterisk+chan_sccp is a little annoying and takes quite a bit of time (especially on my development system), if you like you could give it a try.
The new sccp_mwi.c implementation looks pretty nice and clean (i did however drop the mwi_polling ability that you are currently depending on). If need be, that could be added back in before merging into develop (I kept the interface compatible to a polling implentation).

mscdex, 2019-01-20:
I have not had a chance to try out the issues/442 branch lately. I downloaded from Asterisk 16.1.1 from the Asterisk website at the time, but it should be same contents as the release tarball from their github.

dkgroot, 2019-01-20:
I downloaded and compiled 16.1.1.tgz and will do some tests with it tomorrow. Hopefully we are all good to go, if not I will figure it out. Thanks for the update !

dkgroot, 2019-01-20:
@mscdex
When running 16.1.1.tgz (from gh releases page) i sometimes got the same behaviour where events would stop arriving. I have added a fix in 3f9cfb6. Next to the regular mwi events we also receive subscribe/unsubscribe events over the same event channel (even after having set the filter). On this events i now pull information from the stasis_cache and that seems to fix the issue (please don't ask me why).
I also noticed a small bug when processing mailboxes connected to shared lines, so that also got fixed. We should be pretty much finished with this issue i think. Just waiting for you guys to run your checks/tests.

darkponay, 2019-01-21:
Hi,
just installed asterisk 16.1.1 from the tar.gz. then installed chan-sccp (develop ;-)) and checkout issues/442.

Here is the log (show version at start):
https://pastebin.com/VkRsTTbZ

Lamp, enveloppe and message stayed off all the time.

dkgroot, 2019-01-27:
@mscdex I would like to merge this new version (even if there are a small number of asterisk versions that it still does not apply to). Can you confirm that it works against the asterisk 16 branch correctly (even though 16.1.1 does not) ? If so we should be ok, by the time 16.1.2 is release (in 3-4 weeks).

mscdex, 2019-01-30:
I've been swamped and haven't had time to install another version of Asterisk to test. I'm not sure if I'll be able to test until next week some time maybe.

dkgroot, 2019-01-30:
@mscdex If anybody else tests/checks it before you get around to it and approves it could get merged beforehand, otherwise i guess we will have to wait for you :-)

ojaksch, 2019-01-31:
> Larry also promised to give it run through tomorrow using asterisk-16.1.1 and report back. If he also finds everything ok, I think we should move ahead to merging this one.

...and he did, but it doesn't worked as expected. Got a call on voicemailbox yesterday (Line 21, see below), after setting up module issues/442. I then listened to to the message, deleted it, but MWI led is still lit.

And this is weird:

> +--- MWISubscriptions ---------------------------------------------------------------------------------+
>| Mailbox                        LineName             New Old Sub AstUniq                              |
>+ ============================== ==================== === === === ==================================== +
>| 9019166@default                11                   000 000 YES bc5073e8-bedf-4279-bc0c-c083ef8528b2 |
>| 9019164@default                21                   001 000 YES 2f7bf7f5-1671-4de0-8787-34878c013e96 |
+------------------------------------------------------------------------------------------------------+

Line 11 **still has** an unheared voicemail and it's MWI led is lit, but sccp shows "0".
Line 21 shows "1" waiting message, which was deleted.
All lines are shared ones.

dkgroot, 2019-01-31:
@ojaksch Not so weird if you read the rest of this thread. Some versions of 16.1.1 seem to have this behavior and others do not, or there is a voicemail/stasis setting we have not figured out yet. Using the latest asterisk-16 development branch however behavior returns to normal again. The new sccp_mwi.c implementation does not seem to remedy this odd behavior, but it does seem to run correctly with 16.1.0 and the head of 16.

mpaland, 2019-02-01:
Tested issues/442 with 13.24.1
There is an issue with MWI activation. When I generate a new VM message, the new message is not recognized. `New` is still 000. Should be 001.
```
+--- MWISubscriptions ---------------------------------------------------------------------------------+
| Mailbox                        LineName             New Old Sub AstUniq                              |
+ ============================== ==================== === === === ==================================== +
| 42@default                     21                   000 003 YES a6c3b443-4aa0-41cb-b49b-4fad028e8057 |
| 42@default                     23                   000 003 YES 5e5e8204-e1e0-40a2-a80d-9a705788f0e0 |
```
When I restart asterisk (reload is not enough), it is updated, MWI light is on.
When I hear the msg, light goes off, that's fine.

mpaland, 2019-02-01:
Hi Diederik,
okay, I thought that only 16 is affected, not 13. Going to retest this with the latest 13 branch. Thanks for clarification.

dkgroot, 2019-02-02:
No problem. thanks for retesting. When it does work with the latest stance we are a step closer !

dkgroot, 2019-02-09:
I am going to merge issues/442 because it works and the code looks a lot better. It does not actually solve the issue with regard to this intermediate asterisk version which seems to be a little bit broken. But that should be corrected by the next release.

closed 2019-02-09 by dkgroot

mpaland, 2019-02-09:
That's perfectly fine.
Fixing/working around that kind of asterisk bugs is IMHO out of scope here.
And fixed asterisk rc1 versions are already released.

dkgroot, 2019-02-09:
@mpaland Thanks for the update, i had not even noticed (yet) :-)

dkgroot, 2019-02-12:
chan_sip having [similar issue with mwi](https://issues.asterisk.org/jira/browse/ASTERISK-13145?focusedCommentId=246203&page=com.atlassian.jira.plugin.system.issuetabpanels:comment-tabpanel#comment-246203) after update to latest 13 revision. Confirming the issue.

dkgroot, 2019-02-16:
@mpaland @mscdex @ojaksch 
Releases today:
Asterisk 16.2.0 Now Available
Asterisk 13.25.0 Now Available
Can you guys please recheck when you upgrade to these version.

mpaland, 2019-02-18:
I updated productive to 13.25.0 and latest development version today.
Have to do some tests and observe the behavior. I post the results here. Stand by...

dkgroot, 2019-02-18:
ay ay sir, standing by :-)

mpaland, 2019-02-22:
I couldn't observe any problems with 13.25.0 this week regarding the mailbox and one way audio problems.
Excellent work, Diederik, really appreciate that!

[4 more replies]

commit 353be6d7 (2019-01-03): Fix MWI event handling when used in asterisk-16

fixes #442
commit 471b2b02 (2019-01-11): Add detection for new asterisk function stasis_subscription_set_filter
Update sccp_hint.c and sccp_devstate
Fix sccp_mwi.c to use stasis_subscription_set_filter
Make sure to skip processing stasis_subscription_final_message
Cleanup/Reduce/integrate sccp_mwi.c

fixes #442
commit 5b38b1e2 (2019-01-11): Add detection for new asterisk function stasis_subscription_set_filter
Update sccp_hint.c and sccp_devstate
Fix sccp_mwi.c to use stasis_subscription_set_filter
Make sure to skip processing stasis_subscription_final_message
Cleanup/Reduce/integrate sccp_mwi.c

fixes #442
commit 1ef64ce6 (2019-01-16): Complete rewrite of sccp_mwi.c
- reduced/simplified code
- added modern interface
- Use mailbox->uniqueid throughout, instead of splitting it into mbox/context parts and afterward re-assembling them
- Fix compile warnings/issues for other asterisk versions

fixes #442