chan-sccp/chan-sccp #444: Capture Group problem

mmguglielmi, 2019-01-07:
Hello to everybody!
I'm new in this group, and not very experienced in sccp channels.
Thanks in advance for reading

My PBX: Asterisk 11 (64 bits)
3 networks: phones, trunk, corp network
At first, I tried with SCCP stable version 4.2.3, and then updated to beta 4.3, and the problem persists.

I've a problem with transferences. Extension 100 calls to ext 101. Ext 102, press button "capture group", and the PBX not response anymore. Phones get freeze. And I don't have any way to come back to normallity.

I know that my post is generic, but I want to know, what kind of info/log/debug do you need to give me a hand. The problem's made me crazy.

Thanks a lot
Regards
Marco

dkgroot, 2019-01-07:
Hi Marco,

When you say "capture group" i am interpreting this as gpickup, am i correct ? You say the phones freeze, does asterisk crash, or is it just that one phone that does not react to further input ?

Please make sure you did a ```git clone https://github.com/chan-sccp/chan-sccp``` and you are currently on revision:d8bef763 using ```git log -n1```. That way we know we are looking at the same version. 
There has been a small modification in pickup/gpickup handling in recent weeks, whereby some of the settings moved from 'per device' to 'per line' so have a look at conf/sccp.conf and conf/sccp.conf.annotated to make sure you are using the correct entries. You line definition should look something like:
```
[mylinexxxxx]
callgroup=1                         ; these need to match between the two phones
pickupgroup=1                    ; these need to match between the two phones
directed_pickup=on
directed_pickup_context = "default"    ; or "from-internal" if you are using FreePBX
pickup_modeanswer = yes
```
Please set "sccp debug core, channel, pbx, feature" when doing your testing so we can see what is going on.

mmguglielmi, 2019-01-09:
Hi Diederik
Thanks for your quick answer!

I've to reformulate my problem, because is wasn't well described before.

My OS es 32 bits

At first, I installed with SCCP stable version 4.2.3. At this moment, I'd a problem with transferences. Extension 100 calls to ext 101. Ext 102, press button "capture group" in the phone, and the PBX not response anymore. Phones get freeze. And I don't have any way to come back to normallity. This was the only error, all other funcions worked perfect.

Then I started reading the forum, because SCCP isn't my knowledge area (but I'm very interested in!), and I realized that I needed to update SCCP to get support, so I updated to beta 4.3, and the original problem dessapeared (excellent!), but appeared a new error reffered to codecs, that is not logical, and didn't exist before the update! It's like asterisk couldn't translate from alaw to ulaw, and it's crazy

Do you think that could be a missconfiguration of SCCP?
I repeat, this problem didn't exist before the update

I executed "sccp debug core, channel, pbx, feature", and here the results:
https://pastebin.com/8S0448D0

Please, let me know, if you need any additional information

Thanks a lot for your help

dkgroot, 2019-01-09:
Hi Marco,
Can you show me the git revision you are currenty on ?:
```
git log -n 1
```
Just to make sure we are looking at the same version/revision.
Please do use ```git clone``` as shown in a previous comment and not one of the old .tgz files.
(We should really clean that old stuff up, that might be confusing, sorry about that).

I have looked at the debug log, you attached. And it looks like you are using an older version of chan-sccp. Can you show the output from:
```asterisk -rx "sccp show version"```

Note: Asterisk-11 is really old so you might want to think about upgrading to at least asterisk-13 or asterisk-15.
Note: Also when reporting and issue here, you should include the asterisk version (major+minor). And that you are using FreePBX. Just to be complete.

dkgroot, 2019-01-09:
@mmguglielmi 
If you were already correctly using the github repo, you might have run into an problem with codecs which i accidentally created yesterday (not all the preferences you set, were included during config loading). I just commited a fix (6f46d1e) for that. 
So a ```git pull``` should get you back on track.

mmguglielmi, 2019-01-10:
Thanks a lot!
I'm using old .tgz file you mentioned. So, I'll start with git, and I tell you the result

mmguglielmi, 2019-01-11:
I've upgrated as you said, and voala!
It's working like a charm!
Thanks a lot for your support
Sorry by my mistake in the download

dkgroot, 2019-01-11:
Thanks for the update ! Closing this issue. Happy to help

Note: If you are running on a non-production/home machine, you can keep up to date by just doing ```git pull``` once every couple of days/weeks. On an actual production machine it's a good policy to run behind a couple of weeks and check the issues / git log before pulling. In a production environment i recommend having a staging machine that track the head of develop so you know which revisions are save to pull. We try to keep the development branch as stable as possible, but stuff happens.
Please sign up to our [gitter](https://gitter.im/chan-sccp/chan-sccp) so you can stay in contact / ask questions etc..

closed 2019-01-11 by dkgroot