chan-sccp/chan-sccp #445: Always one way audio on some incoming calls

mpaland, 2019-01-13:
One way audio on incoming call. Caller can hear me, (outbound audio okay), but I can't hear the caller (no inbound audio).
Problem can only be observed on certain callers, but if they call, there is always one way audio (not randomly).
Other callers have no problems (works every time).
Assumption: It happens when calling party is wants/needs alaw codec.

**Further info**
Taking the call on a connected SIP phone has no problem with the caller, so it seems to be SCCP related. (no NAT issue).

**Reproduction Steps:**
Explaining the trace below:

1. [First Step]
Incoming call on (my) extensions 21 and 23
2. [Second Step]
Answer the call on ext23

Setup:
PBX on Gateway, internal IP: 192.168.2.1/16, external IP: 87.153.229.244
ext SCCP/21 IP: 192.168.2.121
ext SCCP/23 IP: 192.168.1.210

**chan-sccp version:**
develop/d8bef76 (01/07/2019)

**asterisk version:**
13.24.1 (LTS release)

**Expected behavior:**
Two way audio

**Addition / Optional Information:**
One way audio trace:
```
<--- Received SIP request (1199 bytes) from UDP:217.0.21.2:5060 --->
INVITE sip:0511xxxxxxx@87.153.229.244:5060 SIP/2.0
Max-Forwards: 52
Via: SIP/2.0/UDP 217.0.21.2:5060;branch=z9hG4bKg3Zqkv7ipcfyfhckf9jfb31oh6rz3uq6o
To: <sip:+49511xxxxxxx@telekom.de;user=phone>
From: <sip:+495336xxxxxx@htp-ngn.de;user=phone>;tag=h7g4Esbg_p65550t1547218486m811652c187403019s1_3763074456-561284339
Call-ID: p65550t1547218486m811652c187403019s2
CSeq: 1 INVITE
Contact: <sip:sgc_c@217.0.21.2;transport=udp>;+g.3gpp.icsi-ref="urn%3Aurn-7%3A3gpp-service.ims.icsi.mmtel"
Record-Route: <sip:217.0.21.2;transport=udp;lr>
Accept-Contact: *;+g.3gpp.icsi-ref="urn%3Aurn-7%3A3gpp-service.ims.icsi.mmtel"
Min-Se: 900
P-Asserted-Identity: <sip:+495336xxxxxx@htp-ngn.de;user=phone>
Session-Expires: 1800
Supported: timer
Supported: histinfo
User-Agent: HTP_GMBH/HTP05 (RZII)
Content-Type: application/sdp
Content-Length: 220
Session-ID: 276a1262e0335713429228a0d9679742
Allow: REGISTER, REFER, NOTIFY, SUBSCRIBE, UPDATE, INVITE, ACK, OPTIONS, CANCEL, BYE

v=0
o=- 1640507250 3763074249 IN IP4 217.0.21.2
s=SIP Call
c=IN IP4 217.0.5.180
t=0 0
m=audio 41976 RTP/AVP 8 101
a=rtpmap:8 PCMA/8000/1
a=rtpmap:101 telephone-event/8000
a=fmtp:101 0-15
a=ptime:30
a=sendrecv

  == Setting global variable 'SIPDOMAIN' to '87.153.229.244'
<--- Transmitting SIP response (452 bytes) to UDP:217.0.21.2:5060 --->
SIP/2.0 100 Trying
[... 260 lines cut ...]
    -- SEP00270DBD6712: Current channel SCCP/23-0000003F state CONNECTED(30)
    -- SEP00270DBD6712: Finish to indicate channel state:ONHOOK on call:SCCP/23-0000003F, lineInstance:1. New channel state:DOWN
    -- SEP00270DBD6712: Device is Requesting CallStatisticsAndClear
    -- SCCP: Cleaning channel SCCP/23-0000003F
  == Spawn extension (ext-group, 20, 1) exited non-zero on 'PJSIP/anonymous-00000028'
    -- SCCP/23-0000003F: Disabling scheduler / Removing Scheduled tasks (digittimeout_id:-1) (hangup_id:-1)
  == SEP00270DBD6712: (sccp_channel_closeAllMediaTransmitAndReceive) Stop All Media Reception and Transmission on channel 63
       > SEP00270DBD6712: Stopping PBX audio rtp transmission on channel SCCP/23-0000003F
    -- SEP00270DBD6712: Set the active channel 0 on device
    -- SEP00270DBD6712: Looking for selected channel (63)
    -- Destroying channel SCCP/23-0000003F
       > SCCP: Stopping PBX audio rtp transmission on channel SCCP/23-0000003F
    -- SCCP: destroying PBX rtp server on channel SCCP/23-0000003F
    -- SEP00270DBD6712: is Onhook (buttonIndex: 0, callid: 0)
    -- SEP00270DBD6712: Getting the active channel on device.
    -- SEP00270DBD6712: No active channel on device.
    -- QualityStats: MLQK=0.0000;MLQKav=0.0000;MLQKmn=0.0000;MLQKmx=0.0000;ICR=0.0000;CCR=0.0000;ICRmx=0.0000;CS=0;SCS=0;MLQKvr=0.95
    -- SEP00270DBD6712: Call Statistics:
       [
         Last Call        : CallID: 63 Packets sent: 412 rcvd: 0 lost: 0 jitter: 0 latency: 0
         Last Quality     : MLQK=0.0000;MLQKav=0.0000;MLQKmn=0.0000;MLQKmx=0.0000;MLQKvr=0.95|ICR=0.0000;CCR=0.0000;ICRmx=0.0000|CS=0;SCS=0
         Mean Statistics  : #Calls: 9 Packets sent: 23160 rcvd: 22994 lost: 4 jitter: 1 latency: 0
         Mean Quality     : MLQK=3.0000;MLQKav=2.9860;MLQKmn=2.7518;MLQKmx=4.5000;MLQKvr=0.95|ICR=0.0000;CCR=0.0003;ICRmx=0.0560|CS=3;SCS=0
       ]
```

dkgroot, 2019-01-13:
I guess i did not test the multiple destination scenario.
I think i know what is happening:
- The codec selection looks pretty good (which was what 420_2 was all about).
- Codecs (ie compressionType) is only one party of sending and receiving rtp though, there is also the assigned rtppayloadtype. In this case the outbound sip call has set this up as type "8". However skinny is using rtppayloadtype "2", this is not compatible. I think this was introduced in revision fb6bde, when David and I tried fixing slin16/wideband. 
- I guess "rtp set debug on" would have shown both devices sending differnt payloads.

Solution path:
- We should only do rtppayload setting if:
   - We are starting an outbound call
   - We are dealing with non-static (dynamic) rtppayload types (anything above type:77) and use the asterisk default for all others.
   - Make an exception for slin16, which requires rtppayloadtype:25

I will have to create a scenario to replicate your situation and see if the solution path is gonna work .without negative side effects.

Continuation #420

mpaland, 2019-01-13:
@dkgroot, thanks a lot for giving feedback to this issue so fast.
As this is still related to #420, I close this, please reopen #420 for further tracking.
Sorry for creating trouble, again.

dkgroot, 2019-01-13:
No trouble at all, i think i am the cause of the trouble not you :-) (I thought i asked you to open a new one in #420). I just added a link to the other issue for reference). It is only partially related to the old #420 issue (AFAIK), same call, same one way audio, but a different reason :-) Continuing with a new issue and a partially new slate.

mpaland, 2019-01-16:
Here is another one way audio trace with rtp information when calling from my mobile.
No inbound audio, outbound working fine.
For testing, I configured both 7975 phones like:
```
disallow = all
allow = alaw
```
to make code negotiation simple.

```
<--- Received SIP request (1882 bytes) from UDP:217.0.21.2:5060 --->
INVITE sip:0511xxxxxxx@87.153.229.244:5060 SIP/2.0
Max-Forwards: 52
Via: SIP/2.0/UDP 217.0.21.2:5060;branch=z9hG4bKg3Zqkv7iuxdot7jyfhu98e1ko6pd9oz39
To: <sip:+49511xxxxxxx@telekom.de;user=phone>
From: <sip:+49177xxxxxxx@tmobile.de;user=phone>;tag=h7g4Esbg_p65550t1547637963m959782c204590401s1_2333428588-1856915906
Call-ID: p65550t1547637963m959782c204590401s2
CSeq: 1 INVITE
Contact: <sip:sgc_c@217.0.21.2;transport=udp>;+g.3gpp.icsi-ref="urn%3Aurn-7%3A3gpp-service.ims.icsi.mmtel"
Record-Route: <sip:217.0.21.2;transport=udp;lr>
Accept-Contact: *;+g.3gpp.icsi-ref="urn%3Aurn-7%3A3gpp-service.ims.icsi.mmtel"
Min-Se: 900
P-Asserted-Identity: <sip:+49177xxxxxxx@tmobile.de;user=phone>
P-Asserted-Identity: <tel:+49177xxxxxxx>
Session-Expires: 1800
Supported: timer
Supported: histinfo
Supported: norefersub
Supported: uui
Content-Type: application/sdp
Content-Length: 813
Session-ID: d5f5d2457c391e4a8d3c1d216f29c315
Request-Disposition: no-fork
Allow: REGISTER, REFER, NOTIFY, SUBSCRIBE, INFO, PRACK, UPDATE, INVITE, ACK, OPTIONS, CANCEL, BYE

v=0
o=- 1473290517 2333428366 IN IP4 217.0.21.2
s=-
c=IN IP4 217.0.5.180
t=0 0
m=audio 46972 RTP/AVP 96 97 9 98 99 100 101 8 102 103
a=rtpmap:96 AMR-WB/16000
a=fmtp:96 mode-set=0,1,2; mode-change-period=2; mode-change-neighbor=1; max-red=0
a=rtpmap:97 AMR-WB/16000
a=fmtp:97 mode-change-capability=2; max-red=0
a=rtpmap:9 G722/8000
a=rtpmap:98 AMR/8000
a=fmtp:98 mode-set=7; max-red=0
a=rtpmap:99 AMR/8000
a=fmtp:99 mode-set=0,1,2,3,4,5,6,7; mode-change-period=2; mode-change-neighbor=1; max-red=0
a=rtpmap:100 AMR/8000
a=fmtp:100 mode-set=0,2,4,7; mode-change-period=2; mode-change-neighbor=1; max-red=0
a=rtpmap:101 AMR/8000
a=fmtp:101 mode-set=0,1,2,3,4,5,6,7; max-red=0
a=rtpmap:8 PCMA/8000
a=rtpmap:102 telephone-event/8000
a=rtpmap:103 telephone-event/16000
a=ptime:20
a=maxptime:30

  == Setting global variable 'SIPDOMAIN' to '87.153.229.244'
<--- Transmitting SIP response (453 bytes) to UDP:217.0.21.2:5060 --->
SIP/2.0 100 Trying
Via: SIP/2.0/UDP 217.0.21.2:5060;rport=5060;received=217.0.21.2;branch=z9hG4bKg3Zqkv7iuxdot7jyfhu98e1ko6pd9oz39
Record-Route: <sip:217.0.21.2;transport=udp;lr>
Call-ID: p65550t1547637963m959782c204590401s2
From: <sip:+49177xxxxxxx@tmobile.de;user=phone>;tag=h7g4Esbg_p65550t1547637963m959782c204590401s1_2333428588-1856915906
To: <sip:+49511xxxxxxx@telekom.de;user=phone>
CSeq: 1 INVITE
Server: Asterisk
[... 892 lines cut ...]
<--- Received SIP response (631 bytes) from UDP:217.0.21.2:5060 --->
SIP/2.0 200 OK
Via: SIP/2.0/UDP 87.153.229.244:5060;received=87.153.229.244;rport=5060;branch=z9hG4bKPjbf9bdc01-a47a-4b6a-bd0b-93b611a5b1b7
To: <sip:+49177xxxxxxx@tmobile.de;user=phone>;tag=h7g4Esbg_p65550t1547639278m793023c204689869s1_3648262396-1793843747
From: <sip:+49511xxxxxxx@telekom.de;user=phone>;tag=5c68fc2e-b7ee-4f6d-bc9d-7e52e9534c26
Call-ID: p65550t1547639278m793023c204689869s2
CSeq: 7741 BYE
Supported: timer
Supported: 100rel
Supported: histinfo
Supported: norefersub
Supported: uui
Content-Length: 0
Allow: REGISTER, REFER, NOTIFY, SUBSCRIBE, INFO, PRACK, UPDATE, INVITE, ACK, OPTIONS, CANCEL, BYE

    -- QualityStats: MLQK=4.5000;MLQKav=4.5000;MLQKmn=4.5000;MLQKmx=4.5000;ICR=0.0000;CCR=0.0000;ICRmx=0.0000;CS=0;SCS=0;MLQKvr=0.95
    -- SEP002699F01CFA: Call Statistics:
       [
         Last Call        : CallID: 114 Packets sent: 1014 rcvd: 996 lost: 0 jitter: 1 latency: 0
         Last Quality     : MLQK=4.5000;MLQKav=4.5000;MLQKmn=4.5000;MLQKmx=4.5000;MLQKvr=0.95|ICR=0.0000;CCR=0.0000;ICRmx=0.0000|CS=0;SCS=0
         Mean Statistics  : #Calls: 11 Packets sent: 734 rcvd: 389 lost: 0 jitter: 0 latency: 0
         Mean Quality     : MLQK=2.4226;MLQKav=2.4226;MLQKmn=2.4226;MLQKmx=4.5000;MLQKvr=0.95|ICR=0.0000;CCR=0.0007;ICRmx=0.0066|CS=0;SCS=0
       ]
```
This is okay, both endpoints support g722, which is taken.

mpaland, 2019-01-16:
And here is just another trace from the German *htp* provider, this time with
```
allow = all
```
but still **only one way audio**. Which is really strange, because alaw codecs are selected.
```
<--- Received SIP request (1355 bytes) from UDP:217.0.21.2:5060 --->
INVITE sip:0511xxxxxxx@87.153.229.244:5060 SIP/2.0
Max-Forwards: 50
Via: SIP/2.0/UDP 217.0.21.2:5060;branch=z9hG4bKg3Zqkv7i9t7npz9skk3bhw11xsu121vwe
To: <sip:+49511xxxxxxx@tel.t-online.de;user=phone>
From: <sip:+49531xxxxxxxx@tel.t-online.de;user=phone>;tag=h7g4Esbg_p65550t1547642569m230250c204909426s1_2643731289-1531114646
Call-ID: p65550t1547642569m230250c204909426s2
CSeq: 1 INVITE
Contact: <sip:sgc_c@217.0.21.2;transport=udp>;+g.3gpp.icsi-ref="urn%3Aurn-7%3A3gpp-service.ims.icsi.mmtel"
Record-Route: <sip:217.0.21.2;transport=udp;lr>
Accept-Contact: *;+g.3gpp.icsi-ref="urn%3Aurn-7%3A3gpp-service.ims.icsi.mmtel"
Min-Se: 900
P-Asserted-Identity: <sip:+49531xxxxxxxx@pstn-icp.de;user=phone>
Session-Expires: 1800
Supported: timer
Supported: 100rel
Supported: histinfo
Content-Type: application/sdp
Content-Length: 174
Session-ID: fce4540e160014febefee8023c75e1d9
Allow: REGISTER, REFER, NOTIFY, SUBSCRIBE, PRACK, UPDATE, PUBLISH, INFO, INVITE, ACK, OPTIONS, CANCEL, BYE
Accept: application/vnd.etsi.sci+xml
Accept: multipart/mixed
Accept: application/vnd.telekom.service_indication+xml
Accept: application/vnd.etsi.cug+xml
Accept: application/sdp

v=0
o=- 1963002989 2643731075 IN IP4 217.0.21.2
s=Basic Session
c=IN IP4 217.0.5.180
t=0 0
m=audio 32808 RTP/AVP 8 99
a=rtpmap:99 telephone-event/8000
a=fmtp:99 0-15

  == Setting global variable 'SIPDOMAIN' to '87.153.229.244'
<--- Transmitting SIP response (464 bytes) to UDP:217.0.21.2:5060 --->
SIP/2.0 100 Trying
Via: SIP/2.0/UDP 217.0.21.2:5060;rport=5060;received=217.0.21.2;branch=z9hG4bKg3Zqkv7i9t7npz9skk3bhw11xsu121vwe
Record-Route: <sip:217.0.21.2;transport=udp;lr>
Call-ID: p65550t1547642569m230250c204909426s2
From: <sip:+49531xxxxxxxx@tel.t-online.de;user=phone>;tag=h7g4Esbg_p65550t1547642569m230250c204909426s1_2643731289-1531114646
To: <sip:+49511xxxxxxx@tel.t-online.de;user=phone>
CSeq: 1 INVITE
Server: Asterisk
Content-Length:  0

    -- Executing [0511xxxxxxx@from-trunk-telekom:1] Verbose("PJSIP/anonymous-0000004a", "1,*** Telekom CALL from  <+49531xxxxxxxx> to 0511xxxxxxx ***") in new stack
 *** Telekom CALL from  <+49531xxxxxxxx> to 0511xxxxxxx ***
    -- Executing [0511xxxxxxx@from-trunk-telekom:2] GotoIf("PJSIP/anonymous-0000004a", "0?black") in new stack
    -- Executing [0511xxxxxxx@from-trunk-telekom:5] ExecIf("PJSIP/anonymous-0000004a", "0 ?Set(CALLERID(name)=)") in new stack
    -- Executing [0511xxxxxxx@from-trunk-telekom:6] Set("PJSIP/anonymous-0000004a", "GROUP()=group-trunk-inbound") in new stack
    -- Executing [0511xxxxxxx@from-trunk-telekom:7] Verbose("PJSIP/anonymous-0000004a", "1,*** Number of incoming calls is 1") in new stack
 *** Number of incoming calls is 1
    -- Executing [0511xxxxxxx@from-trunk-telekom:8] GotoIf("PJSIP/anonymous-0000004a", "1?true:false") in new stack
    -- Goto (from-trunk-telekom,0511xxxxxxx,9)
    -- Executing [0511xxxxxxx@from-trunk-telekom:9] Macro("PJSIP/anonymous-0000004a", "strip-country-code") in new stack
    -- Executing [s@macro-strip-country-code:1] GotoIf("PJSIP/anonymous-0000004a", "1?:intl") in new stack
    -- Executing [s@macro-strip-country-code:2] Set("PJSIP/anonymous-0000004a", "CALLERID(num)=0531xxxxxxxx") in new stack
    -- Executing [s@macro-strip-country-code:3] MacroExit("PJSIP/anonymous-0000004a", "") in new stack
    -- Executing [0511xxxxxxx@from-trunk-telekom:10] ExecIf("PJSIP/anonymous-0000004a", "1 ?Set(CALLERID(name)=0531xxxxxxxx)") in new stack
    -- Executing [0511xxxxxxx@from-trunk-telekom:11] Goto("PJSIP/anonymous-0000004a", "ext-group,20,1") in new stack
    -- Goto (ext-group,20,1)
[... 291 lines cut ...]
Route: <sip:217.0.21.2;transport=udp;lr>
Reason: Q.850;cause=16
Max-Forwards: 70
User-Agent: Asterisk
Content-Length:  0

<--- Received SIP response (555 bytes) from UDP:217.0.21.2:5060 --->
SIP/2.0 200 OK
Via: SIP/2.0/UDP 87.153.229.244:5060;received=87.153.229.244;rport=5060;branch=z9hG4bKPj143bffda-1600-4fb7-ac1b-2088aa2e13cb
To: <sip:+49531xxxxxxxx@tel.t-online.de;user=phone>;tag=h7g4Esbg_p65550t1547642569m230250c204909426s1_2643731289-1531114646
From: <sip:+49511xxxxxxx@tel.t-online.de;user=phone>;tag=30ff123f-e2e2-48ba-89cb-be6d06e2c59e
Call-ID: p65550t1547642569m230250c204909426s2
CSeq: 24875 BYE
Content-Length: 0
Allow: REGISTER, REFER, NOTIFY, SUBSCRIBE, PRACK, UPDATE, PUBLISH, INFO, INVITE, ACK, OPTIONS, CANCEL, BYE

    -- QualityStats: MLQK=0.0000;MLQKav=0.0000;MLQKmn=0.0000;MLQKmx=0.0000;ICR=0.0000;CCR=0.0000;ICRmx=0.0000;CS=0;SCS=0;MLQKvr=0.95
    -- SEP002699F01CFA: Call Statistics:
       [
         Last Call        : CallID: 122 Packets sent: 1343 rcvd: 0 lost: 0 jitter: 0 latency: 0
         Last Quality     : MLQK=0.0000;MLQKav=0.0000;MLQKmn=0.0000;MLQKmx=0.0000;MLQKvr=0.95|ICR=0.0000;CCR=0.0000;ICRmx=0.0000|CS=0;SCS=0
         Mean Statistics  : #Calls: 15 Packets sent: 763 rcvd: 387 lost: 0 jitter: 0 latency: 0
         Mean Quality     : MLQK=2.0766;MLQKav=2.0766;MLQKmn=2.0766;MLQKmx=4.5000;MLQKvr=0.95|ICR=0.0000;CCR=0.0006;ICRmx=0.0066|CS=0;SCS=0
       ]
```

dkgroot, 2019-01-17:
@mpaland From reading the transcript of all three calls it should have worked.
- Call 1: It could have choossen alaw instead of the requested g722, to revent transcoding. But it did end up with transcoding which should have been working as well.
- Call 2: Looks ok
- Call 3: Here we do end up with alaw and no transcoding, but you are still having audio issues.

Things i noticed:
- In the RTP setup we should output the skinny payload used for both rtp directions, that would help with debugging
- Once we have started rtp in and out we should button down out nativecapabilities to only that codec, which we are not doing at the moment. Skinny phones cannot change codecs mid call (like sip phones can).
- PJSIP seems to be slightly different in its behaviour than chan_sip, in adapting to the remote party. I will have to switch to pjsip for a bit.

Question: is it only happening when multiple destinations are involved ? What happens when chan_sccp is the calling phone, does that change matters ?
Please set "rtp set debug on" for any further debug reports, so that i can see who is sending/receiving what.

dkgroot, 2019-01-18:
@mpaland 
I have an initial fix, that seems to work on my system in branch "issues/445". Could you give it a try, just to see if it fixes the issue.
```
git pull
git fetch -a
git checkout issues/445
```
Note: not production quality yet, still some cleanup work to be done !
Also it currently picks any codec the other side has to offer and it might not be the best one. I have already figured a nice way/location to do that, but requires some more work.
Please set
```
core set debug 1
core set verbose 5
sccp debug core, channel, pbx, codec, rtp
rtp set debug on
```
while testing. It might be best to link a url instead of pasting the /var/log/asterisk/full results here, as `rtp set debug on` does generate a lot of output quite quickly.

dkgroot, 2019-01-19:
@mpaland Some time ago asterisk dropped support for setting a bridge handler on a channel via .bridge. I assumed that .earlybridge was also removed at that time, and removed that as well, in retrospect that was a mistake. Has been readded now, which should help with earlyrtp and directrtp issues.

mpaland, 2019-01-23:
Further info: Dialing of any internal numbers like "42" (VM) or "*60" MOH test playback results in a crash:
```
SCCP new debug status: (1048833 -> 1057093) core,rtp,channel,config,pbx,codec
    -- SEP002699F01CFA: Getting the active channel on device.
    -- SEP002699F01CFA: No active channel on device.
SEP002699F01CFA: (getEmptyChannel) on line:21, maybe_c:
SEP002699F01CFA: (getEmptyChannel) getActiveChannel
    -- SEP002699F01CFA: Getting the active channel on device.
    -- SEP002699F01CFA: No active channel on device.
    -- SEP002699F01CFA: Set the active channel 5 on device
    -- 1021: New channel number: 5 on line 21
    -- SCCP: (pbx_channel_allocate) try to allocate Outbound channel on line: 21
    -- SCCP:              cid_num: "21"
    -- SCCP:             cid_name: "Marco"
    -- SCCP:          accountcode: "(null)"
    -- SCCP:                exten: ""
    -- SCCP:              context: "from-internal"
    -- SCCP:             amaflags: "0"
    -- SCCP:            chan/call: "SCCP/21-00000005"
    -- SCCP:combined capabilities: "[slin16, g722/64k, ulaw/64k, alaw/64k, g722/56k, g722/48k, ilbc, g729b, g729ab, g729, g729a]"
    -- SCCP:  reduced preferences: "[slin16, alaw/64k, ulaw/64k, g722/64k, g722/56k, g722/48k]"
    -- SCCP: (allocPBXChannel) Create New Channel with name: SCCP/21-00000005
[Jan 23 17:50:47] ERROR[30595][C-00000001]: astobj2.c:136 INTERNAL_OBJ: FRACK!, Failed assertion bad magic number 0x0 for object 0x13f0860 (0)
[Jan 23 17:50:47] ERROR[30595][C-00000001]:   Got 16 backtrace records
# 0: /usr/sbin/asterisk(__ast_assert_failed+0x68) [0x5fff28]
# 1: /usr/sbin/asterisk(__ao2_ref+0x1a1) [0x462bc1]
# 2: /usr/sbin/asterisk(__ast_format_cap_append+0xa4) [0x51f774]
# 3: /usr/lib/asterisk/modules/chan_sccp.so(+0xd15fa) [0x7fa9b381b5fa]
# 4: /usr/lib/asterisk/modules/chan_sccp.so(+0xd767a) [0x7fa9b382167a]
# 5: /usr/lib/asterisk/modules/chan_sccp.so(+0x3abbe) [0x7fa9b3784bbe]
# 6: /usr/lib/asterisk/modules/chan_sccp.so(+0x23647) [0x7fa9b376d647]
# 7: /usr/lib/asterisk/modules/chan_sccp.so(+0x237dd) [0x7fa9b376d7dd]
# 8: /usr/lib/asterisk/modules/chan_sccp.so(+0x497dc) [0x7fa9b37937dc]
# 9: /usr/lib/asterisk/modules/chan_sccp.so(+0x4a849) [0x7fa9b3794849]
#10: /usr/lib/asterisk/modules/chan_sccp.so(+0x5c89d) [0x7fa9b37a689d]
#11: /usr/lib/asterisk/modules/chan_sccp.so(+0x60bc4) [0x7fa9b37aabc4]
#12: /usr/lib/asterisk/modules/chan_sccp.so(+0x4081c) [0x7fa9b378a81c]
#13: /usr/sbin/asterisk() [0x5fb7dc]
#14: /lib/libpthread.so.0(+0x7ebb) [0x7faa1fc26ebb]
#15: /lib/libc.so.6(clone+0x3f) [0x7faa1f13308f]
```

dkgroot, 2019-01-23:
Hi Marco, This last one was a stupid mistake on my part, i has been corrected this morning (assuming this was using the develop branch (right ?).

dkgroot, 2019-01-24:
Pushed a temporary for for the FRACK! issue.
Please let me know if the current version, fixes the one way audio you got before.

mpaland, 2019-01-25:
Hi Diederik,
yesterday I tested the latest branch version and you are right, the FRACK issue is not happening in every call, I thought it was my problem, but as you confirm it, it makes sense.
Calling/getting called with the htp provider does work now! alaw coded is selected. Very good! 👍 

I tested your latest temp fixes in /445 today and the FRACK problem is gone by it.
Problem is still, that the VM does not accept any pressed digits. I thought this was due to the FRACK issue, but it is not.
I think, this is still a problem of one way audio and missing `theirs: []` codecs, I attach a file calling the VM system (dialing 42).
[VM.txt](https://github.com/chan-sccp/chan-sccp/files/2797618/VM.txt)

Thanks a lot for your excellent work so far!!

dkgroot, 2019-01-25:
Hi Marco,
Sorry it's a bit of a slow process, but we are making progress.
- [x] Codec selection on early progress, seems to work
- [x] Frack issue prevented (but not really solved, in the sense that it is not clear why this destroy was not allowed. The moment of the frack (after hangup) makes it hard to debug why a payloads_destroy at call setup is causing this.
- [ ] The dialing 42 in voicemail sounds like a DTMF issue (AFAIK), correct ? You are able to dial extension 42, but any digits pressed during the call are not received by the PBX/IVR/VM. 

Note: To fix some of the slin16/wideband issues, i have rewritten the payload section in createRtp, that maps codecs to payload types. This is also where DTMF is being setup. What happens if you make alaw you preferred codec (as a test), does that fix the DTMF issues ? I will need to replicate the DTMF issue you are seeing, an i need to know the phone type you are using during the test.

mpaland, 2019-01-25:
Hi Diederik,

locking the phone to the alaw codec doesn't change the DTMF issue.
```
    -- Executing [42@from-internal:1] VoiceMailMain("SCCP/23-00000009", "42@default,g(12)") in new stack
    -- SCCP/23-00000009, Could not find a common prefered codec (yet), using alaw/64k (2)
    -- SCCP/23-00000009: (findBestJoint) returning prefered codec alaw/64k (2)
    -- SCCP/23-00000009: (setNativeAudioFormats) (alaw)
    -- SEP00270DBD6712 - SCCP/23-00000009: (recalculateCodecformat)
        ours: [alaw/64k]
        theirs: []
        picked: alaw/64k
```
dtmfmode" & "force_dtmfmode are both not defined in my sccp.conf, so it will be using the default, I guess.
I'm using a 7975 phone with the following setup:
```
[general]
servername = Asterisk
keepalive = 60

debug = core,config,codec

bindaddr = 192.168.2.1
port = 2000
deny=0.0.0.0/0.0.0.0
permit=192.168.0.0/255.255.0.0
localnet = internal
context = default

nat = auto
directrtp = no
dateformat = D.M.Y
firstdigittimeout = 16
digittimeout = 10
digittimeoutchar = #
simulate_enbloc = yes
autoanswer_ring_time = 1
autoanswer_tone = 0x32
remotehangup_tone = 0x32
transfer_tone = 0
transfer_on_hangup = no
callwaiting_tone = 0x2d
callwaiting_interval = 0
musicclass = default
language = en
callevents = yes
sccp_tos = 0x68
sccp_cos = 4
audio_tos = 0xB8
audio_cos = 6
video_tos = 0x88
video_cos = 5
echocancel = yes
silencesuppression = no
earlyrtp = progress
dndFeature = on
private = yes
mwilamp = on
cfwdall = yes
cfwdbusy = yes
cfwdnoanswer = yes
allowoverlap = no
callanswerorder = oldestfirst
directed_pickup_modeanswer = on
hotline_enabled = no
hotline_context = default
hotline_extension = 555
fallback = no

;=========================
; E X T E N S I O N S
;=========================
[defaultdevice](!)
type = device
park = yes
transfer = on
cfwdall = on
cfwdbusy = on
cfwdnoanswer = on
dndFeature = on
dnd = silent
private = on
directed_pickup = on
directed_pickup_context = default
directed_pickup_modeanswer = on
mwioncall = off
deny=0.0.0.0/0.0.0.0
permit=192.168.0.0/255.255.0.0
directrtp = no
earlyrtp = progress
keepalive = 60

[SEP00270DBD6712](defaultdevice)
description = 23 (xxxxx)
devicetype = 7975
button = line, 23, default                      ; Assign Line 23 to device and use this as default line
button = feature,call forward to Mobile,cfwdAll,0177xxx
button = speeddial, Amt,, 80@ext-hints          ; trunk
softkeyset = softkeyset1
; allow = all
disallow = all
; allow = g722
allow = alaw
; allow = g729
; allow = ilbc
```

I'm going to test your latest fix tomorrow. Great work so far!

dkgroot, 2019-01-25:
Ok thanks for the details... Will wait until you have tested the latest commit to see if something still needs to be done. I think the partial revert of the dtmf code, might already do the trick.

Toi toi toi

mpaland, 2019-01-26:
Hi Diederik,
I've tested latest abe16202b9961b3e291e0045f6b49320dac07151 and DTMF works fine again! Great!

I stay on this /445 version so far and would like to do/get some more phone calls for testing next week. If all works fine, I close this issue.

Getting closer to the next release! 🎉

dkgroot, 2019-01-26:
@mpaland 
I created a temporary "issues/445_1" branch which implements the same behaviour as issues/455 with a minimal number of changes. It fixes the frack issue nicely/correctly. Can you please give this one a try, to see if it works exactly like issues/455 without any negative side effects ? If so, we can replicate it to the other asterisk and merge it.
I took this route as it was faster and did not involve me trying to hunt for the frack issue which was "impossible"

mpaland, 2019-01-27:
Hi Diederik,
I updated to `issues/445_1`; DTMF is working, no FRACK issue. First impression is fine! 👍 
But there seems to be a new issue concerning MWI, upon cleaning the inbox the MWI light stays on. I have to investigate this a little further...
Can test the htp provider not before tomorrow, cause today nobody is in. Keep you informed the next week. Again, thanks a lot!

dkgroot, 2019-01-27:
Hi Marco,
Super, that's good to hear. 
The mwi issue is being handled in a separate issue (#442), was caused by a chang in asterisk and it is already fixed, just waiting for it to be tested/checked.
I will wait for your confirmation tomorrow. In the meantime i will replicate the ast13 changes in 445_1 to the other asterisk versions, so we are ready to merge this back to the main branch.
Thanks for heling out !

dkgroot, 2019-01-27:
@mpaland If you have a pre-java phone (like 7960) hooked up to your setup, can you check if DTMF works correctly with it too (just trying to make sure), if not i would have to connect one and run some trials. I do have a patch in the pipeline, that might do a better job using skinny dtmf on these old devices (not committed yet).

mpaland, 2019-01-28:
@dkgroot, indeed I have an ancient 7960, reactivated it and flashed the latest SCCP firmware.
Can confirm that DTMF is working fine (with latest 445_1 branch).

FYI, I had a general INCOMING audio issue on a G524 phone (but not on 7970/7975). Upon the phone switched to any in-band audio (off-hook, dial a number etc.), there was an extremely loud white noise.
This is fixed by 445_1, too. DTMF is working as well on the G524.

dkgroot, 2019-01-28:
Great, will be merging this version tonight, which will also close this issue !
Thanks a lot for your help, detailed debug and testing !

closed 2019-01-28 by dkgroot

dkgroot, 2019-01-28:
@mpaland Hi Marco,
issues/445_1 has been merged (and issues/445 has been removed)

There are two small 
```
/* untested */
...
/* \untested */
```
sections in pbx_impl/ast113/ast113.c. If you still have the G524 and 7960 still attached, would you mind activating these small sections and give it a go. I am hoping DTMF will still work and use the special CISCO DTMF RTP Frames instead of running real inband. It it works, i would love to activate this code.

mpaland, 2019-01-29:
Hi @dkgroot,
I took https://github.com/chan-sccp/chan-sccp/commit/724b8109922789c8b0f07a40b763c5ebe67089f8 and enabled both *untested* sections.
DTMF is working fine on all phones (SPA524 and 7960 are installed for testing purpose now).
Unfortunately there's a regression concerning the FRACK issue.
I attach a call from the 7960 the the VM, pressing a few numbers and ending the call.
[VM 7960.txt](https://github.com/chan-sccp/chan-sccp/files/2808701/VM.7960.txt)

dkgroot, 2019-01-29:
Hi Marco,
THanks a lot for checking that out. Great that it works, and it also sheds some light on the cause for the FRACK issue from before. I will not enable this just yet, until i figure out the FRACK issue.
Thanks so much for this extra bit of info !

commit 10fd54de (2019-01-28): Replace slin16 with wideband-256K using rtppayload to 25
Update codecs table
Return LOCAL during earlyrtp
Call retrieve_remote_capabilities during progress
Add initial retrieve_remote_capabilities_from_caps implementation (remarked out) for later use in direcrtp
Add remarked out SKINNY DTMF code for older model phones (untested still)

fixes #445
commit 724b8109 (2019-01-28): Replace slin16 with wideband-256K using rtppayload to 25
Update codecs table
Return LOCAL during earlyrtp
Call retrieve_remote_capabilities during progress
Add initial retrieve_remote_capabilities_from_caps implementation (remarked out) for later use in direcrtp
Add remarked out SKINNY DTMF code for older model phones (untested still)

fixes #445