chan-sccp/chan-sccp #446: Transfer fails from incoming pjsip call

ianmca, 2019-01-23:
I can't transfer an incoming SIP call, but a chan-sccp call transfers OK.

Reproduction is reliable for me... if i make an inbound call via PJSIP or via chan SIP into asterisk, onto an SCCP extension. 
1. Answering the inbound call as normal is OK.
2. Press the Transfer button on a 7975 or 7965

Result.
The first call shows on the 7965 screen as if it is on hold.
The phone shows as if it is ready to make the second call.
The phone plays the "Temp Fail" tones instead of dial tone.
The call is hung up and the "Temp Fail" tones replay again.
With the very latest commit Asterisk crashes, although with the develop branch downloaded on friday it doesn't crash.

Environment:   Asterisk version 16.1.1 (also tested with 15.7.1), Chan SCCP develop downloaded this morning (latest commit 30d9c2b). Centos 7 latest.

....

== Last Event Received: 272:SelectSoftKeys
    -- SEPA45630418BFC: Got rtpPort:3500 which the device wants to use for media
    -- SEPA45630418BFC: Sending ConfigStatMessage, lines 1, speeddials 1
    -- SEPA45630418BFC: Device has 12 Capabilities
    -- SEPA45630418BFC: Got Forward Status Request.  Line: 1
    -- SEPA45630418BFC: Device registered; performing post registration tasks...
    -- SEPA45630418BFC: (sccp_devstate_deviceRegisterListener) device registered
  == Setting global variable 'SIPDOMAIN' to '10.1.4.8'
    -- Executing [271@internal-norestrictions:1] Dial("PJSIP/asterisk2end-00000000", "SCCP/271/ringer=inside") in new stack
    -- SCCP: Asterisk request to call SCCP/271-00000001 (dest:271/ringer=inside, timeout: 0)
    -- 271: Asterisk request to call SCCP/271-00000001
    -- SCCP/271-00000001: (sccp_pbx_call) Returning: 0
    -- Called SCCP/271/ringer=inside
    -- SCCP/271-00000001 is ringing
    -- SEPA45630418BFC: Taken Offhook
    -- SEPA45630418BFC: (sccp_channel_answer) Answer Channel SCCP/271-00000001
    -- SEPA45630418BFC: Answer channel SCCP/271-00000001
    -- SEPA45630418BFC: (sccp_channel_answer) Set Called Party
    -- SEPA45630418BFC: Answering channel with state 'Ringing' (5)
  == Using SCCP RTP TOS bits 184
  == Using SCCP RTP CoS mark 6
    -- SEPA45630418BFC: Answered channel SCCP/271-00000001 on line 271
       > 0x7fa6b4003e20 -- Strict RTP learning after remote address set to: 10.1.4.171:16648
    -- SCCP/271-00000001 answered PJSIP/asterisk2end-00000000
       > 0x7fa63401a680 -- Strict RTP learning after remote address set to: 10.1.4.1:19420
    -- Channel SCCP/271-00000001 joined 'softmix' basic-bridge <be9cf56e-d6c1-4f13-8371-f267482bf4f1>
    -- Channel PJSIP/asterisk2end-00000000 joined 'softmix' basic-bridge <be9cf56e-d6c1-4f13-8371-f267482bf4f1>
       > 0x7fa6b4003e20 -- Strict RTP switching to RTP target address 10.1.4.171:16648 as source
       > 0x7fa63401a680 -- Strict RTP switching to RTP target address 10.1.4.1:19420 as source
    -- SEPA45630418BFC: Transfer request from line channel SCCP/271-00000001
    -- SEPA45630418BFC: Hold the channel SCCP/271-00000001
SEPA45630418BFC: (getEmptyChannel) on line:271, maybe_c:
SEPA45630418BFC: (getEmptyChannel) getActiveChannel
  == Using SCCP RTP TOS bits 184
  == Using SCCP RTP CoS mark 6
    -- SCCP/271-00000002: schedule digittimeout 16
       > 0x7fa63401a680 -- Strict RTP learning complete - Locking on source address 10.1.4.1:19420
    -- SCCP: Sending hangupRequest to Call SCCP/271-00000001 (state: CONGESTION)
    -- Channel SCCP/271-00000001 left 'softmix' basic-bridge <be9cf56e-d6c1-4f13-8371-f267482bf4f1>
    -- Channel PJSIP/asterisk2end-00000000 left 'softmix' basic-bridge <be9cf56e-d6c1-4f13-8371-f267482bf4f1>
  == Spawn extension (internal-norestrictions, 271, 1) exited non-zero on 'PJSIP/asterisk2end-00000000'
 SCCP: Timeout for call 'SCCP/271-00000002'. Nothing to dial -> INVALIDNUMBER
  == SEPA45630418BFC: Ending call SCCP/271-00000002 (state:INVALIDNUMBER)
    -- SEPA45630418BFC: Sending hangupRequest to Call SCCP/271-00000002 (state: INVALIDNUMBER)
    -- QualityStats: K=0.0000;MLQKav=0.0000;MLQKmn=0.0000;MLQKmx=0.0000;ICR=0.0000;CCR=0.0000;ICRmx=0.0000;CS=1;SCS=0;MLQKvr=0.95
    -- SEPA45630418BFC: Call Statistics:

dkgroot, 2019-01-23:
@ianmca
I managed to fix the crash issue (that was a slight regression on my part). I was able to perform serveral transfers though without a problem using asterisk-16.1.1 (after committing the fix mentioned). Can you retest with the latest revision ?

ianmca, 2019-01-23:
Hi @dkgroot , I have attached the debug with hopefully debug set correctly. The nothing to dial - invalid number is exactly the behaviour i'm seeing, but i'm not getting an opportunity to enter in a number at all.

[debuglog2019-01-23-01.txt](https://github.com/chan-sccp/chan-sccp/files/2787170/debuglog2019-01-23-01.txt)

dkgroot, 2019-01-23:
I think this is the clue i was looking for "Cannot transfer a dialplan application, bridged channel is required on SCCP/271-00000004" in line 154 of you log.
Now we need to figure out why you are seeing this and how i can reproduce this. Tomorrow, I will create a new branch called "issues/446" and add some debug statements, to figure out how it is your system gets to this state.

ianmca, 2019-01-24:
Thanks Diederik,

I’ll watch out for the branch and give it a run to get the debug..

I think bridging is working in general, because I have the 7975 set on G722 and a G711 trunk so there is definitely bridging and transcoding taking place in asterisk.  I had a bit of transcoding problems when I first installed – it seemed to struggle to negotiate any codecs that were listed in the general section of sccp.conf, but once I placed them into the extension template it all was OK, I also normally set direct_rtp to be off.  Probably unrelated but I though I would mention in case it helps.

Ian.

dkgroot, 2019-01-24:
@ianmca 
FYI: This branch is asterisk-13 only (meaning it's only focused to fix the DIAL(SIP/123&SCCP/321) issue in asterisk-13) for the moment, once we find the solution, we can replicate the changes to the other asterisk versions. Also note that the codec selection issue that marco is having seems to be related to his SIP provider sending back Progress before Ringing, which alters the order of events.
Thanks for helping in with the testing of this branch !

dkgroot, 2019-01-24:
@ianmca Have not had time today, to look into this issue yet (sorry about that). It's curious that i was not able to reproduce the issue using asterisk-16 so far. I will let you know as soon as i have an update.

dkgroot, 2019-01-24:
@ianmca 
Hi Ian,
I added a couple of lines of debug to figure out what is missing during transfer. Can you do
```
git pull
git checkout issues/446
make && make install
```
And run another test (with the same debug setting) please.

ianmca, 2019-01-25:
Hi, I've attached the debug log. Hope it helps.

[debuglog2019-01-25-01.txt](https://github.com/chan-sccp/chan-sccp/files/2795595/debuglog2019-01-25-01.txt)

dkgroot, 2019-01-25:
Hmmm, not as much as i had hoped :-( It does show the function that actually fails (namely: retrieving the bridged channel, which is the remote/opposite side of the call we are trying to transfer). It does however not explain why.
Little question: Which asterisk 16.1.1 version are you using (git clone -b 16 / git releases 16.1.1.tgz or asterisk-16.1.1.tgz from asterisk.org ? If i know which one, i might be able to reproduce the issue over here.
Could you produce a minimal pjsip.conf + sccp.conf that reproduces the issue, without having to send to much information ? If you like you can use my mail-address instead of having to attach it here.

ianmca, 2019-01-25:
Hi,

I downloaded the 16.1.1 tgz from the asterisk website. I also tried the 15.7.1 tgz from the website and had the same result.

I’ll send you the pjsip.conf and sccp.conf privately. I don’t think it’s related to pjsip.conf because I tried moving from pjsip to sip and that didn’t make any difference. I don’t think there’s anything too private (no passwords etc.) so I’m probably OK to send you my entire set of conf files privately. Might be able to do later today or might be Monday.

Thanks again,

Ian.

dkgroot, 2019-01-25:
Can you provide me with the url you used (maybe that would have been the better question :-))

ianmca, 2019-01-28:
Hi Diederik, I've tried building from
http://downloads.asterisk.org/pub/telephony/asterisk/asterisk-16-current.tar.gz

but i have also tried building against the latest asterisk 15 and the lastest asterisk 13, both downloaded off the asterisk website as .tar.gz files and they all give the same result. 

Over the weekend I also installed the old version (asterisk 11) and a very old chan_sccp (2014-11-26T11:43:18+0000 )   just to see if i had something else wrong with the environment in the new centos7 server, but the old versions worked first time with the same config files.

I'm happy to send you all the config files if you let me know the right email to use - don't really want to post them publicly. I tried the one listed in github but it seemed to bounce.

ianmca, 2019-01-28:
Hi,

I've also tried going back to an older release - i'm getting the same issue on release 4.2.3 of chan-sccp with asterisk 13.24.1, so it doesn't seem to be a particularly new problem.

ianmca, 2019-01-28:
Hi. 

Is there a way I can send you the config files without posting them publicly?

dkgroot, 2019-01-29:
Building an asterisk-16 directory using your config files, to see if it is able to reproduce the issue you are seeing.

dkgroot, 2019-01-29:
@ianmca Using your asterisk config files (without changing my cnf.xml files) i can reproduce the behaviour you have been seeing, so that is a step in the right direction, now i need to figure out what is going on, and how to fix it.

ianmca, 2019-01-30:
@dkgroot just wondering if its something related to bridging and transcoding more than something specifically related to the sip channels. I did try both chan sip and Pjsip but they both had exactly the same result. I have all our sccp phones on the same codec but all the sip trunks are G711. I tried adjusting (removing) some cha sccp config file entries (things like incoming call limit) but haven’t found any that makes it work yet. I haven’t yet tried your config files.

dkgroot, 2019-01-31:
Using you config it even happened when doing a transfer of an incoming sccp call. It might wel be a bridge setting (that would make sense based on the error' we are seeing). But I have not had time to figure it out yet.

ianmca, 2019-02-12:
@dkgroot  I've made a bit of progress.... there was a noload bridge_simple on modules.conf on the config files that i sent you. Possibly a typo of mine or something i did while trying to debug a related problem... Making that noload a load fixes at least part of the problem. I'll try a bit more tomorrow

dkgroot, 2019-02-12:
@ianmca That explains a lot. Maybe you should just switch to the automatic loading. The automatic loader checks if it sees a config file for the particular module, if it is not present the module will not be loader. All other dependent modules will.
In the meantime, i should update chan-sccp to require the bridge modules (including the other ones for parking, conference etc), i will check that out.

ianmca, 2019-02-13:
@dkgroot - I'm not sure if i should raise this as a separate issue or continue this one...

I'm now seeing one way audio when i call from a SIP line into an SCCP line where a transcode is needed, and no audio when calling from SCCP to SIP where a transcode is needed.

If i match all the codecs so that there's no transcode needed all works perfectly.

I have more detail on this - would you like me to raise this as a separate issue??

dkgroot, 2019-02-13:
Always open a new issue if it is unrelated :-)
In the meantime i will check transcoding, to see if i can easilty spot what might be causing this..

ianmca, 2019-02-13:
Yes i think it should be closed.

I haven't tried and SCCP to SCCP with transcode so I'll try that on my test system.

closed 2019-02-13 by ianmca

dkgroot, 2019-02-13:
And during the call, please include:
```
sccp show channels
core show channels
core show channel SIP/xxx
core show channel SCCP/xxxx
```
Output

dkgroot, 2019-02-13:
Removed unrelated message
Added a commit to add module requirements for 
- bridge_simple, bridge_holding (transfer)
- bridge_softmix (conference)
To prevent issue #446

commit 06b3940e (2019-02-13): Add extra module requirements to satisfy transfer requirements

fixes #446