chan-sccp/chan-sccp #448: Some sounds not playing

j-wauters, 2019-02-07:
When dialing out I do not here ringing on my sccp channels.  It works fine dialing from a sip client to the same outbound number.

j-wauters, 2019-02-07:
SCCP version 4.3.1 and asterisk 15.5 and freepbx 14

dkgroot, 2019-02-08:
@j-wauters
I have a couple of small questions:
- Which chan-sccp revision are you using ?
- When ringing out to some extension, you don't hear the remote ringing sound through the receiver/hook, correct ?
- What is sccp.conf earlyrtp set to ?
- Which codec are you using during the call (specific codec or any/all) ?
- Can you provide the CLI debug output for one call, with "sccp debug core, device, line, channel, indicate, rtp" set ?

j-wauters, 2019-02-08:
I think I have it fixed thanks to your post.  Early RTP was set to none.  I changed this to progress in SCCP manager and it is working now.  What should this be set to?  

Internal calls were working fine, just sounds external were not.  I noticed that internal messages were not working either.  I would dial an internal extension that didn't exist and I would not hear the "your call cannot be completed as dialed message".  Changing the early RTP seems to have corrected this.

I have sccp 4.3.1

Just an FYI, sccp manager has been an awesome module add on to freepbx for getting phones up and running quickly.  I don't think that this project has anything to do with it.

dkgroot, 2019-02-09:
@j-wauters It should always be set to Progress (unless you know exactly what you are doing and why). Progress makes sense, because that is the event that occurs shortly after dialing a number, when you want to be able to hear messages like "This call is going to cost 50 cents per minute" or other inband ring events.
The sccp_manager event by @PhantomVI is not directly related to chan-sccp but we are working together quite closely.

@PhantomVI Can you please remove the "None" setting from earlyRtp (until further notice, will need some work).

closed 2019-02-09 by dkgroot