chan-sccp/chan-sccp #449: Problem Installing 4.3.1 on Asterisk 16

kbstechnologies, 2019-02-09:
[Short description of the problem here]
Installing 4.3.1 on Asterisk 16

Can you guide me to what I need to modify to get this install to complete? 

**Reproduction Steps:**

1. [First Step]
./configure
2. [Second Step]
make
3. [Other Steps...]
make install 
**chan-sccp version:** (include revision and/or branch if relevant)
 4.3.1
**asterisk version:** (include revision and/or branch if relevant)
Asterisk 16.1.1
**Expected behavior:**

Complete install 

ast114.c:3665:1: error: unknown field ‘nonoptreq’ specified in initializer
 AST_MODULE_INFO(ASTERISK_GPL_KEY, AST_MODFLAG_LOAD_ORDER, SCCP_VERSIONSTR,.load = load_module,.unload = unload_module,.reload = module_reload,.load_pri = AST_MODPRI_DEFAULT,.nonoptreq = "chan_local");
 ^
cc1: warning: unrecognized command line option "-Wno-format-truncation" [enabled by default]

**Observed behavior:**
It errors out on make

Making all in src
make[1]: Entering directory `/usr/src/chan-sccp-4.3.1/src'
make  all-recursive
make[2]: Entering directory `/usr/src/chan-sccp-4.3.1/src'
Making all in pbx_impl
make[3]: Entering directory `/usr/src/chan-sccp-4.3.1/src/pbx_impl'
  CC       pbx_impl.lo
  CCLD     libpbximpl.la
make[3]: Leaving directory `/usr/src/chan-sccp-4.3.1/src/pbx_impl'
Making all in pbx_impl/ast
make[3]: Entering directory `/usr/src/chan-sccp-4.3.1/src/pbx_impl/ast'
  CC       libast_la-ast.lo
  CCLD     libast.la
make[3]: Leaving directory `/usr/src/chan-sccp-4.3.1/src/pbx_impl/ast'
Making all in pbx_impl/ast114/
make[3]: Entering directory `/usr/src/chan-sccp-4.3.1/src/pbx_impl/ast114'
  CC       libast114_la-ast114.lo
In file included from ast114.c:37:0:
ast114.c:3665:1: error: unknown field ‘nonoptreq’ specified in initializer
 AST_MODULE_INFO(ASTERISK_GPL_KEY, AST_MODFLAG_LOAD_ORDER, SCCP_VERSIONSTR,.load = load_module,.unload = unload_module,.reload = module_reload,.load_pri = AST_MODPRI_DEFAULT,.nonoptreq = "chan_local");
 ^
cc1: warning: unrecognized command line option "-Wno-format-truncation" [enabled by default]
make[3]: *** [libast114_la-ast114.lo] Error 1
make[3]: Leaving directory `/usr/src/chan-sccp-4.3.1/src/pbx_impl/ast114'
make[2]: *** [all-recursive] Error 1
make[2]: Leaving directory `/usr/src/chan-sccp-4.3.1/src'
make[1]: *** [all] Error 2
make[1]: Leaving directory `/usr/src/chan-sccp-4.3.1/src'
make: *** [all-recursive] Error 1

Dump from config. 

Operating System : Linux
          Kernel : 3.10.0-957.5.1.el7.x86_64
    Architecture : x86_64
Asterisk Version : 11400 / 114
  Asterisk Repos : TRUNK
      Build user : root
         Version : 4.3.1_TRUNK_runknown
Compiler Version : 4.8.5
              CC : gcc
             CPP : gcc -E
         HOST_CC : gcc
          CFLAGS : -Og -Wall -I. -D_LARGEFILE_SOURCE -D_FILE_OFFSET_BITS=64 -std=gnu11 -fgnu89-inline  -isystem /usr/include -g3 -I/usr/include/libxml2 -g3 -ggdb3 -pthread
        CPPFLAGS : -U_FORTIFY_SOURCE  -isystem /usr/include -g3 -I/usr/include/libxml2  -g3 -ggdb3
         LDFLAGS :      -L/usr/lib64        -lbfd
        PBX_TYPE : Asterisk
      PBX_PREFIX :
         PBX_ETC : /etc/asterisk
         PBX_LIB : /usr/lib64
      PBX_VARLIB : /var/lib/asterisk
     PBX_INCLUDE : /usr/include
Module Directory : /usr/lib64/asterisk/modules

dkgroot, 2019-02-09:
@kbstechnologies
Where did you get this version from (tgz / url) ? You might have noticed that it "detected" asterisk-16 as asterisk-14, which should have cause some alarm bells.

What you should do:
```
git clone https://github.com/chan-sccp/chan-sccp chan-sccp
cd chan-sccp
./configure
make && make install
```
This will get you to the current version and it should be able to detect it and build correctly.

kbstechnologies, 2019-02-09:
I downloaded this one from git hub here. https://github.com/chan-sccp/chan-sccp/archive/v4.3.1.tar.gz

Just did from the clone and when I got to make, it errored with 

  CC       pbx_impl.lo
cc1: warning: command line option "-Wenum-compare" is valid for C++/ObjC++ but not for C
In file included from ../../src/common.h:44,
                 from pbx_impl.c:12:
../../src/forward_declarations.h:56: error: redefinition of typedef ‘sccp_mailbox_t’
../../src/forward_declarations.h:48: note: previous declaration of ‘sccp_mailbox_t’ was here
cc1: warning: unrecognized command line option "-Wno-format-truncation"
make[3]: *** [pbx_impl.lo] Error 1
make[3]: Leaving directory `/var/lib/edge/edgepbx2.X/var/www/html/asterisk_edge/get/packages/chan-sccp/src/pbx_impl'
make[2]: *** [all-recursive] Error 1
make[2]: Leaving directory `/var/lib/edge/edgepbx2.X/var/www/html/asterisk_edge/get/packages/chan-sccp/src'
make[1]: *** [all] Error 2
make[1]: Leaving directory `/var/lib/edge/edgepbx2.X/var/www/html/asterisk_edge/get/packages/chan-sccp/src'
make: *** [all-recursive] Error 1

kbstechnologies, 2019-02-09:
Nevermind I was in the wrong box - Clone version worked.

closed 2019-02-09 by kbstechnologies

dkgroot, 2019-02-09:
Thanks for closing the issue. Hope everything works out for you !