chan-sccp/chan-sccp #450: missing audio when codec translation needed

ianmca, 2019-02-13:
Audio missing when codec translation needed
chan sccp develop
asterisk 16.1.1

I have tried this a number of different ways.. here's the easiest probably to find.
I have two SCCP phones, one as line 271, one as line 273.  I set both to different codecs.
Similar happens if the other device is a SIP trunk to another local asterisk.

Device hosting line 271
No codec specified on the device, or on the template.
[general]
disallow=all
allow=g722

Device hosting line 273
disallow=all
allow=alaw

Call from 273 to 271
No Audio from 271 to 273
Audio OK from 273 to 271

If I call from 271 to 273
I'm Getting no audio either direction

More details in the attachment.  One of more obvious issues is one phone seems to be picking a video codec instead of an audio one.

[codec-translation-problems.txt](https://github.com/chan-sccp/chan-sccp/files/2860196/codec-translation-problems.txt)

dkgroot, 2019-02-13:
Is there any change of you running tests with a stock asterisk config. After the transfer issue caused by missing bridge module, i am a little afraid there might be some other changes in the asterisk config files that might be responsible. I am not seeing these issue with pjsip->sccp , sip->sccp nor sccp->sccp calls.

ianmca, 2019-02-13:
Stage 1: asterisk make samples, and copied over my extensions.conf and sccp.conf... problem still exists.

Stage 2: Delete all in /etc/asterisk.    Asterisk make samples, chan sccp make clean ./configure make make install.
sample extensions.conf added to new context to minimise changes
[iantest]
exten => 271,1,Dial(SCCP/${EXTEN}/ringer=inside)
exten => 273,1,Dial(SCCP/${EXTEN}/ringer=inside)

sample sccp.conf taken from latest chan_sccp.
Just changing the very basics of device mac addresses and line numbers

In [general] ensured no allows or disallows.
against first device:  disallow=all, allow=g722
against second device: disallow=all, allow=alaw

Problem still exists.  When I first built asterisk i did enable the opus codec in menuselect I wonder if it's that.

[codecs2.txt](https://github.com/chan-sccp/chan-sccp/files/2860393/codecs2.txt)

ianmca, 2019-02-13:
core show translation
         Translation times between formats (in microseconds) for one second of data
          Source Format (Rows) Destination Format (Columns)

           ulaw  alaw   gsm  g726 g726aal2 adpcm slin8 slin12 slin16 slin24 slin32 slin44 slin48 slin96 slin192 lpc10  ilbc  g722 testlaw  opus silk8 silk12 silk16
     ulaw     -  9150 15000 15000    15000 15000  9000  17000  17000  17000  17000  17000  17000  17000   17000 15000 15000 17250   15000 23000 15000  23000  23000
     alaw  9150     - 15000 15000    15000 15000  9000  17000  17000  17000  17000  17000  17000  17000   17000 15000 15000 17250   15000 23000 15000  23000  23000
      gsm 15000 15000     - 15000    15000 15000  9000  17000  17000  17000  17000  17000  17000  17000   17000 15000 15000 17250   15000 23000 15000  23000  23000
     g726 15000 15000 15000     -    15000 15000  9000  17000  17000  17000  17000  17000  17000  17000   17000 15000 15000 17250   15000 23000 15000  23000  23000
 g726aal2 15000 15000 15000 15000        - 15000  9000  17000  17000  17000  17000  17000  17000  17000   17000 15000 15000 17250   15000 23000 15000  23000  23000
    adpcm 15000 15000 15000 15000    15000     -  9000  17000  17000  17000  17000  17000  17000  17000   17000 15000 15000 17250   15000 23000 15000  23000  23000
    slin8  6000  6000  6000  6000     6000  6000     -   8000   8000   8000   8000   8000   8000   8000    8000  6000  6000  8250    6000 14000  6000  14000  14000
   slin12 14500 14500 14500 14500    14500 14500  8500      -   8000   8000   8000   8000   8000   8000    8000 14500 14500 14000   14500 14000 14500   6000  14000
   slin16 14500 14500 14500 14500    14500 14500  8500   8500      -   8000   8000   8000   8000   8000    8000 14500 14500  6000   14500 14000 14500  14500   6000
   slin24 14500 14500 14500 14500    14500 14500  8500   8500   8500      -   8000   8000   8000   8000    8000 14500 14500 14500   14500 14000 14500  14500  14500
   slin32 14500 14500 14500 14500    14500 14500  8500   8500   8500   8500      -   8000   8000   8000    8000 14500 14500 14500   14500 14000 14500  14500  14500
   slin44 14500 14500 14500 14500    14500 14500  8500   8500   8500   8500   8500      -   8000   8000    8000 14500 14500 14500   14500 14000 14500  14500  14500
   slin48 14500 14500 14500 14500    14500 14500  8500   8500   8500   8500   8500   8500      -   8000    8000 14500 14500 14500   14500  6000 14500  14500  14500
   slin96 14500 14500 14500 14500    14500 14500  8500   8500   8500   8500   8500   8500   8500      -    8000 14500 14500 14500   14500 14500 14500  14500  14500
  slin192 14500 14500 14500 14500    14500 14500  8500   8500   8500   8500   8500   8500   8500   8500       - 14500 14500 14500   14500 14500 14500  14500  14500
    lpc10 15000 15000 15000 15000    15000 15000  9000  17000  17000  17000  17000  17000  17000  17000   17000     - 15000 17250   15000 23000 15000  23000  23000
     ilbc 15000 15000 15000 15000    15000 15000  9000  17000  17000  17000  17000  17000  17000  17000   17000 15000     - 17250   15000 23000 15000  23000  23000
     g722 15600 15600 15600 15600    15600 15600  9600  17500   9000  17000  17000  17000  17000  17000   17000 15600 15600     -   15600 23000 15600  23500  15000
  testlaw 15000 15000 15000 15000    15000 15000  9000  17000  17000  17000  17000  17000  17000  17000   17000 15000 15000 17250       - 23000 15000  23000  23000
     opus 23500 23500 23500 23500    23500 23500 17500  17500  17500  17500  17500  17500   9000  17000   17000 23500 23500 23500   23500     - 23500  23500  23500
    silk8 15000 15000 15000 15000    15000 15000  9000  17000  17000  17000  17000  17000  17000  17000   17000 15000 15000 17250   15000 23000     -  23000  23000
   silk12 23500 23500 23500 23500    23500 23500 17500   9000  17000  17000  17000  17000  17000  17000   17000 23500 23500 23000   23500 23000 23500      -  23000
   silk16 23500 23500 23500 23500    23500 23500 17500  17500   9000  17000  17000  17000  17000  17000   17000 23500 23500 15000   23500 23000 23500  23500      -
asterisk1*CLI>

dkgroot, 2019-02-13:
@ianmca Adding opus should be absolutely fine. Still wondering why i cannot reproduce this. I will do some more testing. You are running on asterisk-16.1.1 still, or did you switch to asterisk-16 (develop branch) ?

dkgroot, 2019-02-13:
You are running the latest chan-sccp (git pull), right (so that we are looking at the same stuff) ?

dkgroot, 2019-02-13:
Here a call on my asterisk-16.1.1 with chan-sccp revision 06b3940e

moviebox*CLI> sccp call SEPE0D173E11D95  98041
```
Starting Call for Device: SEPE0D173E11D95
SEPE0D173E11D95: (getEmptyChannel) on line:98031, maybe_c:
SEPE0D173E11D95: (getEmptyChannel) getActiveChannel
    -- SEPE0D173E11D95: Getting the active channel on device.
    -- SEPE0D173E11D95: No active channel on device.
    -- SEPE0D173E11D95: Set the active channel 6 on device
    -- 2: New channel number: 6 on line 98031
    -- SCCP: (pbx_channel_allocate) try to allocate Outbound channel on line: 98031
    -- SCCP:              cid_num: "98031"
    -- SCCP:             cid_name: "Diederik-Phone3"
    -- SCCP:          accountcode: "98031"
    -- SCCP:                exten: ""
    -- SCCP:              context: "internal"
    -- SCCP:             amaflags: "0"
    -- SCCP:            chan/call: "SCCP/98031-00000006"
    -- SCCP:combined capabilities: "[slin16, g722/64k, ulaw/64k, alaw/64k, g722/56k, g722/48k, ilbc, g729b, g729ab, g729, g729a]"
    -- SCCP:  reduced preferences: "[alaw/64k, ulaw/64k]"
    -- SCCP: (allocPBXChannel) Create New Channel with name: SCCP/98031-00000006
    -- allocPBXChannel: using nativeformats=(alaw|ulaw) fmt=alaw
    -- SCCP/98031-00000006: rtp server instance created at [::]:15470
  == Using SCCP RTP TOS bits 184
  == Using SCCP RTP TOS bits 184 in TCLASS field.
  == Using SCCP RTP CoS mark 6
  == SCCP/98031-00000006: update rtpmap: format:CISCO-DTMF, payload:101, mime:audio, rate:0
    -- SCCP/98031-00000006: (sccp_rtp_createRtpServer) updated phone Audio RTP destination to : [::ffff:192.168.178.3]:15470, family:IPv4, mapped: True
    -- 2: Allocated asterisk channel SCCP/98031-00000006
    -- SEPE0D173E11D95: Set asterisk state OffHook (2) for call 6
    -- SEPE0D173E11D95: Indicate SCCP new state:SPEEDDIAL, current channel state:DOWN on call:SCCP/98031-00000006, lineInstance:1 (previous channelstate:DOWN)
    -- SEPE0D173E11D95: Finish to indicate channel state:SPEEDDIAL on call:SCCP/98031-00000006, lineInstance:1. New channel state:OFFHOOK
    -- SEPE0D173E11D95: (sccp_pbx_softswitch) New call on line 98031
    -- SEPE0D173E11D95: (sccp_pbx_softswitch) Dial Extension 98041
    -- SEPE0D173E11D95: Indicate SCCP new state:DIALING, current channel state:OFFHOOK on call:SCCP/98031-00000006, lineInstance:1 (previous channelstate:DOWN)
    -- SEPE0D173E11D95: Send the dialed number:98041, callid:6, lineInstance:1
    -- SEPE0D173E11D95: Finish to indicate channel state:DIALING on call:SCCP/98031-00000006, lineInstance:1. New channel state:DIALING
    -- SCCP: (sccp_pbx_softswitch) set variable SKINNY_PRIVATE to: 0
    -- SCCP: (sccp_pbx_softswitch) set variable SKINNY_PRIVATE to: 1
    -- +- pbx extension matcher (98041          ): ---+
    -- |ignore     |exists     |can match  |match more|
    -- | no        |yes        |yes        | no       |
    -- +----------------------------------------------+
 SEPE0D173E11D95: (sccp_pbx_softswitch) channel SCCP/98031-00000006 is dialing number 98041
    -- SEPE0D173E11D95: Set asterisk state Ring (4) for call 6
    -- SCCP/98031-00000006: (pbx_pbx_start) autoloop has started, set requestHangup = requestQueueHangup
 SEPE0D173E11D95: (sccp_pbx_softswitch) pbx started
 SEPE0D173E11D95: (sccp_pbx_softswitch) finished
 SCCP: (sccp_pbx_softswitch) exit
    -- Executing [98041@internal:1] NoOp("SCCP/98031-00000006", "98041") in new stack
    -- Executing [98041@internal:2] NoOp("SCCP/98031-00000006", "internal context called on Server B to 98041") in new stack
    -- Executing [98041@internal:3] Dial("SCCP/98031-00000006", "SCCP/98041, 20") in new stack
    -- SCCP: Asterisk asked us to create a channel with type=SCCP, format=(alaw), lineName=98041, options=
       > SCCP: requested Audio Codec in Skinny Format: G.711 A-law 64k
[Feb 13 14:03:45] NOTICE[3732][C-00000001]: ast116.c:1611 struct ast_channel *sccp_astwrap_request(const char *, struct ast_format_cap *, const struct ast_assigned_ids *, const struct ast_channel *, const char *, int *): Could not match video codec. No Video
       > SCCP: requested Video Codec in Skinny Format: No codec
    -- SCCP: (getSkinnyFormatMultiple) caps (alaw|ulaw)
       > remote audio caps: [alaw/64k]
       > remote video caps: []
    -- 0003: New channel number: 7 on line 98041
    -- SCCP: (pbx_channel_allocate) try to allocate Inbound channel on line: 98041
    -- SCCP:              cid_num: "98041"
    -- SCCP:             cid_name: "PHONE4"
    -- SCCP:          accountcode: "79005"
    -- SCCP:                exten: ""
    -- SCCP:              context: "internal"
    -- SCCP:             amaflags: "0"
    -- SCCP:            chan/call: "SCCP/98041-00000007"
    -- SCCP:combined capabilities: "[slin16, g722/64k, ulaw/64k, alaw/64k, g722/56k, g722/48k, g729b, g729ab, g729, g729a]"
[... 388 lines cut ...]
       > SEPE0D173E11D95: Stopping PBX video rtp transmission on channel SCCP/98031-00000006
    -- SEPE0D173E11D95: Set the active channel 0 on device
    -- SEPE0D173E11D95: Looking for selected channel (6)
    -- Destroying channel SCCP/98031-00000006
       > SCCP: Stopping PBX audio rtp transmission on channel SCCP/98031-00000006
       > SCCP: Stopping PBX video rtp transmission on channel SCCP/98031-00000006
    -- SCCP: destroying PBX rtp server on channel SCCP/98031-00000006
    -- SCCP: destroying PBX vrtp server on channel SCCP/98031-00000006
[... cut ...]
    -- SEP0023043403F9: Call Statistics:
       [
         Last Call        : CallID: 7 Packets sent: 1058 rcvd: 1047 lost: 0 jitter: 0 latency: 0
         Last Quality     : MLQK=4.5000;MLQKav=4.5000;MLQKmn=4.5000;MLQKmx=4.5000;MLQKvr=0.95|ICR=0.0000;CCR=0.0000;ICRmx=0.0000|CS=0;SCS=0
         Mean Statistics  : #Calls: 3 Packets sent: 510 rcvd: 499 lost: 0 jitter: 0 latency: 0
         Mean Quality     : MLQK=1.5000;MLQKav=1.5000;MLQKmn=1.5000;MLQKmx=4.5000;MLQKvr=0.95|ICR=0.0039;CCR=0.0039;ICRmx=0.0084|CS=0;SCS=0
       ]
```

ianmca, 2019-02-13:
It's something on the asterisk side... when I drop back to asterisk 13.24.1 with the same chan_sccp i'm getting transcoding working properly, but i'm seeing an error message appear... 

[Feb 13 15:06:42] ERROR[22683][C-00000000]: translate.c:1384 ast_translator_best_choice: Cannot determine best translation path since one capability supports no formats

However the behaviour is correct. The core show channels show that both are transcoding...

-- General --
           Name: SCCP/271-00000001
           Type: SCCP
       UniqueID: 1550070402.0
       LinkedID: 1550070402.0
      Caller ID: 271
 Caller ID Name: MY CID
Connected Line ID: 273
Connected Line ID Name: MY CID
Eff. Connected Line ID: 273
Eff. Connected Line ID Name: MY CID
    DNID Digits: 273
       Language: en
          State: Up (6)
  NativeFormats: (alaw)
    WriteFormat: slin16
     ReadFormat: slin16
 WriteTranscode: Yes (slin@16000)->(slin@8000)->(alaw@8000)
  ReadTranscode: Yes (alaw@8000)->(slin@8000)->(slin@16000)
 Time to Hangup: 0
   Elapsed Time: 0h7m8s
      Bridge ID: 1319d0dd-89de-4580-9173-fe9c08451b3a
 --   PBX   --
        Context: iantest
      Extension: 273
       Priority: 1
     Call Group: 26
   Pickup Group: 58
    Application: Dial
           Data: SCCP/273/ringer=inside
 Call Identifer: [C-00000000]
      Variables:
BRIDGEPEER=SCCP/273-00000002
DIALEDPEERNUMBER=273/ringer=inside
DIALEDPEERNAME=SCCP/273-00000002
DIALSTATUS=ANSWER
DIALEDTIME=
ANSWEREDTIME=
testvar2=my value
testvar=value
SKINNY_PRIVATE=0
SCCP_DEVICE_TYPE=Cisco 7965
SCCP_DEVICE_IP=10.1.4.171
SCCP_DEVICE_MAC=SEPA45630418BFC
  CDR Variables:
level 1: dnid=273
level 1: clid="MY CID" <271>
level 1: src=271
level 1: dst=273
level 1: dcontext=iantest
level 1: channel=SCCP/271-00000001
level 1: dstchannel=SCCP/273-00000002
level 1: lastapp=Dial
level 1: lastdata=SCCP/273/ringer=inside
level 1: start=1550070402.829189
level 1: answer=1550070409.966023
level 1: end=0.000000
level 1: duration=427
level 1: billsec=420
level 1: disposition=8
level 1: amaflags=3
level 1: accountcode=79011
level 1: uniqueid=1550070402.0
level 1: linkedid=1550070402.0
level 1: sequence=0

asterisk1*CLI> core show channel SCCP/273-00000002
 -- General --
           Name: SCCP/273-00000002
           Type: SCCP
       UniqueID: 1550070402.1
       LinkedID: 1550070402.0
      Caller ID: 273
 Caller ID Name: MY CID
Connected Line ID: 271
Connected Line ID Name: MY CID
Eff. Connected Line ID: 271
Eff. Connected Line ID Name: MY CID
    DNID Digits: (N/A)
       Language: en
          State: Up (6)
  NativeFormats: (g722)
    WriteFormat: slin16
     ReadFormat: slin16
 WriteTranscode: Yes (slin@16000)->(g722@16000)
  ReadTranscode: Yes (g722@16000)->(slin@16000)
 Time to Hangup: 0
   Elapsed Time: 0h7m31s
      Bridge ID: 1319d0dd-89de-4580-9173-fe9c08451b3a
 --   PBX   --
        Context: iantest
      Extension:
       Priority: 1
     Call Group: 26
   Pickup Group: 58
    Application: AppDial
           Data: (Outgoing Line)
 Call Identifer: [C-00000000]
      Variables:
BRIDGEPEER=SCCP/271-00000001
testvar=value
testvar2=my value
DIALEDPEERNUMBER=273/ringer=inside
SCCP_DEVICE_TYPE=Cisco 7965
SCCP_DEVICE_IP=10.1.4.169
SCCP_DEVICE_MAC=SEP002414B37BBD
  CDR Variables:
level 1: clid="MY CID" <273>
level 1: src=273
level 1: dcontext=iantest
level 1: channel=SCCP/273-00000002
level 1: lastapp=AppDial
level 1: lastdata=(Outgoing Line)
level 1: start=1550070402.841513
level 1: answer=1550070409.964925
level 1: end=1550070409.966582
level 1: duration=7
level 1: billsec=0
level 1: disposition=8
level 1: amaflags=3
level 1: accountcode=79011
level 1: uniqueid=1550070402.1
level 1: linkedid=1550070402.0
level 1: sequence=1

asterisk1*CLI>

Going to try asterisk 15

dkgroot, 2019-02-13:
@ianmca I think that might be a helpfull warning. I need to figure out where `ast_translator_best_choice` is being called from. Can you check if that warning appears during the RINGING indication (before you answer the call) (ie: from `pbx_retrieve_remote_capabilities`) ? I think that's the only time it's called.

ianmca, 2019-02-13:
Asterisk 15 (and 16) is now crashing when I put different codecs on the the two SCCP extensions...

   -- SEP002414B37BBD: Getting the active channel on device.
    -- SEP002414B37BBD: No active channel on device.
    -- SEP002414B37BBD: Indicate SCCP new state:RINGING, current channel state:DOWN on call:SCCP/273-00000002, lineInstance:1 (previous channelstate:DOWN)
    -- SEP002414B37BBD: Clear the display notify message
    -- SEP002414B37BBD: Clear the status prompt on line 1 and callid 0
    -- SEP002414B37BBD: Send and Set the call state ring-in(4) on call 2 (visibility:default)
    -- SEP002414B37BBD: Found matching linedevice. Aux parameter =
    -- SEP002414B37BBD: Send ringer mode Inside(2) on device
    -- SEP002414B37BBD: Normal ring occurred.
    -- SEP002414B37BBD: Set softkeyset to RINGIN(3) on line 1  and call 2
    -- SEP002414B37BBD: validKeyMask 4294967295
    -- SEP002414B37BBD: ( sccp_indicate.c:256:__sccp_indicate ) sccp_dev_displayprompt '▒MY CID (271)' for line 1 (8)
    -- SEP002414B37BBD: Display prompt on line 1, callid 2, timeout 8
    -- SEP002414B37BBD: Finish to indicate channel state:RINGING on call:SCCP/273-00000002, lineInstance:1. New channel state:RINGING
    -- SCCP: Set asterisk state Ringing (5) for call 2
    -- SCCP/273-00000002: (sccp_pbx_call) Returning: 0
    -- SCCP: (pbx_indicate) start indicate 'AST_CONTROL_CONNECTED_LINE: Indicate connected line has changed'
    -- SCCP/271-00000001: (pbx_indicate) start indicate 'AST_CONTROL_CONNECTED_LINE: Indicate connected line has changed'
 SEPA45630418BFC: (pbx_indicate) start indicate 'AST_CONTROL_CONNECTED_LINE: Indicate connected line has changed' (22) condition on channel SCCP/271-00000001 (rtp_instance:yes, receiveChannelState:0/No codec, mediaTransmissionState:0/No codec)
 SCCP/271-00000001: 0x7fed24000ca0: (getCallInfoStr):
 - calledParty: MY CID <273>, valid
 - callingParty: MY CID <271>, valid
 - originalCalledParty: MY CID <273>, valid, reason: 0
 - lastRedirectingParty: MY CID <271>, valid, reason: 0
 - presentation: CallerId Presentation Allowed

    -- SEPA45630418BFC: send callInfo on SCCP/271-00000001 with lineInstance: 1
 SEPA45630418BFC: (get_rtp_info) Asterisk requested EarlyRTP peer for channel SCCP/271-00000001
 SEPA45630418BFC: (get_vrtp_info) Asterisk requested EarlyRTP peer for channel SCCP/271-00000001
 SCCP: (get_rtp_info) Asterisk requested EarlyRTP peer for channel SCCP/273-00000002
 SCCP: (get_vrtp_info) Asterisk requested EarlyRTP peer for channel SCCP/273-00000002
    -- SCCP: (pbx_indicate) start indicate 'AST_CONTROL_RINGING: Remote end is ringing'
    -- SCCP/271-00000001: (pbx_indicate) start indicate 'AST_CONTROL_RINGING: Remote end is ringing'
 SEPA45630418BFC: (pbx_indicate) start indicate 'AST_CONTROL_RINGING: Remote end is ringing' (3) condition on channel SCCP/271-00000001 (rtp_instance:yes, receiveChannelState:0/No codec, mediaTransmissionState:0/No codec)
[Feb 13 15:31:38] ERROR[5991][C-00000001]: ast115.c:508 pbx_retrieve_remote_capabilities: FRACK!, Failed assertion user_data is NULL (0)
Got 19 backtrace records

Asterisk 16 now giving me this too... Will have a look again tomorrow, but now really confused.

ianmca, 2019-02-13:
Sorry i didn't see your last comment... will do what you suggest tomorrow morning.

dkgroot, 2019-02-13:
Please give the issues/450 a try tomorrow.
I noticed the crash as well :-) I am looking how to solve that, and will push another update to issue/450 when i find it.

ianmca, 2019-02-13:
i'll do that... do you want me to test issues/450 on asterisk 13 or a later version?

dkgroot, 2019-02-13:
the 13 that you had should be fine, just want to make sure it takes care of the warning message you got before.

ianmca, 2019-02-13:
Had a chance to do it today... I can't see any difference unfortunately. Here's my console output.

I definitely deleted all the old modules and i'm really sure i have built the right version  (chan-sccp-issues-450)

I've sent you more of the log in case it helps get context. It does seem to be during the ringing, and earlyrtp is progress.

Ian.

asterisk1*CLI> sccp call SEP002414B37BBD 271
Starting Call for Device: SEP002414B37BBD
    -- SEP002414B37BBD: Looking for line with instance 1.
    -- SCCP: Found line 273
SEP002414B37BBD: (getEmptyChannel) on line:273, maybe_c:
SEP002414B37BBD: (getEmptyChannel) getActiveChannel
    -- SEP002414B37BBD: Getting the active channel on device.
    -- SEP002414B37BBD: No active channel on device.
 SCCP: Adding channel 1 to line 273
    -- SEP002414B37BBD: Set the active channel 1 on device
    -- SEP002414B37BBD: Set the active line 273
    -- 273: New channel number: 1 on line 273
    -- SCCP: (pbx_channel_allocate) try to allocate Outbound channel on line: 273
    -- SCCP:              cid_num: "273"
    -- SCCP:             cid_name: "MY CID"
    -- SCCP:          accountcode: "79021"
    -- SCCP:                exten: ""
    -- SCCP:              context: "iantest"
    -- SCCP:             amaflags: "0"
    -- SCCP:            chan/call: "SCCP/273-00000001"
    -- SCCP:combined capabilities: "[slin16, g722/64k, ulaw/64k, alaw/64k, g722/56k, g722/48k, ilbc, g729b, g729ab, g729, g729a]"
    -- SCCP:  reduced preferences: "[g722/64k, g722/56k, g722/48k]"
    -- SCCP: (allocPBXChannel) Create New Channel with name: SCCP/273-00000001
    -- 273: Allocated asterisk channel SCCP/273-00000001
    -- SEP002414B37BBD: Set asterisk state OffHook (2) for call 1
    -- SEP002414B37BBD: Indicate SCCP new state:SPEEDDIAL, current channel state:DOWN on call:SCCP/273-00000001, lineInstance:1 (previous channelstate:DOWN)
    -- SEP002414B37BBD: Clear the display notify message
    -- SEP002414B37BBD: Clear the status prompt on line 0 and callid 0
    -- SEP002414B37BBD: Send ringer mode Ring Off(1) on device
    -- SEP002414B37BBD: Send speaker mode 'on'
    -- SEP002414B37BBD: Send and Set the call state offhook(1) on call 1 (visibility:default)
    -- SEP002414B37BBD: Set softkeyset to OFFHOOK(4) on line 1  and call 1
    -- SEP002414B37BBD: validKeyMask 4294967295
    -- SEP002414B37BBD: Send activate call plane on line 1
    -- SEP002414B37BBD: Finish to indicate channel state:SPEEDDIAL on call:SCCP/273-00000001, lineInstance:1. New channel state:OFFHOOK
    -- SEP002414B37BBD: (sccp_pbx_softswitch) New call on line 273
    -- SEP002414B37BBD: Indicate SCCP new state:DIALING, current channel state:OFFHOOK on call:SCCP/273-00000001, lineInstance:1 (previous channelstate:DOWN)
    -- SEP002414B37BBD: Stop tone on line 1 with callid 1
    -- SEP002414B37BBD: Send the dialed number:271, callid:1, lineInstance:1
    -- SEP002414B37BBD: Send and Set the call state proceed(12) on call 1 (visibility:default)
    -- SEP002414B37BBD: Finish to indicate channel state:DIALING on call:SCCP/273-00000001, lineInstance:1. New channel state:DIALING
    -- +- pbx extension matcher (271            ): ---+
    -- |ignore     |exists     |can match  |match more|
    -- | no        |yes        |yes        | no       |
    -- +----------------------------------------------+
 SEP002414B37BBD: (sccp_pbx_softswitch) channel SCCP/273-00000001 is dialing number 271
    -- SEP002414B37BBD: Set asterisk state Ring (4) for call 1
 SEP002414B37BBD: (sccp_pbx_softswitch) finished
 SCCP: (sccp_pbx_softswitch) exit
    -- SCCP: Asterisk asked us to create a channel with type=SCCP, format=(g722), lineName=271, options=ringer=inside
 SCCP: Adding channel 2 to line 271
    -- 1000: New channel number: 2 on line 271
    -- SCCP: (pbx_channel_allocate) try to allocate Inbound channel on line: 271
    -- SCCP:              cid_num: "271"
    -- SCCP:             cid_name: "MY CID"
    -- SCCP:          accountcode: "79011"
    -- SCCP:                exten: ""
    -- SCCP:              context: "iantest"
    -- SCCP:             amaflags: "0"
    -- SCCP:            chan/call: "SCCP/271-00000002"
    -- SCCP:combined capabilities: "[slin16, g722/64k, ulaw/64k, alaw/64k, g722/56k, g722/48k, ilbc, g729b, g729ab, g729, g729a]"
    -- SCCP:  reduced preferences: "[alaw/64k]"
[... 246 lines cut ...]
    -- SEPA45630418BFC: Clear the status prompt on line 0 and callid 0
    -- SEPA45630418BFC: ( sccp_device.c:2020:sccp_dev_check_displayprompt ) sccp_dev_displayprompt '▒' for line 0 (0)
    -- SEPA45630418BFC: Display prompt on line 0, callid 0, timeout 0
    -- SEPA45630418BFC: Set softkeyset to ONHOOK(0) on line 0  and call 0
    -- SEPA45630418BFC: validKeyMask 4294967295
    -- SEPA45630418BFC: Accessory 'Speaker' is 'On Hook'
       > SEPA45630418BFC: keepalive:62, keepaliveinterval:58
    -- QualityStats: K=4.5000;MLQKav=4.5000;MLQKmn=4.5000;MLQKmx=4.5000;ICR=0.0000;CCR=0.0000;ICRmx=0.0000;CS=0;SCS=0;MLQKvr=0.95
    -- SEP002414B37BBD: Call Statistics:
       [
         Last Call        : CallID: 1 Packets sent: 115363840 rcvd: 115495936 lost: 0 jitter: 0 latency: 28416
         Last Quality     : MLQK=0.0000;MLQKav=0.0000;MLQKmn=0.0000;MLQKmx=0.0000;MLQKvr=0.00|ICR=0.0000;CCR=0.0000;ICRmx=0.0000|CS=0;SCS=0
         Mean Statistics  : #Calls: 1 Packets sent: 115363840 rcvd: 115495936 lost: 0 jitter: 0 latency: 28416
[... cut ...]
         Last Call        : CallID: 2 Packets sent: 115584000 rcvd: 115275776 lost: 0 jitter: 0 latency: 28416
         Last Quality     : MLQK=0.0000;MLQKav=0.0000;MLQKmn=0.0000;MLQKmx=0.0000;MLQKvr=0.00|ICR=0.0000;CCR=0.0000;ICRmx=0.0000|CS=0;SCS=0
         Mean Statistics  : #Calls: 1 Packets sent: 115584000 rcvd: 115275776 lost: 0 jitter: 0 latency: 28416
         Mean Quality     : MLQK=0.0000;MLQKav=0.0000;MLQKmn=0.0000;MLQKmx=0.0000;MLQKvr=0.00|ICR=0.0000;CCR=0.0000;ICRmx=0.0000|CS=0;SCS=0
       ]
asterisk1*CLI>

dkgroot, 2019-02-13:
Just now pushed another update which fixes the crash and it looks like it does a better job dealing with the transcoding case as well.

ianmca, 2019-02-14:
Hi, yes no problems connecting to the chat... but here's the initial testing.

Demo config files... Make & receive calls across 2 codecs to enfore transcoding (one SCCP phone forced to G722, one to G711a)... Works

Reloaded own test configs, including SIP trunk to our live asterisk.
Make & receive calls across 2 codecs. Works.
Incoming G711a SIP to G722 SCCP. Works

Move both SCCP phones onto G722...
Incoming SIP G711 call onto G722 only phone. Works.
Transfer SIP call onto second G722 phone. Works

Incoming SIP G711 call onto G722 only phone.
Now blind transfer SIP call onto second G722 phone. Works.

This looks very successful.

dkgroot, 2019-02-14:
@ianmca, sounds like time to merge this one too, thanks for reporting and helping out with the testing.
Did you these tests in both asterisk-13 and asterisk-16 (because of the backport) ?

ianmca, 2019-02-14:
@dkgroot I haven't tested on asterisk 13 I can do.  What should I be looking for on 13? Is it the removal of that warning message?  Thanks so much for your debugging and your development.  Hopefully the fix will help lots of people as well as me!

dkgroot, 2019-02-14:
@ianmca:
- I just merged the changes back into develop (this also removed the 450 branch).
- Removed the temporarily added debug output
- The check for asterisk-13 would just be to make sure there are no regressions

commit 0a10f87d (2019-02-14): Turn off temporary debug

fixes #450
commit 7e777670 (2019-02-14): commit 2e5b7176b7aa5002499a72a1b78ef9b40195c6de
Author: Diederik de Groot <dkgroot@talon.nl>
Date:   Wed Feb 13 18:10:46 2019 +0100

    Revert accidental checking of sccp_channel.c changes (for transfer).
    Fix pbx_retrieve_remote_capabilities when dealing with transcoding.
    Fix ast_translator_best_choice warning
    Fix scanbuild warning.

    fixes #450
commit aefbeaea (2019-02-14): commit 2e5b7176b7aa5002499a72a1b78ef9b40195c6de
Author: Diederik de Groot <dkgroot@talon.nl>
Date:   Wed Feb 13 18:10:46 2019 +0100

    Fix pbx_retrieve_remote_capabilities to deal with transcoding correctly.
    Fix ast_translator_best_choice warning
    Unrelated Fix scanbuild warning.

    fixes #450
closed 2019-02-14 by dkgroot