chan-sccp/chan-sccp #451: Unknown Dependencies

aselafernando, 2019-02-24:
Latest git, v4.3.2-gamma & v4.3.2-beta fail to load in Asterisk 16.2.0 with the error:
```
[Feb 24 10:51:20] ERROR[974]: loader.c:170 module_load_error: chan_sccp has one or more unknown dependencies.
```
v4.3.2-alpha loads fine

Asterisk & Chan SCCP are both cross compiled for Cortex A8 ARM, no errors when cross compiling

modules.conf attached

[modules.txt](https://github.com/chan-sccp/chan-sccp/files/2897428/modules.txt)

dkgroot, 2019-02-25:
@aselafernando 
I changed the required module for chan-sccp to:
- app_voicemail
- bridge_simple
- bridge_native_rtp
- bridge_softmix
- bridge_holding
- res_stasis
- res_stasis_device_state

To prevent other issues some people experienced during transfer/conference etc. So i you can make sure these modules are loaded, you should be ok. I had hoped the asterisk warning about missing requirements would have been a little more informing. I will see if something can be done about this.

dkgroot, 2019-02-25:
I have updated this [wiki page](https://github.com/chan-sccp/chan-sccp/wiki/setup-asterisk) to include the same requirements. Also added this list to the README.md front page.

aselafernando, 2019-02-26:
I just checked the modules on 16.2.0 and res_stasis and res_stasis_device_state are included in the base installation.
I will cross compile the latest git release of Chan SCCP this Thursday and will check if it loads with all the required modules listed.
When you say older version of asterisk - 16.2.0 is the latest :P

Haha yeah it was a bit of a struggle at the start to work out how to cross compile Asterisk, Chan SCCP and their dependencies for ARM, but I have now been running on the ARM for about 4 years. The platform is a BeagleBone Black running ArchLinux ARM.

dkgroot, 2019-02-26:
16.2.0 is latest, woops :-)

res_stasis is required for app_voicemail to be able to communicate correctly. res_stasis_device_state is also required to be able to query/update device state (related to hints). It's a little strange that asterisk was willing to work correctly without those two :-)

FYI: You can also run asterisk:module.conf with autoload=yes. It will load all modules for which there is a valid config file in /etc/asterisk. So if you don't want pjsip you just (re)move it's config file and the module will try to load, and fail because of a missing config file. That might be a better way of managing the modules you run on your system. You can also use "noload => xxxx.so" for modules that you want to deny explicitly. You might want to switch to this system/method in the future.

You should not have to recompile chan_sccp to test this. You only need to add the two modules and restart asterisk for chan_sccp to be able to load. Requirements are not checked during compilation, only during module loading.

BTW i run asterisk+chan_sccp on openwrt (cross compiled for MIPS32). Having chan_sccp work on multiple cpu's is a good source test / check. Would you be interested in writing a little wiki article about cross compiling and running on Arm/BeagleBone ?

aselafernando, 2019-03-06:
Tested and all is working now when loading all the modules you specified :)

I actually created a script some time ago that does the asterisk compile and the chan-sccp compile. To be fair the chan-sccp cross compile is pretty simple once asterisk is compiled.

I use the crosstool-ng for installing the cross compiling toolchain. There's no reason why by modifying the variable TARGETMACH that you could compile for any architecture as long as the cross compiling toolchain is installed.

https://github.com/aselafernando/AsteriskCrossCompile/

closed 2019-03-06 by aselafernando

dkgroot, 2019-03-06:
Thanks for checking and the update !

Very clean looking script :-)
FYI: the script's setting of  asterisk.conf is correct, but the README has " = " instead of " => " for the path entries (which is wrong AFAIK). This is an ancient asterisk-1.8 mistake, which you might want to correct.

aselafernando, 2019-03-06:
Haha thanks and that's right mixed up the = and =>!
Cheers for the pull request. It has been merged :) 👍 

ArchLinux also has the AUR for compiling Asterisk https://aur.archlinux.org/packages/asterisk however I don't use it as compiling on the BeagleBone Black (BBB) would be super slow, and it only has 2GB of internal space haha

What I have done is inserted a microSD card in the BBB which stores the asterisk install. That way I can keep the ArchLinux install on the internal storage mounted as read only and be able to have asterisk write to it's own directories and I can easily update it. By being read only it can easily survive power failures a lot better :)