chan-sccp/chan-sccp #452: EndCall SoftKey on the receiver rejects the calls and hangs the call, but does not hangs up the call initiator

IntergalacticQuasar, 2019-03-02:
When phone A initiates a call to phone B and makes the phone B ring, if phone B wants to reject the call before answering it, by pressing EndCall softkey, phone B stops ringing and the call is hanged up on phone B, but phone A doesn't hang up the call. Even though phone A doesn't immediately hang up the call when EndCall softkey is pressed on phone B, it eventually does so after 10 seconds, and during that 10 seconds phone A is completely silent (no Ringback tone).

On the asterisk console, this manifest it self by the following line:

<pre>
...
0x7f08e455de20 -- Strict RTP learning after remote address set to: 192.168.3.244:21962
...
</pre>
Where `192.168.3.244` is the address of phone A (call initiator)

I found a workaround for this particular problem, but it introduces a new problem.
By changing the **Early RTP** setting to one of _Offhook_, _Dial_, or _Ringout_, the problem goes away and there is noting on the Asterisk console; however doing so causes the call initiator (phone A) to lose its **Ringback tone**.

**Reproduction Steps:**

1. Make a call phone phone A to phone B
2. Reject the call on phone B before answering it by pressing softkey **EndCall** 

**chan-sccp version:** Sccp ver.4.3.2 r433 Revision :10570 Hash :116da34  Develop branck)

**asterisk version:** 13.18.3

**Expected behavior:**

By rejecting the call on the receiver phone (phone B) by pressing softkey EndCall when the phone is ringing, phone B must stop ringing, and the call must be hanged up for both callee (phone B) and caller (phone A).

**Observed behavior:**
Callee's behaviour is normal, but caller's phone doesn't hang up the call after 10 seconds. During this period, it is completely silent with no Ringback tone.
Asterisk console also shows the following line:

```
...
0x7f08e455de20 -- Strict RTP learning after remote address set to: 192.168.3.244:21962
...
```

_complete asterisk console output:_

<pre>
VoipServer*CLI>
  == Using SCCP RTP TOS bits 184
  == Using SCCP RTP CoS mark 6
    -- Executing [101@from-internal:1] Set("SCCP/104-0000020F", "__RINGTIMER=15") in new stack
    -- Executing [101@from-internal:2] Macro("SCCP/104-0000020F", "exten-vm,novm,101,0,0,0") in new stack
    -- Executing [s@macro-exten-vm:1] Macro("SCCP/104-0000020F", "user-callerid,") in new stack
    -- Executing [s@macro-user-callerid:1] Set("SCCP/104-0000020F", "TOUCH_MONITOR=1551550689.558") in new stack
    -- Executing [s@macro-user-callerid:2] Set("SCCP/104-0000020F", "AMPUSER=104") in new stack
    -- Executing [s@macro-user-callerid:3] GotoIf("SCCP/104-0000020F", "0?report") in new stack
    -- Executing [s@macro-user-callerid:4] ExecIf("SCCP/104-0000020F", "1?Set(REALCALLERIDNUM=104)") in new stack
    -- Executing [s@macro-user-callerid:5] Set("SCCP/104-0000020F", "AMPUSER=104") in new stack
    -- Executing [s@macro-user-callerid:6] GotoIf("SCCP/104-0000020F", "0?limit") in new stack
    -- Executing [s@macro-user-callerid:7] Set("SCCP/104-0000020F", "AMPUSERCIDNAME=104 Kishka") in new stack
    -- Executing [s@macro-user-callerid:8] ExecIf("SCCP/104-0000020F", "0?Set(__CIDMASQUERADING=TRUE)") in new stack
    -- Executing [s@macro-user-callerid:9] GotoIf("SCCP/104-0000020F", "0?report") in new stack
    -- Executing [s@macro-user-callerid:10] Set("SCCP/104-0000020F", "AMPUSERCID=104") in new stack
    -- Executing [s@macro-user-callerid:11] Set("SCCP/104-0000020F", "__DIAL_OPTIONS=HhTtr") in new stack
    -- Executing [s@macro-user-callerid:12] Set("SCCP/104-0000020F", "CALLERID(all)="104 Kishka" <104>") in new stack
    -- Executing [s@macro-user-callerid:13] GotoIf("SCCP/104-0000020F", "0?limit") in new stack
    -- Executing [s@macro-user-callerid:14] ExecIf("SCCP/104-0000020F", "0?Set(GROUP(concurrency_limit)=104)") in new stack
    -- Executing [s@macro-user-callerid:15] NoOp("SCCP/104-0000020F", "Macro Depth is 2") in new stack
    -- Executing [s@macro-user-callerid:16] GotoIf("SCCP/104-0000020F", "1?report2:macroerror") in new stack
    -- Goto (macro-user-callerid,s,17)
    -- Executing [s@macro-user-callerid:17] GotoIf("SCCP/104-0000020F", "0?continue") in new stack
    -- Executing [s@macro-user-callerid:18] Set("SCCP/104-0000020F", "__TTL=64") in new stack
    -- Executing [s@macro-user-callerid:19] GotoIf("SCCP/104-0000020F", "1?continue") in new stack
    -- Goto (macro-user-callerid,s,35)
    -- Executing [s@macro-user-callerid:35] Set("SCCP/104-0000020F", "CALLERID(number)=104") in new stack
    -- Executing [s@macro-user-callerid:36] Set("SCCP/104-0000020F", "CALLERID(name)=104 Kishka") in new stack
    -- Executing [s@macro-user-callerid:37] GotoIf("SCCP/104-0000020F", "0?cnum") in new stack
[... 156 lines cut ...]
    -- Executing [s-CHANUNAVAIL@macro-exten-vm:1] GotoIf("SCCP/104-0000020F", "0?exit,1") in new stack
    -- Executing [s-CHANUNAVAIL@macro-exten-vm:2] PlayTones("SCCP/104-0000020F", "congestion") in new stack
    -- Executing [s-CHANUNAVAIL@macro-exten-vm:3] Congestion("SCCP/104-0000020F", "10") in new stack
<b><I>
       > 0x7f08e455de20 -- Strict RTP learning after remote address set to: 192.168.3.244:20640
</I></b>
VoipServer*CLI>
VoipServer*CLI>
VoipServer*CLI>
VoipServer*CLI>
VoipServer*CLI>
  == Spawn extension (macro-exten-vm, s-CHANUNAVAIL, 3) exited non-zero on 'SCCP/104-0000020F' in macro 'exten-vm'
  == Spawn extension (from-internal, 101, 2) exited non-zero on 'SCCP/104-0000020F'
    -- Executing [h@from-internal:1] Macro("SCCP/104-0000020F", "hangupcall") in new stack
    -- Executing [s@macro-hangupcall:1] GotoIf("SCCP/104-0000020F", "1?theend") in new stack
    -- Goto (macro-hangupcall,s,3)
    -- Executing [s@macro-hangupcall:3] ExecIf("SCCP/104-0000020F", "0?Set(CDR(recordingfile)=)") in new stack
    -- Executing [s@macro-hangupcall:4] NoOp("SCCP/104-0000020F", " monior file= ") in new stack
    -- Executing [s@macro-hangupcall:5] AGI("SCCP/104-0000020F", "attendedtransfer-rec-restart.php,,") in new stack
    -- Launched AGI Script /usr/share/asterisk/agi-bin/attendedtransfer-rec-restart.php
    -- <SCCP/104-0000020F>AGI Script attendedtransfer-rec-restart.php completed, returning 0
    -- Executing [s@macro-hangupcall:6] Hangup("SCCP/104-0000020F", "") in new stack
  == Spawn extension (macro-hangupcall, s, 6) exited non-zero on 'SCCP/104-0000020F' in macro 'hangupcall'
  == Spawn extension (from-internal, h, 1) exited non-zero on 'SCCP/104-0000020F'
</pre>

dkgroot, 2019-03-02:
Thanks for the detailed problem explanation. I will try to replicate the issue on Monday/Tuesday and see how we can fix the issue.

IntergalacticQuasar, 2019-03-02:
Thanks for a very quick reply. Really looking forward to your diagnostics.
By the way, really awesome project. Absolutely amazing 👍 :)

dkgroot, 2019-03-02:
Thanks !
Don't worry about the "Strict RTP Learning". That's part of the normal behaviour after starting an RTP-Server (even if there is no connection yet).
The 'normal' behaviour (for chan-sccp) is to play back busy or congestion when the remote party hangs up for about 10 seconds, before hanging up the caller (so that have an audio clue about what is going on). This does seem to be tried by your dialplan (The line before Strict RTP Learning) says 
```
    -- Executing [s-CHANUNAVAIL@macro-exten-vm:2] PlayTones("SCCP/104-0000020F", "congestion") in new stack
    -- Executing [s-CHANUNAVAIL@macro-exten-vm:3] Congestion("SCCP/104-0000020F", "10") in new stack
```

We are seeing a lot of the dialplan debug output, but not a lot of chan-sccp interaction. Can you replace the debug log above with one where "sccp debug core, channel, indicate, device, pbx, rtp, codec" set. I might be able to shed some light without having to reproduce issue (at the moment). I am specifically interested if the caller opened an rtp openreceive channel on the phone so that it can hear what's going on. Normally earlyrtp=progress, will do this during the "Progress" stage of the call, which should happen shortly after "Ringout". The FreePBX dialplan might be doing this differently though. We should make sure we have the ability to hear audio during the 10 second waiting time (during busy/congestion playback etc), so that is something i need to check.

Allowing you to change earlyrtp to some other value is mostly a backward compatibility thing. It should remain on progress for 99% of the cases.

IntergalacticQuasar, 2019-03-02:
Sorry for not including chan_sccp debug logs. Here it is:

**Caller phone (phone A) offhook:**

```
Connected to Asterisk 13.18.3~dfsg-1ubuntu4 currently running on VoipServer (pid = 1521)
VoipServer*CLI>
    -- SEP0022900312B4: Accessory 'Speaker' is 'Off Hook'
    -- SEP0022900312B4: Getting the active channel on device.
    -- SEP0022900312B4: No active channel on device.
    -- SEP0022900312B4: Device State is 'Off Hook'
    -- SCCP: Looking for channel by state '22' on device: SEP0022900312B4
    -- SEP0022900312B4: line: '104'
    -- SEP0022900312B4: Could not find active channel with state RINGING(22) on device
    -- SEP0022900312B4: Looking for line with instance 1.
    -- SCCP: Found line 104
    -- SEP0022900312B4: Getting the active channel on device.
    -- SEP0022900312B4: No active channel on device.
 SCCP: Adding channel 557 to line 104
    -- SEP0022900312B4: Set the active channel 557 on device
    -- SEP0022900312B4: Set the active line 104
    -- 5: New channel number: 557 on line 104
    -- SCCP: (pbx_channel_allocate) try to allocate Outbound channel on line: 104
    -- SCCP:              cid_num: "104"
    -- SCCP:             cid_name: "104 Kishka"
    -- SCCP:          accountcode: "104"
    -- SCCP:                exten: ""
    -- SCCP:              context: "from-internal"
    -- SCCP:             amaflags: "0"
    -- SCCP:            chan/call: "SCCP/104-0000022D"
    -- SCCP:combined capabilities: "[slin16, g722/64k, ulaw/64k, alaw/64k, g722/56k, g722/48k, g729b, g729ab, g729, g729a]"
    -- SCCP:  reduced preferences: "[g722/64k, g722/56k, g722/48k, ulaw/64k, alaw/64k]"
    -- SCCP: (allocPBXChannel) Create New Channel with name: SCCP/104-0000022D
    -- SCCP/104-0000022D: rtp server instance created at 0.0.0.0:14476
    -- SCCP/104-0000022D: (sccp_rtp_createRtpServer) updated phone Audio RTP destination to : 192.168.3.246:14476, family:IPv4, mapped: False
    -- 5: Allocated asterisk channel SCCP/104-0000022D
    -- SEP0022900312B4: Set asterisk state OffHook (2) for call 557
    -- SEP0022900312B4: Indicate SCCP new state:OFFHOOK, current channel state:DOWN on call:SCCP/104-0000022D, lineInstance:1 (previous channelstate:DOWN)
    -- 104 (getLinestate) state:ONHOOK, party:/, calltype:INACTIVE
       > SCCP: (devicestate) sccp_hint returned state:ONHOOK for '104'
    -- SEP0022900312B4: (sccp_device_suppressMWI)
       > SCCP: (devicestate) PBX requests state for '104' - state Not in use
    -- SEP0022900312B4: Send speaker mode 'on'
    -- SEP0022900312B4: Send and Set the call state offhook(1) on call 557 (visibility:default)
    -- SEP0022900312B4: Send activate call plane on line 1
    -- SEP0022900312B4: ( sccp_indicate.c:88:__sccp_indicate ) sccp_dev_displayprompt '� ' for line 1 (8)
    -- SEP0022900312B4: Display prompt on line 1, callid 557, timeout 8
    -- SEP0022900312B4: Sending tone Inside Dial Tone (33) on line 1 with callid 557 (direction: User)
    -- SEP0022900312B4: softkey 'Redial' on OFFHOOK to on
       > SEP0022900312B4: found softkey 'Redial' at 0
    -- SEP0022900312B4: softkey 'Confrn' on OFFHOOK to on
    -- SEP0022900312B4: softkey 'Join' on OFFHOOK to off
    -- SEP0022900312B4: softkey 'ConfList' on OFFHOOK to on
    -- SEP0022900312B4: softkey 'Record' on OFFHOOK to off
    -- SEP0022900312B4: Set softkeyset to OFFHOOK(4) on line 1  and call 557
    -- SEP0022900312B4: validKeyMask 4294967295
    -- SEP0022900312B4: Finish to indicate channel state:OFFHOOK on call:SCCP/104-0000022D, lineInstance:1. New channel state:OFFHOOK
       > SEP0022900312B4: keepalive:120, keepaliveinterval:59
    -- Processing Event 0x7f08e452e130 of Type LineStatus Changed via 0 callback:0x7f0873d56f2b
       > 104 (hint_updateLineState) Update Line Channel State: ONHOOK(1)
       > 104: get speeddial party: ' <>' (calledParty)
       > 104 (hint_updateLineStateForSingleChannel) partyName: , partyNumber:
       > 104 (hint_updateLineStateForSingleChannel) Set singleLineState to OFFHOOK(10)
    -- SCCP: (sccp_hint_notifySubscribersViaPbx) Notify asterisk to set state to sccp channelstate 'OFFHOOK' (10) => asterisk: 'Device is in use' (2) on channel SCCP/104
    -- SCCP: (sccp_hint_notifyLineStateUpdate) Notified asterisk to set state to sccp channelstate 'OFFHOOK' (10) => asterisk: 'Device is in use' (2) on channel SCCP/104
```

**Dialling callee (phone B) from phone A:**
```
VoipServer*CLI>
[... 302 lines cut ...]
    -- SEP0022900312B4: Set softkeyset to ONHOOK(0) on line 0  and call 0
    -- SEP0022900312B4: validKeyMask 4294967295
    -- Destroying channel SCCP/104-0000022D
       > SCCP: Stopping PBX audio rtp transmission on channel SCCP/104-0000022D
    -- SCCP: destroying PBX rtp server on channel SCCP/104-0000022D
    -- 104 (getLinestate) state:ONHOOK, party:/, calltype:INACTIVE
       > SCCP: (devicestate) sccp_hint returned state:ONHOOK for '104'
       > SCCP: (devicestate) PBX requests state for '104' - state Not in use
       > SEP0022900312B4: keepalive:62, keepaliveinterval:59
    -- SEP0022900312B4: Clear the display notify message
    -- SEP0022900312B4: Clear the status prompt on line 0 and callid 0
    -- SEP0022900312B4: ( sccp_device.c:2023:sccp_dev_check_displayprompt ) sccp_dev_displayprompt '�' for line 0 (0)
    -- SEP0022900312B4: Display prompt on line 0, callid 0, timeout 0
    -- SEP0022900312B4: softkey 'Redial' on ONHOOK to on
       > SEP0022900312B4: found softkey 'Redial' at 0
    -- SEP0022900312B4: softkey 'Confrn' on ONHOOK to on
    -- SEP0022900312B4: softkey 'Join' on ONHOOK to off
    -- SEP0022900312B4: softkey 'ConfList' on ONHOOK to on
    -- SEP0022900312B4: softkey 'Record' on ONHOOK to off
    -- SEP0022900312B4: Set softkeyset to ONHOOK(0) on line 0  and call 0
    -- SEP0022900312B4: validKeyMask 4294967295
    -- SEP0022900312B4: Accessory 'Speaker' is 'On Hook'
```

Hope these information was what you were looking for.

dkgroot, 2019-03-02:
Updated the original issue report with your debug output. Thanks ! And no need to be sorry :-)

dkgroot, 2019-03-02:
Looking at your debug log I noticed this:
We did open the receive channel on the caller.
But just before processing the Congestion, we ran sccp_channel_closeAllMediaTransmitAndReceive, closing the RTP stream again. This cause you not to hear any more signalling events. This should be corrected. Thanks for the detailed information. Will be helpfull when correcting the error, beginning of next week. The correction will take place in a branch called issues/452. I will let you know when it is available for testing/checking.

IntergalacticQuasar, 2019-03-02:
WOW! It was super fast. Really appreciate your determination. I am very grateful for your help. Thanks for taking your time looking into this issue.

I'll try it and keep you posted. Just hope not to break anything dramatically 😉

dkgroot, 2019-03-02:
It was a tiny commit, and it just got smaller (pushed another commit), so the side effects should be contained :-)

IntergalacticQuasar, 2019-03-02:
This was my first issue reporting ever. I always was pissed off by people not spending enough time preparing the issue report and expecting a perfect fix in no time, so tried my best to explain the problem as clearly as possible. Really glad that I was able to communicate this across.

Following your directions, I applied the patch (at least I hope I have), but it hasn't made a difference. 
Here are the logs:

**Caller (phone A) off hook:**
```
VoipServer*CLI>
    -- SEP0022900312B4: Accessory 'Speaker' is 'Off Hook'
    -- SEP0022900312B4: Getting the active channel on device.
    -- SEP0022900312B4: No active channel on device.
    -- SEP0022900312B4: Device State is 'Off Hook'
    -- SCCP: Looking for channel by state '22' on device: SEP0022900312B4
    -- SEP0022900312B4: line: '104'
    -- SEP0022900312B4: Could not find active channel with state RINGING(22) on device
    -- SEP0022900312B4: Looking for line with instance 1.
    -- SCCP: Found line 104
    -- SEP0022900312B4: Getting the active channel on device.
    -- SEP0022900312B4: No active channel on device.
 SCCP: Adding channel 570 to line 104
    -- SEP0022900312B4: Set the active channel 570 on device
    -- SEP0022900312B4: Set the active line 104
    -- 5: New channel number: 570 on line 104
    -- SCCP: (pbx_channel_allocate) try to allocate Outbound channel on line: 104
    -- SCCP:              cid_num: "104"
    -- SCCP:             cid_name: "104 Kishka"
    -- SCCP:          accountcode: "104"
    -- SCCP:                exten: ""
    -- SCCP:              context: "from-internal"
    -- SCCP:             amaflags: "0"
    -- SCCP:            chan/call: "SCCP/104-0000023A"
    -- SCCP:combined capabilities: "[slin16, g722/64k, ulaw/64k, alaw/64k, g722/56k, g722/48k, g729b, g729ab, g729, g729a]"
    -- SCCP:  reduced preferences: "[g722/64k, g722/56k, g722/48k, ulaw/64k, alaw/64k]"
    -- SCCP: (allocPBXChannel) Create New Channel with name: SCCP/104-0000023A
    -- SCCP/104-0000023A: rtp server instance created at 0.0.0.0:14278
    -- SCCP/104-0000023A: (sccp_rtp_createRtpServer) updated phone Audio RTP destination to : 192.168.3.246:14278, family:IPv4, mapped: False
    -- 5: Allocated asterisk channel SCCP/104-0000023A
    -- 104 (getLinestate) state:ONHOOK, party:/, calltype:INACTIVE
       > SCCP: (devicestate) sccp_hint returned state:ONHOOK for '104'
       > SCCP: (devicestate) PBX requests state for '104' - state Not in use
    -- SEP0022900312B4: Set asterisk state OffHook (2) for call 570
    -- SEP0022900312B4: Indicate SCCP new state:OFFHOOK, current channel state:DOWN on call:SCCP/104-0000023A, lineInstance:1 (previous channelstate:DOWN)
    -- SEP0022900312B4: (sccp_device_suppressMWI)
    -- SEP0022900312B4: Send speaker mode 'on'
    -- SEP0022900312B4: Send and Set the call state offhook(1) on call 570 (visibility:default)
    -- SEP0022900312B4: Send activate call plane on line 1
    -- SEP0022900312B4: ( sccp_indicate.c:88:__sccp_indicate ) sccp_dev_displayprompt '� ' for line 1 (8)
    -- SEP0022900312B4: Display prompt on line 1, callid 570, timeout 8
    -- SEP0022900312B4: Sending tone Inside Dial Tone (33) on line 1 with callid 570 (direction: User)
    -- SEP0022900312B4: softkey 'Redial' on OFFHOOK to on
       > SEP0022900312B4: found softkey 'Redial' at 0
    -- SEP0022900312B4: softkey 'Confrn' on OFFHOOK to on
    -- SEP0022900312B4: softkey 'Join' on OFFHOOK to off
    -- SEP0022900312B4: softkey 'ConfList' on OFFHOOK to on
    -- SEP0022900312B4: softkey 'Record' on OFFHOOK to off
    -- SEP0022900312B4: Set softkeyset to OFFHOOK(4) on line 1  and call 570
    -- SEP0022900312B4: validKeyMask 4294967295
    -- SEP0022900312B4: Finish to indicate channel state:OFFHOOK on call:SCCP/104-0000023A, lineInstance:1. New channel state:OFFHOOK
       > SEP0022900312B4: keepalive:120, keepaliveinterval:59
    -- Processing Event 0x7f08e4011650 of Type LineStatus Changed via 0 callback:0x7f0873d56f2b
       > 104 (hint_updateLineState) Update Line Channel State: ONHOOK(1)
       > 104: get speeddial party: ' <>' (calledParty)
       > 104 (hint_updateLineStateForSingleChannel) partyName: , partyNumber:
       > 104 (hint_updateLineStateForSingleChannel) Set singleLineState to OFFHOOK(10)
    -- SCCP: (sccp_hint_notifySubscribersViaPbx) Notify asterisk to set state to sccp channelstate 'OFFHOOK' (10) => asterisk: 'Device is in use' (2) on channel SCCP/104
    -- SCCP: (sccp_hint_notifyLineStateUpdate) Notified asterisk to set state to sccp channelstate 'OFFHOOK' (10) => asterisk: 'Device is in use' (2) on channel SCCP/104
```
**Dialling phone B from phone A:**
```
VoipServer*CLI>
[... 298 lines cut ...]
    -- SEP0022900312B4: softkey 'ConfList' on ONHOOK to on
    -- SEP0022900312B4: softkey 'Record' on ONHOOK to off
    -- SEP0022900312B4: Set softkeyset to ONHOOK(0) on line 0  and call 0
    -- SEP0022900312B4: validKeyMask 4294967295
    -- Destroying channel SCCP/104-0000023A
       > SCCP: Stopping PBX audio rtp transmission on channel SCCP/104-0000023A
    -- SCCP: destroying PBX rtp server on channel SCCP/104-0000023A
    -- 104 (getLinestate) state:ONHOOK, party:/, calltype:INACTIVE
       > SCCP: (devicestate) sccp_hint returned state:ONHOOK for '104'
       > SCCP: (devicestate) PBX requests state for '104' - state Not in use
    -- SEP0022900312B4: Clear the display notify message
    -- SEP0022900312B4: Clear the status prompt on line 0 and callid 0
[... cut ...]
       > SEP0022900312B4: found softkey 'Redial' at 0
    -- SEP0022900312B4: softkey 'Confrn' on ONHOOK to on
    -- SEP0022900312B4: softkey 'Join' on ONHOOK to off
    -- SEP0022900312B4: softkey 'ConfList' on ONHOOK to on
    -- SEP0022900312B4: softkey 'Record' on ONHOOK to off
    -- SEP0022900312B4: Set softkeyset to ONHOOK(0) on line 0  and call 0
    -- SEP0022900312B4: validKeyMask 4294967295
    -- SEP0022900312B4: Accessory 'Speaker' is 'On Hook'
       > SEP0022900312B4: keepalive:62, keepaliveinterval:59
```

dkgroot, 2019-03-02:
I noticed this line:
```
    -- SEP0022900312B4: (receiveChannelOpen) Channel is already terminating. Giving up... (CONGESTION)
```
Which should not be possible after the patch. I will test the patch myself, to confirm.

If need be you can create a new test clone, using `git checkout -b issues/452 https://github.com/chan-sccp/chan-sccp chan-sccp_452`. If that is easier :-)

IntergalacticQuasar, 2019-03-02:
Well, it is past 3:00am here and I'm like 75% sleep! No wonder if I've screwed something up!
I'll definitely look into it more carefully tomorrow and keep you posted.

Cheers mate

dkgroot, 2019-03-02:
Sleep tight for now, we will continue tomorrow. Could also be i missed/overlooked something. I was not able to test my patch just yet (other stuff going on).

dkgroot, 2019-03-03:
My Call Log with issue/452:

98031 Calls 98041
```
    -- +- pbx extension matcher (98041          ): ---+
    -- |ignore     |exists     |can match  |match more|
    -- | no        |yes        |yes        | no       |
    -- +----------------------------------------------+
 SEPE0D173E11D95: (sccp_pbx_softswitch) channel SCCP/98031-00000006 is dialing number 98041
    -- SEPE0D173E11D95: Set asterisk state Ring (4) for call 6
    -- SCCP/98031-00000006: (pbx_pbx_start) autoloop has started, set requestHangup = requestQueueHangup
 SEPE0D173E11D95: (sccp_pbx_softswitch) pbx started
    -- SEPE0D173E11D95: softkey 'Redial' on ONHOOK to on
       > SEPE0D173E11D95: found softkey 'Redial' at 0
    -- SEPE0D173E11D95: softkey 'Redial' on OFFHOOK to on
       > SEPE0D173E11D95: found softkey 'Redial' at 0
    -- SEPE0D173E11D95: softkey 'Redial' on OFFHOOKFEAT to on
    -- SEPE0D173E11D95: softkey 'Redial' on ONHOOKSTEALABLE to on
       > SEPE0D173E11D95: found softkey 'Redial' at 0
 SEPE0D173E11D95: (sccp_pbx_softswitch) finished
 SCCP: (sccp_pbx_softswitch) exit
    -- Executing [98041@internal:1] NoOp("SCCP/98031-00000006", "98041") in new stack
    -- Executing [98041@internal:2] NoOp("SCCP/98031-00000006", "internal context called on Server B to 98041") in new stack
    -- Executing [98041@internal:3] Dial("SCCP/98031-00000006", "SCCP/98041, 20") in new stack
    -- SEPE0D173E11D95: Accessory 'Speaker' is 'Off Hook'
[Mar  3 01:46:20] NOTICE[11095][C-00000000]: app_dial.c:2512 dial_exec_full: SCCP/98031-00000006: (app_dial) NativeAudioFormats:(alaw|ulaw)
    -- SCCP: Asterisk asked us to create a channel with type=SCCP, format=(alaw), lineName=98041, options=
    -- SCCP: (getSkinnyFormatMultiple) caps (alaw|ulaw)
       > remote audio caps: [alaw/64k, ulaw/64k]
       > remote video caps: []
       > SCCP: requestedCodec in Skinny Format: audio:alaw/64k (video:)
 SCCP: Adding channel 7 to line 98041
    -- 0003: New channel number: 7 on line 98041
    -- SCCP: (pbx_channel_allocate) try to allocate Inbound channel on line: 98041
    -- SCCP:              cid_num: "98041"
    -- SCCP:             cid_name: "PHONE4"
    -- SCCP:          accountcode: "79005"
    -- SCCP:                exten: ""
    -- SCCP:              context: "internal"
    -- SCCP:             amaflags: "0"
    -- SCCP:            chan/call: "SCCP/98041-00000007"
    -- SCCP:combined capabilities: "[slin16, g722/64k, ulaw/64k, alaw/64k, g722/56k, g722/48k, g729b, g729ab, g729, g729a]"
    -- SCCP:  reduced preferences: "[g722/64k, g722/56k, g722/48k]"
    -- SCCP: (allocPBXChannel) Create New Channel with name: SCCP/98041-00000007
    -- allocPBXChannel: new->nativeformats=(g722) based on audio preferences
    -- allocPBXChannel: requestor->nativeformats=(alaw|ulaw)
    -- allocPBXChannel: joint formats=(nothing)
    -- allocPBXChannel: using nativeformats=(g722) fmt=g722
    -- 0003: Allocated asterisk channel SCCP/98041-00000007
 SCCP: (get_rtp_info) Asterisk requested EarlyRTP peer for channel SCCP/98041-00000007
 SCCP: (get_vrtp_info) Asterisk requested EarlyRTP peer for channel SCCP/98041-00000007
 SEPE0D173E11D95: (get_rtp_info) Asterisk requested EarlyRTP peer for channel SCCP/98031-00000006
 SEPE0D173E11D95: (get_vrtp_info) Asterisk requested EarlyRTP peer for channel SCCP/98031-00000006
    -- SCCP: Asterisk request to call SCCP/98041-00000007 (dest:98041, timeout: 0)
    -- 98041: Asterisk request to call SCCP/98041-00000007
    -- SCCP: (sccp_pbx_call) asterisk callerid='Diederik-Phone3 <98031>'
 SCCP/98041-00000007: 0x7f5090037630: (getCallInfoStr):
 - calledParty: PHONE4 <98041>, valid
 - callingParty: Diederik-Phone3 <98031>, valid
 - presentation: CallerId Presentation Allowed

    -- SEP0023043403F9: Getting the active channel on device.
    -- SEP0023043403F9: No active channel on device.
    -- SEP0023043403F9: Ringing Line: 98041 on device:SEP0023043403F9 using channel:SCCP/98041-00000007, ringermode:Outside
    -- SEP0023043403F9: Indicate SCCP new state:RINGING, current channel state:DOWN on call:SCCP/98041-00000007, lineInstance:1 (previous channelstate:DOWN)
    -- SEP0023043403F9: Clear the display notify message
    -- SEP0023043403F9: Clear the status prompt on line 1 and callid 0
    -- SEP0023043403F9: Send and Set the call state ring-in(4) on call 7 (visibility:default)
    -- SEP0023043403F9: Found matching linedevice. Aux parameter = 
    -- SEP0023043403F9: Send ringer mode Outside(3) on device
[... 163 lines cut ...]
    -- SEPE0D173E11D95: Finish to indicate channel state:ONHOOK on call:SCCP/98031-00000006, lineInstance:1. New channel state:DOWN
    -- SEPE0D173E11D95: Device is Requesting CallStatisticsAndClear
    -- SCCP: Cleaning channel SCCP/98031-00000006
    -- SCCP/98031-00000006: Disabling scheduler / Removing Scheduled tasks (digittimeout_id:-1) (hangup_id:6)
  == SEPE0D173E11D95: (sccp_channel_closeAllMediaTransmitAndReceive) Stop All Media Reception and Transmission on channel 6
       > SEPE0D173E11D95: Stopping PBX audio rtp transmission on channel SCCP/98031-00000006
SEPE0D173E11D95: (allowPickup) on line:98031.
    -- SEPE0D173E11D95: Set the active channel 0 on device
    -- SEPE0D173E11D95: Set the active line (NULL)
    -- SEPE0D173E11D95: Looking for selected channel (6)
    -- SEPE0D173E11D95: Set the active line (NULL)
    -- Destroying channel SCCP/98031-00000006
       > SCCP: Stopping PBX audio rtp transmission on channel SCCP/98031-00000006
[... cut ...]
         Last Call        : CallID: 6 Packets sent: 0 rcvd: 0 lost: 0 jitter: 0 latency: 0
         Last Quality     : MLQK=0.0000;MLQKav=0.0000;MLQKmn=0.0000;MLQKmx=0.0000;MLQKvr=0.00|ICR=0.0000;CCR=0.0000;ICRmx=0.0000|CS=0;SCS=0
         Mean Statistics  : #Calls: 3 Packets sent: 0 rcvd: 0 lost: 0 jitter: 0 latency: 0
         Mean Quality     : MLQK=0.0000;MLQKav=0.0000;MLQKmn=0.0000;MLQKmx=0.0000;MLQKvr=0.00|ICR=0.0000;CCR=0.0000;ICRmx=0.0000|CS=0;SCS=0
       ]
```
This seems to work fine (after the patch). I am guessing you did not switch branches correctly (somehow).

IntergalacticQuasar, 2019-03-03:
Great to here you've fixed it. Congratulations 👍 :) Mine is still broken though. I'm definitely doing something wrong. To be upfront, I'm on FreePBX 14.0.5.25, with SCCP_Manager 13.0.0.4 installed.

Now, I'm sure that I'm on the correct branch:
```
voip@VoipServer:/usr/src/chan-sccp$ git branch
  develop
* issues/452
```
Also, by looking at the timestamp of chan_sccp.so, it seems that it has been correctly replaced by the newer version:
```
voip@VoipServer:/usr/src/chan-sccp$ ls -alh /usr/lib/asterisk/modules/chan_sccp.so
-rwxr-xr-x 1 asterisk asterisk 5.7M Mar  3 12:12 /usr/lib/asterisk/modules/chan_sccp.so
```
But on the asterisk-cli, shows me that asterisk has picked up the old version:
```
VoipServer*CLI> sccp show version
Skinny Client Control Protocol (SCCP). Release: 4.3.2 develop - 116da34M (built by 'root' on '2019-02-26 20:40:00 UTC')
```
I've tried to reload modules, and also restarted the server couple of times, no luck. I'm suspecting I'm making an obvious and embarrassing mistake.

dkgroot, 2019-03-03:
@IntergalacticQuasar 
Intriguing...
Can you please log on to our [gitter chat](https://gitter.im/chan-sccp/chan-sccp) so i can help you directly and see if we can figure this out.

IntergalacticQuasar, 2019-03-03:
Sure, thanks for helping me with my particular case.  Highly appreciate it.

dkgroot, 2019-03-03:
See you in gitter in a moment :-)

dkgroot, 2019-03-03:
Thanks a lot Intergalactic for reporting the issue and all your help/testing/checking etc.
Issue has been resolved and the temporary branch has been merged back into the develop branch.

closed 2019-03-03 by dkgroot

IntergalacticQuasar, 2019-03-04:
My absolute pleasure. It was really great interacting with such a great and smart person about a fantastic project, and I'm really glad you found the bug worthy of fixing and helping me and others. 
Keep up the good work mate :-)

dkgroot, 2019-03-04:
No problem and happy we found a long lasting one in the process :-)

commit cf78ec10 (2019-03-03): Allow Congestion during OpenReceive, when RingOut call is terminated by remote party using EndCall Softkey

- Handling OpenReceiveChannel was cancelled when the remote party had already finished the call.
  This caused the inband congestion signal not to be heard by dialing party.
- Also fixes a long standing bug where a device w/could run out of rtp channels after some time
  (logically after having indicated congestion 2-4 times).

fixes #452
commit da511f5e (2019-03-03): Allow Congestion during OpenReceive, when RingOut call is terminated by remote party using EndCall Softkey

- Handling OpenReceiveChannel was cancelled when the remote party had already finished the call.
  This caused the inband congestion signal not to be heard by dialing party.
- Also fixes a long standing bug where a device w/could run out of rtp channels after some time
  (logically after having indicated congestion 2-4 times).

fixes #452