chan-sccp/chan-sccp #453: One way audio with g722 and g729

darkponay, 2019-03-06:
Hi everyone!
I have a little problem with using two codecs. My set up is a Alaw only trunk, Cisco 7975 and 7945 with XML enabled G722 and several phones over the network (some OpenVPN interconnected pfSense gateways).
One of us has a low bandwidth internet connection, and we have sometimes cracks, and a lot more while using conference. We tried to use G729 and it was a lot better for the quality of the communication and for his bandwidth.

**My goal :** Every phone in G722, except the ones behind a low bandwidth have to be in G729.

**The config:**
in sccp.conf, under _[general]_ : 
_disallow=all
allow=g722,g729_

**Nothing else in the config except one test device :** 
_disallow=all
allow=g729_

**What i have tried :** 
Say we have G729 phone and G722 one (in fact, allowed for G722 and 729)
**-Voicemail :** Ok (G729 only for this phone)
**-Outbound call :** Ok (G729 transcoded in Alaw for my trunk)
**-Internal (from G722 to G729) :** Quite ok. Both directions (but in G722, even on the G729-only phone)
**-Internal (from G729 to G722) :** Nok. One way audio. The G729 phone is muted, not the 722 one. G729 is shown on the G729 one, G722 on the other one.

**Asterisk version :** 
_Asterisk 16.2.1 built by root @ voice15 on a x86_64 running Linux on 2019-03-05 11:30:37 UTC_

**Chan-SCCP version :** 
_Release: 4.3.2 develop - a3d1b17M (built by 'root' on '2019-03-05 12:37:47 UTC')_

Logs are arriving soon

darkponay, 2019-03-06:
Separate message to avoid messing everything up...

**sccp show channels :**
```
+--- Channels ----------------------------------------------------------------------------------------------------------------------------------------------+
| ID    Name                      LineName   DeviceName       NumCalled  PBX State  SCCP State ReadCodec  WriteCodec RTPPeer                Direct DTMFmode |
+ ===== ========================= ========== ================ ========== ========== ========== ========== ========== ====================== ====== ======== +
| 21    SCCP/8941-00000015        8941       SEP00230432DD23             Up         CONNECTED  g722/64k   g722/64k       172.17.1.107:30454 no     RFC2833  |
| 20    SCCP/9912-00000014        9912       SEP00230433A9B9  8941       Up         CONNECTED  g729       g729            172.17.1.95:19012 no     RFC2833  |
+-----------------------------------------------------------------------------------------------------------------------------------------------------------+
```

**core show channels :** 
```
Channel              Location             State   Application(Data)
SCCP/8941-00000015   (None)               Up      AppDial((Outgoing Line))
SCCP/9912-00000014   8941@default:1       Up      Dial(SCCP/8941/ringer=inside,3
2 active channels
1 active call
12 calls processed
```

**core show channel SCCP/8941-00000015 :**
```
 -- General --
           Name: SCCP/8941-00000015
           Type: SCCP
       UniqueID: 1551839388.33
       LinkedID: 1551839388.32
      Caller ID: 8941
 Caller ID Name: ~~~~~~~~~~~~
Connected Line ID: 991202
Connected Line ID Name: ~~~~~~~~~~~~
Eff. Connected Line ID: 991202
Eff. Connected Line ID Name: ~~~~~~~~~~~~
    DNID Digits: (N/A)
       Language: fr
          State: Up (6)
  NativeFormats: (g729|g722)
    WriteFormat: g729
     ReadFormat: g722
 WriteTranscode: No
  ReadTranscode: No
 Time to Hangup: 0
   Elapsed Time: 0h7m30s
      Bridge ID: b0cacbc0-4046-420f-b9eb-8f1e1fa9aac4
 --   PBX   --
        Context: default
      Extension:
       Priority: 1
     Call Group: 26
   Pickup Group: 58
    Application: AppDial
           Data: (Outgoing Line)
 Call Identifer: [C-00000002]
      Variables:
BRIDGEPEER=SCCP/9912-00000014
cidnam=~~~~~~~~~~~~
cidout=+~~~~~~~~~~~~
cidint=8941
DIALEDPEERNUMBER=8941/ringer=inside
SCCP_DEVICE_TYPE=Cisco 7975
SCCP_DEVICE_IP=172.17.1.107
SCCP_DEVICE_MAC=SEP00230432DD23
  CDR Variables:
level 1: clid="~~~~~~~~~~~~" <8941>
level 1: src=8941
level 1: dcontext=default
level 1: channel=SCCP/8941-00000015
level 1: lastapp=AppDial
level 1: lastdata=(Outgoing Line)
level 1: start=1551839388.193695
level 1: answer=1551839389.712813
level 1: end=1551839389.725201
level 1: duration=1
level 1: billsec=0
level 1: disposition=8
level 1: amaflags=3
level 1: accountcode=9912
level 1: uniqueid=1551839388.33
level 1: linkedid=1551839388.32
level 1: sequence=22

 -- Streams --
Name: audio
    Type: audio
    State: sendrecv
    Group: -1
    Formats: (g729|g722)
    Metadata:
```

**core show channel SCCP/9912-00000014 :**
```
 -- General --
           Name: SCCP/9912-00000014
           Type: SCCP
       UniqueID: 1551839388.32
       LinkedID: 1551839388.32
      Caller ID: 991202
 Caller ID Name: ~~~~~~~~
Connected Line ID: 8941
Connected Line ID Name: ~~~~~~~~
Eff. Connected Line ID: 8941
Eff. Connected Line ID Name: ~~~~~~~~
    DNID Digits: 8941
       Language: fr
          State: Up (6)
  NativeFormats: (g729)
    WriteFormat: g722
     ReadFormat: g729
 WriteTranscode: Yes (g722@16000)->(slin@8000)->(g729@8000)
  ReadTranscode: No
 Time to Hangup: 0
   Elapsed Time: 0h9m24s
      Bridge ID: b0cacbc0-4046-420f-b9eb-8f1e1fa9aac4
 --   PBX   --
        Context: default
      Extension: 8941
       Priority: 1
     Call Group: 26
   Pickup Group: 58
    Application: Dial
           Data: SCCP/8941/ringer=inside,35
 Call Identifer: [C-00000002]
      Variables:
BRIDGEPEER=SCCP/8941-00000015
DIALEDPEERNUMBER=8941/ringer=inside
DIALEDPEERNAME=SCCP/8941-00000015
DIALSTATUS=ANSWER
DIALEDTIME=
ANSWEREDTIME=
cidnam=~~~~~~~~
cidout=+~~~~~~~~
cidint=9912
SKINNY_PRIVATE=0
SCCP_DEVICE_TYPE=Cisco 7975
SCCP_DEVICE_IP=172.17.1.95
SCCP_DEVICE_MAC=SEP00230433A9B9
  CDR Variables:
level 1: dnid=8941
level 1: clid="~~~~~~~~" <991202>
level 1: src=991202
level 1: dst=8941
level 1: dcontext=default
level 1: channel=SCCP/9912-00000014
level 1: dstchannel=SCCP/8941-00000015
level 1: lastapp=Dial
level 1: lastdata=SCCP/8941/ringer=inside,35
level 1: start=1551839388.174247
level 1: answer=1551839389.723060
level 1: end=0.000000
level 1: duration=564
level 1: billsec=563
level 1: disposition=8
level 1: amaflags=3
level 1: accountcode=9912
level 1: uniqueid=1551839388.32
level 1: linkedid=1551839388.32
level 1: sequence=21

 -- Streams --
Name: audio
    Type: audio
    State: sendrecv
    Group: -1
    Formats: (g729)
    Metadata:
```

dkgroot, 2019-03-06:
@darkponay 
Create an initial (ast1.6 only) fix for the issue.
- Have only looked at the remote codec retrieval (remoteCapabilities) part for now. It's now allowed from both sides of the call.
- Have not yet focused on preventing having differing incoming and outgoing codecs which happened on `SCCP/9912-00000014`.

I pushed this fix to a new branch "issues/453". Can you give it a go ?
```
git fetch -a
git pull
git checkout issues/453
git log -n1
./configure ....
make && make reload
```

darkponay, 2019-03-07:
Hello,
You are so efficient! Thank you for helping with this problem!

Just compiled issue/453. It's not yet perfect, but a lot better. 

**What i have tried :** 
Say we have G729 phone and G722 one (in fact, allowed for G722 and 729)
**-Voicemail :** Still Ok (G729 only for this phone)
**-Outbound call :** Still Ok (G729 transcoded in Alaw for my trunk)
**-Internal (from G722 to G729) :** Still quite ok. Both directions (but in G722, even on the G729-only phone)
**-Internal (from G729 to G722) :** Now Ok!!! Both directions (both phones in G729)

It seems the last problem is that the G729-only choice is not respected if the caller has a G722 phone.
At least we dont have one way audio.

**This is the log :** 
```
    -- SEP00230432DD23: Got Softkey: NewCall (2) line=0 callid=0
    -- SEP00230432DD23: Looking for line with instance 1.
    -- SCCP: Found line 8941
    -- SEP00230432DD23: Getting the active channel on device.
    -- SEP00230432DD23: No active channel on device.
SEP00230432DD23: (getEmptyChannel) on line:8941, maybe_c:
SEP00230432DD23: (getEmptyChannel) getActiveChannel
    -- SEP00230432DD23: Getting the active channel on device.
    -- SEP00230432DD23: No active channel on device.
 SCCP: Adding channel 13 to line 8941
    -- SEP00230432DD23: Set the active channel 13 on device
    -- SEP00230432DD23: Set the active line 8941
    -- 8941: New channel number: 13 on line 8941
    -- SCCP: (pbx_channel_allocate) try to allocate Outbound channel on line: 8941
    -- SCCP:              cid_num: "8941"
    -- SCCP:             cid_name: "~~~~~~~~~~~"
    -- SCCP:          accountcode: "8941"
    -- SCCP:                exten: ""
    -- SCCP:              context: "default"
    -- SCCP:             amaflags: "0"
    -- SCCP:            chan/call: "SCCP/8941-0000000D"
    -- SCCP:combined capabilities: "[slin16, g722/64k, ulaw/64k, alaw/64k, g722/56k, g722/48k, ilbc, g729b, g729ab, g729, g729a]"
    -- SCCP:  reduced preferences: "[g722/64k, g722/56k, g722/48k, g729, g729a, g729b, g729ab]"
    -- SCCP: (allocPBXChannel) Create New Channel with name: SCCP/8941-0000000D
    -- allocPBXChannel: using nativeformats=(g722|g729) fmt=g722
    -- SCCP/8941-0000000D: rtp server instance created at 0.0.0.0:16110
  == Using SCCP RTP TOS bits 184
  == Using SCCP RTP CoS mark 6
  == SCCP/8941-0000000D: update rtpmap: format:CISCO-DTMF, payload:101, mime:audio, rate:0
    -- SCCP/8941-0000000D: (sccp_rtp_createRtpServer) updated phone Audio RTP destination to : 172.17.1.210:16110, family:IPv4, mapped: False
    -- 8941: Allocated asterisk channel SCCP/8941-0000000D
    -- SEP00230432DD23: Set asterisk state OffHook (2) for call 13
    -- SEP00230432DD23: Indicate SCCP new state:OFFHOOK, current channel state:DOWN on call:SCCP/8941-0000000D, lineInstance:1 (previous channelstate:DOWN)
    -- SEP00230432DD23: Send speaker mode 'on'
    -- SEP00230432DD23: Send and Set the call state offhook(1) on call 13 (visibility:default)
    -- SEP00230432DD23: Send activate call plane on line 1
    -- SEP00230432DD23: ( sccp_indicate.c:88:__sccp_indicate ) sccp_dev_displayprompt '▒ ' for line 1 (8)
    -- SEP00230432DD23: Display prompt on line 1, callid 13, timeout 8
    -- SEP00230432DD23: Sending tone Inside Dial Tone (33) on line 1 with callid 13 (direction: User)
    -- SEP00230432DD23: Set softkeyset to OFFHOOK(4) on line 1  and call 13
    -- SEP00230432DD23: validKeyMask 4294967295
    -- SEP00230432DD23: Finish to indicate channel state:OFFHOOK on call:SCCP/8941-0000000D, lineInstance:1. New channel state:OFFHOOK
    -- SCCP/8941-0000000D: schedule digittimeout 16
       > SEP00230432DD23: keepalive:120, keepaliveinterval:54
    -- SEP00230432DD23: Accessory 'Speaker' is 'Off Hook'
    -- SEP00230432DD23: EnblocCall, party: 9912, lineInstance 1
    -- SEP00230432DD23: Getting the active channel on device.
    -- SCCP/8941-0000000D: stop schedule digittimeout 7
    -- SEP00230432DD23: (sccp_pbx_softswitch) New call on line 8941
    -- SEP00230432DD23: (sccp_pbx_softswitch) Dial Extension 9912
    -- SEP00230432DD23: Indicate SCCP new state:DIALING, current channel state:OFFHOOK on call:SCCP/8941-0000000D, lineInstance:1 (previous channelstate:DOWN)
    -- SEP00230432DD23: Stop tone on line 1 with callid 13
    -- SEP00230432DD23: Send the dialed number:9912, callid:13, lineInstance:1
[... 574 lines cut ...]
level 1: lastapp=AppDial
level 1: lastdata=(Outgoing Line)
level 1: start=1551926231.608415
level 1: answer=1551926232.496557
level 1: end=1551926232.502655
level 1: duration=0
level 1: billsec=0
level 1: disposition=8
level 1: amaflags=3
level 1: accountcode=8941
level 1: uniqueid=1551926231.27
level 1: linkedid=1551926231.26
level 1: sequence=18

 -- Streams --
Name: audio
    Type: audio
    State: sendrecv
    Group: -1
    Formats: (g722|g729)
    Metadata:
```

I hope it will help you.
Thanks again!

dkgroot, 2019-03-07:
Fixed: Codec selection on inbound sharedline->channel could end up using an inappropiate codec, which was not permitted in the devices preferences.
Extended debug logging for codec selection, to show the difference between channel and device preferences.

darkponay, 2019-03-08:
Hi everyone,

chan-sccp updated.

**The config:**
in sccp.conf, under _[general]_ : 
_disallow=all
allow=g722,g729_

**Nothing else in the config regarding codecs except one test device :** 
_disallow=all
allow=g729_

**Note :** The test device in G729 has one shared line. This line is shared between the G729-only phone and other phones without the codec restriction.

**What i have tried :** 
Say we have G729 phone and G722 one (in fact, allowed for G722 and 729)
**-Voicemail :** Still Ok (G729 only for this phone)
**-Outbound call :** Still Ok (G729 transcoded in Alaw for my trunk)
**-Internal (from G722 to G729) :** Not okay. One way audio (The G729 phone can't play the sound from G722 phone)
**-Internal (from G729 to G722) :**  Ok. Both directions (both phones in G729)

**The call log :**
```
    -- SEP00230432DD23: Got Softkey: NewCall (2) line=0 callid=0
    -- SEP00230432DD23: Looking for line with instance 1.
    -- SCCP: Found line 8941
    -- SEP00230432DD23: Getting the active channel on device.
    -- SEP00230432DD23: No active channel on device.
SEP00230432DD23: (getEmptyChannel) on line:8941, maybe_c:
SEP00230432DD23: (getEmptyChannel) getActiveChannel
    -- SEP00230432DD23: Getting the active channel on device.
    -- SEP00230432DD23: No active channel on device.
 SCCP: Adding channel 9 to line 8941
    -- SEP00230432DD23: Set the active channel 9 on device
    -- SEP00230432DD23: Set the active line 8941
    -- 8941: New channel number: 9 on line 8941
    -- SCCP: (pbx_channel_allocate) try to allocate Outbound channel on line: 8941
    -- SCCP:              cid_num: "8941"
    -- SCCP:             cid_name: "~~~~~~~~~~~"
    -- SCCP:          accountcode: "8941"
    -- SCCP:                exten: ""
    -- SCCP:              context: "default"
    -- SCCP:             amaflags: "0"
    -- SCCP:            chan/call: "SCCP/8941-00000009"
    -- SCCP:combined capabilities: "[slin16, g722/64k, ulaw/64k, alaw/64k, g722/56k, g722/48k, ilbc, g729b, g729ab, g729, g729a]"
    -- SCCP:  reduced preferences: "[g722/64k, g722/56k, g722/48k, g729, g729a, g729b, g729ab]"
    -- SCCP: (allocPBXChannel) Create New Channel with name: SCCP/8941-00000009
    -- allocPBXChannel: using nativeformats=(g722|g729) fmt=g722
    -- SCCP/8941-00000009: rtp server instance created at 0.0.0.0:12410
  == Using SCCP RTP TOS bits 184
  == Using SCCP RTP CoS mark 6
  == SCCP/8941-00000009: update rtpmap: format:CISCO-DTMF, payload:101, mime:audio, rate:0
    -- SCCP/8941-00000009: (sccp_rtp_createRtpServer) updated phone Audio RTP destination to : 172.17.1.210:12410, family:IPv4, mapped: False
    -- 8941: Allocated asterisk channel SCCP/8941-00000009
    -- SEP00230432DD23: Set asterisk state OffHook (2) for call 9
    -- SEP00230432DD23: Indicate SCCP new state:OFFHOOK, current channel state:DOWN on call:SCCP/8941-00000009, lineInstance:1 (previous channelstate:DOWN)
    -- SEP00230432DD23: Send speaker mode 'on'
    -- SEP00230432DD23: Send and Set the call state offhook(1) on call 9 (visibility:default)
    -- SEP00230432DD23: Send activate call plane on line 1
    -- SEP00230432DD23: ( sccp_indicate.c:88:__sccp_indicate ) sccp_dev_displayprompt '▒ ' for line 1 (8)
    -- SEP00230432DD23: Display prompt on line 1, callid 9, timeout 8
    -- SEP00230432DD23: Sending tone Inside Dial Tone (33) on line 1 with callid 9 (direction: User)
    -- SEP00230432DD23: Set softkeyset to OFFHOOK(4) on line 1  and call 9
    -- SEP00230432DD23: validKeyMask 4294967295
    -- SEP00230432DD23: Finish to indicate channel state:OFFHOOK on call:SCCP/8941-00000009, lineInstance:1. New channel state:OFFHOOK
    -- SCCP/8941-00000009: schedule digittimeout 16
       > SEP00230432DD23: keepalive:120, keepaliveinterval:56
    -- SEP00230432DD23: Accessory 'Speaker' is 'Off Hook'
    -- SEP00230432DD23: EnblocCall, party: 9912, lineInstance 1
[... 576 lines cut ...]
level 1: src=991202
level 1: dcontext=default
level 1: channel=SCCP/9912-0000000C
level 1: lastapp=AppDial
level 1: lastdata=(Outgoing Line)
level 1: start=1552046412.437307
level 1: answer=1552046413.308335
level 1: end=1552046413.314006
level 1: duration=0
level 1: billsec=0
level 1: disposition=8
level 1: amaflags=3
level 1: accountcode=8941
level 1: uniqueid=1552046412.19
level 1: linkedid=1552046412.18
level 1: sequence=13

 -- Streams --
Name: audio
    Type: audio
    State: sendrecv
    Group: -1
    Formats: (g722|g729)
    Metadata:
```

dkgroot, 2019-03-09:
orig | dest | trans | works | note
 ------ | ------- | ----- | ----- | -------
A | B | no | yes | G722>G722
A | C | yes | yes | G722>G729
B | A | no | yes | G729>G729 should be G722
C | A | yes | yes | G729>G729
B | C | yes | no | G729>G729
C | B | yes | no | G729>G729

dkgroot, 2019-03-09:
A = G722

B = G722
C = G729
B and C = shared line

orig | dest | trans | works | note | new state
 ------ | ------- | ----- | ----- | ------- | --------
A | B | no | yes | G722>G722 | still fine
A | C | yes | yes | G722>G729 | still fine
B | A | no | yes | G729>G729 should be G722 | fixed
C | A | yes | yes | G729>G722! | still fine
B | C | yes | no | G729>G729 | fixed
C | B | yes | no | G729>G729 | fixed

dkgroot, 2019-03-09:
@ianmca @mpaland
Could you guys give this branch a try, to see if there is any regressions ? It looks like it does a better job at late codec selection especially in mixed / sharedline instances.

darkponay, 2019-03-09:
Hi everyone,
I did new tests with b7f4e39.

There is still a problem : 

A = G722

B = G722
C = G729
B and C = shared line

orig | dest | trans | works | note
------ | ------- | ----- | ----- | -------
A | B | no | yes | G722>G722
A | C | yes | yes | G722>G729
B | A | no | yes | G722>G729 **should be G722 on both sides**
C | A | yes | yes | G729>G729
B | C | yes | no | G722>G729
C | B | yes | no | G729>G729

darkponay, 2019-03-10:
Here are logs corresponding to my last post for C>A : 

8941 is the line on A
9912xx is the shared line (the C phone is the G729 one)

As we can see on the phone status and here, both side are in G729.

**sccp show channels :**
```
+--- Channels ----------------------------------------------------------------------------------------------------------------------------------------------+
| ID    Name                      LineName   DeviceName       NumCalled  PBX State  SCCP State ReadCodec  WriteCodec RTPPeer                Direct DTMFmode |
+ ===== ========================= ========== ================ ========== ========== ========== ========== ========== ====================== ====== ======== +
| 8     SCCP/8941-00000008        8941       SEP00230432DD23             Up         CONNECTED  g729       g729           172.17.1.107:19750 no     RFC2833  |
| 7     SCCP/9912-00000007        9912       SEP00230433A9B9  8941       Up         CONNECTED  g729       g729            172.17.1.95:19936 no     RFC2833  |
+-----------------------------------------------------------------------------------------------------------------------------------------------------------+
```

**core show channels :**
```
Channel              Location             State   Application(Data)
SCCP/8941-00000008   (None)               Up      AppDial((Outgoing Line))
SCCP/9912-00000007   8941@default:1       Up      Dial(SCCP/8941/ringer=inside,3
2 active channels
1 active call
4 calls processed
```

**core show channel SCCP/8941-00000008 :**
```
 -- General --
           Name: SCCP/8941-00000008
           Type: SCCP
       UniqueID: 1552185977.10
       LinkedID: 1552185977.9
      Caller ID: 8941
 Caller ID Name: ~~~~~~~~~~~~
Connected Line ID: 991202
Connected Line ID Name: ~~~~~~~~~~~~
Eff. Connected Line ID: 991202
Eff. Connected Line ID Name: ~~~~~~~~~~~~
    DNID Digits: (N/A)
       Language: fr
          State: Up (6)
  NativeFormats: (g729)
    WriteFormat: g729
     ReadFormat: g729
 WriteTranscode: No
  ReadTranscode: No
 Time to Hangup: 0
   Elapsed Time: 0h4m58s
      Bridge ID: c549cd40-752c-4d36-965f-f6e49267350a
 --   PBX   --
        Context: default
      Extension:
       Priority: 1
     Call Group: 26
   Pickup Group: 58
    Application: AppDial
           Data: (Outgoing Line)
 Call Identifer: [C-00000002]
      Variables:
BRIDGEPEER=SCCP/9912-00000007
cidnam=~~~~~~~~~~~~
cidout=+~~~~~~~~~~~~
cidint=8941
DIALEDPEERNUMBER=8941/ringer=inside
SCCP_DEVICE_TYPE=Cisco 7975
SCCP_DEVICE_IP=172.17.1.107
SCCP_DEVICE_MAC=SEP00230432DD23
  CDR Variables:
[... 419 lines cut ...]
    -- SEP00230433A9B9: Set the active channel 7 on device
    -- SEP00230433A9B9: Set the active line 9912
    -- SEP00230433A9B9: Starting Phone RTP/UDP Transmission (State: CONNECTED[30])
    -- SCCP/9912-00000007: Using dtmf rtp_code : 101
    -- SEP00230433A9B9: Tell Phone to send RTP/UDP media from 172.17.1.95:19936 to 172.17.1.210:10766 (NAT: (Auto)Off)
    -- SEP00230433A9B9: Using codec:G.729(11), TOS:184, Silence Suppression:OFF for call with PassThruId:4294967288 and CallID:7
    -- SEP00230433A9B9: send callInfo on SCCP/9912-00000007 with lineInstance: 1
    -- SEP00230433A9B9: Stop tone on line 1 with callid 7
    -- SCCP: (pbx_indicate) start indicate 'AST_CONTROL_PROD: Kick remote channel'
    -- SCCP/8941-00000008: (pbx_indicate) start indicate 'AST_CONTROL_PROD: Kick remote channel'
 SEP00230432DD23: (pbx_indicate) start indicate 'AST_CONTROL_PROD: Kick remote channel' (-1) condition on channel SCCP/8941-00000008 (rtp_instance:yes, receiveChannelState:2/G.729, mediaTransmissionState:1/G.729)
  == SEP00230432DD23: (pbx_indicate) finish: send indication (res:-1)
    -- SEP00230432DD23: Got startMediaTransmission ACK. Status:'Media Status: OK' (0), Remote RTP/UDP:'172.17.1.107:19750', Type:Indirect RTP, PassThruPartyId:4294967287, CallID:8, CallID1:8
    -- SEP00230432DD23: Getting the active channel on device.
    -- SEP00230432DD23: Media Transmission Started (State: CONNECTED[30])
    -- SEP00230432DD23: Set asterisk state Up (6) for call 8
       > 0x7fc840010160 -- Strict RTP switching to RTP target address 172.17.1.107:19750 as source
    -- SEP00230433A9B9: Got startMediaTransmission ACK. Status:'Media Status: OK' (0), Remote RTP/UDP:'172.17.1.95:19936', Type:Indirect RTP, PassThruPartyId:4294967288, CallID:7, CallID1:7
    -- SEP00230433A9B9: Getting the active channel on device.
    -- SEP00230433A9B9: Media Transmission Started (State: CONNECTED[30])
    -- SEP00230433A9B9: Set asterisk state Up (6) for call 7
       > 0x7fc854010700 -- Strict RTP switching to RTP target address 172.17.1.95:19936 as source
       > 0x7fc840010160 -- Strict RTP learning complete - Locking on source address 172.17.1.107:19750
       > 0x7fc854010700 -- Strict RTP learning complete - Locking on source address 172.17.1.95:19936
```

darkponay, 2019-03-10:
Sorry for multipost, i less the mess like this :)

Now there are logs corresponding to B>A :

8941 is the line on A, still the G722 phone
9912xx is the shared line (the B phone is the G722 one)

On the phone status, we can clearly see B is in G722 and A is in G729.

**sccp show channels :**
```
+--- Channels ----------------------------------------------------------------------------------------------------------------------------------------------+
| ID    Name                      LineName   DeviceName       NumCalled  PBX State  SCCP State ReadCodec  WriteCodec RTPPeer                Direct DTMFmode |
+ ===== ========================= ========== ================ ========== ========== ========== ========== ========== ====================== ====== ======== +
| 12    SCCP/8941-0000000C        8941       SEP00230432DD23             Up         CONNECTED  g729       g729           172.17.1.107:24004 no     RFC2833  |
| 11    SCCP/9912-0000000B        9912       SEP0022555DA4F7  8941       Up         CONNECTED  g722/64k   g722/64k        172.17.1.96:24294 no     RFC2833  |
+-----------------------------------------------------------------------------------------------------------------------------------------------------------+
```

**core show channels :**
```
Channel              Location             State   Application(Data)
SCCP/8941-0000000C   (None)               Up      AppDial((Outgoing Line))
SCCP/9912-0000000B   8941@default:1       Up      Dial(SCCP/8941/ringer=inside,3
2 active channels
1 active call
6 calls processed
```

**core show channel SCCP/8941-0000000C :**
```
 -- General --
           Name: SCCP/8941-0000000C
           Type: SCCP
       UniqueID: 1552186840.16
       LinkedID: 1552186840.15
      Caller ID: 8941
 Caller ID Name: ~~~~~~~~~~~~
Connected Line ID: 991201
Connected Line ID Name: ~~~~~~~~~~~~
Eff. Connected Line ID: 991201
Eff. Connected Line ID Name: ~~~~~~~~~~~~
    DNID Digits: (N/A)
       Language: fr
          State: Up (6)
  NativeFormats: (g729)
    WriteFormat: slin16
     ReadFormat: slin16
 WriteTranscode: Yes (slin@16000)->(slin@8000)->(g729@8000)
  ReadTranscode: Yes (g729@8000)->(slin@8000)->(slin@16000)
 Time to Hangup: 0
   Elapsed Time: 0h7m28s
      Bridge ID: 4d5e8010-222b-4d9f-9044-6ee500224bf8
 --   PBX   --
        Context: default
      Extension:
       Priority: 1
     Call Group: 26
   Pickup Group: 58
    Application: AppDial
           Data: (Outgoing Line)
 Call Identifer: [C-00000001]
      Variables:
BRIDGEPEER=SCCP/9912-0000000B
cidnam=~~~~~~~~~~~~
cidout=+~~~~~~~~~~~~
cidint=8941
DIALEDPEERNUMBER=8941/ringer=inside
SCCP_DEVICE_TYPE=Cisco 7975
SCCP_DEVICE_IP=172.17.1.107
[... 545 lines cut ...]
       > 0x7fc83400c400 -- Strict RTP learning after remote address set to: 172.17.1.96:24294
    -- SCCP: (setPhoneRTPAddress) Update PBX to send RTP/UDP media to '172.17.1.96:24294' (new remote) (NAT: no)
    -- SEP0022555DA4F7: Tell PBX   to send RTP/UDP media from 172.17.1.210:10022 to 172.17.1.96:24294 (NoNat)
    -- SEP0022555DA4F7: Opened Receive Channel (State: CONNECTED[30])
    -- SEP0022555DA4F7: Set the active channel 11 on device
    -- SEP0022555DA4F7: Set the active line 9912
    -- SEP0022555DA4F7: Starting Phone RTP/UDP Transmission (State: CONNECTED[30])
    -- SCCP/9912-0000000B: Using dtmf rtp_code : 101
    -- SEP0022555DA4F7: Tell Phone to send RTP/UDP media from 172.17.1.96:24294 to 172.17.1.210:10022 (NAT: (Auto)Off)
    -- SEP0022555DA4F7: Using codec:G.722 64k(6), TOS:184, Silence Suppression:OFF for call with PassThruId:4294967284 and CallID:11
    -- SEP0022555DA4F7: send callInfo on SCCP/9912-0000000B with lineInstance: 1
    -- SEP0022555DA4F7: Stop tone on line 1 with callid 11
    -- SCCP: (pbx_indicate) start indicate 'AST_CONTROL_PROD: Kick remote channel'
    -- SCCP/8941-0000000C: (pbx_indicate) start indicate 'AST_CONTROL_PROD: Kick remote channel'
 SEP00230432DD23: (pbx_indicate) start indicate 'AST_CONTROL_PROD: Kick remote channel' (-1) condition on channel SCCP/8941-0000000C (rtp_instance:yes, receiveChannelState:2/G.729, mediaTransmissionState:2/G.729)
  == SEP00230432DD23: (pbx_indicate) finish: send indication (res:-1)
       > 0x7fc840010430 -- Strict RTP switching to RTP target address 172.17.1.107:24004 as source
    -- SEP0022555DA4F7: Got startMediaTransmission ACK. Status:'Media Status: OK' (0), Remote RTP/UDP:'172.17.1.96:24294', Type:Indirect RTP, PassThruPartyId:4294967284, CallID:11, CallID1:11
    -- SEP0022555DA4F7: Getting the active channel on device.
    -- SEP0022555DA4F7: Media Transmission Started (State: CONNECTED[30])
    -- SEP0022555DA4F7: Set asterisk state Up (6) for call 11
       > 0x7fc83400c400 -- Strict RTP switching to RTP target address 172.17.1.96:24294 as source
       > 0x7fc840010430 -- Strict RTP learning complete - Locking on source address 172.17.1.107:24004
       > 0x7fc83400c400 -- Strict RTP learning complete - Locking on source address 172.17.1.96:24294
```

dkgroot, 2019-03-10:
When looking at the preference for A in the first example. it looks like A has both G722 and G729 enabled, so that changes the outset and it makes the choice a valid one (it's not similar to the case we were testing before (One pref per device)).

With the second call, we are transcoding and both parties can do both G722 and G729. There was no real need to transcode in this case. So i agree. we could have done a little better.

I am already pretty happy that all 6 situations result in two way communication.

Note: The idea to move the setting of codec preference from the device to the line-level (especially in the shared line case), makes a lot of sense and i think we should move for that some time in the future. It removes the need for some of the assuming / guessing we currently have to do.

darkponay, 2019-03-10:
I put the three phones, just in case :)

**Phone A :**
```
--- SCCP channel driver device settings ----------------------------------------------------------------------------------
MAC-Address                                    : SEP00230432DD23
Protocol Version                               : Supported '22', In Use '22'
Protocol In Use                                : SCCP Version 22
Device Features                                : 0x7285,000000000111001010000101
Tokenstate                                     : None
Keepalive                                      : 60
Registration state                             : OK
State                                          : On Hook
Addons                                         :
MWI state                                      : off (0/0)
MWI light-type                                 : On
MWI During call                                : keep on
Description                                    : TEST941
Config Phone Type                              : 7975
Skinny Phone Type                              : Cisco 7975(437)
Softkey support                                : yes
Softkeyset                                     : default => default (0x560bc6efa370)
BTemplate support                              : yes
linesRegistered                                : yes
Image Version                                  : SCCP75.9-4-2SR3-1S
Timezone Offset                                : 0
Audio Capabilities                             : [slin16, g722/64k, ulaw/64k, alaw/64k, g722/56k, g722/48k, ilbc, g729b, g729ab, g729, g729a]
Auduo Preferences                              : [g722/64k, g722/56k, g722/48k, g729, g729a, g729b, g729ab, g729/annex/b]
Audio TOS                                      : 184
Audio COS                                      : 6
Video TOS                                      : 136
Video COS                                      : 5
DND Feature enabled                            : yes
DND Status                                     : Disabled
DND Action                                     : Disabled / Cycle
Can Transfer                                   : on
Can Park                                       : on
Can CFWDALL                                    : on
Can CFWBUSY                                    : off
Can CFWNOANSWER                                : off
Allow ringin notification                      : no
Private softkey                                : on
Dtmf mode                                      : RFC2833
Nat                                            : (Auto)Off
Videosupport?                                  : no
Direct RTP                                     : off
Trust phone ip (deprecated)                    : off
Bind Address                                   : 172.17.1.107:49951
Server Address                                 : 172.17.1.210:54543
Deny/Permit                                    : deny:0.0.0.0/0.0.0.0,permit:172.0.0.0/255.224.0.0,
PermitHosts                                    :
Early RTP                                      : Progress
Accessory State                                : None
Last dialed number                             : 9912 (1)
Default line instance                          : 1
Custom Background Image                        : http://172.17.1.212/cisco/7975.png
Custom Ring Tone                               : ---
Use Placed Calls                               : off
PendingUpdate                                  : off
PendingDelete                                  : off
allow_conference                               : on
conf_play_general_announce                     : off
conf_play_part_announce                        : off
conf_mute_on_entry                             : off
conf_music_on_hold_class                       : (null)
conf_show_conflist                             : on
conflist_active                                : off
CallHistory Answered Elsewhere                 : Ignore

+--- Buttons ----------------------------------------------------------------------------------------------------------+
[... 241 lines cut ...]
| 1    9912                    (+)02:                                                      None                        |
+----------------------------------------------------------------------------------------------------------------------+

+--- SpeeddialButtons -------------------------------------------------------------------------------------------------+
| Id   Name                    Number                 Hint                                                             |
+ ==== ======================= ====================== ================================================================ +
+----------------------------------------------------------------------------------------------------------------------+

+--- FeatureButtons ---------------------------------------------------------------------------------------------------+
| Id   Name                    Status                 Options                                                          |
+ ==== ======================= ====================== ================================================================ +
+----------------------------------------------------------------------------------------------------------------------+

+--- ServiceURLButtons ------------------------------------------------------------------------------------------------+
[... cut ...]
| Type     Calls    PcktSnt  PcktRcvd Lost     Jitter   Latency  Quality  avgQual  meanQual maxQual  rConceal sConceal |
+ ======== ======== ======== ======== ======== ======== ======== ======== ======== ======== ======== ======== ======== +
| LAST     0        0        0        0        0        0        0.000000 0.000000 0.000000 0.000000 0.000000 0        |
| AVG      0        0        0        0        0        0        0.000000 0.000000 0.000000 0.000000 0.000000 0        |
+----------------------------------------------------------------------------------------------------------------------+
```

dkgroot, 2019-03-10:
@darkponay @mpaland @ianmca 
I created a second branch called "issues/453_1" which incorporates the idea of moving the codec preference to the line level instead of at device level. This initial version, still allows for either setting, but when line preferences have been set, they overrule the device ones. The new version seems to be very succesfull at reducing complexity when dealing with shared line instance, without complicating the non shared version. The amount of addition code is remarkable small. Once the device level would/will be removed (via deprecation at first) the net amount of change would be a reduction, whilst maintaining functionality and a much clearer and simpler codec selection.
the 453_1 branch was build ontop of the changes already done for 453. So functionality should be equal but easier to reason about.

ianmca, 2019-03-10:
Hi Diederik.
Sorry I was away from email.

Would you like me to test the issues/453_1 branch?

I’m not using g729 anywhere but am using G722 and G711a

Do I need to adjust my config then to move the codec preferences from the devices to the lines?

dkgroot, 2019-03-10:
Hi Ianmca,
When you have checked out the 453_1 branch and you leave the sccp.conf like it is now, everything should still work correctly as it did before (might make a better codec decision here or there).
You can change the sccp.conf and move some or all device preferences to the line level if/when you like. This change is most benefitial for any sharedlines you might have.

darkponay, 2019-03-11:
Hi,
For these tests, i was all the time in issues/453_1 - **d706d0c** except for VM tests which has been made in **3f6d892**.

Some of the tests have been done again after changing revision.
Asterisk has been restarted between tests/revision change.

With the device config : 
**- general :** disallow=all / allow=g722,g729
**- Phone A :** no codec pref, classic line x
**- Phone B :** no codec pref, shared line y + classic line z
**- Phone C :** disallow=all / allow=g729, shared line y
_No line has codec preference_

orig   | dest    | trans | works | caller>called
------ | ------- | ----- | ----- | -------
Ax     | By      | no    | yes   | G722>G722
Ax     | Bz      | no    | yes   | G722>G722
Ax     | Cy      | yes   | yes   | G722>G729
By     | Ax      | no    | yes   | G722>G722
By     | Cy      | yes   | yes   | G722>G729
Bz     | Ax      | no    | yes   | G722>G722
Bz     | Cy      | yes   | yes   | G722>G729
Cy     | Ax      | yes   | yes   | G729>G722
Cy     | By      | yes   | yes   | G729>G722
Cy     | Bz      | yes   | yes   | G729>G722
Ax     | VM      | x     | yes   | G722
By     | VM      | x     | yes   | G722
Bz     | VM      | x     | yes   | G722
Cy     | VM      | x     | yes   | G729

It seems to be absolutely perfect.

--------------------------------------------------------------------
Now i totally remove the device config for codecs to put it on the shared line:
**- general :** disallow=all / allow=g722,g729
**- Phone A :** no codec pref, classic line x
**- Phone B :** no codec pref, shared line y + classic line z
**- Phone C :** no codec pref, shared line y
_No device codec config, only shared line y has codec config (disallow=all, allow=g729)_

orig   | dest    | trans | works | codecs
------ | ------- | ----- | ----- | -------
Ax     | By      | no    | yes   | G729>G729
Ax     | Bz      | no    | yes   | G722>G722
Ax     | Cy      | no    | yes   | G729>G729
By     | Ax      | no    | yes   | G729>G729
By     | Cy      | no    | yes   | G729>G729
Bz     | Ax      | no    | yes   | G722>G722
Bz     | Cy      | no    | yes   | G729>G729
Cy     | Ax      | no    | yes   | G729>G729
Cy     | By      | no    | yes   | G729>G729
Cy     | Bz      | no    | yes   | G729>G729
Ax     | VM      | x     | yes   | G722
By     | VM      | x     | yes   | G729
Bz     | VM      | x     | yes   | G722
Cy     | VM      | x     | yes   | G729

It is perfectly working in this case too

--------------------------------------------------------------------
The same config as before, but the only line that has codec config is a non-shared line.
**- general :** disallow=all / allow=g722,g729
**- Phone A :** no codec pref, classic line x
**- Phone B :** no codec pref, shared line y + classic line z
**- Phone C :** no codec pref, shared line y
_No device codec config, only shared line x has codec config (disallow=all, allow=g729)_

orig   | dest    | trans | works | codecs
------ | ------- | ----- | ----- | -------
Ax     | By      | no    | yes   | G729>G729
Ax     | Bz      | no    | yes   | G729>G729
Ax     | Cy      | no    | yes   | G729>G729
By     | Ax      | no    | yes   | G729>G729
By     | Cy      | no    | yes   | G722>G722
Bz     | Ax      | no    | yes   | G729>G729
Bz     | Cy      | no    | yes   | G722>G722
Cy     | Ax      | no    | yes   | G729>G729
Cy     | By      | no    | yes   | G722>G722
Cy     | Bz      | no    | yes   | G722>G722
Ax     | VM      | x     | yes   | G729
By     | VM      | x     | yes   | G722
Bz     | VM      | x     | yes   | G722
Cy     | VM      | x     | yes   | G722

Everything seems to be working now.

darkponay, 2019-03-11:
We are actually playing (without logging every actions) with some functions like hold, conferencing, transfer attended or not, Dirtrfr, and it works all the time. Actual config is g722,g729 allowed in general, and g729 forced on the line of my interlocutor.

dkgroot, 2019-03-15:
Reported by @ianmca in the chat :
1. I've been running it successfully. I've generally had no complaints from users;
2. however I did have one asterisk crash yesterday afternoon, but I haven't been able to get to the bottom of exactly what happened....
3. I have moved codec choices over to the lines in a template (after the crash), and that seems to be all OK.
4. Only two things i'm noticing... on an sccp reload all the lines are reporting major line change detected, and so all are reloading,
5. I'm also getting a deprecated config for disallow= but it reports an incorrect line on the config file - might be related to a template, but i don't think i have any disallow statements now that aren't against lines.

My Reply:
1.    sounds pretty good in all
2. the crash would have to be researched / backtrace etc. without it there is only possible causation, but not confirmation (it could be, but it could also be anything else). I had chan_iax2 blow up yesterday :-)
3. codecs on lines (which is the main reason for the patch) sounds really good as well
4. The sccp reload needs investigation. thanks for that, so i will give that a couple of runs through to see what might be causing that. It's not a main show stopper for me. It can and will be fixed.
5. yes the change to have codec prefs on the line, does deprecate codec prefs on devices (just to persuade users to move to the line method). disallow and allow are parsed in one go. to pull that off we collect all disallow/allow lines per config segment and when we have read the last one, we process them all. So yes the reported line would be off by a couple. There is not really anything i can do about them. Let's call the reported line somewhat indeterministic. This processing method has not changed however, it was also there in the previous version. It just has been reused for the line->(dis)allow. The deprecation of device->(dis)allow might have brought it to the foreground however.

All in all i think we can and should move forward. I will look into the sccp reload issue, so thank you for reporting that.
Thanks you for reporting in detail and being willing to test

dkgroot, 2019-03-15:
Thanks everyone working on this... the eagle has landed :-)

commit 181573ac (2019-03-15): Fix: one way audio on sharedline which non matching preference members

- Fix sccp_astwrap_getSkinnyFormatMultiple
- Fix: Codec selection on inbound sharedline->channel could end up using an inappropiate codec, which was
- Allow pbx_find_channel_by_linkid to be used by both channels in a call (from either side)
- Call iPbx.retrieve_remote_capabilities from sccp_channel_answer, when we don't have a valid set of
  remoteCapabilities yet not permitted in the devices preferences.
- Fix sccp_line_copyCodecSetsFromLineToChannel: when there is no possible matching codecs we should combine
  not reduc [... cut]
closed 2019-03-15 by dkgroot