chan-sccp/chan-sccp #454: ConfList: initiator name replaced by conference name

darkponay, 2019-03-08:
Hi!

A little bug seems to happen with the conflist, like in the screenshot below :
[image]

As you can see, the initiator name is replaced by "Conference 101". Hopefully everything works, it is just a problem in displaying the name. All the functions work properly.

Changing Asterisk or chan-sccp version didn't change anything.

dkgroot, 2019-03-08:
@darkponay 
Can you give issue/454 a try please
Merci bien !

darkponay, 2019-03-09:
Hello!
For the moment, it seems perfect, I tried basic 2 members and 3 members conference, some hold/resume operations, promote as moderator and demote, and it is still the right name shown for the initiator!

dkgroot, 2019-03-09:
Thanks for reporting the issue and testing the result. Solution has been merged !

commit 62d9a50c (2019-03-09): Fix/Change conference moderator callerid info

fixes #454
closed 2019-03-09 by dkgroot