chan-sccp/chan-sccp #455: Call Forward on Shared line

dkgroot, 2019-03-09:
Latest asterisk 13 and 16
Only happens if one of the shared line members forwards the call
Generates a Frack ! in current or subsequent channels

Feb 21 09:25:17 weiphatu asterisk[23902]: ERROR[7062][C-0000000a]: astobj2.c:136 in INTERNAL_OBJ: FRACK!, Failed assertion bad magic number 0x0 for object 0x2fd1270 (0)
Feb 21 09:25:17 weiphatu asterisk[23902]: ERROR[7062][C-0000000a]: :0 in : Got 19 backtrace records
# 0: /usr/sbin/asterisk() [0x454a85]
# 1: /usr/sbin/asterisk() [0x454ae4]
# 2: /usr/sbin/asterisk(ast_frdup+0x100) [0x4ef4b0]
# 3: /usr/sbin/asterisk(ast_bridge_channel_queue_frame+0x58) [0x470e08]
# 4: /usr/sbin/asterisk(ast_bridge_queue_everyone_else+0x40) [0x471350]
# 5: /usr/lib64/asterisk/modules/bridge_simple.so(+0xa49) [0x7f07418b9a49]
# 6: /usr/sbin/asterisk() [0x46f9d2]
# 7: /usr/sbin/asterisk(bridge_channel_internal_join+0x82c) [0x4733dc]
# 8: /usr/sbin/asterisk(ast_bridge_join+0x2de) [0x461c8e]
# 9: /usr/sbin/asterisk(ast_bridge_call_with_flags+0xbf) [0x4e020f]
#10: /usr/lib64/asterisk/modules/app_dial.so(+0xc0b8) [0x7f07484bb0b8]
#11: /usr/lib64/asterisk/modules/app_dial.so(+0xcf66) [0x7f07484bbf66]
#12: /usr/sbin/asterisk(pbx_exec+0xb9) [0x5415f9]
#13: /usr/sbin/asterisk() [0x539c74]
#14: /usr/sbin/asterisk() [0x53ac8e]
#15: /usr/sbin/asterisk() [0x53c14b]
#16: /usr/sbin/asterisk() [0x5a01ee]
#17: /lib64/libpthread.so.0(+0x7dd5) [0x7f07bb2eadd5]
#18: /lib64/libc.so.6(clone+0x6d) [0x7f07babbdead]

dkgroot, 2019-03-09:
Create a branch for testing: "issues/cfwdsharedline"
Contains initial fix for the issue. Still needs more work

KCross84, 2019-04-18:
I got this error as a result of trying to use the WB2 paging method as I am not yet able to install PAMI. I installed this branch and things are finally working and seem stable for the moment.

dkgroot, 2019-04-27:
@KCross84 Is this still the case, did we not move you back to the development branch ?

dkgroot, 2019-08-19:
Updated branch "issues/cfwdsharedline"

***Git***
_Checkout this temporary branch_
```
git pull
git fetch -a
git branch
git checkout issues/cfwdshareline
./configure [....]
make -j4 && make install && make reload
```

_move back to develop afterwards:_
```
git checkout develop
```

dkgroot, 2019-08-19:
@nikolaciprich Please have a look/update if you can

dkgroot, 2019-08-31:
Issue should be fixed, including the continues ringing reported.

dkgroot, 2019-09-30:
Reimplemented handling of sharedline callforward:
-  Fix sharedline callforward
-  Fix ringing channel
-  Fix ao2_reference count
-  Fix ast_module_ref/unref
----
Created new branch "issues/cfwdsharedline2" for testing

Git
Checkout this temporary branch

git pull
git fetch -a
git branch
git checkout issues/cfwdsharedline2
./configure [....]
make -j4 && make install && make reload

move back to develop afterwards:

git checkout develop

closed 2019-10-04 by dkgroot

dkgroot, 2019-10-04:
@kc2vrj Thanks for helping to test this
@nikolaciprich The new cfwdsharedline branch has been merged into develop so you should be ok.

commit 3e649076 (2019-08-19): Fix pbx_impl use pbx_channel_ref / pbx_channel_unref instead of ast_channel_(un)ref
Fix pbx_impl getChannelByName returning a reffed channel
Fix warning about missing device on early hangup before media transmission
Fix ast_channel_varshead handling in sccp_astwrap_call

Initial fix to the call forward issue when dealing with shared line instances. ->
Integrate old/new asterisk handling of sccp_pbx_answer when dealing with forward on a shared line

This commit gets rid of the problem in the forwarded call and subsequent channels

fixes #455
commit 79582475 (2019-08-19): Fix pbx_impl use pbx_channel_ref / pbx_channel_unref instead of ast_channel_(un)ref
Fix pbx_impl getChannelByName returning a reffed channel
Fix warning about missing device on early hangup before media transmission
Fix ast_channel_varshead handling in sccp_astwrap_call

Initial fix to the call forward issue when dealing with shared line instances. ->
Integrate old/new asterisk handling of sccp_pbx_answer when dealing with forward on a shared line

This commit gets rid of the problem in the forwarded call and subsequent channels

fixes #455
commit 3eaa8498 (2019-08-23): Fix pbx_impl use pbx_channel_ref / pbx_channel_unref instead of ast_channel_(un)ref
Fix pbx_impl getChannelByName returning a reffed channel
Fix warning about missing device on early hangup before media transmission
Fix ast_channel_varshead handling in sccp_astwrap_call

Initial fix to the call forward issue when dealing with shared line instances. ->
Integrate old/new asterisk handling of sccp_pbx_answer when dealing with forward on a shared line

This commit gets rid of the problem in the forwarded call and subsequent channels

fixes #455
commit ed679291 (2019-09-01): Fix pbx_impl use pbx_channel_ref / pbx_channel_unref instead of ast_channel_(un)ref
Fix pbx_impl getChannelByName returning a reffed channel
Fix warning about missing device on early hangup before media transmission
Fix ast_channel_varshead handling in sccp_astwrap_call

Initial fix to the call forward issue when dealing with shared line instances. ->
Integrate old/new asterisk handling of sccp_pbx_answer when dealing with forward on a shared line

This commit gets rid of the problem in the forwarded call and subsequent channels

fixes #455