chan-sccp/chan-sccp #456: PickUp and GPickUp softkeys problem with shared lines

darkponay, 2019-03-14:
Hi everyone,

We have found a little problem with PickUp and GPickUp softkeys while you are on call on a shared line.

Say we have two phones sharing the same line.
1. I call someone (no matter who, internal or not) from Phone A. (It also work if receive the call)
2. I can see on Phone B the caller ID i am with from Phone A.
3. I press the Phone B's line button, which will show me the right softkeys for holding, intercept, etc, and also PickUp and GPickUp. At this moment the bottom status line show "Hold"
4. I press PickUp or GPickUp on Phone B, and it asks for a number.
5. My active call on Phone A is now on hold.

It does not happen if you skip step 3.

I dont know it is is related, but after some tests, Asterisk fracked.

**chan-sccp version:** Actually i'm on issues/453_1, but i can reproduce this behaviour on an old october '18 installation. Truth be told, i thought it was a normal behaviour.

**asterisk version:** 16.2.1, but also my FreePBX which is on Asterisk 14.7

**Expected behavior:** 
When pressing PickUp and GPickUp softkeys on Phone B, the active call on phone A should remain active.

**Logs :**

`SCCP new debug status: (1339773 -> 1341821) core,rtp,device,line,action,channel,config,softkey,indicate,pbx,socket,buttontemplate,codec
`

```
    -- SEP00230432DD23: Got Softkey: Redial (1) line=0 callid=0
    -- SEP00230432DD23: Looking for line with instance 1.
    -- SCCP: Found line 8941
    -- SEP00230432DD23: (sccp_getSoftkeyMap_by_SoftkeyEvent) default: 0x7f4831cee1a0, softkeyset: 0x55b877f72340, softkeyCbMap: (nil)
    -- SEP00230432DD23: Handling Softkey: Redial on line: 8941 and channel: UNDEF
    -- SEP00230432DD23: SoftKey Redial Pressed
    -- SEP00230432DD23: Get ready to redial number 9912 lineInstance: 1
    -- SEP00230432DD23: Looking for line with instance 1.
    -- SCCP: Found line 8941
SEP00230432DD23: (getEmptyChannel) on line:8941, maybe_c:
SEP00230432DD23: (getEmptyChannel) getActiveChannel
    -- SEP00230432DD23: Getting the active channel on device.
    -- SEP00230432DD23: No active channel on device.
 SCCP: Adding channel 42 to line 8941
    -- SEP00230432DD23: Set the active channel 42 on device
    -- SEP00230432DD23: Set the active line 8941
    -- 8941: New channel number: 42 on line 8941
    -- SCCP: (pbx_channel_allocate) try to allocate Outbound channel on line: 8941
    -- SCCP:              cid_num: "8941"
    -- SCCP:             cid_name: "~~~~~~~~~~"
    -- SCCP:          accountcode: "8941"
    -- SCCP:                exten: ""
    -- SCCP:              context: "default"
    -- SCCP:             amaflags: "0"
    -- SCCP:            chan/call: "SCCP/8941-0000002A"
    -- SCCP:combined capabilities: "[slin16, g722/64k, ulaw/64k, alaw/64k, g722/56k, g722/48k, ilbc, g729b, g729ab, g729, g729a]"
    -- SCCP:  reduced preferences: "[g722/64k, g722/56k, g722/48k, g729, g729a, g729b, g729ab]"
    -- SCCP: (allocPBXChannel) Create New Channel with name: SCCP/8941-0000002A
    -- allocPBXChannel: using nativeformats=(g722|g729) fmt=g722
    -- SCCP/8941-0000002A: rtp server instance created at 0.0.0.0:19402
  == Using SCCP RTP TOS bits 184
  == Using SCCP RTP CoS mark 6
  == SCCP/8941-0000002A: update rtpmap: format:CISCO-DTMF, payload:101, mime:audio, rate:0
    -- SCCP/8941-0000002A: (sccp_rtp_createRtpServer) updated phone Audio RTP destination to : 172.17.1.210:19402, family:IPv4, mapped: False
    -- 8941: Allocated asterisk channel SCCP/8941-0000002A
    -- SEP00230432DD23: Set asterisk state OffHook (2) for call 42
    -- SEP00230432DD23: Indicate SCCP new state:SPEEDDIAL, current channel state:DOWN on call:SCCP/8941-0000002A, lineInstance:1 (previous channelstate:DOWN)
    -- SEP00230432DD23: Clear the display notify message
    -- SEP00230432DD23: Clear the status prompt on line 0 and callid 0
    -- SEP00230432DD23: Send ringer mode Ring Off(1) on device
    -- SEP00230432DD23: Send speaker mode 'on'
    -- SEP00230432DD23: Send and Set the call state offhook(1) on call 42 (visibility:default)
[... 961 lines cut ...]
       > SCCP: Stopping PBX audio rtp transmission on channel SCCP/8941-0000002A
    -- SCCP: destroying PBX rtp server on channel SCCP/8941-0000002A
    -- SEP00230432DD23: Clear the display notify message
    -- SEP00230432DD23: Clear the status prompt on line 0 and callid 0
    -- SEP00230432DD23: ( sccp_device.c:2023:sccp_dev_check_displayprompt ) sccp_dev_displayprompt '▒' for line 0 (0)
    -- SEP00230432DD23: Display prompt on line 0, callid 0, timeout 0
    -- SEP00230432DD23: softkey 'Redial' on ONHOOK to on
       > SEP00230432DD23: found softkey 'Redial' at 0
    -- SEP00230432DD23: softkey 'Confrn' on ONHOOK to on
    -- SEP00230432DD23: softkey 'Join' on ONHOOK to off
    -- SEP00230432DD23: softkey 'ConfList' on ONHOOK to on
    -- SEP00230432DD23: softkey 'Record' on ONHOOK to off
    -- SEP00230432DD23: Set softkeyset to ONHOOK(0) on line 0  and call 0
    -- SEP00230432DD23: validKeyMask 4294967295
    -- SEP00230432DD23: Accessory 'Speaker' is 'On Hook'
       > SEP00230432DD23: keepalive:72, keepaliveinterval:59
    -- QualityStats: MLQK=3.7001;MLQKav=3.6911;MLQKmn=3.6545;MLQKmx=3.7001;ICR=0.0000;CCR=0.0006;ICRmx=0.0066;CS=2;SCS=0;MLQKvr=0.95
    -- SEP00230432DD23: Call Statistics:
       [
         Last Call        : CallID: 42 Packets sent: 2451 rcvd: 2453 lost: 0 jitter: 0 latency: 0
         Last Quality     : MLQK=3.7001;MLQKav=3.6911;MLQKmn=3.6545;MLQKmx=3.7001;MLQKvr=0.95|ICR=0.0000;CCR=0.0006;ICRmx=0.0066|CS=2;SCS=0
         Mean Statistics  : #Calls: 7 Packets sent: 2351 rcvd: 2321 lost: 0 jitter: 0 latency: 0
         Mean Quality     : MLQK=3.6999;MLQKav=3.4950;MLQKmn=2.9685;MLQKmx=3.7001;MLQKvr=0.95|ICR=0.0000;CCR=0.0148;ICRmx=0.3879|CS=1;SCS=0
       ]
```

dkgroot, 2019-03-14:
Pretty minor issue, but should not be hard to fix. We can just disable the softkey at this stage, preventing the press in the first place.

closed 2019-03-14 by dkgroot

dkgroot, 2019-03-14:
Pushed commit a534c76 on your issue (#456)
It fixes three things:
1. It checks if the device performing the pickup is the same one as the one dealing with the call (sharedline issue)
2. general shared issue: before putting a previous channel on hold to create a new call, we should check if the event came in through the device requesting the new call (related to no 1).
3. Allow you to do a pickup/gpickup when you have selected a line before pressing pickup (broken before: found that out while testing).

Thanks so much for pointing this issue out. It's the minor details that sometimes uncover related stuff.

commit a534c766 (2019-03-14): (g)pickup: Check if device currently linked to the provided channel is the same device as the on initiating the pickup
(g)pickup: Allow pickup/gpickup on an active (shared)line with an empty channel present
emptyChannel: perform the same check before putting a potential remote call on hold because of an unrelated action

fixes #456