chan-sccp/chan-sccp #458: 7940_7914 BLF+Pickup Great Problem

eyatsko, 2019-03-24:
Cisco Phone 7940 with addon 7914 doesn't work as expected: when monitored line gets a call pressing 7914 button doesn't pick up the call instead it shows "External Call" on 7960's display.
At the same time pressing the speeddial button on 7960 with THE SAME configuration shows "Call to ..." and if to press Dial SoftKey it actually picks up! Different behavior on coupled devices.

chan_sccp 4.3.0_TRUNK
Asterisk 11.7.0~dfsg-1ubuntu1
CP-7960G firmware 8.1.2SR2
CP-7914 firmware 5.0.4

===Reproduction Steps===
SCCP.CONF:

[SEP00070E578F4F]
description = RODITELI No 101
devicetype = 7960
park = off
button = line, 101
type = device
keepalive = 60
tzoffset = +3
transfer = on
park = on
cfwdall = off
cfwdbusy = off
cfwdnoanswer = off
dnd = reject
directrtp = off
earlyrtp = progress
private = on
mwilamp = on
mwioncall = off
cfwdall = on
button = speeddial, Delia, **102#, 102@BLF
button = speeddial, Kitchen, **103#, 103@BLF
button = speeddial, MamaMob, 89160714452
button = speeddial, DeliaMob, 89169040990
addon = 7914
button = empty
button = speeddial, Test1, **104, 104@BLF
button = speeddial, Test2, **105, 105@BLF
button = speeddial, Kitchen, **103, 103@BLF

[101]
id = 101
type = line
pin = 1234
label = Roditeli
description = Roditeli 101
mailbox = 10021
cid_name = Roditeli
cid_num = 101
accountcode = 101
callgroup = 1
pickupgroup = 1
directed_pickup = on
directed_pickup_context = Permit-Internal
directed_pickup_modeanswer = on
context = Permit-National
incominglimit = 2
transfer = on
vmnum = 600
trnsfvm = 1000
secondary_dialtone_digits = 9
secondary_dialtone_tone = 0x22
musicclass = default
language = ru
echocancel = on
silencesuppression = off

EXTENSIONS.CONF:
[Permit-Internal]
. . .
exten => _XXX,1,Macro(DialInternal,${EXTEN},180,tTwW)
exten => _XXX,n,Hangup

exten => _**XXX,1,Pickup()
exten => _**XXX,n,Macro(DialInternal,${EXTEN:2},180,tTwW)
;include => BLF ; Tried with and without it with no effect

[BLF]
exten => 101,hint,SCCP/101
exten => 102,hint,SCCP/102
exten => 103,hint,SCCP/103
exten => 104,hint,SIP/104
exten => 105,hint,SIP/105

===Expected behavior===
When I am making call 104 -> 105 or 104 -> 103 it should lead to 7914's button '104' lights RED and 105 or 103 SHOULD blinking AMBER.
When I press 103/104 button on 7914 it MUST pick up the call. In initial state all thre "lamps' should light GREEN (?)

===Observed  behavior===
During above calls both calling and called buttons are RED. When I press any of them nothing happens incluiding Asterisk Console (debug is on at this moment) and 7960 shows "EXTERNAL CALL".
Little addition: when I am trying to pickup 103 call (SCCP destination, called party in SCCP) 7914 behaves as 7960 actually. The abscence of actions concerned (it seems) just monitored SIP phones.

===Platform===
VMWare VM ESXi 6.5.0
Ubuntu 14.04.6 LTS (3.19.0-47-generic #53)

dkgroot, 2019-03-24:
@eyatsko
Can you tell me the branch and revision of chan-sccp you are using (sccp show version). Have you tried the latest current version (ie: `git clone https://github.com/chan-sccp/chan-sccp chan-sccp_develop`) ?

Is the SEP00070E578F4F the 7940 or 7960 (FYI: In the devicetype you declared it as 7960). As far as i know the 7940 and 7960 should be (mostly) the same hardware and firmware except for two line buttons missing on the 7940. That would also mean they should function in the exact same way. I don't have any 7940's so i cannot replicate your exact setup though. I do know *all* buttons on the addon stay red if there is a problem with the connection between the phone and the first addon. All buttons stay amber on the second if there is a bad connection.

Questions/Tests in order:
- When the device is registered are the buttons assigned correctly in `sccp show device SEP00070E578F4F` ? 
- If you set `sccp debug core, action, softkey, feature` do you see anything in the debug log when you press one of the speeddials buttons on the addon ?
- Have you tried a factory reset on the phone to see if that makes a difference (do make sure you have the necessary firmware files before you attempt this)
- Have you tried downgrading and then upgrading the firmware (do make sure you have the necessary firmware files before you attempt this)

dkgroot, 2019-03-24:
@eyatsko Maybe you could join us in the [chan-sccp gitter chat](https://gitter.im/chan-sccp/chan-sccp), and ask for @PhantomVI. He is also from Russia and has a number of 7960 and 7940 with addon modules. That could help.
Also have you tried just assigning a simple line to the addon module instead of depening on speeddials + dialplan + hints. Might be good to start with something simple and easy and test that first.

eyatsko, 2019-03-24:
Hello dkgroot!

1) sccp show version gives:
  Skinny Client Control Protocol (SCCP). Release: 4.3.2 develop - aaace51M (built by 'root' on '2019-03-23 23:16:41 UTC')

2) SEP00070E578F4F is 7960 + 7914, and is seems that connection is good. I double checked.

3) sccp show device SEP00070E578F4F
`+--- Buttons ----------------------------------------------------------------------------------------------------------+
| Id   Inst TypeStr                                  Type                                  pendUpdt pendDel  Default   |
+ ==== ==== ======================================== ===================================== ======== ======== ========= +
| 1    1    Line                                     0                                     No       No       Yes       |
| 2    2    Speeddial                                1                                     No       No       No        |
| 3    3    Speeddial                                1                                     No       No       No        |
| 4    1    Speeddial                                1                                     No       No       No        |
| 5    2    Speeddial                                1                                     No       No       No        |
| 6    0    Empty                                    4                                     No       No       No        |
| 7    4    Speeddial                                1                                     No       No       No        |
| 8    5    Speeddial                                1                                     No       No       No        |
| 9    6    Speeddial                                1                                     No       No       No        |
+----------------------------------------------------------------------------------------------------------------------+

+--- LineButtons ------------------------------------------------------------------------------------------------------+
| Id   Name                    SubCid                Label                                 CfwdType   CfwdNumber       |
+ ==== ======================= ===================== ===================================== ========== ================ +
| 1    101                                           Roditeli                              None                        |
+----------------------------------------------------------------------------------------------------------------------+

+--- SpeeddialButtons -------------------------------------------------------------------------------------------------+
| Id   Name                    Number                 Hint                                                             |
+ ==== ======================= ====================== ================================================================ +
| 2    Delia                   **102#                 102@BLF                                                          |
| 3    Kitchen                 **103#                 103@BLF                                                          |
| 4    MamaMob                 89160714452                                                                             |
| 5    DeliaMob                89169040990                                                                             |
| 7    Test1                   **104                  104@BLF                                                          |
| 8    Test2                   **105                  105@BLF                                                          |
| 9    Kitchen                 **103                  103@BLF                                                          |
+----------------------------------------------------------------------------------------------------------------------+
`

4) sccp debug core, action, softkey, feature - doesn't show anything while I' pressing during a call.
SCCP new debug status: (16761 -> 19321) core,device,line,action,channel,config,feature,softkey,socket
srvats*CLI>
srvats*CLI>
srvats*CLI>
srvats*CLI>
srvats*CLI>
srvats*CLI>
srvats*CLI>
  == Using SIP RTP CoS mark 5
    -- SEP00070E578F4F: Send and Set the call state call remote multiline(13) on call 0 (visibility:default)
    -- SEP00070E578F4F: softkey 'Confrn' on INUSEHINT to on
    -- SEP00070E578F4F: softkey 'Join' on INUSEHINT to off
    -- SEP00070E578F4F: softkey 'ConfList' on INUSEHINT to on
    -- SEP00070E578F4F: softkey 'Record' on INUSEHINT to off
    -- SEP00070E578F4F: Set softkeyset to INUSEHINT(10) on line 4  and call 0
    -- SEP00070E578F4F: validKeyMask 4294967295
    -- Executing [105@Permit-National:1] Macro("SIP/104-0000001b", "DialInternal,105,180,tTwW") in new stack
    -- Executing [s@macro-DialInternal:1] NoOp("SIP/104-0000001b", "NOT_INUSE") in new stack
    -- Executing [s@macro-DialInternal:2] GotoIf("SIP/104-0000001b", "0?sccp:sip") in new stack
    -- Goto (macro-DialInternal,s,3)
    -- Executing [s@macro-DialInternal:3] Dial("SIP/104-0000001b", "SIP/105,180,tTwW") in new stack
  == Using SIP RTP CoS mark 5
    -- SEP00070E578F4F: Send and Set the call state call remote multiline(13) on call 0 (visibility:default)
    -- SEP00070E578F4F: softkey 'Confrn' on INUSEHINT to on
    -- SEP00070E578F4F: softkey 'Join' on INUSEHINT to off
    -- SEP00070E578F4F: softkey 'ConfList' on INUSEHINT to on
    -- SEP00070E578F4F: softkey 'Record' on INUSEHINT to off
    -- SEP00070E578F4F: Set softkeyset to INUSEHINT(10) on line 5  and call 0
[... cut ...]
5) I have tried a factory reset on the phone. Sure! "123456789*0#" ,DHCP opt. 150, TFTP and so on.
No effect.

6) I've tried downgrading and then upgrading the firmware. These versions:
cmterm-7940-7960-sccp.6-0-5.zip
cmterm-7940-7960-sccp.7-2-3.zip
cmterm-7940-7960-sccp.8-0-1.zip
cmterm-7940-7960-sccp.8-0-10.zip
cmterm-7940-7960-sccp.8-0-3.zip
cmterm-7940-7960-sccp.8-1-1.zip
cmterm-7940-7960-sccp.8-1-2.zip
cmterm-7940-7960-sccp.8-1-2SR1.zip
cmterm-7940-7960-sccp.8-1-2SR2.zip

7) I'v join "chan-sccp gitter chat". Waiting for reply.

dkgroot, 2019-03-25:
@eyatsko Was Alex (@PhantomVI) able to help you, figure out what went wrong ?

eyatsko, 2019-03-25:
Hello!

I'm waiting for his reply. :-) I became able to ask just this evening. An hour ago.

dkgroot, 2019-03-25:
Ok good luck.. He can be quite busy some days :-)

dkgroot, 2019-04-16:
@eyatsko Was this issue resolved / or moved to the sccp_manager site in the meantime ? Can i close this issue report here ?

dkgroot, 2019-08-31:
Any progress ?

dkgroot, 2019-09-15:
Closing issue report for now. Please reopen if/when required.

closed 2019-09-15 by dkgroot