chan-sccp/chan-sccp #460: Oops! on editing extension/phone

KCross84, 2019-03-31:
When you create a new phone in SCCP Manager or edit a phone marked as *NEW*, settings are saved. However, when you click the pencil icon to re-edit a phone, you get the following error:

Whoops \ Exception \ ErrorException (E_WARNING)
Invalid argument supplied for foreach()

**Reproduction Steps:**

1. Create a phone in SCCP manager or connect a phone with Hotline service activated.
2. Enter the settings and save.
3. Click the pencil icon above the trash icon.

**chan-sccp version:** (include revision and/or branch if relevant)
CHAN_SCCP Develop, SCCP Manager Develop.

**asterisk version:** (include revision and/or branch if relevant)
Current PBX Version:14.0.5.25
Current System Version:12.7.5-1902-3.sng7  

**Expected behavior:**
Saves new settings for phone.

**Observed behavior:**

Whoops! Whoops \ Exception \ ErrorException (E_WARNING)
Invalid argument supplied for foreach() in

   echo '<ul class="sortable">';
                                foreach ($codec_list as $codec => $codec_state) {
                                    $codec_trans = _($codec);

(Screenshots which show reproduction steps and/or demonstrate the problem)
http://jensencross.com/whoops.png

**Addition / Optional Information:**
- Problem started happening recently, didn't happen in an older version of Chan-SCCP: [Yes]
- Problem can be reliably reproduced, doesn't happen randomly: [Yes]
- Platform:
  - [Intel Celeron Dual Core]
  - [64 Bit]
  - [Big/Small Endian]
  - [Samgoma CentOS]
  - [7]
- Direct Contact information: [Kirk@jensencross.com or on here as KCross84.]

**BTW: Consider making a donation via paypal:**
- https://www.paypal.com/cgi-bin/webscr?&cmd=_donations&business=chan.sccp.b.pp%40gmail.com
[Absolutely!]

PhantomVl, 2019-03-31:
This problem is caused by an error or bug in "sccp_manager". 
Most likely this is due to some change in the internal formats of codecs freepbx.
If it is convenient for you, let's move the discussion of this problem to the chat https://gitter.im/sccp_manager/community

dkgroot, 2019-03-31:
@PhantomVl Thanks for picking up on this one. I was a little confused about the content, because it is not related directly to chan-sccp :-) Can/should we move this issue to https://github.com/PhantomVl/sccp_manager ?

dkgroot, 2019-04-16:
Closing this issue on here, for now ! I think it was already resolved on sccp_manager.

closed 2019-04-16 by dkgroot