chan-sccp/chan-sccp #462: Disable cfwdall from CLI

dkulemin, 2019-04-26:
Hi!
Is there a way to disable call forward all for line from CLI? Tried command "sccp set device SEPXXXXXXXXXXXX cfwdall off", but it just makes CFwdALL service unavailable from device and call forwarding is still enabled.

Thanks!

dkgroot, 2019-04-27:
That sounds like a regression to me.. I will have a look at it tomorrow.
In the meantime you could consider doing:
```
CLI> database show
CLI> database deltree SCCP/SEPxxxxx cfwdAll
CLI> database deltree SCCP/SEPxxxxx cfwdBusy
CLI> sccp restart SEPxxxxx
```
For all the devices in question, that should also do the trick.
Sorry about the inconvenience

dkgroot, 2019-04-27:
@dkulemin There currently is no method implemented to set/remove callforward from device/lines on the CLI. You mentioned being able to use "sccp set device SEPXXXXXXXXXXXX cfwdall off" but that is not implemented either". In retrospect i think that's a flaw that should be corrected, but it will take me a little bit to add it. Thanks for bringing it to my attention. I will let you know as soon as we have something.

dkulemin, 2019-04-29:
@dkgroot Thanks for the answer! And thanks for inviting in chatroom :)

dkgroot, 2019-04-29:
I will try to make some time to add this little feature somewhere this week. You should be ok, using the database solution for now i guess. 

Background Info: the astdb is used to store these settings temporarily so that we can recover them after an asterisk restart. So to get rid of them, you have to clean them out manually and restart chan-sccp or the device in question.

dkgroot, 2019-05-01:
Added CLI sccp callforward <lineName> [deviceid] [none|all|busy] [destination]
Added AMI SCCPCallForward: Line, Device, Type, Destination

Example:
> sccp callforward 98051 none
> sccp callforward 98051 all 0714343433
> sccp callforward 98051 SEP232332223 none
> sccp callforward 98051 SEP232332223 busy 0714343433

commit e1d94527 (2019-05-01): Add CLI sccp callforward <lineName> [deviceid] [none|all|busy] [destination]
Add AMI SCCPCallForward: Line, Device, Type, Destination

Example:
> sccp callforward 98051 none
> sccp callforward 98051 all 0714343433
> sccp callforward 98051 SEP232332223 none
> sccp callforward 98051 SEP232332223 busy 0714343433

Fixes: #462
closed 2019-05-01 by dkgroot