chan-sccp/chan-sccp #464: Conference barge issue after hangup

darkponay, 2019-05-08:
Hi,

We have a little issue with barging.
Say we have three phones and two lines. For example A1, B1 and C2 (where letter is the phone and number is the line).

B1 and C2 are discussing each other about their all new Chan-SCCP install and it's cool features.

A1, see the line button with a red light, press on it, and then press barge softkey. The A phone will go off hook and join the B1/C2 conversation very quickly. Sound works perfectly.

When A1 leave the conversation, and B1/C2 stays on call, the A1 line button goes off and at this moment it become impossible to barge again until B1/C2 hangup and call back.

Except this, barge seems to work flawlessly !
Thank you a lot for this feature !

Version:
Skinny Client Control Protocol (SCCP). Release: 4.3.2 develop - a5e4b15M (built by 'root' on '2019-05-08 15:52:21 UTC')

Running on Asterisk 16.3.0

dkgroot, 2019-05-15:
Thanks for the entertaining narrative :-)

I need a little time to fix up the exception cases.

dkgroot, 2019-05-15:
Fixed the issue (for shared line barge). Now the channel can be barged into again, after the first time.
Thanks for reporting the issue.

commit 373d8668 (2019-05-15): Fix resetting channelstate after barge hangup, so a channel can be barged on again

fixes #464
closed 2019-05-15 by dkgroot