chan-sccp/chan-sccp #465: Interfaces glitches

darkponay, 2019-05-08:
Hi,

We noticed some interface glitches in some situations :

1. Barge
When you barge into a call and then you leave, the remaining users (the "caller" of the initial call) will see "Barge From XXXX" in the bottom line until the end of their call. It should probably be "Barge from XXXX" and showed only when necessary. It should disappear after the barge end.

2. Call Waiting
When you are on the phone, if someone else calls you, you will see "Call WaitingName (Num)" in the bottom line. A space is missing and other problem, this indication will stay there until the end of the call.

3. Recording.
My users reported the bottom line shows "Recording" until you do anything with the phone. They would prefer the indication to disappear after a short delay since the recording button is lit, a second indication is not needed.

4. Conference.
When you are in a conference, if you dont show the ConfList, the right softkey on a 7975 (the 5th one) will be grayed, but shows an X.

These are present on this version : 
Skinny Client Control Protocol (SCCP). Release: 4.3.2 develop - a5e4b15M (built by 'root' on '2019-05-08 15:52:21 UTC')

Asterisk 16.3.0

dkgroot, 2019-05-08:
In all of these instances we do display the prompt with a timeout already, and on my phones this timeout is followed. Meaning the message goes away after 5-8 seconds. We need to figure out what might be causing this effect on your phones.

dkgroot, 2019-08-26:
@darkponay
Regarding recording calls, you can also use/add the 'monitor' softkey. This softkey will switch to show you the state of recording.
The number of supported icons on Feature buttons is very limited so we had to use what was available. Did not know they would not be supported on the Addon at all.

dkgroot, 2020-01-17:
Regarding "_Recording button icon_" I don't think there is a way to correct this. This is a firmware issue. We should only reduce the use of the 'Boxes Icon' on feature button like record.