chan-sccp/chan-sccp #466: ATA devices don't receive MWI

TheCranston, 2019-05-13:
registering ATA186 via chan_sccp - MWI stutter tone is not active.

dkgroot, 2019-05-13:
Create a new branch called "issues/466_mwi"
Note1: Solution has not been tested by me.
Note2: I did not know which dialtone to use to create a "stutter". For now i picked "SKINNY_TONE_RECALLDIALTONE" (which was not yet in use for anything).

***Git***
_Checkout this temporary branch_
```
git pull
git fetch -a
git branch
git checkout issues/466_mwi
./configure [....]
make -j4 && make install && make reload
```

_move back to develop afterwards:_
```
git checkout develop
```

Thanks for reporting the issue. Let's see if this quick solution does the trick, or more work is needed.

dkgroot, 2019-05-15:
https://sourceforge.net/p/chan-sccp-b/mailman/chan-sccp-b-discussion/thread/CAMZpLiUF0DqPHxY_zrd%2BLKEZSVeOejpysN9Qk9Lco_HEhxA7KA%40mail.gmail.com/
https://www.cisco.com/c/en/us/td/docs/voice_ip_comm/cucm/admin/11_0_1/featureConfig/cucm_b_feature-configuration-guide-1101/cucm_b_testmap_chapter_01110.html#CUCM_TK_C9B41467_00

I had a look in the Local Files (g3-tones.xml / g4-tones.xml and gateway-tones.xml).
They contain either amwi (audible message waiting indication) or "CPTONE_STUTT_DIALTONE"
So if we can find out if the ATA tries to download them, that would be a bit of progress
Note: I updated the US g?-tones.xml on the provision_sccp site (not sure if you are using that to provision your phones). The US tones were incorrect copies from the dutch versions (And apparently we don't use these stutter tones in NL)

dkgroot, 2019-05-15:
Little progress: @TheCranston got CallerID working by setting the `CallerID Method` in the ataxxxxxx.cfg file to US standard settings. No fix for the mwi yet.

dkgroot, 2019-07-29:
@TheCranston Have you had a change to give this a try ?

TheCranston, 2019-07-29:
Diederik,
I gave up on the Audible WMI (Stutter tone when the handset goes off hook)… SCCP handles this very differently compared to a SIP endpoint.  One day I’ll revisit it..

-Ben

dkgroot, 2019-07-29:
Should we close this issue for now (so you can reopen it in 2025 :-)

PS. Haven't [heard from | seen you in the chat] in a while, hope everything is ok !

closed 2019-07-29 by dkgroot

TheCranston, 2019-07-30:
Please close it for now, yes.  Perhaps in 2025 I’ll get back around to it.  Just trying to keep my head above water right now.

dkgroot, 2019-07-30:
@TheCranston Sorry to hear you are in a spot of trouble (sounds like), lots of luck with that. When you do have a little more time, do give me a shout in the chatroom if you like ! Keeping fingers crossed for you !

dkgroot, 2019-08-29:
@TheCranston The patch created above was pretty small and should not have any negative side effects. So i merged it into the main tree. When/if you get a chance to have another look at MWI on the ATA, to see if stutter works, give me a shout.