chan-sccp/chan-sccp #467: Asterisk segfault

SeSti4, 2019-06-13:
Hello, i'm using:

FreePBX 14.0.5.25
Asterisk 13.22.0
Sccp ver.4.3.1 r433 Revision :10561
Sccp Manager V.13.0.0.4

On phones used last firmware from cisco.com

Very often there is a problem.
Asterisk going to segfault ...

Asterisk Log is:
[2019-06-03 15:28:49] VERBOSE[10094][C-00000000] app_dial.c: Called PJSIP/telnumber@AddPac-Murom
[2019-06-03 15:28:49] VERBOSE[10094][C-00000000] app_dial.c: PJSIP/AddPac-Murom-00000083 is making progress passing it to SCCP/1604-0000054D
[2019-06-03 15:28:49] ERROR[10094][C-00000000] translate.c: Cannot determine best translation path since one capability supports no formats
[2019-06-03 15:28:49] ERROR[10094][C-00000000] astobj2.c: FRACK!, Failed assertion user_data is NULL (0)
[2019-06-03 15:28:49] VERBOSE[10094][C-00000000] logger.c: Got 28 backtrace records
[2019-06-03 15:28:49] VERBOSE[10094][C-00000000] logger.c: #0: [0x5fcc55] asterisk __ast_assert_failed() (0x5fcbd1+84)
[2019-06-03 15:28:49] VERBOSE[10094][C-00000000] logger.c: #1: [0x45ccf4] asterisk <unknown>()
[2019-06-03 15:28:49] VERBOSE[10094][C-00000000] logger.c: #2: [0x45d3bd] asterisk <unknown>()
[2019-06-03 15:28:49] VERBOSE[10094][C-00000000] logger.c: #3: [0x45d816] asterisk __ao2_ref() (0x45d7e8+2E)
[2019-06-03 15:28:49] VERBOSE[10094][C-00000000] logger.c: #4: [0x7f347e11390f] ast113/ast113.c:510 pbx_retrieve_remote_capabilities()
[2019-06-03 15:28:49] VERBOSE[10094][C-00000000] logger.c: #5: [0x7f347e114186] ast113/ast113.c:657 sccp_astwrap_indicate()
[2019-06-03 15:28:53] Asterisk 13.22.0 built by mockbuild @ jenkins7 on a x86_64 running Linux on 2018-07-25 22:30:39 UTC

dmesq is:
Jun  3 15:28:49 pbx kernel: asterisk[10094]: segfault at 8 ip 000000000051a98a sp 00007f34a28492a0 error 4 in asterisk[400000+38d000]

Can help?

SeSti4, 2019-06-14:
Ok! Thanks to help.
I will try to update the module, but I can check it only in two weeks.

and ... 729 codec i don't use. Only 711 (ulaw alaw)

dkgroot, 2019-06-14:
Ok... Sorry for the inconvenience. Do monitor your system in the meantime, to see if there is some logical set of circumstance that would cause this error. If you notice a way to reproduce the issue, let us know.

dkgroot, 2019-07-29:
@SeSti4 Any progress ?

SeSti4, 2019-07-30:
@dkgroot Hello, i`m update to "Release: 4.3.2 develop - 3dbf2ceM"
3 week without segfault =)

dkgroot, 2019-07-30:
@SeSti4 
Thanks for the update, that's really nice/good to hear. I did make a two small commit's since then, so you might want to give them a try in good time (preferably on a separate machine/instance). 

I guess we can close this bug report then (for now).

dkgroot, 2019-07-30:
@davidded The current `develop` branch is what @SeSti4 was using (2 commit's behind) and he was also having some trouble. You both moved to the head and are now trouble free, so the special/temp branch i created for you (David) might not be required.

closed 2019-07-30 by dkgroot

SeSti4, 2020-03-19:
Hello.
Again an error occurs

Asterisk | 13.22.0
Sccp ver.4.3.2 r432 Revision :10678 Hash :3dbf2ce

`[2020-03-15 10:02:20] VERBOSE[15547][C-00003b92] pbx.c: Executing [1144@from-queue:1] Set("Local/1144@from-queue-0001f640;2", "QAGENT=1144") in new stack
[2020-03-15 10:02:20] VERBOSE[15547][C-00003b92] pbx.c: Executing [1144@from-queue:2] Set("Local/1144@from-queue-0001f640;2", "__FROMQ=true") in new stack
[2020-03-15 10:02:20] VERBOSE[15547][C-00003b92] pbx.c: Executing [1144@from-queue:3] GotoIf("Local/1144@from-queue-0001f640;2", "0?hangup") in new stack
[2020-03-15 10:02:20] VERBOSE[15547][C-00003b92] pbx.c: Executing [1144@from-queue:4] GotoIf("Local/1144@from-queue-0001f640;2", "1?1190,1:hangup") in new stack
[2020-03-15 10:02:20] VERBOSE[15547][C-00003b92] pbx_builtins.c: Goto (from-queue,1190,1)
[2020-03-15 10:02:20] VERBOSE[15547][C-00003b92] pbx.c: Executing [1190@from-queue:1] Goto("Local/1144@from-queue-0001f640;2", "from-internal,1144,1") in new stack
[2020-03-15 10:02:20] VERBOSE[15547][C-00003b92] pbx_builtins.c: Goto (from-internal,1144,1)
[2020-03-15 10:02:20] VERBOSE[15547][C-00003b92] pbx.c: Executing [1144@from-internal:1] GotoIf("Local/1144@from-queue-0001f640;2", "1?ext-local,1144,1:followme-check,1144,1") in new stack
[2020-03-15 10:02:20] VERBOSE[15547][C-00003b92] pbx_builtins.c: Goto (ext-local,1144,1)
[2020-03-15 10:02:20] VERBOSE[15547][C-00003b92] pbx.c: Executing [1144@ext-local:1] Set("Local/1144@from-queue-0001f640;2", "__RINGTIMER=15") in new stack
[2020-03-15 10:02:20] VERBOSE[15547][C-00003b92] pbx.c: Executing [1144@ext-local:2] Macro("Local/1144@from-queue-0001f640;2", "exten-vm,novm,1144,0,0,0") in new stack
[2020-03-15 10:02:20] VERBOSE[15547][C-00003b92] pbx.c: Executing [s@macro-exten-vm:1] Macro("Local/1144@from-queue-0001f640;2", "user-callerid,") in new stack
[2020-03-15 10:02:20] VERBOSE[15547][C-00003b92] pbx.c: Executing [s@macro-user-callerid:1] Set("Local/1144@from-queue-0001f640;2", "TOUCH_MONITOR=1584255740.434063") in new stack
[2020-03-15 10:02:20] VERBOSE[15547][C-00003b92] pbx.c: Executing [s@macro-user-callerid:2] Set("Local/1144@from-queue-0001f640;2", "AMPUSER=7900XXXXXXX") in new stack
[2020-03-15 10:02:20] VERBOSE[15547][C-00003b92] pbx.c: Executing [s@macro-user-callerid:3] GotoIf("Local/1144@from-queue-0001f640;2", "1?report") in new stack
[2020-03-15 10:02:20] VERBOSE[15547][C-00003b92] pbx_builtins.c: Goto (macro-user-callerid,s,16)
[2020-03-15 10:02:20] VERBOSE[15547][C-00003b92] pbx.c: Executing [s@macro-user-callerid:16] NoOp("Local/1144@from-queue-0001f640;2", "Macro Depth is 2") in new stack
[2020-03-15 10:02:20] VERBOSE[15547][C-00003b92] pbx.c: Executing [s@macro-user-callerid:17] GotoIf("Local/1144@from-queue-0001f640;2", "1?report2:macroerror") in new stack
[2020-03-15 10:02:20] VERBOSE[15547][C-00003b92] pbx_builtins.c: Goto (macro-user-callerid,s,18)
[2020-03-15 10:02:20] VERBOSE[15547][C-00003b92] pbx.c: Executing [s@macro-user-callerid:18] GotoIf("Local/1144@from-queue-0001f640;2", "0?continue") in new stack
[2020-03-15 10:02:20] VERBOSE[15547][C-00003b92] pbx.c: Executing [s@macro-user-callerid:19] ExecIf("Local/1144@from-queue-0001f640;2", "1?Set(__CALLEE_ACCOUNCODE=)") in new stack
[2020-03-15 10:02:20] VERBOSE[15547][C-00003b92] pbx.c: Executing [s@macro-user-callerid:20] Set("Local/1144@from-queue-0001f640;2", "__TTL=63") in new stack
[2020-03-15 10:02:20] VERBOSE[15547][C-00003b92] pbx.c: Executing [s@macro-user-callerid:21] GotoIf("Local/1144@from-queue-0001f640;2", "1?continue") in new stack
[2020-03-15 10:02:20] VERBOSE[15547][C-00003b92] pbx_builtins.c: Goto (macro-user-callerid,s,37)
[2020-03-15 10:02:20] VERBOSE[15547][C-00003b92] pbx.c: Executing [s@macro-user-callerid:37] Set("Local/1144@from-queue-0001f640;2", "CALLERID(number)=7900XXXXXXX") in new stack
[2020-03-15 10:02:20] VERBOSE[15547][C-00003b92] pbx.c: Executing [s@macro-user-callerid:38] Set("Local/1144@from-queue-0001f640;2", "CALLERID(name)=7900XXXXXXX") in new stack
[2020-03-15 10:02:20] VERBOSE[15547][C-00003b92] pbx.c: Executing [s@macro-user-callerid:39] GotoIf("Local/1144@from-queue-0001f640;2", "0?cnum") in new stack
[2020-03-15 10:02:20] VERBOSE[15547][C-00003b92] pbx.c: Executing [s@macro-user-callerid:40] Set("Local/1144@from-queue-0001f640;2", "CDR(cnam)=7900XXXXXXX") in new stack
[2020-03-15 10:02:20] VERBOSE[15547][C-00003b92] pbx.c: Executing [s@macro-user-callerid:41] Set("Local/1144@from-queue-0001f640;2", "CDR(cnum)=7900XXXXXXX") in new stack
[2020-03-15 10:02:20] VERBOSE[15547][C-00003b92] pbx.c: Executing [s@macro-user-callerid:42] Set("Local/1144@from-queue-0001f640;2", "CHANNEL(language)=ru") in new stack
[2020-03-15 10:02:20] VERBOSE[15547][C-00003b92] pbx.c: Executing [s@macro-exten-vm:2] Set("Local/1144@from-queue-0001f640;2", "RingGroupMethod=none") in new stack
[2020-03-15 10:02:20] VERBOSE[15547][C-00003b92] pbx.c: Executing [s@macro-exten-vm:3] Set("Local/1144@from-queue-0001f640;2", "__EXTTOCALL=1144") in new stack
[2020-03-15 10:02:20] VERBOSE[15547][C-00003b92] pbx.c: Executing [s@macro-exten-vm:4] Set("Local/1144@from-queue-0001f640;2", "__PICKUPMARK=1144") in new stack
[2020-03-15 10:02:20] VERBOSE[15547][C-00003b92] pbx.c: Executing [s@macro-exten-vm:5] Set("Local/1144@from-queue-0001f640;2", "RT=") in new stack
[2020-03-15 10:02:20] VERBOSE[15547][C-00003b92] pbx.c: Executing [s@macro-exten-vm:6] Gosub("Local/1144@from-queue-0001f640;2", "sub-record-check,s,1(exten,1144,dontcare)") in new stack
[... cut ...]
[2020-03-15 10:04:05] VERBOSE[15547][C-00003b92] logger.c: #24: [0x7f473cf362d4] app_macro.so <unknown>()
[2020-03-15 10:04:05] VERBOSE[15547][C-00003b92] logger.c: #25: [0x7f473cf37552] app_macro.so <unknown>()
[2020-03-15 10:04:05] VERBOSE[15547][C-00003b92] logger.c: #26: [0x582ef6] asterisk pbx_exec() (0x582dda+11C)`

in log:

Mar 15 10:04:07 pbx kernel: asterisk[15547]: segfault at 8 ip 000000000062afda sp 00007f4714c39560 error 4 in asterisk[400000+38d000]

Can help?
This can happen once a month or several times a day.

dkgroot, 2020-03-19:
@SeSti4 
That's  a lot of dialplan output to go through.

However:
```
[2020-03-15 10:02:48] VERBOSE[15547][C-00003b92] bridge_channel.c: Channel Local/1144@from-queue-0001f640;2 joined 'simple_bridge' basic-bridge <5559ca83-a777-49c7-8931-e19919ea9fe3>
[2020-03-15 10:04:05] ERROR[15547][C-00003b92] astobj2.c: Excessive refcount 100000 reached on ao2 object 0x2f6de68
[2020-03-15 10:04:05] ERROR[15547][C-00003b92] astobj2.c: FRACK!, Failed assertion Excessive refcount 100000 reached on ao2 object 0x2f6de68 (0)
```
So it's the Local channel and app_dial code that's causing troubles here.

And in the backtrace these is no mention chan-sccp at all. So chan-sccp does not seem to be involved.

Time to update you asterisk release ?

dkgroot, 2020-03-19:
Sorry i  am not much help in this case.