chan-sccp/chan-sccp #468: undefined symbol: __ast_test_status_update

LeePorte, 2019-06-26:
having downloaded the chan-sccp package from  http://ppa.launchpad.net/chan-sccp-b/ppa/ubuntu it is unable to be loaded by Asterisk 16

I have checked ownership and permissions on `/usr/lib/asterisk/modules/chan_sccp.so` and these have been corrected.

**Reproduction Steps:**

1. Install Asterisk 16 from ubuntu repos
2. install chan-sccp from PPA
3. disable chan-skinny in /etc/asterisk/modules.conf
4. restart asterisk

**chan-sccp version:** (include revision and/or branch if relevant)
4.3.2.10626.a5e4b15~bionic from http://ppa.launchpad.net/chan-sccp-b/ppa/ubuntu bionic/main amd64 Packages

**asterisk version:** (include revision and/or branch if relevant)
1:16.3.0-1 from http://gb.archive.ubuntu.com/ubuntu bionic/universe amd64 Packages

**Expected behavior:**

chan-sccp to be loaded by asterisk

**Observed behavior:**

chan-sccp is not loaded with the following error;

```
Error loading module 'chan_sccp.so': /usr/lib/asterisk/modules/chan_sccp.so: undefined symbol: __ast_test_status_update
```

**Screenshots**

(Screenshots which show reproduction steps and/or demonstrate the problem)
[url]

**Addition / Optional Information:**
- Problem started happening recently, didn't happen in an older version of Chan-SCCP: [Unknown]
- Problem can be reliably reproduced, doesn't happen randomly: [Yes]
- Platform:
  - [CPU] Intel(R) Core(TM) i7 CPU  870  @ 2.93GHz
  - [32/64 Bit] 64 Bit 
  - [Operating System] Ubuntu
  - [Operating System Version] 18.04

dkgroot, 2019-06-26:
Hi Lee,

To be honest it has been a while since i tested the ppa's on my local system :-)

When using a package from ppa/launchpad, you need to make sure you use the asterisk version this module was compiled against. And it has to be an exact match.
So looking at the [build log](https://launchpadlibrarian.net/422794979/buildlog_ubuntu-bionic-amd64.chan-sccp_4.3.2.10626.a5e4b15~bionic_BUILDING.txt.gz) You can see that the exact asterisk version used to build the module is:
http://ftpmaster.internal/ubuntu bionic/universe amd64 asterisk amd64 1:13.18.3~dfsg-1ubuntu4 

So the simplest way would be to use this repo for your asterisk source (maybe we should update the README.md to notify people about this. But i guess you are the first one to actually use the chan-sccp ppa repo and running into this.

Another option is to just compile chan-sccp locally from the git repo. It's a very simple build and takes about 1 minute to complete. So that might be the faster and simpler option.

LeePorte, 2019-06-26:
I've built it locally and it's starting now.

I think updating the Readme would be a good idea. Any thoughts about deprecating the PPA?

Cheers

Lee

dkgroot, 2019-06-26:
I have been using the ppa build system mostly as a sort CI, because it's so easy to build for. Give it a debian pkg description and off it goes. And it builds it for a lot of platforms :-). So i actually use it, but not for what it is meant for. I will try out the chan-sccp.deb file this week, to see if it is actually usable as intended and then update the readme.md and wiki to give some guidelines. Thanks for pointing out the issue.
Hope you locally build module works like a dream.
BTW: If you like you can join us on [gitter chat](https://gitter.im/chan-sccp/chan-sccp)

closed 2019-06-26 by dkgroot