chan-sccp/chan-sccp #469: checkACL denies device over VPN when it should be permitted

Bikkies, 2019-07-06:
I'll pop the explanation below, but I'll chuck the problem and my awful hacky solution up here to get to the point quicker.

**Problem:** For some reason the checkACL function was denying a phone that should have been allowed according to the rules. The only thing that I can think of is because it was connected over a VPN, but the server should basically just see it as just being on the same LAN right? So no idea why that would do it.

**"Solution":** in src/sccp_device.c made  static boolean_t sccp_device_checkACL always return TRUE.

**More info:**

I have phones connecting successfully on my local network, but I wanted to connect one over a VPN.

My FreePBX server is on a 192.168.33.0/24 network and the phone is on a 192.168.48.0/24 network that is being routed over the VPN.

The VPN is L2TP over two Mikrotik routers with routes from the external router pointing to my 192.168.33.0/24 network over the VPN tunnel and routes from my router pointing to the 192.168.48.0/24 network over the tunnel. Everything routes correctly between the two LAN networks, machines can see each other on either side.

The phone is connected on the remote side and gets DHCP which points it to the TFTP server in the 192.168.33.0/24 network, and it downloads all the files correctly. As soon as it tries to register with the server though:

```
[2019-07-05 23:03:03] VERBOSE[14002] :     -- SEPXXXXXXXXXXXX: Allocating device to session (40) 192.168.48.101
[2019-07-05 23:03:03] VERBOSE[14002] :     -- SEPXXXXXXXXXXXX: not allowed by deny/permit list (deny:0.0.0.0/0.0.0.0,permit:127.0.0.0/255.0.0.0,permit:10.0.0.0/255.0.0.0,permit:172.0.0.0/255.224.0.0,permit:192.168.0.0/255.255.0.0,permit:192.168.0.0/255.255.0.0,permit:192.168.48.0/255.255.255.0,). Checking permithost list...
[2019-07-05 23:03:03] VERBOSE[14002] :     -- SEPXXXXXXXXXXXX: checkACL returning FALSE
[2019-07-05 23:03:03] NOTICE[14002] sccp_actions.c: SEPXXXXXXXXXXXX: Rejecting device: Ip address '192.168.48.101' denied (deny + permit/permithosts).
```

I had permitted 192.168.0.0/255.255.0.0 which should have let it through, but didn't, so I specifically also permitted 192.168.48.0/255.255.255.0 and then went crazy and added a bunch of stuff in there and even just allowed 0.0.0.0/0.0.0.0 ... no luck.

Tried every firewall and allow/deny setting I could find, nothing worked. So I decided to see what happens if I just skip past this specific check. So I made the change in the solution I mentioned at the top, and BAM, the phone instantly connected and was working perfectly over the VPN.

I'm not entirely sure how the checkACL function works from my quick look at it, but it doesn't make sense to me why it would be denying that IP address if it's within the allow list, unless it doesn't just check IP address, or somehow it was getting the IP address of the VPN tunnel and using that as the actual comparison, not sure what the actual issue is.

dkgroot, 2019-07-06:
Hi Bikkies,
Thanks for the report, and sorry to see you ran into this. It will be a little hard for me to exactly replicate your circumstance / network layout etc. But i will have a look at the code (tomorrow) and see if i can find anything. If i do, i will create a seperate branch called 'issues/469' and report back to you. That way we can work on the issue together (without bothering anyone) until we figure out the actual problem and the fix.
Did you set the permit/deny rule globally (in the [general] section) or per device ?

dkgroot, 2019-07-06:
Hi Bikkies,
I have re-enabled / added some debug output, so we can trace what's going on, and how chan-sccp is coming to the conclusion the devices ip-address it outside the deny/permit range.

***Git***
_Please checkout this temporary branch_
```
git pull
git fetch -a
git branch
git checkout issues/469
./configure [....]
make -j4 && make install && make reload
```
And set:
```
sccp debug core, device, socket, high
```
Then restart the device and post the device registration CLI output from the first socket connection to the checkACL being denied. If need be you can send the debug output via e-mail instead of posting here.

I am looking for something like this:
```
 SEPE0D173E11DXX: is registering, Instance: 1, UserId: 0, Type: Cisco 8945 (585), Version: 22 (loadinfo 'SCCP894x.9-4-2SR1-2')
    -- SEPE0D173E11D95: Our Session IP4 Address 192.168.178.6:0
    -- SEPE0D173E11D95: Allocating device to session (19) ::ffff:192.168.178.6
    -- deny:0.0.0.0/0.0.0.0
  == SCCP: (apply_netmask) applying netmask to 192.168.178.6/0.0.0.0
  == SCCP: (apply_netmask) result applied netmask 0.0.0.0
    -- SCCP: apply_ha_default: result: 0.0.0.0
    -- SCCP: apply_ha_default: current_ha->netaddr: 0.0.0.0
    -- permit:127.0.0.0/255.0.0.0
  == SCCP: (apply_netmask) applying netmask to 192.168.178.6/255.0.0.0
  == SCCP: (apply_netmask) result applied netmask 192.0.0.0
    -- permit:10.0.0.0/255.0.0.0
  == SCCP: (apply_netmask) applying netmask to 192.168.178.6/255.0.0.0
  == SCCP: (apply_netmask) result applied netmask 192.0.0.0
    -- permit:172.0.0.0/255.224.0.0
  == SCCP: (apply_netmask) applying netmask to 192.168.178.6/255.224.0.0
  == SCCP: (apply_netmask) result applied netmask 192.160.0.0
    -- permit:192.168.0.0/255.255.0.0
  == SCCP: (apply_netmask) applying netmask to 192.168.178.6/255.255.0.0
  == SCCP: (apply_netmask) result applied netmask 192.168.0.0
    -- SCCP: apply_ha_default: result: 192.168.0.0
    -- SCCP: apply_ha_default: current_ha->netaddr: 192.168.0.0
    -- SEPE0D173E11D95: checkACL returning TRUE
    -- permit:127.0.0.0/255.0.0.0
  == SCCP: (apply_netmask) applying netmask to 192.168.178.6/255.0.0.0
  == SCCP: (apply_netmask) result applied netmask 192.0.0.0
    -- permit:10.0.0.0/255.0.0.0
  == SCCP: (apply_netmask) applying netmask to 192.168.178.6/255.0.0.0
  == SCCP: (apply_netmask) result applied netmask 192.0.0.0
    -- permit:172.0.0.0/255.224.0.0
  == SCCP: (apply_netmask) applying netmask to 192.168.178.6/255.224.0.0
  == SCCP: (apply_netmask) result applied netmask 192.160.0.0
    -- permit:192.168.0.0/255.255.0.0
  == SCCP: (apply_netmask) applying netmask to 192.168.178.6/255.255.0.0
  == SCCP: (apply_netmask) result applied netmask 192.168.0.0
    -- SCCP: apply_ha_default: result: 192.168.0.0
    -- SCCP: apply_ha_default: current_ha->netaddr: 192.168.0.0
    -- SCCP: searching for our capability for device protocol version 22
    -- SCCP: searching for our capability for device protocol SCCP
    -- SCCP: found protocol version '22' at 22
```
In the output above, you can see we take the ip-address of the device trying to register. We walk throught the list of deny/permits and apply the netmask to the ip-address provided. This results in an applicable range, we then check if the range matched the one provided in the deny/permit entry. We do continue, even after we get a match, because there my be:
1. a better match
2. a later defined 'deny' which would cancel the earlier permit-match we got.

_move back to develop afterwards:_
```
git checkout develop
```

Bikkies, 2019-07-06:
I've checked out that branch and here is the output from the log for two connection attempts. They both look the same, but I noticed both do a netmask check right at the start and accept the connection, but then go on to deny it a bit after.

Also yeah I did try setting some network allows on the device itself but that didn't help and a set it back to inherit afterwards.

Here's the output I got, I don't think I'm seeing the debug line showing the result after the netmask stuff that's present in your example though:

```
[2019-07-06 23:05:15] VERBOSE[12617] :   == SCCP: (apply_netmask) applying netmask to 192.168.48.101/0.0.0.0
[2019-07-06 23:05:15] VERBOSE[12617] :   == SCCP: (apply_netmask) result applied netmask 0.0.0.0
[2019-07-06 23:05:15] VERBOSE[12617] :   == SCCP: (apply_netmask) applying netmask to 192.168.48.101/255.0.0.0
[2019-07-06 23:05:15] VERBOSE[12617] :   == SCCP: (apply_netmask) result applied netmask 192.0.0.0
[2019-07-06 23:05:15] VERBOSE[12617] :   == SCCP: (apply_netmask) applying netmask to 192.168.48.101/255.0.0.0
[2019-07-06 23:05:15] VERBOSE[12617] :   == SCCP: (apply_netmask) result applied netmask 192.0.0.0
[2019-07-06 23:05:15] VERBOSE[12617] :   == SCCP: (apply_netmask) applying netmask to 192.168.48.101/255.224.0.0
[2019-07-06 23:05:15] VERBOSE[12617] :   == SCCP: (apply_netmask) result applied netmask 192.160.0.0
[2019-07-06 23:05:15] VERBOSE[12617] :   == SCCP: (apply_netmask) applying netmask to 192.168.48.101/255.255.0.0
[2019-07-06 23:05:15] VERBOSE[12617] :   == SCCP: (apply_netmask) result applied netmask 192.168.0.0
[2019-07-06 23:05:15] VERBOSE[12617] :   == SCCP: (apply_netmask) applying netmask to 192.168.48.101/255.255.0.0
[2019-07-06 23:05:15] VERBOSE[12617] :   == SCCP: (apply_netmask) result applied netmask 192.168.0.0
[2019-07-06 23:05:15] VERBOSE[12617] :   == SCCP: (apply_netmask) applying netmask to 192.168.48.101/255.255.255.0
[2019-07-06 23:05:15] VERBOSE[12617] :   == SCCP: (apply_netmask) result applied netmask 192.168.48.0
[2019-07-06 23:05:15] VERBOSE[12617] :     -- SCCP: Accepted Client Connection from 192.168.48.101:35227
[2019-07-06 23:05:15] VERBOSE[12617] :     -- SCCP: Connected on server via 192.168.33.33:55349
[2019-07-06 23:05:15] VERBOSE[12617] :        > 192.168.33.33:55349: keepalive:62, keepaliveinterval:60
[2019-07-06 23:05:15] VERBOSE[12819] :        > SCCP: set poll timeout 60 for session 41
[2019-07-06 23:05:15] VERBOSE[12819] :        > SCCP: set poll timeout 60 for session 41
[2019-07-06 23:05:15] VERBOSE[12819] :        > SCCP: set poll timeout 60 for session 41
[2019-07-06 23:05:15] VERBOSE[12819] :     -- <?xml version="1.0" encoding="UTF-8"?><x-cisco-alarm><Alarm Name="LastOutOfServiceInformation"><ParameterList><String name="DeviceName">SEP5ca48afe60a7</String><String name="DeviceIPv4Address">192.168.48.101/24</String><String name="IPv4DefaultGateway">192.168.48.1</String><String name="DeviceIPv6Address"></String><String name="IPv6DefaultGateway"></String><String name="ModelNumber">CP-8941</String><String name="NeighborIPv6Address"></String><String name="NeighborIPv4Address">192.168.48.1</String><String name="NeighborDeviceID">MikroTik</String><String name="NeighborPortID">Bridge_LAN</String><Enum name="DHCPv4Status">1</Enum><Enum name="DHCPv6Status"></Enum><Enum name="TFTPCfgStatus">1</Enum><Enum name="DNSStatusUnifiedCM1">4</Enum><String name="VoiceVLAN">4095</String><String name="UnifiedCMIPAddress">192.168.33.33</String><String name="LocalPort">39561</String><String name="TimeStamp">1562418288</String><Enum name="ReasonForOutOfService">24</Enum><String name="LastProtocolEventSent">0x1</String><String name="LastProtocolEventReceived">0x0</String></ParameterList></Alarm></x-cisco-alarm>
[2019-07-06 23:05:15] VERBOSE[12819] :        > SCCP: set poll timeout 60 for session 41
[2019-07-06 23:05:15] VERBOSE[12819] :     -- <?xml version="1.0" encoding="UTF-8"?>
[2019-07-06 23:05:15] VERBOSE[12819] :     -- <x-cisco-alarm>
[2019-07-06 23:05:15] VERBOSE[12819] :     -- <Alarm Name="DeviceTLInfo">
[2019-07-06 23:05:15] VERBOSE[12819] :   == Alarm Type: DeviceTLInfo
[2019-07-06 23:05:15] VERBOSE[12819] :     -- <ParameterList>
[2019-07-06 23:05:15] VERBOSE[12819] :     -- <String name="DeviceName">SEP5ca48afe60a7</String>
[2019-07-06 23:05:15] VERBOSE[12819] :     -- <String name="IPv4Address">192.168.48.101</String>
[2019-07-06 23:05:15] VERBOSE[12819] :     -- <String name="CTL_Signature">Not Installed</String>
[2019-07-06 23:05:15] VERBOSE[12819] :     -- <String name="CTL_TFTP_Server">N/A</String>
[2019-07-06 23:05:15] VERBOSE[12819] :     -- <String name="ITL_Signature">Not Installed</String>
[2019-07-06 23:05:15] VERBOSE[12819] :     -- <String name="ITL_TFTP_Server">N/A</String>
[2019-07-06 23:05:15] VERBOSE[12819] :     -- </ParameterList>
[2019-07-06 23:05:15] VERBOSE[12819] :     -- </Alarm>
[2019-07-06 23:05:15] VERBOSE[12819] :     -- </x-cisco-alarm>
[2019-07-06 23:05:15] VERBOSE[12819] :        > SCCP: set poll timeout 60 for session 41
[2019-07-06 23:05:15] VERBOSE[12819] :  SEP5CA48AFE60A7: is registering, Instance: 1, UserId: 0, Type: Cisco 8941 (586), Version: 22 (loadinfo 'SCCP894x.9-4-2-8')
[2019-07-06 23:05:15] VERBOSE[12819] :     -- SEP5CA48AFE60A7: Allocating device to session (41) 192.168.48.101
[2019-07-06 23:05:15] VERBOSE[12819] :   == SCCP: (apply_netmask) applying netmask to 192.168.48.101/0.0.0.0
[2019-07-06 23:05:15] VERBOSE[12819] :   == SCCP: (apply_netmask) result applied netmask 0.0.0.0
[... cut ...]
[2019-07-06 23:05:41] VERBOSE[12858] :   == SEP5CA48AFE60A7: checking buttonconfig for pendingDelete (index:3, type:Line (0), pendingDelete:False, pendingUpdate:False)
[2019-07-06 23:05:41] VERBOSE[12858] :   == SCCP: Unregister Device SEP5CA48AFE60A7
[2019-07-06 23:05:41] VERBOSE[12858] :     -- SEP5CA48AFE60A7: (sccp_devstate_deviceRegisterListener) device unregistered
[2019-07-06 23:05:41] VERBOSE[12858] :     -- SEP5CA48AFE60A7: Registration State is 'None'
[2019-07-06 23:05:41] VERBOSE[12858] :     -- SEP5CA48AFE60A7: Registration State is 'None'
[2019-07-06 23:05:41] VERBOSE[12858] :     -- SCCP: Destroy Session 192.168.48.101
```

dkgroot, 2019-07-06:
Could it be that you forgot to set the "sccp debug" after reloading , i am missing some output, that i was expecting ? If you like you can also set the debug=xxx line in the sccp.conf.

dkgroot, 2019-07-07:
```
  SEP5CA48AFXXXX: is registering, Instance: 1, UserId: 0, Type: Cisco 8941 (586), Version: 22 (loadinfo 'SCCP894x.9-4-2-8')
     -- SEP5CA48AFXXXX: Our Session IP4 Address 192.168.48.101:0
     -- SEP5CA48AFXXXX: Allocating device to session (27) 192.168.48.101
     -- deny:0.0.0.0/0.0.0.0
   == SCCP: (apply_netmask) applying netmask to 192.168.48.101/0.0.0.0
   == SCCP: (apply_netmask) result applied netmask 0.0.0.0
     -- SCCP: apply_ha_default: result: 0.0.0.0
     -- SCCP: apply_ha_default: current_ha->netaddr: 0.0.0.0
     -- permit:192.168.33.0/255.255.255.0
```
The phones's ip is masked by the vpn connection, the origin of the connection is now coming from the FreeBSD machine providing the VPN service
```
   == SCCP: (apply_netmask) applying netmask to 192.168.48.101/255.255.255.0
   == SCCP: (apply_netmask) result applied netmask 192.168.48.0
     -- SEP5CA48AFXXXX: ip address 192.168.48.101 not allowed by deny/permit list (deny:0.0.0.0/0.0.0.0,permit:127.0.0.0/255.0.0.0,permit:10.0.0.0/255.0.0.0,permit:172.0.0.0/255.224.0.0,permit:192.168.0.0/255.255.0.0,permit:192.168.0.0/255.255.0.0,permit:192.168.48.0/255.255.255.0,). Checking permithost list...
     -- SEP5CA48AFXXXX: checkACL returning FALSE
 sccp_actions.c: SEP5CA48AFXXXX Rejecting device: Ip address '192.168.48.101' denied (deny + permit/permithosts).
 sccp_actions.c: SEP5CA48AFEXXXX: Phone's physical ipv4 address: 192.168.48.101:0.
 ```
If you add 192.168.33.0/255.255.255.0 the device would have gotten through. Although i must admit the debug output is a little confusing. From the asterisk perspective the connecting is coming in from 192.168.33.0. Not all VPN's work in the same way (depends on the layer they operate on and how they wrap packets, and if they do any packet re-writing in the process).

dkgroot, 2019-07-07:
Ok we figured it out. There was both a /etc/asterisk/sccp_hardware.conf and an entry in the realtimedb. Apparently chan-sccp took both, and added them together. In the old sccp_hardware.conf, the port was still blocked and the conclusion made by chan-sccp was the right one in the end.

closed 2019-07-07 by dkgroot