chan-sccp/chan-sccp #471: 7942 phone not responding properly after sitting for some time

hedgehog1029, 2019-08-26:
After sitting for some time untouched, pressing "New Call" or one of the line buttons does not properly start a new call - a small speaker icon 

**Reproduction Steps:**

1. Wait some time without touching the phone
2. Try pressing a line button or the New Call softbutton
3. Observe that a normal call is not started, the line button is not lit, and a small speaker icon is present in the bottom left corner.
4. Wait approximately 5 seconds.
5. The line button now lights up and the phone switches to the on-call interface, however no dialtone is heard and numbers cannot be dialled
6. In this state, attempting to dial numbers is silent phone-side; only log messages are printed indicating button presses are received

**chan-sccp version:** 4.3.2 TRUNK (Fedora 28 package)  
**asterisk version:** 15.3.0 (Fedora 28 package)  

**Expected behavior:** Calls can be started normally  
**Observed behavior:** Calls fail to be started normally.  

**Addition / Optional Information:**
- Problem started happening recently, didn't happen in an older version of Chan-SCCP: No, problem has always been present
- Problem can be reliably reproduced, doesn't happen randomly: No
- Platform:
  - Pentium E5300
  - 64 Bit
  - Fedora 28 (Linux)
- Log File: [https://paste.dimdev.org/azunutomiq](https://paste.dimdev.org/azunutomiq)

Some discussion of this issue in the gitter room has taken place. One thing to note is that I also saw this behavior with chan_skinny, so this is probably a workaround discussion rather than a bug discussion.

hedgehog1029, 2019-08-26:
Here's the log with `message` switched on: https://paste.dimdev.org/efegehegoq

The time before this behavior is seen again seems random - sometimes it takes hours, sometimes minutes, but if you're actively using the phone it doesn't seem to reoccur.

dkgroot, 2019-08-26:
I just notice you are running asterisk-15, is that intentional ? Can you tell me the current chan-sccp revision number you are using. I just want to make sure that's up to date. You said you where running the pre-build chan-sccp rpm's (right ?)

dkgroot, 2019-08-26:
Can you please attach the same debug log but with 'message' as extra debug param. That might shed some light.
You say 'After sitting for some time untouched'... How long would that normally be ?

hedgehog1029, 2019-08-26:
here's my sccp.conf: https://gist.github.com/hedgehog1029/f76b027afabe37b99e910679111037ef

dkgroot, 2019-08-26:
Hmmm. i am not seeing the `message` output in there. I should look something like this:
```
[Aug 26 21:17:16] NOTICE[2633]: sccp_session.c:1036 int sccp_session_send2(const sccp_session_t *const, sccp_msg_t *): SEP381C1AB6B976: Send Message: Define Time Date(0x0094) 40 bytes length
 SCCP: packet hex dump:
 00000000 - 28 00 00 00 00 00 00 00  94 00 00 00 E3 07 00 00  - (...............
 00000010 - 08 00 00 00 01 00 00 00  1A 00 00 00 15 00 00 00  - ................
 00000020 - 11 00 00 00 10 00 00 00  00 00 00 00 3C 30 64 5D  - ............<0d]
    -- SEP381C1AB6B976: >> Got message Keep Alive Message (0x0)
```
For each and every TCP message send to and from the phone

dkgroot, 2019-08-26:
Ooh i am seeing the sccp_session_send2 part, but not the hex dumps underneath. They got filtered out somewhere.

dkgroot, 2019-08-26:
looking at it again: It looks like the device send us a 'Accessory Status Message'... And apparently it's expecting us to respond to that somehow. That could make sense.

dkgroot, 2019-08-26:
Reading a little further down, stuff is getting worse
```
[Aug 26 19:24:11] NOTICE[13776][C-00000001]: sccp_actions.c:122 check_session_message_device: SEP18339D142010: SCCP Handle Message: Off-Hook Message(0x0006) 20 bytes length
    -- SEP18339D142010: Getting the active channel on device.
    -- SEP18339D142010: (session_dissect_msg) Incoming message is smaller(1852) than known size(2012).
    -- SEP18339D142010: >> Got message Update Capabilities Message (0x30)
[Aug 26 19:24:12] NOTICE[13776][C-00000001]: sccp_actions.c:122 check_session_message_device: SEP18339D142010: SCCP Handle Message: Update Capabilities Message(0x0030) 1852 bytes length
    -- SEP18339D142010: Device has 9 Audio Capabilities, RTPPayloadFormat=0 (V2)
```
The Update Capabilities message normally only comes ones, shortly after registering, not in the middle of a call. Also just before it's complaining that it's not receiving the normal amount of data. I think we might have a different problem all together.

A TCP stream has to be a continues stream of data (without any errors). Once the server and the client get out of sync (for whatever reason) the server and the client would be reading junk. I am not 100% that is the case here, but it does smell a little like it. Although most of the data seems to be legible.

dkgroot, 2019-08-26:
Can you post your sccp.conf here as well, there might be something in there, that sheds a light. I don't have a 7942, but the 7962 is close enough (i think) and could try your config on it.

dkgroot, 2019-08-26:
Nice and clean / short... Nothing in it really. Just the way i like it. 

Did not see anything that stood out.. If you like you can remove the link, if you don't want it fully public. Thanks

dkgroot, 2019-08-26:
The tcpdump might clarify what is going on. I am sorry to put you through all this hassle.

dkgroot, 2019-09-11:
@hedgehog1029 Hi Henry, any update ?

hedgehog1029, 2020-01-03:
hi, sorry for disappearing for a long time! other commitments got in the way! i've got a basic tcpdump of the problem but haven't matched it up to a debug log yet - hopefully the problem occurs again today and I can get those logs combined :)

dkgroot, 2020-01-03:
No need to be sorry :-)

hedgehog1029, 2020-01-04:
i've got both the logs matched up now:
SCCP debug log: https://paste.dimdev.org/ovukekeleb
tcpdump: https://paste.dimdev.org/bijahinivi

the tcpdump definitely looks abnormal; I compared it to a normal, working session and there definitely seems to be a lot more null bytes

dkgroot, 2020-01-04:
@hedgehog1029 I was hoping for a pcap file, so i could run in through wireshark :-)

hedgehog1029, 2020-01-04:
oh no problem! did another capture this morning, here's the pcap file: [phone.pcap](https://witch.press/phone.pcap)

corresponding log: https://paste.dimdev.org/isucekibij.sql

dkgroot, 2020-01-06:
Looking at the logs and pcap, everything looks normal. Except for the "Update Capabilities Message" entry. That is a bit unexpected. Normally we only get that message when the device has just registered, not in the middle of a call. I think the device might be in power save mode and takes a long time to wake up. When it finally wakes up, we are already stopping the call because of timeout.

Can you have a look at your SEPxxx.cnf.xml to see if changing the powersave features has an effect ?

I currently don't have an idea what i could change in the source to remmedy this situation.

hedgehog1029, 2020-01-08:
is there documentation on the power save features somewhere? I don't think I have any enabled. I do know that the problem can occur even after browsing the settings menu, loading services, etc. so not sure how to detect if the phone is in a power-save state.

dkgroot, 2020-01-08:
@hedgehog1029 
These are some of the power save values your can set in SEPxxxx.cnf.xml
```xml
<vendorConfig>
...
    <daysDisplayNotActive>1,7</daysDisplayNotActive>
    <displayOnTime>08:30</displayOnTime>
    <displayOnDuration>11:30</displayOnDuration>
    <displayIdleTimeout>01:00</displayIdleTimeout>
    <displayOnWhenIncomingCall>1</displayOnWhenIncomingCall>
    <enableCdpSwPort>1</enableCdpSwPort>
    <enableCdpPcPort>1</enableCdpPcPort>
    <enableLldpSwPort>1</enableLldpSwPort>
    <enableLldpPcPort>1</enableLldpPcPort>
    <lldpAssetId>Cisco Phone</lldpAssetId>
    <powerPriority>1</powerPriority>
....
</vendorConfig>
```
I am not sure if this is related, but it could be based on seeing the 'UpdateCapabilities' message.

Are you using PowerOverEthernet (POE) to power your phones ? In that case the switch might have off-hours and reduce power output.

If you have time, please sign up to our [gitter chat](https://gitter.im/chan-sccp/chan-sccp) and ask some other members if they have a 7942 and are having similar issues.