chan-sccp/chan-sccp #472: Capture ext between SCCP and SIP

mmguglielmi, 2019-09-03:
I can't capture between SCCP and SIP

**Reproduction Steps:**

1. ext 101 ring (SCCP)
2. 101 answer, then transfer ext 200 (SIP)
3. ext 212 (SCCP) try to capture (with *8 or CISCO capture  button), and not work
I attach asterisk log of both ( *8 and capture button)

Should I have any special consideration in config?

[with ast 8.txt](https://github.com/chan-sccp/chan-sccp/files/3567295/with.ast.8.txt)
[with CISCO button.txt](https://github.com/chan-sccp/chan-sccp/files/3567296/with.CISCO.button.txt)

Thanks in advance for your help
Regards
Marco

dkgroot, 2019-09-03:
Hi @mmguglielmi ,

If by "capture" you mean gpickup/pickup. You can use so called directed pickup (pickup softkey) by pressing the pickup button and entering the number of the RINGING extension you would like to answer. Or if you want to use gpickup, you need to make sure that:
the sccp device you are using is a call pickupgroup which matches the (named)callgroup of the phone you would like to answer. So to answer a remote SIP call, you need to 
sip.conf
```
[sipuser]
...
callgroup=1
```

sccp.conf
```
[SEPxxxxx]
...
pickupgroup=1
```
This will allow you to gPickup a call from a remote RINGING sip call. You can also use named call/pickup groups (see asterisk documentation.

If by "capture" you mean something else then please re-explain.

dkgroot, 2019-09-11:
@mmguglielmi Any update ? Can this issue report be closed ?

mmguglielmi, 2019-09-11:
Hi @dkgroot 
Thanks for asking
We're doing few more test, and I'll update
The problem continue, but I want to give you more info...
Thanks a lot

mmguglielmi, 2019-09-18:
Hi @dkgroot
We were making a mistake in the config you suggested
It's working now
I really appreciate your answer
Thanks a lot

dkgroot, 2019-09-18:
Ok, thanks for the update. Good luck using chan-sccp !
Closing this issue

closed 2019-09-18 by dkgroot