chan-sccp/chan-sccp #473: Caller ID not transferring on blind transfer using transfer softkey

rofferman, 2019-09-09:
When I transfer a call using the transfer softkey the caller id of the inbound caller is not sent with the call, instead the name of internal extension is sent.
If I use ## to transfer the call is transferred with the caller id information correctly

I am using freepbx with asterisk 16. 

This was working up until a few weeks ago.  I cannot figure out what has changed

dkgroot, 2019-09-09:
@rofferman 
I am not sure which revision might have caused this. Most resent commit merges are only done after quite a bit of testing, to prevent regressions. I am not a 100% sure, why you mean by `internal extension` to be honest. An example (either as image or text) of the shown callinfo would have been more helpfull.

Would you be able to do a `git bisect` to figure out which commit cause the regression. To find the revision you were on previously (a couple of weeks ago) you can use `git reflog`.

rofferman, 2019-09-09:
For example  a call comes in with the caller id of the incoming caller.. The caller is transferred to another phone in the building. The original caller id showing the callee is overwritten by the extension that is transferring the call. So the phone that the call was transferred to is showing the extension of the phone that transferred the call. 

I can get some log information uploaded tomorrow and get the version of the module I am on.
Would the asterisk logs showing this transfer be of help?

dkgroot, 2019-09-09:
If you set `sccp debug core, channel, callinfo` the debug logs should be helpfull. I would have to see the before and after though (so the old situation and the new). And please do include some details about the devices and lines taking part in the call. Otherwise it's difficult to figure out what is what (for someone outside your network).

rofferman, 2019-09-10:
[sccptosccp.txt](https://github.com/chan-sccp/chan-sccp/files/3596928/sccptosccp.txt)
[sccptosip.txt](https://github.com/chan-sccp/chan-sccp/files/3596931/sccptosip.txt)
 I am attaching the two asterisk logs i captured and am working on getting some other logs.
When i transfer from SCCP to SIP the callerid transfers correctly, it seems only be when i am transferring SCCP to SCCP that it does not. I am using Cisco 7975 and 7942 phones

rofferman, 2019-09-10:
[sccptosccpdegugon.txt](https://github.com/chan-sccp/chan-sccp/files/3596949/sccptosccpdegugon.txt)
Sorry I forgot to have debug on.. Here is the SCCPto SCCP log with debug turned on

Here is the version I am using
Skinny Client Control Protocol (SCCP). Release: 4.3.2 develop - ed9fbdbM (built by 'root' on '2019-09-09 15:48:43 UTC')

dkgroot, 2019-09-10:
@rofferman Thanks for the debug logs, but i do need a little more help.

Per debug log, i need to know who is calling who (which phone to which phone). Then which phone is transfering the call, and what the callerid is that you are seeing and what you would have expected.

Example : SIP:A is calling SCCP:B at extension 111. B transfers the call blindly to C using Transfer softkey. When A is connected to B, the correct callinfo is displayed. When the transfer is in progress (RINGING) the incorrect callinfo is displayed on C. etc.

You might have noticed that there are a lot of callerid exchanges happening during a transfer (in the debug log). Also freepbx does some connectedline updates during dialplan handling as well, making this even more difficult (especially for me as an outsider).

rofferman, 2019-09-11:
I am calling 812-492-3538 which is the DID for  SCCP phone 3538 from my cell phone. Blind Transferring the call to SIP 3591 using transfer softkey on SCCP phone 3538. when the transfer is in progress it is showing the correct caller id from my cell phone(original incoming call) on SIP 3595. 

I am using the same structure as above to test from SCCP to SCCP. The only difference is that SIP 3591 changes to SCCP 3595. In this scenario the caller id is not transferred from my cell phone. Instead the call transfers to SCCP 3595 but shows SCCP 3538 as the caller

I hope that helps clear it up.. Please let me know if there are any parts I am missing.

I have not used git bisect much but I will look and see if I can find anything with it

rofferman, 2019-09-11:
I have found that this way works, but it used to simpler than this

**If you really want to maintain visibility of the Caller ID, the simplest way we have found is to put the call on hold, grab a line and call the party the call is for, and if they want it, hang up and blind-transfer the call.**

anutator, 2019-10-22:
I also have an issue with softkey - softkey "Answer" during ringing.
When I try to press it nothing happens and I see "this line is not active"
Phone - 7911
softkeyset - [softkey2]
....
ringin=answer,idivert

I pressed two times and could answer only when onhook phone.

[softkey-answer.txt]
(https://github.com/chan-sccp/chan-sccp/files/3757734/softkey-answer.txt)

dkgroot, 2019-10-22:
@rofferman 
> I have found that this way works, but it used to simpler than this
> If you really want to maintain visibility of the Caller ID, the simplest way we have found is to put the call on hold, grab a line and call the party the call is for, and if they want it, hang up and blind-transfer the call.

That does not really sounds like a sensible way of having to do things, i agree. Did your bisect have any result ?

dkgroot, 2019-10-22:
@anutator 
That is completely unrelated to this issue, and could be very confusing (and get lost easily).

Please let's create a new issue for it. Please include some answers to these questions:

Does it only happen on the 7911 Phone. Was this also present in older chan-sccp versions ? Can you show the output of 'Answer' key not working  ( with 'sccp debug core, device, line, channel, softkey, action' set). Does it work correctly with the standard softkeyset ?

anutator, 2019-10-24:
I found th answer on Cisco forum - Answer key depends on model:

> The determining factor for using the "Answer Softkey" with the Handset On Hook is whether or not the phone model has a speakerphone. If Yes then Answer Softkey is supported, if No (like the 7911,7912,7905) the the Answer Softkey with the Handset On Hook is not supported..You will see that if the Handset is Off-Hook the Answer Softkey is displayed on an Incoming call :)

dkgroot, 2019-10-24:
@anutator 

:-)
i never thought about or realised that there were skinny phones without speaker mode. But without a speakermode it makes sense to also suppress the 'answer' softkey. I guess that answers that question. I will have a look at the source code to see if any ajustments should be made to suppress the creation of the Answer SoftKey in this case.

rofferman, 2019-10-28:
I did a bisect and was unable to get to find a working commit
I just got back from vacation and will dig in and try to find some helpdul info in the logs that may point in the right direction.. It is a weird issue.

I did find I think before I left where it is treating the blind transfer as an attended transfer in the debug logs. I tried changing the transfer on hangup option and got the same result.

I will do some more looking and get back. Is there anything you can think of where I should look for more information.

dkgroot, 2019-10-28:
Enjoy your vacation !
By any chance are you also using FreePBX like @tabbermj in the other issue report ? That might explain somethings and might actually help towards a solution as well, if we find out what they are trying to achieve.

rofferman, 2019-10-30:
I am not sure what happened, maybe a freepbx update but my caller id is transferring now like it should. 
I have not made any changes or pulled any updates from chan sccp, just freepbx and today it is transferring as it should. Very Strange. 
So when a caller calls (A) and the receptionist answers the call (B) when the receptionist pushes the transfer softkey, then presses  the BLF (C) to transfer it to, and hangs up. The original caller id (A) is transferred to pressed BLF line (C).

PS. I hope that made sense. I have been on vacation lol

Just thought I would update

dkgroot, 2019-10-30:
I guess they fixed something upstream which corrected the issue. Chan-sccp and FreePBX do quite a bit of massaging of the callerid information to try and get the right information. I guess we differ in who we think should be responsible for this though :-)

I had already updated the statement above where i said 'enjoy your vacation' after re-reading your previous comment :-)

I guess we can close this issue, right ? Or did i miss something ?

tabbertmj, 2019-10-31:
I'm on FreePBX 14.0.13.6.  I do not see an update listed.  @rofferman, What version are you on?

tabbertmj, 2019-11-05:
I will give that a try tonight.  Thanks!

stoyanmarinov, 2019-11-05:
Had the same issue, dug a bit deeper. The issue is with FreePBX, seems to be fixed in git already about 3 weeks ago. It's an issue with macro-dial-one. If you want to patch yours, go edit /var/www/html/admin/modules/core/functions.inc/macro-dial-one.php

Currently at line 93

replace:
        $ext->add($mcontext,$exten,'', new ext_set('D_OPTIONS', '${D_OPTIONS}I'));
with:
        $ext->add($mcontext,$exten,'', new ext_set('D_OPTIONS', '${D_OPTIONS}'));

(note the missing I, like in uppercase i)

after this rebuild your dialplan (fwconsole reload I guess)

dkgroot, 2019-11-05:
@stoyanmarinov Wow nice find !

dkgroot, 2019-12-17:
Closing this issue, as it seems to have been resolve in FreePBX (correct). Please reopen if/when necessary.

closed 2019-12-17 by dkgroot