chan-sccp/chan-sccp #474: Russian letters in phone description, year - four digits

anutator, 2019-09-18:
I have russian locale with buttons in Russian.
Button labes in Russian, everything is Ok.
But I can't get phone description in russian - get dirty symbols. So I have to white it in English (Surname and phone number).

The second question: tried D/M/YY but can't get year 2019, only 19.

**chan-sccp version:**  - the latest
**asterisk version:** Asterisk 16.5.0
phones - 7911, firmware - the latest SCCP11.9-4-2SR3-1S

dkgroot, 2019-09-18:
@anutator Thank you for reporting the issue.

I have looked at the code (sccp_action.c:1553 / sccp_protocol.c:1312) and we use the same locale conversion code for the description and the line button labels. . It could be that the 7911 is just broken in this way. I don't have any 7911 around anymore. I do have 7960/7970 phones, which displays the description correctly (in other locale's). I do know you can transliterate words/names in russian, so i guess you might have to resort to that :-(.

Can someone out there using a different locale, verify that the description works (or doesn't) on 7911 / older skinny phones ?

Where did you set the D/M/YY, you need to make sure it is set both in the sccp.conf and the SEPxxx.cnf.xml file. Some of the phone types use one and some the other.

ojaksch, 2019-09-18:
Yes, I'm using a 7911/sccp phone in my kitchen and can check this out later 
this day.
I setup the date field in sccp.conf as dateformat = D.M.Y in [general] and as 
<dateTemplate>D.M.YY</dateTemplate> in *.cnf.xml which displays the date 
"HH:MM DD.MM.YY" at least on my 7960s and 7970s using locale German_Germany/
de_DE
-
Oliver

On Wednesday, 18 September 2019, 13:21:09 CEST you wrote:
> @anutator Thank you for reporting the issue.
>
> I have looked at the code (sccp_action.c:1553 / sccp_protocol.c:1312) and we
> use the same locale conversion code for the description and the line button [...]

dkgroot, 2019-09-18:
@ojaksch 
Thanks Oliver. Maybe you can add "öß" to the description on that phone, to see if they come out correctly (when testing).

Note to @anutator: In germany they separate dates using "." so that format is the norm there. If i am not mistaken, this is also the case in the Russian Federation (correct?).

ojaksch, 2019-09-18:
Will name the device "Küche" (Kitchen) later then. But if I remember correctly 
I did this at first setup of the 7911 which hasn't worked - that's why it's 
name is "Kueche" at the moment.

dkgroot, 2019-09-18:
@ojaksch Isn't transliteration wonderfull :-) Thanks for helping out / checking BTW !

ojaksch, 2019-09-18:
It IS, indeed! And always at service as you know :)

On Wednesday, 18 September 2019, 15:12:23 CEST you wrote:
> @ojaksch Isn't transliteration wonderfull :-) Thanks for helping out /
> checking BTW !

dkgroot, 2019-09-18:
Current status on 7911 (provided by @PhantomVl)
[image]
[image]

Top Row DisplayLabel (ie: device description) should have said: "Дима"
We are sending the string correctly in UTF8 format, which java should be able to display.

dkgroot, 2019-09-18:
@anutator
We did some more tests (with german locale).
On the 7970 (java) description works (UTF8)
On the 7960 (pre-java) description works as well (UTF8 Converted to CodePage)
On the 7911 (java) description doesn't work (send in UTF8, which should have worked) 

To be honest, with the low number of people using this type of phone, and no real idea about what might be causing this, we should maybe just skip this issue. I would recommend using transliteration for the description on this phones, or just leaving it empty.

If you agree, you can just close the issue. If you have a lot of these phones (100+) and you would be willing to invest some time to do more research, we can continue.

anutator, 2019-09-18:
> @ojaksch
> Thanks Oliver. Maybe you can add "öß" to the description on that phone, to see if they come out correctly (when testing).
>
> Note to @anutator: In germany they separate dates using "." so [...]

Yes. I tested D.M.YY and also see 18.09.19 instead of 18.09.2019.
And in upper right corner as on screenshot (provided by @PhantomVl). Also use utf-8 in SEP file.

dkgroot, 2019-09-18:
> Besides on CUCM we only have the number on upper right corner but I'd like to see surname using command sccp show devices.

Aah so on CUCM, that corner is filled with the `primary line number` ie: ascii only. That explains a lot. i did not know that. They might use the shorter date format, to preserve space, it is a pretty small display already.

dkgroot, 2019-09-25:
Shall we close this issue report, in the assumption that the display is limited and cisco made these decisions for us ?

dkgroot, 2019-10-05:
@anutator Any update ?

anutator, 2019-10-07:
Cisco 7925 has other issue - 1919 instead of 2019.
You can close issue. I think that old models have bugs.

dkgroot, 2019-10-07:
@anutator the 7925 issue sounds really wrong. Can you check the cisco firmware page for this device. it should show that this issue exists (i would think)

dkgroot, 2020-01-10:
Closing issue for the moment, because of lack of response.

anutator, 2020-01-10:
Hi… I registered all phones on pjsip and use smartphone with Grandstream wave instead of..7925… ..

dkgroot, 2020-01-11:
Ooh ok. I guess we can close this issue for now then. Sorry to hear you are not using chan-sccp anymore. But thanks for reporting back !

closed 2020-01-17 by dkgroot