chan-sccp/chan-sccp #475: Autoprovision

anutator, 2019-09-18:
Simple script to make SEP files from sccp.conf
Not an issue))
Maybe you can make it better ant correct my script.
You must have templates in /tftpboot for different phone models:
SEP_sccp7911.cnf.xml
SEP_sccp7925.cnf.xml
etc.

In sccp.conf at first I make tamplates for phones and lines.
[7925](!)
option=....

[7911](!)
option=...

Then I add phones with mac addresses and lines.
[SEPmac-address1](7911)   ; my comment
description=.... etc.

[SEPmac-address2](7925)  ; my comment
description=.... etc.

In script I search for strings which begin with [SEP  using egrep. Then I use awk to extract text from brackets [...] and () and save the result in file temp:
SEPmac-address1 7911
SEPmac-address2 7925

Now I use read to use mac and model of phone for creating soft links to template xml files.

So the name of model, for example 7911, is the name of template in sccp.conf and also the part of name in /tftpboot.
```
 #!/bin/bash
# Anna Toropova
egrep '\[SEP(.*?)\]' /etc/asterisk/sccp.conf | awk -F '[][]|[()]' '{print $2,$4}' > temp

while read mac model;do
echo "ln -sf SEP_sccp$model.cnf.xml $mac.cnf.xml"
ln -sf SEP_sccp$model.cnf.xml $mac.cnf.xml
done < temp

rm -f temp
```

dkgroot, 2019-09-18:
@anutator
This might be something for our wiki (if you like). 

I just had a little idea for you, to make it easier to parse things separately:
sccp.conf
```
[general]
...
#include /etc/asterisk/sccp/SEP*.conf
#include /etc/asterisk/sccp/LINE*.conf
```
You can then can create a SEPxxxxx.conf file per device and a LINExxxx.conf file per line.

Also you can set 
```
[SEPxxxxxxxxxx]
type = device
devicetype = 7961
...
```
Putting this in seperate files makes it really easy to verify if you have all of them, and sync them if necessary.

dkgroot, 2019-09-18:
```
#!/usr/bin/env bash
# generate SEPmac-address.cnf.xml in /tftpboot from sccp.conf
# Anna Toropova
set -e                      # immediately quit if anything goes wrong, during execution
CONFFILE=/usr/local/asterisk-13-branch/etc/asterisk/sccp.conf
TFTPDIR=/tftpboot
if [ ! -f ${CONFFILE} ]; then
   echo "ERROR: ${CONFFILE} could not be found"
   exit 1
fi
if [ ! -d ${TFTPDIR} ]; then
   echo "ERROR: ${TFTPDIR} could not be found"
   exit 2
fi

pushd $TFTPDIR           # always use pushd, so you can return to the calling site after the script, using popd
egrep '\[SEP(.*?)\]' ${CONFFILE} | awk -F '[][]|[()]' '{print $2,$4}' | while read mac model
do
        if [ -f SEP_sccp${model}.cnf.xml ]; then
                echo "ln -sf SEP_sccp$model.cnf.xml $mac.cnf.xml"
                ln -sf SEP_sccp$model.cnf.xml $mac.cnf.xml
        else
                echo "ERROR: SEP_sccp${model}.cnf.xml template file, could not be found in ${TFTPDIR}"
                exit 3
        fi
done
popd
```
I kept all the functionality of your script, just added some safefy checks and clean handling.

anutator, 2019-09-18:
Thank you for ideas!

I use this options when I create a template:
```
[7965](!)
type = device
devicetype = 7961
...
```

dkgroot, 2019-09-18:
Yes that would work, and nice to see somebody understanding and using asterisk config templates :-)
I have my config in the realtime database (postgres), so can't really use your script. But i have something similar.

dkgroot, 2019-09-25:
Thank you for the script / insight. I think the best way forward for it would be a page on our wiki (if you need help let me know), or merging this type of solution into [provision_sccp](https://github.com/dkgroot/provision_sccp). 
Will be closing this issue in 1 week, as it is not really an issue (to be honest).

dkgroot, 2019-10-05:
@anutator
Thank you for your contribution. Nobody will find the script sample in our issue database though. So please consider writing a post on our wiki or move this script into the provision_sccp repository.

Closing this issue report for now.

anutator, 2019-11-07:
Forgotten...
You must add a string to delete old links:
`find ${TFTPDIR} -maxdepth 1 -type l -delete`
This will help if we deleted phone from sccp.conf

dkgroot, 2019-11-07:
Nice addition ! What line should the `find` go on ?
Added '^' to the front of the egrep, that should take care of the remarking.

anutator, 2019-11-07:
I added line with find...  before pushd.
```
find ${TFTPDIR} -maxdepth 1 -type l -delete
pushd ${TFTPDIR}  
```
But I think we can add it after pushd and find in current directory
```
pushd ${TFTPDIR}  
find . -maxdepth 1 -type l -delete
```

dkgroot, 2019-11-07:
Updated the example file :-) Added `-iname "SEP*.cnf.xml"`  to make sure we don't delete other softlinked files

anutator, 2019-11-08:
Today I found special option **-lname**? so you don't have to use to options.
```
-lname pattern
File is a symbolic link whose contents match shell pattern pattern.  The metacharacters do not treat `/' or `.' specially.  If the -L option or the -follow option is in effect, this test returns false unless the symbolic link is broken.
```
So you should change line to:
`find ${TFTPDIR} -maxdepth 1 -lname 'SEP*' -delete`

dkgroot, 2019-11-08:
@anutator 
Thanks, updated the script, as you are the originator of this issue report, you should also be able to edit the script above (i think)

dkgroot, 2020-01-10:
Closing issue. Not sure what to do with the script atm

closed 2020-01-10 by dkgroot