chan-sccp/chan-sccp #476: Everyone is busy

anutator, 2019-09-23:
[Short description of problem here]
**Reproduction Steps:**

Last week registered SCCP phones/ For example line 1125 — two phones, 7911 and 7925.
Today I can't call them (restarted Asterisk) —  Everyone is busy/congested at this time (1:1/0/0)
When I change incominglimit=2 to incominglimit=10, and sccp reload, I can call 1125. When I return incominglimit=2, busy signal again. Why two lines are busy?

```
    -- Executing [1125@sccp:1] Verbose("PJSIP/cucm-00000015", "2,"Вызов с номера (91) 4416 на 1125"") in new stack
  == "Вызов с номера (91) 4416 на 1125"
    -- Executing [1125@sccp:2] Dial("PJSIP/cucm-00000015", "SCCP/1125,,TtRrU(record^(91) 4416^1125)") in new stack
    -- SCCP: Asterisk asked us to create a channel with type=SCCP, format=(alaw), lineName=1125, options=
 SCCP: Adding channel 16 to line 1125
    -- 0001: New channel number: 16 on line 1125
    -- SCCP: (pbx_channel_allocate) try to allocate Inbound channel on line: 1125
    -- SCCP:             cid_num: 1125
    -- SCCP:            cid_name: Харламов В.B.
    -- SCCP:         accountcode: (null)
    -- SCCP:               exten: 
    -- SCCP:             context: operators
    -- SCCP:            amaflags: 0
    -- SCCP:           chan/call: SCCP/1125-00000010
    -- SCCP: combined audio caps: [slin16, ulaw/64k, alaw/64k, ilbc, g729b, g729ab, g729, g729a]
    -- SCCP: reduced audio prefs: [ulaw/64k, alaw/64k, g729, g729a, g729b, g729ab, slin16, ilbc]
    -- SCCP: combined video caps: []
    -- SCCP: reduced video prefs: []
    -- SCCP: (allocPBXChannel) Create New Channel with name: SCCP/1125-00000010
    -- 0001: Allocated asterisk channel SCCP/1125-00000010
    -- SCCP: Asterisk request to call SCCP/1125-00000010 (dest:1125, timeout: 0)
    -- 1125: Asterisk request to call SCCP/1125-00000010
    -- Incoming calls limit (2) reached on SCCP/1125... sending busy
    -- Called SCCP/1125
    -- SCCP/1125-00000010 is busy
    -- (null): Current channel SCCP/1125-00000010 state DOWN(0)
 SCCP: Removing channel 16 from line 1125
    -- SCCP: Cleaning channel SCCP/1125-00000010
    -- SCCP/1125-00000010: Disabling scheduler / Removing Scheduled tasks (digittimeout_id:-1) (hangup_id:-1)
    -- Destroying channel SCCP/1125-00000010
  == Everyone is busy/congested at this time (1:1/0/0)
    -- Executing [1125@sccp:3] Hangup("PJSIP/cucm-00000015", "") in new stack
  == Spawn extension (sccp, 1125, 3) exited non-zero on 'PJSIP/cucm-00000015'
```

sccp.conf
```

;templates
[line](!)
type=line
language=ru
context=operators  ; контекст по умолчанию
incominglimit=2
transfer=on
musicclass=default
echocancel=on

[79XX](!)
type=device
keepalive=60
;tzoffset = +3
park=on
cfwdall=yes
;cfwdbusy=yes
cfwdnoanswer=yes
directed_pickup=on
directed_pickup_context=operators
directed_pickup_modeanswer=on
directrtp=off
earlyrtp=progress
dndfeature=on
mwilamp=on
mwioncall=off
meetme=on
meetmeopts=qxd

[7911](!,79XX)
devicetype=7911
imageversion=SCCP11.9-4-2SR3-1S

[7925](!,79XX)
devicetype=7925
imageversion=CP7925G-1.4.8SR1.5

; phones

[SEP68BDABA41D18](7911)
description=Kharlamov 1125
button=line,1125

[SEPE02F6D63D9FE](7925)
description=Kharlamov 1125
button=line,1125

[1125](line)
label=Харламов B.B.
cid_name=Харламов В.B.
cid_num=1125
description=Харламов В.В.
mailbox=1125
```

**chan-sccp version:**latest
**asterisk version:** 16.5.0
**Expected behavior:** idle instead of busy.

dkgroot, 2019-09-23:
@anutator 
Can you show `sccp show lines`, which shows the number of channels ('Ch' column) before you make the call.
I changed the debug output in the last commit (above) to display the number of calls and the incominglimit, before checking. I did not alter the measurement or the action taken.

I don't know exactly what `TtRrU(record^(91) 4416^1125)` might have on the dial, but i will check in the meantime.
Tr = related to transfer, Rr = related to rining, U(x^args) = Execute via Gosub the routine x for the called channel before connecting to the calling channel.

dkgroot, 2019-09-23:
We could make this change, to only count incoming channels on this line and then compare that against the l->incominglimit. That might be more accurate. Still not taking into account shared line members though, as that would be rather complex.
```diff
diff --git a/src/sccp_pbx.c b/src/sccp_pbx.c
index fe1cd89a..be81344b 100644
--- a/src/sccp_pbx.c
+++ b/src/sccp_pbx.c
@@ -200,13 +200,19 @@ int sccp_pbx_call(sccp_channel_t * c, char *dest, int timeout)
 
        sccp_log((DEBUGCAT_CORE)) (VERBOSE_PREFIX_3 "%s: Asterisk request to call %s\n", l->name, iPbx.getChannelName(c));
 
-       /* if incoming call limit is reached send BUSY */
+       int incoming_calls = 0;
+       sccp_channel_t *count_channel = NULL;
        SCCP_LIST_LOCK(&l->channels);
-       int currentcalls = SCCP_LIST_GETSIZE(&l->channels);
+       SCCP_LIST_TRAVERSE(&l->channels, count_channel, list) {
+               if (count_channel != c && count_channel->calltype == SKINNY_CALLTYPE_INBOUND) {
+                       incomingcalls++;
+               }
+       }
        SCCP_LIST_UNLOCK(&l->channels);
-       sccp_log((DEBUGCAT_CORE))(VERBOSE_PREFIX_3 "SCCP/%s: Number of Calls:%d, Incoming calls limit: %d\n", l->name, currentcalls, l->incominglimit);
-       if(l->incominglimit && currentcalls > l->incominglimit) {
-               sccp_log((DEBUGCAT_CORE))(VERBOSE_PREFIX_3 "SCCP/%s: Number of Calls: %d, Incoming calls limit (%d) -> sending BUSY\n", l->name, currentcalls, l->incominglimit);
+       sccp_log((DEBUGCAT_CORE))(VERBOSE_PREFIX_3 "SCCP/%s: Incoming calls:%d, Incoming calls limit: %d\n", l->name, incomingcalls, l->incominglimit);
+       /* if incoming call limit is reached send BUSY */
+       if(l->incominglimit && incomingcalls >= l->incominglimit) {
+               sccp_log((DEBUGCAT_CORE))(VERBOSE_PREFIX_3 "SCCP/%s: Incoming calls: %d, Incoming calls limit (%d) -> sending BUSY\n", l->name, incomingcalls, l->incominglimit);
                iPbx.queue_control(c->owner, AST_CONTROL_BUSY);
                //iPbx.set_callstate(c, AST_STATE_BUSY);
                pbx_channel_set_hangupcause(c->owner, AST_CAUSE_USER_BUSY);
```

anutator, 2019-09-24:
```
aster16*CLI> sccp show lines

+--- Lines ------------------------------------------------------------------------------------------------------------------------------------------------------+
| Ext           Suffix    Label                          Description      Device           MWI  Chs  Active Channel                                              |
+ ============= ========= ============================== ================ ================ ==== ==== =========================================================== +
| 1126                    Глазков A.              Глазков Алексей --               OFF  0    --         --                                               |
| 1125          (+)       Харламов B.B.          Харламов В.В. SEP68BDABA41D18  OFF  4    --         --                                               |
|  +--          (+)       Харламов B.B.          Харламов В.В. SEPE02F6D63D9FE  OFF  4    --         --                                               |
| 98011                   Phone 1 Line 1                 Line 98011       --               OFF  0    --         --                                               |
|               Variable: testvar2                       = my value                                                                                              |
| 98012                   Phone 1 Line 2                 Line 98012       --               OFF  0    --         --                                               |
| 98021                   Phone 2 Line 1                 Line 98021       --               OFF  0    --         --                                               |
|               Variable: testvar2                       = my value                                                                                              |
| 98031                   Phone 3 Line 1                 Line 98031       --               OFF  0    --         --                                               |
| Hotline                 Hotline                        (null)           --               OFF  0    --         --                                               |
+----------------------------------------------------------------------------------------------------------------------------------------------------------------+
aster16*CLI> sccp show channels

+--- Channels --------------------------------------------------------------------------------------------------------------------------------------+
| ID    Name                      LineName   DeviceName       NumCalled  PBX State  SCCP State AudioR AudioW RTPPeer                Direct DTMFmode |
+ ===== ========================= ========== ================ ========== ========== ========== ====== ====== ====================== ====== ======== +
| 1     SCCP/1125-00000001        1125       SCCP                        (none)     DOWN                                            no     RFC2833  |
| 2     SCCP/1125-00000002        1125       SCCP                        (none)     DOWN                                            no     RFC2833  |
| 3     SCCP/1125-00000003        1125       SCCP                        (none)     DOWN                                            no     RFC2833  |
| 6     SCCP/1125-00000006        1125       SCCP                        (none)     DOWN                                            no     RFC2833  |
+---------------------------------------------------------------------------------------------------------------------------------------------------+
    -- SCCP: getnameinfo(): ai_family not supported 
    -- SCCP: getnameinfo(): ai_family not supported 
    -- SCCP: getnameinfo(): ai_family not supported 
    -- SCCP: getnameinfo(): ai_family not supported 
```

dkgroot, 2019-09-24:
Ok that shows the problem. You have four channels that have not really been ended/hungup/removed. This is not a good sign. You will have to check the asterisk debug file, to see what happened with these channels to figure out why. If you reload chan-sccp and try again, the number of calls will be 0 again, and it would work.

Note: we need to figure out what caused these channels not to be hungup (completely).

dkgroot, 2019-09-24:
BTW: I noticed that the `sccp show channels` doesn't look very nice when using russian locale, so i pushed a fix for that (12a2928) :-)

dkgroot, 2019-10-05:
@anutator Any update ?

anutator, 2019-10-07:
Yes. I installed CentOS 8 (many chaneg including MariaDB, difficulties with repo), tftp using systemd socket, Asterisk 16.6.0 rc (release candidate) and the latest chan_sccp. I  registered only one phone without Wi-Fi phones. Maybe something wrong in configuration
I can ring but if I try to make call to SCCP phone I get error, phone is busy.
```

 == "Вызов с номера (91)4416 на 1127"
    -- Executing [1127@sccp:2] Dial("PJSIP/cucm-00000009", "SCCP/1127,,TtRrU(record^(91)4416^1127)") in new stack
    -- SCCP: Asterisk asked us to create a channel with type=SCCP, format=(alaw), lineName=1127, options=
 SCCP: Adding channel 8 to line 1127
    -- 0005: New channel number: 8 on line 1127
    -- SCCP: (pbx_channel_allocate) try to allocate Inbound channel on line: 1127
    -- SCCP:             cid_num: 1127
    -- SCCP:            cid_name: Торопова А.В.
    -- SCCP:         accountcode: (null)
    -- SCCP:               exten: 
    -- SCCP:             context: operators
    -- SCCP:            amaflags: 0
    -- SCCP:           chan/call: SCCP/1127-00000008
    -- SCCP: combined audio caps: [slin16, ulaw/64k, alaw/64k, ilbc, g729b, g729ab, g729, g729a]
    -- SCCP: reduced audio prefs: [alaw/64k, ulaw/64k, g729, g729a, g729b, g729ab]
    -- SCCP: combined video caps: []
    -- SCCP: reduced video prefs: []
    -- SCCP: (allocPBXChannel) Create New Channel with name: SCCP/1127-00000008
    -- 0005: Allocated asterisk channel SCCP/1127-00000008
    -- SCCP: Asterisk request to call SCCP/1127-00000008 (dest:1127, timeout: 0)
    -- 1127: Asterisk request to call SCCP/1127-00000008
    -- SCCP/1127: Incoming calls:4, Incoming calls limit: 2
    -- SCCP/1127: Incoming calls: 4, Incoming calls limit (2) -> sending BUSY
    -- Called SCCP/1127
    -- SCCP/1127-00000008 is busy
    -- (null): Current channel SCCP/1127-00000008 state DOWN(0)
 SCCP: Removing channel 8 from line 1127
    -- SCCP: Cleaning channel SCCP/1127-00000008
    -- SCCP/1127-00000008: Disabling scheduler / Removing Scheduled tasks (digittimeout_id:-1) (hangup_id:-1)
    -- Destroying channel SCCP/1127-00000008
  == Everyone is busy/congested at this time (1:1/0/0)
    -- Executing [1127@sccp:3] Hangup("PJSIP/cucm-00000009", "") in new stack
  == Spawn extension (sccp, 1127, 3) exited non-zero on 'PJSIP/cucm-00000009'
```

```
ast16*CLI> sccp show channels

+--- Channels --------------------------------------------------------------------------------------------------------------------------------------+
| ID    Name                      LineName   DeviceName       NumCalled  PBX State  SCCP State AudioR AudioW RTPPeer                Direct DTMFmode |
+ ===== ========================= ========== ================ ========== ========== ========== ====== ====== ====================== ====== ======== +
| 1     SCCP/1127-00000001        1127       SCCP                        (none)     DOWN                                            no     RFC2833  |
| 3     SCCP/1127-00000003        1127       SCCP                        (none)     DOWN                                            no     RFC2833  |
| 4     SCCP/1127-00000004        1127       SCCP                        (none)     DOWN                                            no     RFC2833  |
| 5     SCCP/1127-00000005        1127       SCCP                        (none)     DOWN                                            no     RFC2833  |
+---------------------------------------------------------------------------------------------------------------------------------------------------+
    -- SCCP: getnameinfo(): ai_family not supported 
    -- SCCP: getnameinfo(): ai_family not supported 
    -- SCCP: getnameinfo(): ai_family not supported 
    -- SCCP: getnameinfo(): ai_family not supported 

```

anutator, 2019-10-07:
Debug
```
[Oct  7 13:58:32] NOTICE[4672][C-0000000e]: ast117.c:1674 sccp_astwrap_request:     -- SCCP: Asterisk asked us to create a channel with type=SCCP, format=(alaw), lineName=1127, options=
[Oct  7 13:58:32] NOTICE[4672][C-0000000e]: ast117.c:1688 sccp_astwrap_request:        > SCCP: requested Audio Codec in Skinny Format: G.711 A-law 64k
[Oct  7 13:58:32] NOTICE[4672][C-0000000e]: ast117.c:1739 sccp_astwrap_request:        > remote audio caps: [alaw/64k, ulaw/64k, g729]
[Oct  7 13:58:32] NOTICE[4672][C-0000000e]: ast117.c:1742 sccp_astwrap_request:        > remote video caps: []
 sccp_line.c    :0930 (sccp_line_find_byname              ) --------->             refcount increased 09  +> 10 for       line: 1127 (0x3425420)
 sccp_channel.c :0103 (sccp_channel_allocate              ) ---------->            refcount increased 10  +> 11 for       line: 1127 (0x3425420)
[Oct  7 13:58:32] NOTICE[4672][C-0000000e]: sccp_refcount.c:353 sccp_refcount_object_alloc:  SCCP: (alloc_obj) Creating new channel SCCP/1127-0000000C (0x7fcc0c055310) inside 0x7fcc0c0552d0 at hash: 32
[Oct  7 13:58:32] NOTICE[4672][C-0000000e]: sccp_callinfo.c:110 callinfo_Constructor:  SCCP: callinfo constructor: 0x7fcc0c0194e0
 sccp_line.c    :0716 (sccp_line_addChannel               ) ----------->           refcount increased 11  +> 12 for       line: 1127 (0x3425420)
 sccp_line.c    :0721 (sccp_line_addChannel               ) ->                     refcount increased 01  +> 02 for    channel: SCCP/1127-0000000C (0x7fcc0c0552d0)
[Oct  7 13:58:32] NOTICE[4672][C-0000000e]: sccp_line.c:725 sccp_line_addChannel:  SCCP: Adding channel 12 to line 1127
 sccp_line.c    :0716 (sccp_line_addChannel               ) <-----------           refcount decreased 11  <- 12 for       line: 1127 (0x3425420)
[Oct  7 13:58:32] NOTICE[4672][C-0000000e]: sccp_line.c:390 sccp_line_copyCodecSetsFromLineToChannel:     -- SCCP/1127-0000000C: (copyCodecSetsFromLineToChannel) channel capabilities:[slin16, ulaw/64k, alaw/64k, ilbc, g729b, g729ab, g729, g729a]
[Oct  7 13:58:32] NOTICE[4672][C-0000000e]: sccp_line.c:391 sccp_line_copyCodecSetsFromLineToChannel:     -- SCCP/1127-0000000C: (copyCodecSetsFromLineToChannel) channel preferences:[alaw/64k, ulaw/64k, g729, g729a, g729b, g729ab]
[Oct  7 13:58:32] NOTICE[4672][C-0000000e]: sccp_channel.c:218 sccp_channel_allocate:     -- 0005: New channel number: 12 on line 1127
 sccp_pbx.c     :0098 (sccp_requestChannel                ) -->                    refcount increased 02  +> 03 for    channel: SCCP/1127-0000000C (0x7fcc0c0552d0)
 sccp_pbx.c     :0072 (sccp_requestChannel                ) <--                    refcount decreased 02  <- 03 for    channel: SCCP/1127-0000000C (0x7fcc0c0552d0)
 sccp_pbx.c     :0057 (sccp_requestChannel                ) <----------            refcount decreased 10  <- 11 for       line: 1127 (0x3425420)
 sccp_pbx.c     :0694 (sccp_pbx_channel_allocate          ) -->                    refcount increased 02  +> 03 for    channel: SCCP/1127-0000000C (0x7fcc0c0552d0)
 sccp_pbx.c     :0708 (sccp_pbx_channel_allocate          ) ---------->            refcount increased 10  +> 11 for       line: 1127 (0x3425420)
[Oct  7 13:58:32] NOTICE[4672][C-0000000e]: sccp_pbx.c:715 sccp_pbx_channel_allocate:     -- SCCP: (pbx_channel_allocate) try to allocate Inbound channel on line: 1127
 sccp_pbx.c     :0738 (sccp_pbx_channel_allocate          ) --->                   refcount increased 03  +> 04 for     device: SEPDC7B94777357 (0x341f940)
[Oct  7 13:58:32] NOTICE[4672][C-0000000e]: sccp_pbx.c:809 sccp_pbx_channel_allocate:     -- SCCP:             cid_num: 1127
[Oct  7 13:58:32] NOTICE[4672][C-0000000e]: sccp_pbx.c:810 sccp_pbx_channel_allocate:     -- SCCP:            cid_name: Торопова А.В.
[Oct  7 13:58:32] NOTICE[4672][C-0000000e]: sccp_pbx.c:811 sccp_pbx_channel_allocate:     -- SCCP:         accountcode: (null)
[Oct  7 13:58:32] NOTICE[4672][C-0000000e]: sccp_pbx.c:812 sccp_pbx_channel_allocate:     -- SCCP:               exten: 
[Oct  7 13:58:32] NOTICE[4672][C-0000000e]: sccp_pbx.c:813 sccp_pbx_channel_allocate:     -- SCCP:             context: operators
[Oct  7 13:58:32] NOTICE[4672][C-0000000e]: sccp_pbx.c:814 sccp_pbx_channel_allocate:     -- SCCP:            amaflags: 0
[Oct  7 13:58:32] NOTICE[4672][C-0000000e]: sccp_pbx.c:815 sccp_pbx_channel_allocate:     -- SCCP:           chan/call: SCCP/1127-0000000C
[Oct  7 13:58:32] NOTICE[4672][C-0000000e]: sccp_pbx.c:816 sccp_pbx_channel_allocate:     -- SCCP: combined audio caps: [slin16, ulaw/64k, alaw/64k, ilbc, g729b, g729ab, g729, g729a]
[Oct  7 13:58:32] NOTICE[4672][C-0000000e]: sccp_pbx.c:817 sccp_pbx_channel_allocate:     -- SCCP: reduced audio prefs: [alaw/64k, ulaw/64k, g729, g729a, g729b, g729ab]
[Oct  7 13:58:32] NOTICE[4672][C-0000000e]: sccp_pbx.c:818 sccp_pbx_channel_allocate:     -- SCCP: combined video caps: []
[Oct  7 13:58:32] NOTICE[4672][C-0000000e]: sccp_pbx.c:819 sccp_pbx_channel_allocate:     -- SCCP: reduced video prefs: []
 ast117.c       :1159 (sccp_astwrap_allocPBXChannel       ) ----------->           refcount increased 11  +> 12 for       line: 1127 (0x3425420)
[Oct  7 13:58:32] NOTICE[4672][C-0000000e]: ast117.c:1175 sccp_astwrap_allocPBXChannel:     -- SCCP: (allocPBXChannel) Create New Channel with name: SCCP/1127-0000000C
 ast117.c       :1186 (sccp_astwrap_allocPBXChannel       ) --->                   refcount increased 03  +> 04 for    channel: SCCP/1127-0000000C (0x7fcc0c0552d0)
[... cut ...]
    -- Executing [1127@sccp:3] Hangup("PJSIP/cucm-0000000d", "") in new stack
[Oct  7 13:58:32] NOTICE[2440]: sccp_hint.c:1523 sccp_hint_getLinestate:     -- 1127 (getLinestate) state:CONNECTED, party:/, calltype:Outbound
[Oct  7 13:58:32] NOTICE[2440]: ast117.c:345 sccp_astwrap_devicestate:        > SCCP: (devicestate) sccp_hint returned state:CONNECTED for '1127'
[Oct  7 13:58:32] NOTICE[2440]: ast117.c:410 sccp_astwrap_devicestate:        > SCCP: (devicestate) PBX requests state for '1127' - state In use

```

anutator, 2019-10-07:
I think I've found the problem.
chan_sccp doesn't work with string
`allow=!all,alaw,ulaw,g729`
It reques only:
```
disallow=all
allow=alaw
allow=ulaw
allow=g729
```

dkgroot, 2019-10-07:
@anutator 
Sorry to hear you had so much trouble setting up your new system :-(

Did changing "allow=!all,alaw,ulaw,g729" to 
```
disallow=all
allow=alaw
allow=ulaw
allow=g729
```
Fix the issue. I will check out the allow/disallow string parser to see what is going on. I personnally don't use the abbreviated for (to be hones).

dkgroot, 2019-10-07:
Where did you set these disallow/allow values, at the [general], [device] or [line] level. Recently we moved the [device] codecs to the [line] level (and currently still maintain backward compatibility). When creating new lines, it's advisable to set disallow/allow on the line level instead of the device.

dkgroot, 2019-10-07:
Hi @anutator I changed the disallow/allow codec parser to support 'allow=!xxx,xxx'.

***Git***
_Checkout this temporary branch_
```
git pull
git fetch -a
git branch
git checkout issues/476
./configure [....]
make -j4 && make install && make reload
```

_move back to develop afterwards:_
```
git checkout develop
```

That should hopefully improve/fix the issue `allow=!all,ulaw,alaw` part of the issue. I need a little time to figure out the other part, where channels are not terminated correctly.

dkgroot, 2019-10-08:
I looked through your last debug log.. And it does look correct, in the sense that on an incoming call, a new sccp_channel object is created (which get's tied to the backing asterisk channel), this sccp_channel is needed so we can store extra data needed for sccp handling. Once the refcount for this sccp_channel goes down to 0, it is finalized and destroyed. This happens nicely during the hangup of the call. So in this case (based on your log) i would determine that there should nog be a channel remaining (for this call at least).

dkgroot, 2019-10-08:
If you send me you sccp.conf (and extensions.conf if not using freepbx) then i can see if i can run your config on my machine, to see what is going on. I don't use freepbx, so simulating that would be (to) difficult.

dkgroot, 2019-10-10:
@anutator 
I pushed another fix to the 'issues/476' that will make sure that channels are ended correctly, even if/when codecs are not able to be handled correctly. So let's say you have:
```
allow=!all,*,~,none,blabla
```
Which would be something we are not able to understand (even with the new parser '!' rules which were added recently). 
In that case we should still make sure the newly created channel is hungup and removed correctly. This latest commit should take care of that. It will not fix the codec issue or make the call go through, but it does cleanup nicely after reporting the error.

Please checkout this temporary branch:
```
git pull
git fetch -a
git branch
git checkout issues/476
./configure [....]
make -j4 && make install && make reload
```

And give it a go. Just want to make sure that this branch does not introduce any regressions before merging it back into develop.

anutator, 2019-10-15:
How to release channel? We reboot phone but it can't register on Asterisk.

```
ast16*CLI> sccp show channels

+--- Channels --------------------------------------------------------------------------------------------------------------------------------------+
| ID    Name                      LineName   DeviceName       NumCalled  PBX State  SCCP State AudioR AudioW RTPPeer                Direct DTMFmode |
+ ===== ========================= ========== ================ ========== ========== ========== ====== ====== ====================== ====== ======== +
| 254   SCCP/4471-000000FE        4471       SCCP                        Ringing    RINGING                                         no     RFC2833  |
+---------------------------------------------------------------------------------------------------------------------------------------------------+
    -- SCCP: getnameinfo(): ai_family not supported 
```

dkgroot, 2019-10-15:
Where did this issue come from.
Is this with the "issues/476" branch i provided, or a seperate issue ?

I cannot keep track of this many problems at the same time.

anutator, 2019-10-15:
> 
> 
> Where did this issue come from.
> Is this with the "issues/476" branch i provided, or a seperate issue ?
> 
> I cannot keep track of this many problems at the same time.

Channel is stuck. I don't want to find this errors.

dkgroot, 2019-10-15:
There is no way to remove it, if you already tried hanging it up.

I don't know what is causing the issue though, i need the call history (debug log) to see what went wrong.

I am sorry that i don't have any 7925, to help figure this out.
When i do have access to your machine i will look into all the issues. So let's see if we can fix the VPN / SSH stuff.

anutator, 2019-10-15:
> There is no way to remove it, if you already tried hanging it up.
>
> I don't know what is causing the issue though, i need the call history (debug log) to see what went wrong.
>
> I am sorry that i don't [...]

channel request hangup SCCP/4471-000000FE - command to remove channel

dkgroot, 2019-10-15:
If it is not causing any problems, you can just leave it. It should not have any side effects.

I created the "issues/476" branch to fix one of these issues, with channels left open. But i'm not sure if you are running on that branch now or not.

closed 2019-10-15 by dkgroot

anutator, 2019-10-15:
Thank you for helping online! I think everything would work))

dkgroot, 2019-10-15:
Issue resolved

commit 7902fe75 (2019-10-10): Missing/broken/faulty codec disallow/allow codecs would cause leftover channels to remain behind (with refcount 1)
Fix: Make sure that a FALSE value returned from sccp_pbx_channel_allocate is handled correctly.

Fixes #476
commit 995c4418 (2019-10-10): Missing/broken/faulty codec disallow/allow codecs would cause leftover channels to remain behind (with refcount 1)
Fix: Make sure that a FALSE value returned from sccp_pbx_channel_allocate is handled correctly.

Fixes #476
commit a7f3740b (2019-10-15): Missing/broken/faulty codec disallow/allow codecs would cause leftover channels to remain behind (with refcount 1)
Fix: Make sure that a FALSE value returned from sccp_pbx_channel_allocate is handled correctly.

Fixes #476
commit 9c7b975b (2019-10-15): Make sure ast_channel_answer was able to answer the channel before continuing with the masquerade
Missing/broken/faulty codec disallow/allow codecs would cause leftover channels to remain behind (with refcount 1)
Fix: Make sure that a FALSE value returned from sccp_pbx_channel_allocate is handled correctly.
Add Support for 'allow=!all,xxxx', as shorthand for disallow/allow

Fixes: #476

Issue reported by @anutator