chan-sccp/chan-sccp #477: Crash with IPv6

berean-sc, 2019-09-26:
Using latest development version of chan-sccp, Asterisk crashes when calls are initiated or received from the SCCP endpoint using IPv6. Following the log, it appears to happen as soon as media is engaged.

Log end:

```
    -- SEPx: Opened Receive Channel (State: CONNECTED[30])
    -- SEPx: Set the active channel 1 on device
    -- SEPx: Set the active line 1104
    -- SEPx: Starting Phone RTP/UDP Transmission (State: CONNECTED[30])
    -- SCCP/1104-00000001: Using dtmf rtp_code : 101
    -- SEPx: Tell Phone to send RTP/UDP media from [PHONE-GLOBAL-IPV6-ADDR]:29862 to [SERVER-GLOBAL-IPV6-ADDR]:10342 (NAT: (Auto)Off)
    -- SEPx: Using codec:G.722 64k(6), TOS:184, Silence Suppression:OFF for call with PassThruId:4294967294 and CallID:1
    -- SEPx: send callInfo on SCCP/1104-00000001 with lineInstance: 1
    -- SEPx: Stop tone on line 1 with callid 1
    -- SEPx: Got startMediaTransmission ACK. Status:'Media Status: OK' (0), Remote RTP/UDP:'[PHONE-GLOBAL-IPV6-ADDR]:29862', Type:Indirect RTP, PassThruPartyId:4294967294, CallID:1, CallID1:1
    -- SEPx: Getting the active channel on device.
    -- SEPx: Media Transmission Started (State: CONNECTED[30])

(crash)
```

gdb bt full from core file:

```
(gdb) bt full
#0  0x00007fdb72cbe067 in __GI_raise (sig=sig@entry=6) at ../nptl/sysdeps/unix/sysv/linux/raise.c:56
        resultvar = 0
        pid = 15535
        selftid = 15722
#1  0x00007fdb72cbf448 in __GI_abort () at abort.c:89
        save_stage = 2
        act = {__sigaction_handler = {sa_handler = 0x7ffeb50a9edc, sa_sigaction = 0x7ffeb50a9edc}, sa_mask = {__val = {140580501767209, 
              140579095788736, 388, 4, 140578756041552, 163390233680, 140578756041616, 4294967296, 0, 0, 0, 21474836480, 140580501766735, 
              140578756041704, 140580506337280, 140580501782296}}, sa_flags = 521223480, sa_restorer = 0x7fdb1f113ee0}
        sigs = {__val = {32, 0 <repeats 15 times>}}
#2  0x00007fdb72cb7266 in __assert_fail_base (fmt=0x7fdb72deff18 "%s%s%s:%u: %s%sAssertion `%s' failed.\n%n", 
    assertion=assertion@entry=0x7fdb1f113d38 "a->addr.sa_family == PJ_AF_INET || a->addr.sa_family == PJ_AF_INET6", 
    file=file@entry=0x7fdb1f113cc0 "../src/pj/sock_common.c", line=line@entry=388, 
    function=function@entry=0x7fdb1f113ee0 "pj_sockaddr_get_len") at assert.c:92
        str = 0x7fdb34013e90 "\200\025"
        total = 4096
#3  0x00007fdb72cb7312 in __GI___assert_fail (assertion=0x7fdb1f113d38 "a->addr.sa_family == PJ_AF_INET || a->addr.sa_family == PJ_AF_INET6", 
    file=0x7fdb1f113cc0 "../src/pj/sock_common.c", line=388, function=0x7fdb1f113ee0 "pj_sockaddr_get_len") at assert.c:101
No locals.
#4  0x00007fdb1f10a97b in pj_sockaddr_get_len () from /usr/lib/libpj.so.2
No symbol table info available.
#5  0x00007fdb20906788 in __rtp_recvfrom (instance=0x7fdb0ad11ce0, buf=0x7fdb3000e3d8, sa=0x7fdb0ad11ff0, rtcp=0, flags=<optimized out>, 
    size=<optimized out>) at res_rtp_asterisk.c:2067
        combined = {ptr = 0x7fdb34036f38 "[MY-GLOBAL-IPV6-ADDRESS]:27104", slen = 46}
        address = {addr = {sa_family = 6}, ipv4 = {sin_family = 6, sin_port = 0, sin_addr = {s_addr = 0}, 
            sin_zero = "\006\274\035\274fU\000"}, ipv6 = {sin6_family = 6, sin6_port = 0, sin6_flowinfo = 0, sin6_addr = {
              s6_addr = "\006\274\035\274fU\000\000@\212\003\064\333\177\000", u6_addr32 = {3156065286, 21862, 872647232, 32731}}, 
            sin6_scope_id = 2073604352}}
        status = 181476576
        len = 172
        rtp = 0x7fdb3000e220
        loop = 0x7fdb30010968
#6  0x00007fdb2090d59c in rtp_recvfrom (sa=<optimized out>, flags=<optimized out>, size=<optimized out>, buf=<optimized out>, 
    instance=<optimized out>) at res_rtp_asterisk.c:2097
No locals.
#7  ast_rtp_read (instance=0x7fdb30009cd8, rtcp=805364696) at res_rtp_asterisk.c:4346
        rtp = 0x7fdb3000e220
        addr = {ss = {ss_family = 10, __ss_align = 5853274625505034534, 
            __ss_padding = "\276\026\365\377\376\372\316=\000\000\000\000\002", '\000' <repeats 27 times>, "\360!\321\n\333\177\000\000\030 \321\n\333\177\000\000B\024\353\v\333\177\000\000P \353\v\333\177\000\000J\002\000\000\002\000\000\000  \321\n\333\177\000\000B\024\353\v\333\177\000\000P \353\v\333\177\000\000Q\002\000\000\333\177\000"}, len = 28}
        res = 805426832
        ext = 8
        rtpheader = 0x7fdb3000e3d8
        seqno = 805364696
        payload = {asterisk_format = 0, format = {id = 0, fattr = {format_attr = {181478432, 32731, 181477968, 32731, 3156706460, 21862, 
                181477760, 32731, 13, 2, 0, 0, 3156830145, 21862, 0, 0, 232, 21862, 24, 48, 181478192, 32731, 181477824, 32731, 12, 1, 
                467204, 0, 0, 0, 0, 0, 805325000, 32731, 19, 32731, 336, 0, 181477888, 32731, 11, 1, 1569467665, 0, 0, 0, 0, 0, 0, 0, 110, 
                119, 0, 0, 181477952, 32731, 3155784112, 21862, 4, 0, 2073604352, 4195091838, 0, 0}, rtp_marker_bit = 30 '\036'}}, 
---Type <return> to continue, or q <return> to quit---
          rtp_code = 21862, payload = -1}
        remote_address = {ss = {ss_family = 0, __ss_align = 0, __ss_padding = '\000' <repeats 111 times>}, len = 0}
        __PRETTY_FUNCTION__ = "ast_rtp_read"
#8  0x00007fdb0be77307 in sccp_astwrap_rtp_write (ast=0x6, frame=0x7fdb30003ef0) at ast111.c:877
        c = 0x0
        res = 0
        __PRETTY_FUNCTION__ = "sccp_astwrap_rtp_write"
#9  0x00005566bc1a7ce0 in ast_tvnow () at /build/asterisk-11.13.1~dfsg/include/asterisk/time.h:135
        t = <error reading variable t (Cannot access memory at address 0xffffffffffffffb6)>
#10 ast_waitfor_nandfds (c=<error reading variable: Cannot access memory at address 0xffffffffffffff7e>, 
    n=<error reading variable: Cannot access memory at address 0xffffffffffffff92>, 
    fds=<error reading variable: Cannot access memory at address 0xffffffffffffff5e>, nfds=<optimized out>, 
    exception=<error reading variable: Cannot access memory at address 0xffffffffffffff6e>, 
    outfd=<error reading variable: Cannot access memory at address 0xffffffffffffff56>, 
    ms=<error reading variable: Cannot access memory at address 0x16>) at channel.c:3328
        start = <error reading variable start (Cannot access memory at address 0xffffffffffffff76)>
        pfds = <error reading variable pfds (Cannot access memory at address 0xffffffffffffffae)>
        res = <optimized out>
        rms = <optimized out>
        x = <optimized out>
        y = <optimized out>
        max = <optimized out>
        sz = <optimized out>
        now = <optimized out>
        whentohangup = <optimized out>
        diff = <optimized out>
        winner = 0x0
        fdmap = <error reading variable fdmap (Cannot access memory at address 0xffffffffffffffa6)>
        __PRETTY_FUNCTION__ = "ast_waitfor_nandfds"
(gdb) 
```

This is on an older version of Asterisk that ships with Debian 8: Asterisk 11.13.1~dfsg-2+deb8u6 . I understand if you do not want to support this version. Using strictly IPv4, it works fine.

In sccp.conf, I tried setting `bind = ::` and `bind = (full global IPv6 address)` and both produced the same crash.

dkgroot, 2019-09-26:
@berean-sc 
I have been doing some refactoring in the last couple of days/weeks, so this one might have slipped through.
Can you help me out, by doing a `git bisect` to find out the exact revision introducting the issue ?

dkgroot, 2019-09-26:
The backtrace supplied is not really usable to be honest.
For example:
`#8  0x00007fdb0be77307 in sccp_astwrap_rtp_write (ast=0x6,`, an `ast` pointer should be a pointer not some random integer. So something must have gone wrong between the crash and the backtrace (for example updating and recompiling the source).

berean-sc, 2019-09-26:
gdb tells me: warning: .dynamic section for "/usr/lib/asterisk/modules/chan_sccp.so" is not at the expected address (wrong library or version mismatch?)

However, this core does correspond to the version of asterisk and the chan_sccp that were and are now in place.

The crash happens every call. 

I first tested ipv6 yesterday on the chan-sccp dev version I had compiled 22 August. When it crashed, I git pull'ed the latest and recompiled.

dkgroot, 2019-09-26:
So the 22 august version crash and the current one crashes every time ?
BTW: `git log | grep bc3e370` doesn't give me any result, so not really sure what branch you are on.

berean-sc, 2019-09-26:
Oh no! I accidentally merged something local into my git pull yesterday. Let me clean it up and try again to make sure I didn't cause all this.

dkgroot, 2019-09-26:
Aah ok, that explains it :-)

dkgroot, 2019-09-26:
If you like we can switch to gitter.im/chan-sccp/chan-sccp and work on it live

berean-sc, 2019-09-26:
Now using the clean version, new bt as follows seems to have more useful information. Thanks for the offer of live debugging but unfortunately I have to stop for now and come back to this task later. I will ping you on gitter later, thanks again!

```
root@bell:/tmp# gdb -se "asterisk" -ex "bt full" -ex "thread apply all bt" --batch -c core.bell-2019-09-26T10:05:41-0400
[New LWP 6927]
[New LWP 6795]
[New LWP 6797]
[New LWP 6798]
[New LWP 6802]
[New LWP 6801]
[New LWP 6805]
[New LWP 6804]
[New LWP 6807]
[New LWP 6806]
[New LWP 6808]
[New LWP 6809]
[New LWP 6810]
[New LWP 6814]
[New LWP 6813]
[New LWP 6823]
[New LWP 6815]
[New LWP 6825]
[New LWP 6819]
[New LWP 6821]
[New LWP 6824]
[New LWP 6827]
[New LWP 6828]
[New LWP 6831]
[New LWP 6830]
[New LWP 6832]
[New LWP 6833]
[New LWP 6834]
[New LWP 6838]
[New LWP 6836]
[New LWP 6841]
[New LWP 6842]
[New LWP 6837]
[New LWP 6839]
[New LWP 6844]
[New LWP 6840]
[New LWP 6846]
[New LWP 6847]
[New LWP 6850]
[New LWP 6848]
[New LWP 6853]
[New LWP 6820]
[New LWP 6854]
[New LWP 6796]
[New LWP 6843]
[New LWP 6855]
[New LWP 6822]
[New LWP 6858]
[New LWP 6859]
[New LWP 6892]
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".

warning: the debug information found in "/usr/lib/debug//usr/lib/asterisk/modules/app_voicemail.so" does not match "/usr/lib/asterisk/modules/app_voicemail.so" (CRC mismatch).

warning: the debug information found in "/usr/lib/debug/usr/lib/asterisk/modules/app_voicemail.so" does not match "/usr/lib/asterisk/modules/app_voicemail.so" (CRC mismatch).

Core was generated by `/usr/sbin/asterisk -f -U asterisk -G asterisk -vvvg -c'.
Program terminated with signal SIGABRT, Aborted.
#0  0x00007f6c0dfdc067 in __GI_raise (sig=sig@entry=6) at ../nptl/sysdeps/unix/sysv/linux/raise.c:56
56	../nptl/sysdeps/unix/sysv/linux/raise.c: No such file or directory.
#0  0x00007f6c0dfdc067 in __GI_raise (sig=sig@entry=6) at ../nptl/sysdeps/unix/sysv/linux/raise.c:56
        resultvar = 0
        pid = 6795
        selftid = 6927
#1  0x00007f6c0dfdd448 in __GI_abort () at abort.c:89
[... 482 lines cut ...]
#6  0x0000563fcab3ca24 in main (argc=<optimized out>, argv=<optimized out>) at asterisk.c:4466

Thread 1 (Thread 0x7f6ba6018700 (LWP 6927)):
#0  0x00007f6c0dfdc067 in __GI_raise (sig=sig@entry=6) at ../nptl/sysdeps/unix/sysv/linux/raise.c:56
#1  0x00007f6c0dfdd448 in __GI_abort () at abort.c:89
#2  0x00007f6c0dfd5266 in __assert_fail_base (fmt=0x7f6c0e10df18 "%s%s%s:%u: %s%sAssertion `%s' failed.\n%n", assertion=assertion@entry=0x7f6bba2a0d38 "a->addr.sa_family == PJ_AF_INET || a->addr.sa_family == PJ_AF_INET6", file=file@entry=0x7f6bba2a0cc0 "../src/pj/sock_common.c", line=line@entry=388, function=function@entry=0x7f6bba2a0ee0 "pj_sockaddr_get_len") at assert.c:92
#3  0x00007f6c0dfd5312 in __GI___assert_fail (assertion=0x7f6bba2a0d38 "a->addr.sa_family == PJ_AF_INET || a->addr.sa_family == PJ_AF_INET6", file=0x7f6bba2a0cc0 "../src/pj/sock_common.c", line=388, function=0x7f6bba2a0ee0 "pj_sockaddr_get_len") at assert.c:101
#4  0x00007f6bba29797b in pj_sockaddr_get_len () from /usr/lib/libpj.so.2
#5  0x00007f6bbba93788 in __rtp_recvfrom (instance=0x7f6ba6012ce0, buf=0x7f6bc8024638, sa=0x7f6ba6012ff0, rtcp=0, flags=<optimized out>, size=<optimized out>) at res_rtp_asterisk.c:2067
#6  0x00007f6bbba9a59c in rtp_recvfrom (sa=<optimized out>, flags=<optimized out>, size=<optimized out>, buf=<optimized out>, instance=<optimized out>) at res_rtp_asterisk.c:2097
#7  ast_rtp_read (instance=0x7f6bc801f278, rtcp=-939375048) at res_rtp_asterisk.c:4346
#8  0x00007f6ba70045b2 in sccp_astwrap_rtp_read (ast=0x7f6bc8008198) at ast111.c:485
#9  0x0000563fcab8fe42 in __ast_read (chan=0x7f6bc8008198, dropaudio=0) at channel.c:4054
#10 0x0000563fcab91068 in ast_read (chan=<optimized out>) at channel.c:4408
#11 __ast_answer (chan=0x7f6bc8008198, delay=0, cdr_answer=0) at channel.c:2987
#12 0x0000563fcac19201 in pbx_builtin_answer (chan=0x7f6bc8008198, data=0x563fcac7b7c1 "") at pbx.c:11053
#13 0x0000563fcac1ca47 in pbx_exec (c=0x7f6bc8008198, app=0x563fcc4f16b0, data=0x7f6ba6013ca0 "") at pbx.c:1622
#14 0x0000563fcac29f76 in pbx_extension_helper (c=0x7f6bc8008198, con=<optimized out>, context=<optimized out>, exten=<optimized out>, priority=<optimized out>, label=<optimized out>, callerid=0x563fcc9771a0 "1104", action=E_SPAWN, found=0x7f6ba6017d4c, combined_find_spawn=1) at pbx.c:4915
#15 0x0000563fcac2c017 in ast_spawn_extension (combined_find_spawn=<optimized out>, found=<optimized out>, callerid=<optimized out>, priority=<optimized out>, exten=<optimized out>, context=<optimized out>, c=<optimized out>) at pbx.c:6037
#16 __ast_pbx_run (c=0x7f6bc8008198, args=0x1b0f, args@entry=0x0) at pbx.c:6512
#17 0x0000563fcac2d5ab in pbx_thread (data=data@entry=0x7f6bc8008198) at pbx.c:6842
#18 0x0000563fcac6d38c in dummy_start (data=<optimized out>) at utils.c:1192
#19 0x00007f6c0ccfc064 in start_thread (arg=0x7f6ba6018700) at pthread_create.c:309
#20 0x00007f6c0e08f62d in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:111
```

dkgroot, 2019-09-26:
that looks a lot better, will look into it. And currently trying to switch my 7970 to ipv6 mode (and it doesn't want to read my new SEP :-)).
See you later on gitter !

dkgroot, 2019-09-26:
If you have to change to update asterisk-11 (latest revision) or even upgrade to >= asterisk-13, that would be my advice as first step. Based only on the backtrace you provided i think it might be an asterisk/pjproject issue.

berean-sc, 2019-09-26:
I agree. I will plan a system upgrade to include a supported version of Asterisk. It is not worth further investigation to make this work on an obsolete Asterisk system. Thank you. I will start a new ticket if my problems persist on modern Asterisk.

closed 2019-09-26 by berean-sc

dkgroot, 2019-09-26:
There might be a quick path, if you are also behind on asterisk-11 revisions, maybe try latest. I do know ipv6 used to already work on asterisk-11 back-when.
Another thing you might be able to do, is check is sccp->sccp or sccp->chan_sip works (circumventing pjproject).

But yes, asterisk-11 is out of support, and we only maintain a `compiling` version. I don't regularly test anything on it (just being honest). While you are at it, debian 8.0 is also a little old (time to move to "buster", buster:-))

So time to move along and get asterisk-13 or maybe be smart and jump to 16. Asterisk-17 is planned for a release later this year and 13 is scheduled to be scrapped (next year i think). Config wise there should be 0 surprises (AFAIK) when moving from 11 directly to 16.