chan-sccp/chan-sccp #478: 'video_codec' undeclared

billsimon, 2019-09-26:
Compiling latest pull from branch develop (4d4927a29feb7ad5f4c095ee1b84bce7fc96b88a) against Asterisk 16:

```
Making all in pbx_impl/ast116/
make[3]: Entering directory '/usr/local/src/chan-sccp/src/pbx_impl/ast116'
  CC       libast116_la-ast116.lo
ast116.c: In function ‘sccp_astwrap_request’:
ast116.c:1733:26: error: ‘video_codec’ undeclared (first use in this function); did you mean ‘audio_codec’?
   videoCapabilities[0] = video_codec;
                          ^~~~~~~~~~~
                          audio_codec
ast116.c:1733:26: note: each undeclared identifier is reported only once for each function it appears in
make[3]: *** [Makefile:574: libast116_la-ast116.lo] Error 1
```

billsimon, 2019-09-26:
Looks like the declaration of `skinny_codec_t video_codec` is wrapped in an #if macro (line 1638) but line 1733 is not, resulting in the error.

billsimon, 2019-09-26:
Confirmed that wrapping ast116.c line 1733 with `#if ... #endif` produces a successful build.

dkgroot, 2019-09-26:
@billsimon 
Thanks for reporting the issue (and researching the cause) !
Committed the fix.

commit c4a49700 (2019-09-26): Fix undeclared video_codec compiler warning

Fixes #478
closed 2019-09-26 by dkgroot