chan-sccp/chan-sccp #479: Building Debian Package

bfranske, 2019-09-30:
I know you used to provide Debian packages through the OpenSUSE build service but it doesn't look like they are available for Debian 10 / Asterisk 16.2.1~dfsg-1+deb10u1.

I am trying to 1) Build chan_sccp for this environment and then
2) Build it as a dpkg so I can deploy it easily to several other systems

But I'm having difficulty with both of these things. ./configure is working correctly as far as I can tell but when I try to run a `make `or `make -j2` I get the error:

```
make[1]: Entering directory '/home/ben/chan-sccp-develop/src'
make[1]: *** [Makefile:1433: .revision] Error 1
make[1]: Leaving directory '/home/ben/chan-sccp-develop/src'
make: *** [Makefile:610: all-recursive] Error 1
```

If I try a `make deb` I get:
```
dpkg-source: error: can't build with source format '3.0 (native)': native package version may not have a revision
dpkg-buildpackage: error: dpkg-source -b . subprocess returned exit status 255
make: *** [Makefile:1093: deb] Error 255
```

Any assistance would be greatly appreciated!

bfranske, 2019-09-30:
Figured out the make problem was something corrupt in my source archive. I have re-pulled the git repo and it then built correctly. The `make deb` is still not working though.

dkgroot, 2019-09-30:
@bfranske 

Not sure what is going on there. it build fine on my ubuntu system, which ought to be compatible for the most part. What are the exact errors when you do `dpkg-buildpackage -rfakeroot -us -uc` on the command line from the chan-sccp directory.

BTW: we also have a Launchpad PPA repo, which might work for debian as well. Note you will have to read the ppa build log, to figure out the exact asterisk package it's is build against.

I will look into debian-10 on openSuSe BuildService, to see if that is available.

dkgroot, 2019-09-30:
OpenSuSE Build Service is building 'Debian-10' packages
Updated the README to show the path

Note: You need to make sure you use the exact same asterisk package used by the openSuSE Build Service, so do check the build logs !

Last build was using : 'asterisk-dev-1:16.2.1~dfsg-1+deb10u1' so make sure you have that exact same version installed.

bfranske, 2019-09-30:
Awesome! I'm glad to have pre-built packages available for Debian again! I can fumble my way around Debian package creation but it's definitely not my area of expertise. I'm going to be using this package in class with students in a few weeks so it's much nicer to have a pre-built package they can download.

For what it's worth I did play around with it for a couple more hours and got it to build manually on my system (though I haven't actually tested to ensure it works with actual phones yet though the module did load successfully in Asterisk). It turns out that because the debian/source/format file specifies 3.0 (native) instead of 3.0 (quilt) the build system requires there be no - in the version number e.g. it requires 4.3.2 not 4.3.2-1ubuntu1. I eventually discovered it was reading that version number from the debian/changelog file. I modified the version number in there and then it built successfully.

Anyway, thanks for getting the automated builds for Debian working again!

dkgroot, 2019-09-30:
Just a couple more seconds before you start downloading.. it's still (re-building) latest revision...

dkgroot, 2019-09-30:
> For what it's worth I did play around with it for a couple more hours and got it to build manually on my system (though I haven't actually tested to ensure it works with actual phones yet though the [...]

I was expecting something like that would be the case. Debian and Ubuntu are mostly the same, but there a small differences. On the opensuse buildserver we generate a fake changelog on the fly, so we don't actually use the one from source.
Not sure what would be the nicest way to update our Makefile, to include this distinction.

bfranske, 2019-09-30:
I teach an "IT Services & Applications" course at a local college in a Computer Networking & Security program. 1/3 of that course is related to VoIP services, 1/3 to Virtualization, and 1/3 to Cloud Services. In previous years I have done the VoIP portion with Cisco CUCM but want to transition to using Asterisk, mostly because I prefer open source but also because it's likely many of them will be working in small or mid-sized businesses where Asterisk may be a better choice anyway. I already have a large number of 79XX phones from our CUCM labs and previous experience from my consulting practice has been that chan-sccp provides a better user experience than converting the 79XX phones to SIP. I do have a smaller number of new Digium phones they'll work with as well but not enough for all the groups to have multiple phones so we'll need the 79XX phones as well.

dkgroot, 2019-09-30:
Very interesting use case :-)
Setting-Up and configuring chan-sccp is a lot simpler than getting CUCM to do your bidding, so i guess the student will welcome the reprief :-)

I found a little issue in Makefile.am causing an error during the generation of revision.h, causing all builds on build server to fail (with the latest package). Trying to fix it right now.

dkgroot, 2019-10-01:
It took quite a number of tries, to figure out what was wrong with the build script regarding .revision and revision.h but it's building again.

dkgroot, 2019-10-01:
Reopen to check if we need to look into the "make deb" issue on debian, or just leave it as is ?

dkgroot, 2019-10-05:
@bfranske 
I updated the debian/changelog file, and it should now work for both debian and ubuntu. Closing this issue for now. If there is still issue when you use `make deb` on debian, please let me know.

closed 2019-10-05 by dkgroot

bfranske, 2019-11-14:
FWIW we have now gotten past this point in the class and all my students were able to successfully download, install the dpkg, and configure chan-sccp with a mix of 79x0, 79x1, and 79x2 phones.

dkgroot, 2019-11-14:
Nice, great to hear ! Looking forward to hearing how you and your students are progressing :-)

dkgroot, 2019-12-17:
Noticed the links in http://wiki.ihitc.net/w/Franske_ITC-2300_Schedule pointing to chan-sccp :-)

commit ee7174bc (2019-10-01): Final Fix make revision.h

Fixes #479