chan-sccp/chan-sccp #482: Suddenly SCCP phones reload

anutator, 2019-10-08:
There is very critical problem. Reloaded today several times. There are only several phones (less than 10) registered on Asterisk and I can't add all phones due to this problem. Sometimes all phones  (even idle) are reloaded and we can't answer a call in queue.

Cisco 7911 phones.
**chan-sccp version:** the latest (installed last friday)
**asterisk version:** (Asterisk 16.6.0-rc2 )
**Expected behavior:** Phones work without disconnecting from Asterisk/

Unfortunately I didn't keep asterisk cli open during reload.
Can I find some info in logs? Is this due to keepalive=60?

dkgroot, 2019-10-08:
@anutator 
>Unfortunately I didn't keep asterisk cli open during reload. Can I find some info in logs?

It depends if you have the logger.c: `full` log enabled and what your default `sccp debug flags` are (to monitor devices / sessions and keepalive, do set 'sccp debug core, device, socket').

> Is this due to keepalive=60?

No it should not be related.

Skinny phones use TCP as there connection medium (different then most other voip protocols which use UDP). Because of this, the phones needs to remain connected all the time. To make sure the connection stays up, we expect a keepalive message every 60 seconds from the phone. 

Note: Interuptions in the network connection could/would cause problems in this regard.

You can try to change the keepalive to 120 and set 
```
<devicePool>
    ....
    <connectionMonitorDuration>120</connectionMonitorDuration>
</devicePool>
```
 in your SEPxxxx.cnf.xml

anutator, 2019-10-08:
Is it critical - incoming message is smaller?
```
[Oct  8 17:27:23] NOTICE[16665]: sccp_session.c:253 session_buffer2msg:     -- SEPDC7B94777357: (session_dissect_msg) Incoming message is smaller(12) than known size(13).
[Oct  8 17:27:23] NOTICE[16665]: sccp_actions.c:239 sccp_handle_message:     -- SEPDC7B94777357: >> Got message Keep Alive Message (0x0)
[Oct  8 17:27:23] NOTICE[16665]: sccp_actions.c:122 check_session_message_device: SEPDC7B94777357: SCCP Handle Message: Keep Alive Message(0x0000) 12 bytes length
 sccp_session.c :1225 (sccp_session_getDevice             ) --->                   refcount increased 03  +> 04 for     device: SEPDC7B94777357 (0x258f310)
[Oct  8 17:27:23] NOTICE[16665]: sccp_session.c:1038 sccp_session_send2: SEPDC7B94777357: Send Message: Keep Alive Acknowledge(0x0100) 4 bytes length
 sccp_actions.c :0241 (sccp_handle_message                ) <---                   refcount decreased 03  <- 04 for     device: SEPDC7B94777357 (0x258f310)
[Oct  8 17:27:23] NOTICE[16665]: sccp_session.c:633 sccp_session_device_thread:        > SEPDC7B94777357: set poll timeout 51 for session 32
```

dkgroot, 2019-10-08:
Related #424

anutator, 2019-10-09:
It seems that Asterisk was reloading.
I found errors in /var/log/asterisk/messages:
```
[Oct  8 09:42:58] ERROR[12182] loader.c: Error loading module 'res_ari_mailboxes.so', missing dependency: res_stasis_mailbox
[Oct  8 09:42:58] ERROR[12182] loader.c: Error loading module 'res_stasis_mailbox.so', missing dependency: res_mwi_external
[Oct  8 09:42:58] ERROR[12182] loader.c: Error loading module 'res_mwi_external_ami.so', missing dependency: res_mwi_external
```
Added noload for this modules to modules.conf.

[Oct  9 09:55:43] WARNING[18269] sccp_config.c: SCCP: Unknown param at general:24:dtmfmode='rfc2833'

I found errors in sccp.conf  from /var/log/asterisk/messages
[Oct  9 09:55:43] WARNING[18269] sccp_config.c: SCCP: Unknown param at general:24:dtmfmode='rfc2833'

...
```
[Oct  9 10:42:41] NOTICE[18551] sccp_config.c: SCCP: deprecated config param at directed_pickup='on' in line 104
 - enable/disable Pickup button to do directed pickup from a specific extension. (Deprecated: use line->directed_pickup instead)
[Oct  9 10:42:41] NOTICE[18551] sccp_config.c: SCCP: deprecated config param at directed_pickup_context='operators' in line 105
 - context where direct pickup search for extensions. If not set, current context will be use. (Deprecated: use line->directed_pickup_context instead)
[Oct  9 10:42:41] NOTICE[18551] sccp_config.c: SCCP: deprecated config param at directed_pickup_modeanswer='on' in line 106
 - Automatically Answer when using Directed Pickup. (default=on) (Deprecated: use line->pickup_modeanswer instead)
```
I had:
```
directed_pickup=on
directed_pickup_context=operators
directed_pickup_modeanswer=on

```
Example:
```
callgroup = ""                                                                    ; We are in caller groups 1,3,4. Valid for all lines
pickupgroup = ""                                                                  ; We can do call pick-p for call group 1,3,4,5. Valid for all lines
directed_pickup_context = ""                                                      ; context where direct pickup search for extensions. If not set, the current context of the picking line will be used.
;pickup_modeanswer = yes                                                          ; Automatically Answer when using Directed Pickup. (default=on)
```

dkgroot, 2019-10-09:
@anutator 
It's always a good idea, to check the sccp debug output, when loading a 'new' sccp.conf file or a newer version of the chan_sccp module. Things might have changed, and we try to let the administrator know. First we deprecate support for a parameter, and later that parameter is made obsolete.

The `conf/sccp.conf.annotated` is just to let you know which parameters are available, not to be used or copied directly. I think i should add a better warning at the top of this file to make that a little clearer.
Please do not have 'empty' or paramaters with "" in your config. If you are using the default value for a parameter then you can just remove it from your config (and you will get the default value). If a parameter is not clear to you, just remove it from your sccp.conf and you will get the sensible default.

force_dtmfmode should not be used in general (ie: left to default) and is a replacement for the old/obsolete dtmfmode. The default setting for dtmfmode is now always rfc2833. For old (pre-java) devices that do not support rfc2833 we will automatically simulate the rfc2833 behaviour for it.

Some of the pickup parameters have changes names and have been moved from device level to the line level as noted in the warnings you got.

If you want you can generate a sccp.conf.annotated from your current chan_sccp module, which will show you all names and locations of parameters, specific to your current version. Use:
```
asterisk -rx "sccp config generate sccp.conf.annotated"
```

Have you taken a look at issue : #424 . I think that matches the problem you are seeing on your 7911's

dkgroot, 2019-10-09:
The changes you made should not have any direct inpact on the devices restarting. You need to have a look at the issue #424 an change the firmware on the 7911 to match theirs to resolve the issue.

dkgroot, 2019-10-09:
@anutator 
Does `sccp show device SEPxxxx` show 'Image Version' for the 7911 ? If so we can use it to detect and check the minimal Firmware Image Version we require to get around the '87 (ie:0x57)' message issue reported in #424.

dkgroot, 2019-10-09:
Added a new wiki page about [obsolete and deprecated parameters](https://github.com/chan-sccp/chan-sccp/wiki/Deprecated---Obsoleted-Parameters), which might help people sort this out.

anutator, 2019-10-10:
Today when SCCP and Asterisk reloaded:
```
[Oct 10 11:25:50] NOTICE[5707][C-00000004] sccp_pbx.c: SEP001D70FAECC6: initialize cfwdAll for line 4472
[Oct 10 11:25:50] NOTICE[5707][C-00000004] sccp_channel.c: SCCP: (sccp_channel_forward) channel SCCP/4472-0000000B is dialing number 4471
[Oct 10 11:25:59] NOTICE[5708][C-00000004] ast117.c: SCCP: (masqueradeHelper) answer temp: SCCP/4472-0000000A
[Oct 10 11:25:59] NOTICE[5708][C-00000004] ast117.c: SCCP: (masqueradeHelper) replace pbxTmpChannel: SCCP/4472-0000000A with SCCP/4471-0000000C (move)
[Oct 10 11:25:59] NOTICE[5708][C-00000004] ast117.c: Fixup: nativeformats=(alaw)
[Oct 10 11:25:59] NOTICE[5708][C-00000004] ast117.c: Fixup: nativeformats=(alaw)
[Oct 10 11:25:59] NOTICE[5708][C-00000004] ast117.c: SCCP: (masqueradeHelper) move succeeded. Hanging up orphan: SCCP/4472-0000000A
[Oct 10 11:26:04] Asterisk 16.6.0-rc2 built by root @ localhost.localdomain on a x86_64 running Linux on 2019-10-03 11:59:28 UTC
[Oct 10 11:26:04] NOTICE[5725] loader.c: 302 modules will be loaded.
[Oct 10 11:26:04] NOTICE[5725] res_odbc.c: Registered ODBC class 'asterisk' dsn->[asteriskCDR]
[Oct 10 11:26:04] NOTICE[5725] cdr.c: CDR simple logging enabled.
[Oct 10 11:26:04] NOTICE[5725] res_smdi.c: No SMDI interfaces are available to listen on, not starting SMDI listener.
[Oct 10 11:26:04] WARNING[5725] file.c: ignoring unknown format 'wav49'
[Oct 10 11:26:04] WARNING[5725] file.c: ignoring unknown format 'gsm'
[Oct 10 11:26:04] WARNING[5725] file.c: ignoring unknown format 'wav'
[Oct 10 11:26:04] WARNING[5725] file.c: no known formats found in format list (wav49|gsm|wav)
[Oct 10 11:26:04] ERROR[5725] app_voicemail.c: Error processing format string, defaulting to format 'wav'
[Oct 10 11:26:04] NOTICE[5725] chan_sccp.c: preloading pbx module
[Oct 10 11:26:04] NOTICE[5725] sccp_mwi.c: SCCP: (mwi::module_start)
[Oct 10 11:26:04] NOTICE[5725] sccp_config.c: Config file 'sccp.conf' loaded.
[Oct 10 11:26:04] WARNING[5725] config_options.c: Attempted to set internal_sample_rate=auto, but set it to 0 instead due to default)
[Oct 10 11:26:04] NOTICE[5725] confbridge/conf_config_parser.c: Adding default_menu menu to app_confbridge
[Oct 10 11:26:04] NOTICE[5725] cel_custom.c: No mappings found in cel_custom.conf. Not logging CEL to custom CSVs.
[Oct 10 11:26:08] WARNING[5823] sccp_device.c: SCCP: Iconv: illegal character.
[Oct 10 11:26:18] WARNING[5862][C-00000001] app_dial.c: Unable to create channel of type 'SCCP' (cause 44 - Requested channel not available)
[Oct 10 11:26:21] WARNING[5863][C-00000002] app_dial.c: Unable to create channel of type 'SCCP' (cause 44 - Requested channel not available)
```

dkgroot, 2019-10-10:
This does not look like a reload, but more like an asterisk crash. We would need to have a look at the backtrace from the coredump left behind  (hopefully you have dumpcore enabled).

From the last line of the debug log before the crash, it looks the new 'callforward shared line' code might be involved. This change only recently got merged.

Can you please have a look at my answers regarding the three open issues we have already, before looking into this 'new' issue.

anutator, 2019-10-10:
> 
> 
> @anutator
> Does `sccp show device SEPxxxx` show 'Image Version' for the 7911 ? If so we can use it to detect and check the minimal Firmware Image Version we require to get around the '87 (ie:0x57)' message issue reported in #424.

Image version SCCP11.9-4-2SR3-1S.
```
ast16*CLI> sccp show device SEP001D70FAECBB

--- SCCP channel driver device settings ----------------------------------------------------------------------------------
MAC-Address                                    : SEP001D70FAECBB
Protocol Version                               : Supported '22', In Use '22'
Protocol In Use                                : SCCP Version 22
Device Features                                : 0x7285,000000000111001010000101
Tokenstate                                     : None
Keepalive                                      : 60
Registration state                             : OK
State                                          : On Hook
Addons                                         : 
MWI state                                      : off (0/0)
MWI light-type                                 : On
MWI During call                                : turn off
Description                                    : Glazkov 4471
Config Phone Type                              : 7911
Skinny Phone Type                              : Cisco 7911(307)
Softkey support                                : yes
Softkeyset                                     : default => default (0x2ccc060)
BTemplate support                              : yes
linesRegistered                                : yes
Image Version                                  : SCCP11.9-4-2SR3-1S
Timezone Offset                                : 0
Audio Capabilities                             : [slin16, ulaw/64k, alaw/64k, ilbc, g729b, g729ab, g729, g729a]
Audio Preferences                              : [alaw/64k, ulaw/64k, g729, g729a, g729b, g729ab]
Audio TOS                                      : 184
Audio COS                                      : 6
Video TOS                                      : 136
Video COS                                      : 5
DND Feature enabled                            : yes
DND Status                                     : Disabled
DND Action                                     : Disabled / Cycle
Can Transfer                                   : on
Can Park                                       : on
Can CFWDALL                                    : on
Can CFWBUSY                                    : on
Can CFWNOANSWER                                : on
Allow ringin notification                      : no
Private softkey                                : off
Dtmf mode                                      : RFC2833
Nat                                            : (Auto)Off
Videosupport?                                  : no
Direct RTP                                     : off
Trust phone ip (deprecated)                    : off
Bind Address                                   : 10.112.14.16:49798
Server Address                                 : 10.112.0.103:35892
Deny/Permit                                    : deny:0.0.0.0/0.0.0.0,permit:127.0.0.0/255.0.0.0,permit:10.0.0.0/255.0.0.0,permit:172.0.0.0/255.224.0.0,permit:192.168.0.0/255.255.0.0,
PermitHosts                                    :  
Early RTP                                      : None
Accessory State                                : None
Last dialed number                             : 8405222 (1)
Default line instance                          : 1
Custom Background Image                        : ---
Custom Background Thumbnail                    : ---
Custom Ring Tone                               : ---
Use Placed Calls                               : off
PendingUpdate                                  : off
PendingDelete                                  : off
allow_conference                               : on
conf_play_general_announce                     : on
conf_play_part_announce                        : on
conf_mute_on_entry                             : off
[... 14 lines cut ...]
| 1    4471                                                 > лазков А.А.                          None                        |
+----------------------------------------------------------------------------------------------------------------------+

+--- SpeeddialButtons -------------------------------------------------------------------------------------------------+
| Id   Name                    Number                 Hint                                                             |
+ ==== ======================= ====================== ================================================================ +
+----------------------------------------------------------------------------------------------------------------------+

+--- FeatureButtons ---------------------------------------------------------------------------------------------------+
| Id   Name                    Status                 Options                                                          |
+ ==== ======================= ====================== ================================================================ +
+----------------------------------------------------------------------------------------------------------------------+

+--- ServiceURLButtons ------------------------------------------------------------------------------------------------+
[... cut ...]
| Type     Calls    PcktSnt  PcktRcvd Lost     Jitter   Latency  Quality  avgQual  meanQual maxQual  rConceal sConceal |
+ ======== ======== ======== ======== ======== ======== ======== ======== ======== ======== ======== ======== ======== +
| LAST     31       3860     3859     0        0        0        4.500000 4.490700 4.321200 4.500000 0.000500 2        |
| AVG      20       3065     2833     0        0        0        3.239115 3.293225 3.175655 4.500000 0.111050 3        |
+----------------------------------------------------------------------------------------------------------------------+
```

dkgroot, 2019-10-10:
Ok did you update it recently or was it running this firmware before the 'reloading phones' issue. Is the reloading this happening at the moment ?

dkgroot, 2019-10-10:
If the reloading/restarting is still happening, we will have to start a tcpdump to monitor them + `sccp debug core, socket, device` then when the issue occurs we should be able to figure out, what is causing it. The question is, if we are sending something to the phones causing them to restart or if the phone is sending us something we don't know how to handle.

dkgroot, 2019-10-10:
> Image version SCCP11.9-4-2SR3-1S.
> It doesn't show russian letter «Г» in label :

I don't know where the `> ` in front of it is coming from. Because the button definition has to be split up in to it's part, we might be doing it wrong in respect to parsing UTF8. I will have a look at that, in the meantime.

anutator, 2019-10-10:
sccp.conf (two phones with one number):

```
[general]
debug= core, config, action, socket, device, line, channel
servername= Asterisk
keepalive= 60
context= operators
dateformat=D.M.Y
bindaddr= 0.0.0.0
port= 2000
disallow=all
allow=alaw
allow=ulaw
allow=g729
firstdigittimeout= 16
digittimeout= 8
autoanswer_ring_time= 1
autoanswer_tone= Zip
remotehangup_tone= Zip
transfer=on
transfer_tone= 0
transfer_on_hangup= off
dnd_tone= 0x0
callwaiting_tone= Call Waiting Tone
musicclass=default
language=ru
deny=0.0.0.0/0.0.0.0
permit=internal
localnet= internal
dndFeature= on
sccp_tos= 0x68
sccp_cos= 4
audio_tos= 0xB8
audio_cos= 6
video_tos= 0x88
video_cos= 5
echocancel= on
silencesuppression= off
private= on
callanswerorder=oldestfirst
pickup_modeanswer=on

[79XX](!)
type=device
keepalive=60
park=on
cfwdall=yes
cfwdnoanswer=yes
directrtp=no
earlyrtp=none 
dndfeature=on
mwilamp=on
mwioncall=off
meetme=on
meetmeopts=qxd

[7911](!,79XX)
devicetype=7911
imageversion=SCCP11.9-4-2SR3-1S

[7925](!,79XX)
devicetype=7925
imageversion=CP7925G-1.4.8SR1.5
[line](!)
type=line
language=ru
context=operators
incominglimit=2

[SEP001D70FAECBB](7911)
description=Glazkov 4471
button=line,4471

[SEP30F70D4D7ABA](7925)
description=Glazkov 4471
button=line,4471

[4471](line)
label=Глазков А.А.
cid_name=Глазков А.А.
cid_num=4471

```

dkgroot, 2019-10-10:
Ok thanks for the config file, I will check that out later today.

How often does the device restart happen. Would it be ok, to run a tcpdump on your system for today, so we can catch the problem when it happens ?

anutator, 2019-10-10:
coredump - Asterisk crash 11:26 today attached.
[astdump.tar.gz](https://github.com/chan-sccp/chan-sccp/files/3712503/astdump.tar.gz)

And I also will try to make tcpdump.
It doesn't happen often because we have only several phones registered on asterisk and a little amount of calls.

dkgroot, 2019-10-10:
@anutator 
I can't do anything with a dump file, you need to run gdb backtrace on your local machine using the coredump.
You move to where the core.dump file is. You run `gdb asterisk core`, wait for it to finish loading and you end up at a prompt. Now you can run `bt` and `bt full` on the gdb command line. The output from these commands is what we are interested in.

dkgroot, 2019-10-10:
If you like i could help out over an ssh connection, if that makes things a little easier. We could then look into both issues at the same time.

anutator, 2019-10-10:
> If you like i could help out over an ssh connection, if that makes things a little easier. We could then look into both issues at the same time.

Tomorrow I can give you access to my comp using TeamViewer.  Or I can give you remote access (VPN) so you would have access to ssh and web 9090 cockpit. Please give me contact, I'll send you details.

dkgroot, 2019-10-10:
I don't have nor like teamviewer to be honest. We can however work together using `ssh` + `screen` so we can both interact with the system. The ssh might be a better fit.
You can contact me via dkdegroot@gmail.com to send me the VPN & ssh details.

anutator, 2019-10-10:
I'll add user details for vpn on OpenBSD and send you info.

dkgroot, 2019-10-10:
You can still start the tcpdump though, than we have something to use.

dkgroot, 2019-10-11:
Only the traffic between the server and the phones should be enough. You can include port 2000 but that is a little tricky.

dkgroot, 2019-10-11:
BTW: i added a CLI function to generate a wiki page, that shows all the sccp.conf options and their description. [See sccp.conf options page](https://github.com/chan-sccp/chan-sccp/wiki/sccp.conf-options), it might be helpfull when fixing your `module load chan_sccp` warnings later on.

dkgroot, 2019-10-12:
Problems related to the crash + coredump are possibly linked #455 ( 'callforward on shared line')

anutator, 2019-10-14:
Today I installed sccp again. Old and new chan_sccp versions make Asterisk crush during call to phone that is call-forwarded when it has second phone with thw same number registered (7911 and 7925 phones). As new version didn't help I'll register 7925 phones with other numbers and give them the same Caller ID as 7911. Then add extensions to numbering plan which dial 7911 and 7925 simultaneously.

Besides now after reloading new chan_sccp we have other problem with softkeys - we  could callforward phone but can't disable call forward - some button names are in English with name «Empty». Did yuou change button templates?

On 7911:
Повтор  Empty Соообщ Еще    - 1 screen
Empty Empty Empty Empty - 2 screen
Empty Empty -  3 screen

dkgroot, 2019-10-14:
@anutator If i had the segfault you got with the develop branch i would have been able to figure out what was going on by now. Let's focus on fixing the problem, instead of making new ones

BTW: The branch/cfwdshareline does not exists anymore, it was dissolved during a previous merge.

Can you check the VPN situation (replied to your mail).

dkgroot, 2019-10-15:
Thanks for helping out resolving the cfwdall issue on sharedline ! Let's hope the fix will be enough !

dkgroot, 2019-10-15:
We still have to analyse the phone reloading issue.. We have not cleared that one up. Or did you get that to stop ?

anutator, 2019-10-22:
Today I had problems with chan_sccp again. Channel stack in Ringing state.
Other phones (except 4471) were good. But when I released channel using command "channel request hangup..." all phones didn't work - no Ringing tone, But all of them where registered.

```
ast16*CLI> sccp show channels

+--- Channels --------------------------------------------------------------------------------------------------------------------------------------+
| ID    Name                      LineName   DeviceName       NumCalled  PBX State  SCCP State AudioR AudioW RTPPeer                Direct DTMFmode |
+ ===== ========================= ========== ================ ========== ========== ========== ====== ====== ====================== ====== ======== +
| 265   SCCP/4471-00000109        4471       SCCP                        Ringing    RINGING                                         no     RFC2833  |
+---------------------------------------------------------------------------------------------------------------------------------------------------+
[Oct 22 09:41:26] WARNING[6563]: sccp_session.c:1116 sccp_session_check_crossdevice: Session(0x7f1ab000ad60) and Device Session(0x7f1ab0004770) are out of sync.
[Oct 22 09:41:26] WARNING[6563]: sccp_actions.c:833 handle_register: SEP30F70D4D7ABA: Cleaning previous session, come back later, state:OK
```

dkgroot, 2019-10-22:
Hi Anna,
Can you please filter out everything related to this device, it's lines and channels from your debug logs and send them to me. I would like to figure out what happened before this moment, to cause this type of behaviour.

anutator, 2019-10-22:
Where can I find logs? I didn't expect the error.
In sccp debug = core, channel, pbx

dkgroot, 2019-10-22:
If you had debug=core,channel,pbx that would be perfect. You can find the logs in /var/log/asterisk/full.
I hope you can do some of the filtering for me. If not just compress the file and mail it to me.

anutator, 2019-10-22:
I tied to tar this log because in tail -n3000 I didn't get ny interval but now it is binary. So I can't get this log.
Only in /var/log/messages:
```
[Oct 22 09:21:54] ERROR[4186][C-0000004a] sccp_channel.c: SEP001D70FAECBB: Putting Active Channel 4471 OnHold failed -> Cancelling new CaLL
[Oct 22 09:22:00] NOTICE[4186][C-0000004a] sccp_channel.c: SEP001D70FAECBB: Call SCCP/4471-00000109 is already being answered by someone else
[Oct 22 09:22:01] NOTICE[4186][C-0000004a] sccp_channel.c: SEP001D70FAECBB: Call SCCP/4471-00000109 is already being answered by someone else
[Oct 22 09:22:13] NOTICE[4186][C-0000004a] sccp_channel.c: SEP001D70FAECBB: Call SCCP/4471-00000109 is already being answered by someone else
[Oct 22 09:22:13] NOTICE[4186][C-0000004a] sccp_channel.c: SEP001D70FAECBB: Call SCCP/4471-00000109 is already being answered by someone else
[Oct 22 09:22:18] NOTICE[4186][C-0000004a] sccp_channel.c: SEP001D70FAECBB: Call SCCP/4471-00000109 is already being answered by someone else
[Oct 22 09:22:20] NOTICE[4186][C-0000004a] sccp_channel.c: SEP001D70FAECBB: Call SCCP/4471-00000109 is already being answered by someone else
[Oct 22 09:22:23] NOTICE[4186][C-0000004a] sccp_channel.c: SEP001D70FAECBB: Call SCCP/4471-00000109 is already being answered by someone else
[Oct 22 09:22:24] NOTICE[4186][C-0000004a] sccp_channel.c: SEP001D70FAECBB: Call SCCP/4471-00000109 is already being answered by someone else
[Oct 22 09:22:32] NOTICE[4186][C-0000004a] sccp_channel.c: SEP001D70FAECBB: Call SCCP/4471-00000109 is already being answered by someone else
[Oct 22 09:22:32] NOTICE[4186][C-0000004a] sccp_channel.c: SEP001D70FAECBB: Call SCCP/4471-00000109 is already being answered by someone else
[Oct 22 09:22:33] NOTICE[4186][C-0000004a] sccp_channel.c: SEP001D70FAECBB: Call SCCP/4471-00000109 is already being answered by someone else
[Oct 22 09:22:34] NOTICE[4186][C-0000004a] sccp_channel.c: SEP001D70FAECBB: Call SCCP/4471-00000109 is already being answered by someone else
[Oct 22 09:22:34] NOTICE[4186][C-0000004a] sccp_channel.c: SEP001D70FAECBB: Call SCCP/4471-00000109 is already being answered by someone else
[Oct 22 09:22:34] NOTICE[4186][C-0000004a] sccp_channel.c: SEP001D70FAECBB: Call SCCP/4471-00000109 is already being answered by someone else
[Oct 22 09:22:34] NOTICE[4186][C-0000004a] sccp_channel.c: SEP001D70FAECBB: Call SCCP/4471-00000109 is already being answered by someone else
[Oct 22 09:22:34] NOTICE[4186][C-0000004a] sccp_channel.c: SEP001D70FAECBB: Call SCCP/4471-00000109 is already being answered by someone else
[Oct 22 09:22:34] NOTICE[4186][C-0000004a] sccp_channel.c: SEP001D70FAECBB: Call SCCP/4471-00000109 is already being answered by someone else
[Oct 22 09:22:35] NOTICE[4186][C-0000004a] sccp_channel.c: SEP001D70FAECBB: Call SCCP/4471-00000109 is already being answered by someone else
[Oct 22 09:22:35] NOTICE[4186][C-0000004a] sccp_channel.c: SEP001D70FAECBB: Call SCCP/4471-00000109 is already being answered by someone else
[Oct 22 09:22:35] NOTICE[4186][C-0000004a] sccp_channel.c: SEP001D70FAECBB: Call SCCP/4471-00000109 is already being answered by someone else
[Oct 22 09:22:35] NOTICE[4186][C-0000004a] sccp_channel.c: SEP001D70FAECBB: Call SCCP/4471-00000109 is already being answered by someone else
[Oct 22 09:23:11] NOTICE[4186][C-0000004a] sccp_channel.c: SEP001D70FAECBB: Call SCCP/4471-00000109 is already being answered by someone else
[Oct 22 09:23:13] NOTICE[4186][C-0000004a] sccp_channel.c: SEP001D70FAECBB: Call SCCP/4471-00000109 is already being answered by someone else
[Oct 22 09:23:14] NOTICE[4186][C-0000004a] sccp_channel.c: SEP001D70FAECBB: Call SCCP/4471-00000109 is already being answered by someone else
[Oct 22 09:23:14] NOTICE[4186][C-0000004a] sccp_channel.c: SEP001D70FAECBB: Call SCCP/4471-00000109 is already being answered by someone else
[Oct 22 09:23:14] NOTICE[4186][C-0000004a] sccp_channel.c: SEP001D70FAECBB: Call SCCP/4471-00000109 is already being answered by someone else
[Oct 22 09:23:15] NOTICE[4186][C-0000004a] sccp_channel.c: SEP001D70FAECBB: Call SCCP/4471-00000109 is already being answered by someone else
[Oct 22 09:23:15] NOTICE[4186][C-0000004a] sccp_channel.c: SEP001D70FAECBB: Call SCCP/4471-00000109 is already being answered by someone else
[Oct 22 09:23:15] NOTICE[4186][C-0000004a] sccp_channel.c: SEP001D70FAECBB: Call SCCP/4471-00000109 is already being answered by someone else
[Oct 22 09:23:15] NOTICE[4186][C-0000004a] sccp_channel.c: SEP001D70FAECBB: Call SCCP/4471-00000109 is already being answered by someone else
[Oct 22 09:23:15] NOTICE[4186][C-0000004a] sccp_channel.c: SEP001D70FAECBB: Call SCCP/4471-00000109 is already being answered by someone else
[Oct 22 09:23:15] NOTICE[4186][C-0000004a] sccp_channel.c: SEP001D70FAECBB: Call SCCP/4471-00000109 is already being answered by someone else
[Oct 22 09:23:16] NOTICE[4186][C-0000004a] sccp_channel.c: SEP001D70FAECBB: Call SCCP/4471-00000109 is already being answered by someone else
[Oct 22 09:23:16] NOTICE[4186][C-0000004a] sccp_channel.c: SEP001D70FAECBB: Call SCCP/4471-00000109 is already being answered by someone else
[Oct 22 09:23:16] NOTICE[4186][C-0000004a] sccp_channel.c: SEP001D70FAECBB: Call SCCP/4471-00000109 is already being answered by someone else
[Oct 22 09:23:17] NOTICE[4186][C-0000004a] sccp_channel.c: SEP001D70FAECBB: Call SCCP/4471-00000109 is already being answered by someone else
[... cut ...]
[Oct 22 09:44:21] WARNING[6571] sccp_session.c: Session(0x7f1ab0004c60) and Device Session(0x7f1ab0004770) are out of sync.
[Oct 22 09:44:21] WARNING[6571] sccp_actions.c: SEP30F70D4D7ABA: Cleaning previous session, come back later, state:OK
[Oct 22 09:44:23] WARNING[6569][C-0000003a] app_dial.c: Unable to write frametype: 2
[Oct 22 09:44:53] WARNING[6572] sccp_session.c: Session(0x7f1ab0005110) and Device Session(0x7f1ab0004770) are out of sync.
[Oct 22 09:44:53] WARNING[6572] sccp_actions.c: SEP30F70D4D7ABA: Cleaning previous session, come back later, state:OK
```

dkgroot, 2019-10-22:
From the piece that you send, i can't really figure out what is going on here. I had hoped you still had a higher log level something like "sccp debug core, device, line, channel, pbx". Maybe you can set that. And please check that full log is being written and rotated regularly (check to see if CLI> `logger rotate` works). It should switch log files and compress old ones using gzip.

Does SEP30F70D4D7ABA have a callforward set ? That might be good to know.

dkgroot, 2019-10-22:
> @anutator  : 4471 never activated call forward. He sad he had call on second line, he answered second line and then tried to come back to the first line. But softkey was "not assigned".

Are you able to replicate that behaviour on one of your phones in the test environment ? If we can reproduce the situation, we can fix it.

dkgroot, 2019-10-22:
Please run your production system with a higher debug setting like the one suggested above. The impact of a higher debug setting is quite minimal, it only prints during call setup and tear down, not when the call is established. You just need to make sure you have enough filesystem space (so do keep any eye on that). If you like you can set /etc/asterisk/logger.conf to rotate more frequently.

dkgroot, 2019-12-11:
@anutator Can you switch to issue #496 as you are having the same/similar trouble. Closing this issue report for now, so we can keep a little bit of overview.

closed 2019-12-11 by dkgroot