chan-sccp/chan-sccp #483: Phones are registering but lines are not.

rofferman, 2019-10-09:
Cisco 7900 phones are regisering thru tftp, but the lines are not registering. Is this a problem with my realtime database?

asterisk 16.51
FreePBX 15
chan-sccp  Release: 4.3.2 develop - d2a6837M 

I will get some debug info and post it in a minute

rofferman, 2019-10-09:
-- SEP00260BD75EFB: Unregister request Received (Reason: Powersave)
    -- SEP00260BD75EFB: Registration State is 'None'
 SCCP: Clean Device SEP00260BD75EFB, remove from global:Yes, restart_device:No
    -- SEP00260BD75EFB: Registration State is 'Cleaning'
    -- Removed device 'XXX00260BD75EFB' from Glob(devices)
  == SCCP: Unregister Device XXX00260BD75EFB
    -- XXX00260BD75EFB: (sccp_devstate_deviceRegisterListener) device unregistered
    -- XXX00260BD75EFB: Registration State is 'None'
    -- XXX00260BD75EFB: Registration State is 'None'
 XXX00260BD75EFB: Destroying Device
    -- XXX00260BD75EFB: Device Destroyed

 <Alarm Name="LastOutOfServiceInformation">
  == Alarm Type: LastOutOfServiceInformation
    -- <ParameterList>
    -- <String name="DeviceName">SEP00260BD75EFB</String>
    -- <String name="DeviceIPv4Address">192.168.2.90/24</String>
    -- <String name="IPv4DefaultGateway">192.168.2.1</String>
    -- <String name="DeviceIPv6Address"></String>
    -- <String name="IPv6DefaultGateway"></String>
    -- <String name="ModelNumber">CP-7942G</String>
    -- <String name="NeighborIPv4Address"></String>
    -- <String name="NeighborIPv6Address"></String>
    -- <String name="NeighborDeviceID">UnifiSwitch1</String>
    -- <String name="NeighborPortID">Camera above E Lane</String>
    -- <Enum name="DHCPv4Status">3</Enum>
    -- <Enum name="DHCPv6Status">0</Enum>
    -- <Enum name="TFTPCfgStatus">1</Enum>
    -- <Enum name="DNSStatusUnifiedCM1">4</Enum>
    -- <Enum name="DNSStatusUnifiedCM2">4</Enum>
    -- <Enum name="DNSStatusUnifiedCM3">4</Enum>
    -- <String name="VoiceVLAN">2</String>
    -- <String name="UnifiedCMIPAddress">192.168.2.25</String>
    -- <String name="LocalPort">49292</String>
    -- <String name="TimeStamp">1570652396134</String>
    -- <Enum name="ReasonForOutOfService">20</Enum>
  == Reason Enum: 20
    -- <String name="LastProtocolEventSent">0:KeepAlive</String>
  == Last Event Sent: 0:KeepAlive
    -- <String name="LastProtocolEventReceived">256:KeepAliveAck</String>
  == Last Event Received: 256:KeepAliveAck
    -- </ParameterList>
    -- </Alarm>
    -- </x-cisco-alarm>
 SEP00260BD75EFB: is registering, Instance: 2, UserId: 0, Type: Cisco 7942 (434), Version: 22 (loadinfo 'SCCP42.9-4-2SR3-1S')
    -- SCCP: Device 'SEP00260BD75EFB' found in realtime table 'sccpdevice'
    -- SCCP: Create Device
    -- SEP00260BD75EFB: Device State is 'On Hook'
    -- Init MessageStack
  == buttonconfig status before check
  == buttonconfig status after check
    -- Any Previous ButtonConfig will be discared during post-process
  == buttonconfig status after adding new buttons
SCCP: Parse Other Value: auto -> 0
SCCP: Parse Other Value: Missed Calls -> 3
       > Splitting '0.0.0.0' into...
       > ...host '0.0.0.0' and port ''.
  == SCCP: (sccp_sockaddr_storage_parse) addr:0.0.0.0
  == SCCP: (sccp_append_ha) mask:0.0.0.0
       > Splitting '0.0.0.0' into...
       > ...host '0.0.0.0' and port ''.
  == SCCP: (sccp_sockaddr_storage_parse) addr:0.0.0.0
  == SCCP: (sccp_append_ha) strmask:0.0.0.0, netmask:0.0.0.0
  == SCCP: (apply_netmask) applying netmask to 0.0.0.0/0.0.0.0
  == SCCP: (apply_netmask) result applied netmask 0.0.0.0
  == 0.0.0.0/0.0.0.0 sense 0 appended to acl for peer
       > Splitting '127.0.0.0' into...
       > ...host '127.0.0.0' and port ''.
  == SCCP: (sccp_sockaddr_storage_parse) addr:127.0.0.0
  == SCCP: (sccp_append_ha) mask:255.0.0.0
[... 124 lines cut ...]
    -- SEP00260BD75EFB: Clear the status prompt on line 0 and callid 0
    -- SEP00260BD75EFB: ( sccp_device.c:2151:sccp_dev_check_displayprompt ) sccp_dev_displayprompt '' for line 0 (0)
    -- SEP00260BD75EFB: Display prompt on line 0, callid 0, timeout 0
    -- SEP00260BD75EFB: Set softkeyset to ONHOOK(0) on line 0  and call 0
    -- SEP00260BD75EFB: validKeyMask 4294967295
    -- SEP00260BD75EFB: Finish DisplayPrompt
    -- SEP00260BD75EFB: Accessory 'Headset' is 'On Hook'
    -- SEP00260BD75EFB: Accessory 'Handset' is 'Off Hook'
    -- SEP00260BD75EFB: Accessory 'Speaker' is 'Off Hook'
    -- SEP00260BD75EFB: Getting the active channel on device.
    -- SEP00260BD75EFB: No active channel on device.
    -- SEP00260BD75EFB: Taken Offhook
    -- SEP00260BD75EFB: Device State is 'Off Hook'
[2019-10-09 15:21:40] NOTICE[26983]: sccp_actions.c:2432 handle_offhook: No lines registered on SEP00260BD75EFB for take OffHook
    -- SEP00260BD75EFB: ( sccp_actions.c:2433:handle_offhook ) sccp_dev_displayprompt 'No lines registered!' for line 0 (5)
    -- SEP00260BD75EFB: Display prompt on line 0, callid 0, timeout 5
    -- SEP00260BD75EFB: Sending tone Beep Bonk (51) on line 0 with callid 0 (direction: User)
    -- SEP00260BD75EFB: Clear the display notify message
    -- SEP00260BD75EFB: Clear the status prompt on line 0 and callid 0
    -- SEP00260BD75EFB: ( sccp_device.c:2151:sccp_dev_check_displayprompt ) sccp_dev_displayprompt '' for line 0 (0)
    -- SEP00260BD75EFB: Display prompt on line 0, callid 0, timeout 0
    -- SEP00260BD75EFB: Set softkeyset to ONHOOK(0) on line 0  and call 0
    -- SEP00260BD75EFB: validKeyMask 4294967295
    -- SEP00260BD75EFB: Finish DisplayPrompt
    -- SEP00260BD75EFB: Accessory 'Speaker' is 'On Hook'

dkgroot, 2019-10-09:
@rofferman 
Checks to do:
1. check if the devices in the realtime database are accessible, using `realtime load sccpdevice name SEPxxx`
2. check if it is returning the content of the sccpdevices table or the sccpdeviceconfig table (the one generated by the view (as intented). Ie: it should be containing the 'button' entries.
3. check if the lines in the realtime database are accessible, using `realtime load sccpline name 1234`
4. Check if the 'button' in 1. line up to the lines in 3.

rofferman, 2019-10-09:
realtime load sccpline name 3595
                   Column Name  Column Value
          --------------------  --------------------
                            id  86
                           pin
                         label  setup
                   description  setup <3595>
                       context  from-internal
                 incominglimit  6
                      transfer
                       mailbox
                         vmnum  *97
                      cid_name  setup
                       cid_num  3595
                      disallow  all
                         allow
                       trnsfvm
     secondary_dialtone_digits
       secondary_dialtone_tone  0x22
                    musicclass  default
                      language
                   accountcode  3595
                    echocancel  on
            silencesuppression  off
                     callgroup
                   pickupgroup
                namedcallgroup
              namedpickupgroup
                   adhocNumber
                        meetme
                     meetmenum
                    meetmeopts
                      regexten
               directed_pickup
       directed_pickup_context  from-internal
             pickup_modeanswer
                      amaflags
                           dnd  reject
                     videomode  auto
                        setvar
                          name  3595

dkgroot, 2019-10-09:
Was realtime working for you before, or are you just starting to use it now ? Did you make any alteration to the tables, if so you need to drop the sccpdeviceconfig view and regenerate it (That does not happen automatically).

rofferman, 2019-10-09:
realtime load sccpdevice name SEP00260BD75EFB
                   Column Name  Column Value
          --------------------  --------------------
                        button
                          type  7942
                         addon
                  _description  Test Setup
                      tzoffset  00
                  imageversion
                          deny
                        permit
                      earlyrtp
                       mwilamp
                     mwioncall
                    dndFeature
                      transfer
                       cfwdall
                      cfwdbusy
                       private  off
                       privacy  full
                           nat
                     directrtp
                    softkeyset  default
                     audio_tos  0xB8
                     audio_cos  6
                     video_tos  0x88
                     video_cos  5
                    conf_allow  on
    conf_play_general_announce  on
       conf_play_part_announce  on
            conf_mute_on_entry  off
      conf_music_on_hold_class  default
            conf_show_conflist  on
                force_dtmfmode  auto
                        setvar
               backgroundImage
           backgroundThumbnail
                      ringtone
                          name  SEP00260BD75EFB
callhistory_answered_elsewhere  Missed Calls
                       _hwlang  null
                    _loginname
                    _profileid  0
                 useRedialMenu  on
            transfer_on_hangup
                 phonecodepage  null

dkgroot, 2019-10-09:
> ```
realtime load sccpdevice name SEP00260BD75EFB
Column Name Column Value
-------------------- --------------------
button
type 7942
```

This last part is coming from the correct view name sccpdeviceconfig, but it looks like it cannot find a match between the sccpdevice table and the buttonconfig table, to provide any button entries.
Please check your buttonconfig table.

rofferman, 2019-10-09:
Yes it had been working fine. Then today one of the users had the time wrong on the phone and reset it and it never came back up fully. Now whrn a phone reboots it does not come back up 

Can you refresh me on how to drop and regenerate the sccpdeviceconfig view

rofferman, 2019-10-09:
ok ill check the button config table now

dkgroot, 2019-10-09:
Depends on which database backend are you using ?

rofferman, 2019-10-09:
button config table is empty 
Not sure how that happend. Can it be regenerated. I am using maraidb/mysql

dkgroot, 2019-10-09:
mysql:
```
CREATE OR REPLACE
ALGORITHM = MERGE
VIEW sccpdeviceconfig AS
        SELECT GROUP_CONCAT( CONCAT_WS( ',', buttonconfig.type, buttonconfig.name, buttonconfig.options )
        ORDER BY instance ASC
        SEPARATOR ';' ) AS button, sccpdevice.*
        FROM sccpdevice
        LEFT JOIN buttonconfig ON ( buttonconfig.device = sccpdevice.name )
GROUP BY sccpdevice.name;
```
postgresql:
```
CREATE OR REPLACE VIEW sccpdeviceconfig AS
        SELECT 
                (SELECT textcat_column(bc.type || ',' || bc.name || COALESCE(',' || bc.options, '') || ';') FROM (SELECT * FROM buttonconfig WHERE device=sccpdevice.name ORDER BY instance) bc ) as button,
                sccpdevice.*
        FROM sccpdevice
;
```

dkgroot, 2019-10-09:
Ooh that is ... unfortunate.. I don't know. I think you would have to check the mariadb logs, to see what is going on there. You might have to repair the table or something.

rofferman, 2019-10-09:
Ok I will look at the logs and see where they all went.

dkgroot, 2019-10-09:
myisamchk --recover tbl_name
or
maria_chk --recover tbl_name   (i think aria was a typo on that page).

Might do the trick

dkgroot, 2019-10-09:
After you get everything back up and running, it might be a good idea to start planning some FS and DB backup.

rofferman, 2019-10-09:
The repair did not work, I am using Sccp manager also is there any way to recreate the tables with either of these softwares

dkgroot, 2019-10-09:
Were you able to find any reason in the mariadb logs that would help ? 

Is the buttonconfig table still there, or is it empty ? If is there, it might have just been cleaned out for some reason (although normally the relationsships in other table should have prevented that).

If the whole table is not there anymore, then we would have to recreate it. We would first have to check if the backing db files are already removed (e.g. check the location where mariadb stores it's tables).

dkgroot, 2019-10-09:
**!! It would be a good idea to first make a FS and DB backup before continuing and doing any restoration work. There might be another cause that we have not found yet. !!**

To recreate the table:
```
CREATE TABLE IF NOT EXISTS `buttonconfig` (
  `device` varchar(15) NOT NULL default '',
  `instance` tinyint(4) NOT NULL default 0,
  `type` enum('line','speeddial','service','feature','empty') NOT NULL default 'line',
  `name` varchar(36) default NULL,
  `options` varchar(100) default NULL,
  PRIMARY KEY  (`device`,`instance`,`type`),
  KEY `device` (`device`),
  FOREIGN KEY (device) REFERENCES sccpdevice(name) ON DELETE CASCADE ON UPDATE CASCADE
) ENGINE=INNODB DEFAULT CHARACTER SET utf8 COLLATE utf8_general_ci;

DROP TRIGGER IF EXISTS trg_buttonconfig;
DELIMITER $$
CREATE TRIGGER trg_buttonconfig BEFORE INSERT ON buttonconfig
FOR EACH ROW
BEGIN
        IF NEW.`type` = 'line' THEN
                IF (SELECT COUNT(*) FROM `sccpline` WHERE `sccpline`.`name` = NEW.`name`) = 0
                THEN
                        UPDATE `Foreign key contraint violated: line does not exist in sccpline` SET x=1;
                END IF;
        END IF;
END$$
DELIMITER ;
```
Note: this would only work if there is no such table (it will not auto-drop it), and if there are not backing db files in the way, preventing mariadb from writing there.

rofferman, 2019-10-09:
the table is there it just has no data in it

dkgroot, 2019-10-09:
I just noticed that there is a 'DELETE CASCADE ON UPDATE CASCADE' in the buttonconfig table definition, so if you have recently dropped any devices and then recreated it, the entries for that table would also be removed from the buttonconfig table.

dkgroot, 2019-10-09:
if you have trouble recreating/remembering the data for the buttonconfig table, there might be some information we can retrieve from the asterisk message log, depending on your default sccp debug settings.

rofferman, 2019-10-09:
Yeah I do not remember the setup for button config

rofferman, 2019-10-09:
I have a back up that is a few weeks old from an old phone server.. I just switched to a virtual machine..
most of the entries would be correct

rofferman, 2019-10-09:
I have this and I think it has the sql tables inside

[sccp_instal_backup20190709.zip](https://github.com/chan-sccp/chan-sccp/files/3709480/sccp_instal_backup20190709.zip)

rofferman, 2019-10-09:
I finally got it back up, Thank you for your help.. 
I issued a fwconsole ma install sccp_manager command and it reinstalled sccp manager and recreated the buttonconfig.. actually it was sccpbuttonconfig but it worked
Still not sure how this all happened

dkgroot, 2019-10-09:
Congrats on getting everything back up and running.

Now start working on a little backup script and put it on your crontab. It sjhould be simple to do, nothing fancy required.
Maybe there is a module for freepbx that can take care of this ? If not that is a little strange, it tries to do everything else.

Closing the issue report.

closed 2019-10-09 by dkgroot