chan-sccp/chan-sccp #484: Asterisk Crashed

ksivramkrish, 2019-10-09:
Hello,

I am using chan_sccp but some reason asterisk got crashed and unable to find the cause from the core dump. I am trying to simulate the issue but not sure in which scenario it's causing.

I am using Asterisk 13.16.0 with chan_sccp 4.3.2 develop - ad141e4M

Program terminated with signal 11, Segmentation fault.
#0  ast_format_parse_sdp_fmtp (format=0x123e7a8, attributes=0x5e8df5 "") at format.c:332
332     format.c: No such file or directory.
Missing separate debuginfos, use: debuginfo-install asterisk-13.16.1-0.x86_64
(gdb) bt
#0  ast_format_parse_sdp_fmtp (format=0x123e7a8, attributes=0x5e8df5 "") at format.c:332
#1  0x0000000000570e24 in ast_rtp_codecs_payloads_set_rtpmap_type_rate (codecs=0x7ff104b1dd70, instance=0x0, pt=8, mimetype=0x7ff1197046ba "audio", mimesubtype=<optimized out>,
    options=(unknown: 0), sample_rate=8000) at rtp_engine.c:838
#2  0x00007ff1196831aa in process_sdp_a_audio (a=0x7ff16cf8f3b6 "rtpmap:8 PCMA/8000", p=0x7ff19c97a320, newaudiortp=0x7ff104b1dd70, last_rtpmap_codec=0x7ff104b1defc) at chan_sip.c:11828
#3  0x00007ff11968cbd4 in process_sdp (p=0x7ff19c97a320, req=0x7ff104b200b0, t38action=2) at chan_sip.c:11135
#4  0x00007ff1196cc3ff in handle_response_invite (p=0x7ff19c97a320, resp=200, rest=0x7ff16cf8f124 "OK", req=0x7ff104b200b0, seqno=101) at chan_sip.c:25057
#5  0x00007ff1196d7a4f in handle_response (p=<optimized out>, resp=200, rest=0x7ff16cf8f124 "OK", req=0x7ff104b200b0, seqno=101) at chan_sip.c:26158
#6  0x00007ff1196f4779 in handle_incoming (p=0x7ff19c97a320, req=0x7ff104b200b0, addr=0x7ff104b20b20, recount=0x7ff104b2005c, nounlock=0x7ff104b20058) at chan_sip.c:32687
#7  0x00007ff1196f5cea in handle_request_do (req=0x7ff104b200b0, addr=0x7ff104b20b20) at chan_sip.c:32997
#8  0x00007ff1196f75a6 in sipsock_read (id=<optimized out>, fd=<optimized out>, events=<optimized out>, ignore=<optimized out>) at chan_sip.c:32928
#9  0x00000000005147a2 in ast_io_wait (ioc=0x2121530, howlong=<optimized out>) at io.c:295
#10 0x00007ff1196b62b8 in do_monitor (data=<optimized out>) at chan_sip.c:33906
#11 0x00000000005c834b in dummy_start (data=<optimized out>) at utils.c:1238
#12 0x00007ff1b8eeddd5 in start_thread () from /lib64/libpthread.so.0
#13 0x00007ff1b84ccead in clone () from /lib64/libc.so.6
(gdb) frame
#0  ast_format_parse_sdp_fmtp (format=0x123e7a8, attributes=0x5e8df5 "") at format.c:332
332     in format.c
(gdb) where
#0  ast_format_parse_sdp_fmtp (format=0x123e7a8, attributes=0x5e8df5 "") at format.c:332
#1  0x0000000000570e24 in ast_rtp_codecs_payloads_set_rtpmap_type_rate (codecs=0x7ff104b1dd70, instance=0x0, pt=8, mimetype=0x7ff1197046ba "audio", mimesubtype=<optimized out>,
    options=(unknown: 0), sample_rate=8000) at rtp_engine.c:838
#2  0x00007ff1196831aa in process_sdp_a_audio (a=0x7ff16cf8f3b6 "rtpmap:8 PCMA/8000", p=0x7ff19c97a320, newaudiortp=0x7ff104b1dd70, last_rtpmap_codec=0x7ff104b1defc) at chan_sip.c:11828
#3  0x00007ff11968cbd4 in process_sdp (p=0x7ff19c97a320, req=0x7ff104b200b0, t38action=2) at chan_sip.c:11135
#4  0x00007ff1196cc3ff in handle_response_invite (p=0x7ff19c97a320, resp=200, rest=0x7ff16cf8f124 "OK", req=0x7ff104b200b0, seqno=101) at chan_sip.c:25057
#5  0x00007ff1196d7a4f in handle_response (p=<optimized out>, resp=200, rest=0x7ff16cf8f124 "OK", req=0x7ff104b200b0, seqno=101) at chan_sip.c:26158
#6  0x00007ff1196f4779 in handle_incoming (p=0x7ff19c97a320, req=0x7ff104b200b0, addr=0x7ff104b20b20, recount=0x7ff104b2005c, nounlock=0x7ff104b20058) at chan_sip.c:32687
#7  0x00007ff1196f5cea in handle_request_do (req=0x7ff104b200b0, addr=0x7ff104b20b20) at chan_sip.c:32997
#8  0x00007ff1196f75a6 in sipsock_read (id=<optimized out>, fd=<optimized out>, events=<optimized out>, ignore=<optimized out>) at chan_sip.c:32928
#9  0x00000000005147a2 in ast_io_wait (ioc=0x2121530, howlong=<optimized out>) at io.c:295
#10 0x00007ff1196b62b8 in do_monitor (data=<optimized out>) at chan_sip.c:33906
#11 0x00000000005c834b in dummy_start (data=<optimized out>) at utils.c:1238
#12 0x00007ff1b8eeddd5 in start_thread () from /lib64/libpthread.so.0
#13 0x00007ff1b84ccead in clone () from /lib64/libc.so.6
(gdb) info frame
Stack level 0, frame at 0x7ff104b1d400:
 rip = 0x500ac2 in ast_format_parse_sdp_fmtp (format.c:332); saved rip 0x570e24
 called by frame at 0x7ff104b1d480
 source language c.
 Arglist at 0x7ff104b1d3d8, args: format=0x123e7a8, attributes=0x5e8df5 ""
 Locals at 0x7ff104b1d3d8, Previous frame's sp is 0x7ff104b1d400
 Saved registers:
  rbx at 0x7ff104b1d3e0, rbp at 0x7ff104b1d3e8, r12 at 0x7ff104b1d3f0, rip at 0x7ff104b1d3f8
  
(gdb) info locals
interface = 0x40

dkgroot, 2019-10-09:
@ksivramkrish
Based on the backtrace it looks like an asterisk sipsock issue. Which makes this completely unrelated to chan-sccp.

Please do note, that latest asterisk-13 is 13.29.0 (since today), so you are quite a bit behind. This means that reporting the issue will not be taken up / fixed. First step would be to update your asterisk version.

Sorry i can't help you in this case :-( Please close the issue after reading.

ksivramkrish, 2019-10-09:
Thanks for your quick response. I will check it out.

closed 2019-10-09 by ksivramkrish

dkgroot, 2019-10-09:
Thanks for closing the issue, and sorry to see this happen and not being able to help out !