chan-sccp/chan-sccp #485: sccp_enum generation fails with 'mawk' NOT-A-NUMBER

snuffy44, 2019-10-11:
Using Debian 8.11 with default mawk/awk.
\[:xdigit:\] is not recognized causing compile error.

**Reproduction Steps:**

1. Force use of mawk
2. Try to generate sccp_enum.[ch]
3. Resulting sccp_enum.[ch] produces whole lot of NOT-A-NUMBER

**chan-sccp version:** develop branch

**asterisk version:** (include revision and/or branch if relevant)

**Expected behavior:**

Should correctly create the .c/.h files

**Observed behavior:**

inside sccp_enum.c:
ie. uint32_t value_map[] = {NOT-A-NUMBER,NOT-A-NUMBER,NOT-A-NUMBER,NOT-A-NUMBER,NOT-A-NUMBER,NOT-A-NUMBER,NOT-A-NUMBER,NOT-A-NUMBER,NOT-A-NUMBER,NOT-A-NUMBER,NOT-A-NUMBER,};

**Addition / Optional Information:**
- Problem started happening recently, didn't happen in an older version of Chan-SCCP: Yes
- Problem can be reliably reproduced, doesn't happen randomly: Yes
- Platform:
 - Debian 8.11  amd64

snuffy44, 2019-10-11:
Related to this new commit I believe: ee1087e656f4561419f68506c9b86f587dfd7947

dkgroot, 2019-10-11:
@snuffy44 
The enum generation script is based on the GNU awk version (gawk). Can you check if `gawk` is available on your system.
You can run the generation script manually from the command line, using:
`gawk -f "tools/gen_sccp_enum.awk" < src/sccp_enum.in`

snuffy44, 2019-10-11:
Tested - Works fine with gawk

dkgroot, 2019-10-11:
> Related to this new commit I believe: [ee1087e](https://github.com/chan-sccp/chan-sccp/commit/ee1087e656f4561419f68506c9b86f587dfd7947)

Aah ok, super. Then we need to fix the configure script, to search for gawk instead of awk/mawk. This will be a little annoying on other platforms like BSD though.

We can have another look at making the script generic awk compatible later.

snuffy44, 2019-10-11:
mm wouldn't doing just [0-9a-fA-F] work instead of xdigit?

dkgroot, 2019-10-11:
I think you are right. 
Can you give that a go ? And see if it works with standard awk/mawk ?

snuffy44, 2019-10-11:
changing line 27 to  /^0[xX][0-9A-Fa-f]+$/ works fine..
Have actual values again.

dkgroot, 2019-10-11:
Looks like your fix works. Will make the change in a couple of seconds.

FYI: It's so easy to fall for these nice little `enhancements` provided by some tools, and they slowly lure you away from the minimal standards.

dkgroot, 2019-10-11:
Pushed your fix ! Thanks a lot

commit 2b92e89a (2019-10-11): Fix enum generation by making it standards compatible again (mawk/awk and gawk)

Solution provided by @snuffy44

Fixes #485
closed 2019-10-11 by dkgroot