chan-sccp/chan-sccp #486: CallForward by timeout

bknv, 2019-10-17:
Hello all.
I did not find the option to enable forwarding by timeout. Is it possible to implement this in the module chan-sccp in CLI?

Right now I'm doing it like this, i added options and enable forward in module followme:

Enable:
database put AMPUSER/2327/followme grplist  89123456798# - mobile number
database put AMPUSER/2327/followme prering  10 - timeout callforward
database put AMPUSER/2327/followme strategy ringall - Ring Strategy
database put AMPUSER/2327/followme  grptime 30 - Ring Time
database put AMPUSER/2327/followme ddial DIRECT - Enable followme/forward

Disable:
database put AMPUSER/2327/followme ddial EXTENSION

dkgroot, 2019-10-17:
I created a branch for this to work on it, called "feature/cfwd_noanswer"

At the moment it works, but there is still an issue with a asterisk channel being left behind, for every forwarded call, so that is not right yet.

***Git***
_Checkout this temporary branch_
```
git pull
git fetch -a
git branch
git checkout feature/cfwd_noanswer
./configure [....]
make -j4 && make install && make reload
```

_move back to develop afterwards:_
```
git checkout develop
```

Note: the asterisk channel being left behind causes trouble wit unloading/loading modules and restarting asterisk. So if you do want to give it a go, do this only on a non production server.

dkgroot, 2019-10-17:
Note: Because this is an implementation of a new functionality, please take care when testing. If you have a test server (instance) than that would be a good place.

Addition to sccp.conf:
```
[general]
cfwdnoanswer_timeout = 30     ; default value = 30
```
I have added `cfwdall`, `cfwdbusy` and `cfwdnoanswer` to the `onhook`/`offhook` softkeyset entries. 

**Warning: Do not use in production yet.**

dkgroot, 2019-10-21:
Quote: @bknv 

> @dkgroot Hello. Yes I have Virtual Machine for test. Today i perfomed next step:
> 1) git pull
> git fetch -a
> git branch
> git checkout feature/cfwd_noanswer
> ./configure
> make -j4 && make install && make reload
> -- And restart core asterisk
> 
> 2) In sccp.conf adding cfwdnoanswer_timeout = 9
> 
> 3) In CLI adding : sccp callforward 2883 cfwdnoanswer 2881
> Set/Unset CallForward to cfwdNoAnswer:
> 
>     on line:2883 and device:SEP001B54350F95
> 
> 4) Test calling on 2883, but after 9 sec, callforward no worked.
> Did I do everything right?

It looks correct.
Did you check if the callforward was actually performed (sccp show line 2883). I set the cfwdnoanswer on the phone.
If you set 'sccp debug core, channel, pbx' you should be able to see that the cfwdnoanswer is being schedule while the call is ringing, and later the scheduled command should be executed.
On thing is important, namely that 'DIAL(xxx, timeout, xxx) outlasts the cfwdnoanswer_timeout.

anutator, 2019-10-22:
4471 never activated call forward. He sad he had call on second line, he answered second line and then tried to come back to the first line. But softkey was "not assigned".

dkgroot, 2019-10-24:
> In CLI adding : sccp callforward 2883 cfwdnoanswer 2881

should have been:
 `sccp callforward 2883 noanswer 2881`
or
`sccp callforward 2883 SEPxxxxx noanswer 2881`

dkgroot, 2019-10-25:
wiki pages to check/update
https://github.com/chan-sccp/chan-sccp/wiki/Button-Configuration
https://github.com/chan-sccp/chan-sccp/wiki/CLI-Functions
https://github.com/chan-sccp/chan-sccp/wiki/sccp.conf-options
https://github.com/chan-sccp/chan-sccp/wiki/Softkeys

dkgroot, 2019-10-25:
Branch "feature/cfwd_noanswer" has been merged back into the "develop" branch, and will be deleted shortly.

@bknv thanks you for reporting the feature requirement and checking/testing,

But please do check our wiki and see if there is something that needs a little work.

Closing the issue report for now.

closed 2019-10-25 by dkgroot