chan-sccp/chan-sccp #487: Caller ID only sending cid_num field

tabbertmj, 2019-10-26:
When calling another SCCP extension, The called extension number shows as caller ID if cid_num is set to the extension number.  If I is set to a name and number, it displays properly.
I am using the realtime configuration with MySQL.  This is a new build.  I have not run an aterisk box for several years so I am not sure when this started.

Skinny Client Control Protocol (SCCP). Release: 4.3.2 develop - b72d050 
Asterisk 13.27.1

-- 25: New channel number: 35 on line 25
    -- SCCP: (pbx_channel_allocate) try to allocate Outbound channel on line: 25
    -- SCCP:             cid_num: 25
    -- SCCP:            cid_name: Mike T
.
.
.
-- SCCP/32-00000024: SCCP/32-00000024 : CallInfo:
        |- calledParty: Mike 7961 <Mike 7961 x32>, valid
        |- callingParty:  <32>, valid
        |- presentation: CallerId Presentation Allowed

It seems that the callingParty should be the originating cid_name and cid_num.
Here is the full output from the console.

https://pastebin.com/0c5LSrdT

dkgroot, 2019-10-27:
@tabbertmj 

Is the phones display information also wrong ? Or are you just looking at this bit of the logging ?

Looking at your debug:
Outgoing:
```
    -- SCCP: (pbx_channel_allocate) try to allocate Outbound channel on line: 25
    -- SCCP:             cid_num: 25
    -- SCCP:            cid_name: Mike T
    -- SCCP:         accountcode: 25
    -- SCCP:               exten:
    -- SCCP:             context: from-internal
    -- SCCP:            amaflags: 0
    -- SCCP:           chan/call: SCCP/25-00000023
```
Incoming side:
```
-  SCCP: Adding channel 36 to line 32
    -- 32: New channel number: 36 on line 32
    -- SCCP: (pbx_channel_allocate) try to allocate Inbound channel on line: 32
    -- SCCP:             cid_num: Mike 7961 x32
    -- SCCP:            cid_name: Mike 7961
    -- SCCP:         accountcode: 32
    -- SCCP:               exten:
    -- SCCP:             context: from-office
    -- SCCP:            amaflags: 0
    -- SCCP:           chan/call: SCCP/32-00000024
...
    -- SCCP/32: Incoming calls:0, Incoming calls limit: 2
    -- SCCP/32-00000024: SCCP/32-00000024 : CallInfo:
        |- calledParty: Mike 7961 <Mike 7961 x32>, valid
        |- callingParty:  <32>, valid
        |- presentation: CallerId Presentation Allowed
```
calledParty on the incoming side: is the incoming side
callingParty on the incoming side: is the incoming extension from the dialplan (at this moment)
Once we get connected_line update, this will change when we know the remote party exists and can be queried.
You need to set: "sccp debug core, channel, pbx, indicate, callinfo"
To see all callinfo updates to get the actual picture.

tabbertmj, 2019-10-27:
It is displayed wrong on the phone.  It shows as coming from it's own extension.

Here is the output from the debug.
https://pastebin.com/a3ixYhPe

Everything is correct on the outbound portion of the call starting on line 106.
When the new channel is created starting on line 116 it has lost the calling party information.  

Thanks.

dkgroot, 2019-10-27:
@tabbertmj Ooh that's not good. I will check to see if i can reproduce the issue and see what needs to be altered to fix. Thanks for reporting the issue.

rofferman, 2019-10-28:
This sounds kind of like the issue I am also having

tabbertmj, 2019-10-28:
I can open another issue if needed, but when a call is ongoing between two SCCP endpoints, should it be the "connected" softkeys showing?  Currently it displays the list from "conntrans".

Thanks,
Mike.

dkgroot, 2019-10-28:
@tabbertmj 
I am seeing this : SEPE0D1 with line 98031 calls SEP0023 on line 98041
```
moviebox*CLI> sccp debug core channel pbx callinfo indicate 
SCCP new debug status: (0 -> 2103361) core,channel,indicate,pbx,callinfo
    -- SEPE0D173E11D95: Got stimulus=Line (9) for instance=1, callreference=0, status=0
    -- SEPE0D173E11D95: Line Key press on line 98031
...
    -- SCCP: (pbx_channel_allocate) try to allocate Outbound channel on line: 98031
    -- SCCP/98031-00000001: (sccp_callinfo_setter) : CallInfo:
        |- callingParty: Diederik-Phone3 <98031>, valid
        |- lastRedirectingParty: Diederik-Phone3 <98031>, valid, reason: 0
        |- presentation: CallerId Presentation Allowed

    -- SCCP:             cid_num: 98031
    -- SCCP:            cid_name: Diederik-Phone3
    -- SCCP:         accountcode: 98031
    -- SCCP:               exten: 
    -- SCCP:             context: internal
    -- SCCP:            amaflags: 0
    -- SCCP:           chan/call: SCCP/98031-00000001
...
    -- 2: Allocated asterisk channel SCCP/98031-00000001
...
    -- SEPE0D173E11D95: (sccp_pbx_softswitch) New call on line 98031
    -- SEPE0D173E11D95: (sccp_pbx_softswitch) Dial Extension 98041
    -- SEPE0D173E11D95: Indicate SCCP new state:DIALING, current channel state:OFFHOOK on call:SCCP/98031-00000001, lineInstance:1 (previous channelstate:DOWN)
    -- SCCP/98031-00000001: (sccp_callinfo_setter) : CallInfo:
        |- calledParty:  <98041>, valid
        |- callingParty: Diederik-Phone3 <98031>, valid
        |- lastRedirectingParty: Diederik-Phone3 <98031>, valid, reason: 0
        |- presentation: CallerId Presentation Allowed

    -- SEPE0D173E11D95: Send the dialed number:98041, callid:1, lineInstance:1
...
 SCCP: (sccp_pbx_softswitch) exit
    -- Executing [98041@internal:1] NoOp("SCCP/98031-00000001", "98041") in new stack
    -- Executing [98041@internal:2] NoOp("SCCP/98031-00000001", "internal context called on Server B to 98041") in new stack
    -- Executing [98041@internal:3] Set("SCCP/98031-00000001", "CHANNEL(video)=auto") in new stack
  == SCCP/98031-00000001: (setVideoMode) Setting Video Mode to Auto
    -- Executing [98041@internal:4] Dial("SCCP/98031-00000001", "SCCP/98041,20,R") in new stack
    -- SCCP: Asterisk asked us to create a channel with type=SCCP, format=(h264|g722), lineName=98041, options=
 SCCP: callinfo constructor: 0x7f3064017830
    -- 0003: New channel number: 2 on line 98041
    -- SCCP: (pbx_channel_allocate) try to allocate Inbound channel on line: 98041
    -- SCCP/98041-00000002: (sccp_callinfo_setter) : CallInfo:
        |- calledParty: PHONE4 <98041>, valid
        |- presentation: CallerId Presentation Allowed

    -- SCCP:             cid_num: 98041
    -- SCCP:            cid_name: PHONE4
    -- SCCP:         accountcode: 79005
    -- SCCP:               exten: 
    -- SCCP:             context: internal
    -- SCCP:            amaflags: 0
    -- SCCP:           chan/call: SCCP/98041-00000002
...
    -- SCCP: (allocPBXChannel) Create New Channel with name: SCCP/98041-00000002
    -- 0003: Allocated asterisk channel SCCP/98041-00000002
    -- SCCP/98041-00000002: (sccp_callinfo_setter) : CallInfo:
        |- calledParty: PHONE4 <98041>, valid
        |- callingParty: Diederik-Phone3 <98031>, valid
        |- presentation: CallerId Presentation Allowed

 ...
 SEPE0D173E11D95: (get_rtp_info) Asterisk requested EarlyRTP peer for channel SCCP/98031-00000001
 SEPE0D173E11D95: (get_vrtp_info) Asterisk requested EarlyRTP peer for channel SCCP/98031-00000001
    -- SCCP: Asterisk request to call SCCP/98041-00000002 (dest:98041, timeout: 0)
    -- 98041: Asterisk request to call SCCP/98041-00000002
    -- SCCP/98041: Incoming calls:0, Incoming calls limit: 10
    -- SCCP: (sccp_pbx_call) asterisk callerid='Diederik-Phone3 <98031>'
    -- SCCP/98041-00000002: SCCP/98041-00000002 : CallInfo:
        |- calledParty: PHONE4 <98041>, valid
        |- callingParty: Diederik-Phone3 <98031>, valid
        |- presentation: CallerId Presentation Allowed

    -- SEP0023043403F9: Getting the active channel on device.
```

tabbertmj, 2019-10-28:
I am running FreePBX.  Version 14.0.13.6.

Strange, your's appears fine.  

sccp.conf is here.
https://pastebin.com/6aibQX9A

dkgroot, 2019-10-28:
FreePBX dialplan is really complex and convoluted, so i am not 100% what it is doing to cause this.

dkgroot, 2019-10-28:
> I can open another issue if needed, but when a call is ongoing between two SCCP endpoints, should it be the "connected" softkeys showing? Currently it displays the list from "conntrans".
> 
> Thanks,
> Mike.

`Conntrans` is used when we have a connected call that is transferable. When we are in transfer, we switch to `OffhookFeat` (bit of a misnomer, but helps 69xx phones, notice that we are in a transfer). Note the newer phones like 69xx and 89xx have altered the way we have to do things quite a bit. The old phone would just do what the server tells them to do. The new phones expect certain events to have a preset order and they update part of the display and handling by the selves.
We used to have `Connected` during a normal call, and switch to `ConnTrans` during transfer.

tabbertmj, 2019-10-28:
Here is the full debug output.  
https://pastebin.com/QCK9Se3M

As for the settings in the sccp.conf, I used this as a reference.
https://github.com/chan-sccp/chan-sccp/wiki/Realtime-Configuration

For the softkeys, I have not worked with these phones, (79xx), or Asterisk in a few years.  It's not like riding a bicycle.  I'll try to update my softkey sets appropriately for  Conntrans.
I see this, https://github.com/chan-sccp/chan-sccp/wiki/Softkeys,
I think what might also be helpful is to list the phone states for the ones that are not intuitive.  Onhook, or offhook, easy to guess, Conntrans, I wasn't sure what state that was.  But it makes sense now.

Let me know if there are other captures or files you need.

dkgroot, 2019-10-28:
> As for the settings in the sccp.conf, I used this as a reference.
> https://github.com/chan-sccp/chan-sccp/wiki/Realtime-Configuration

Ooh my bad, I will fix that in a minute :-) Thanks for pointing it out to me,
I will have a look at your debug output in a little bit.

If you are only using 79xx phone, you can remove all the softkeysets and just use the defaults normally.

dkgroot, 2019-10-28:
Had a little peek already:
And we now can be sure it is freepbx messing with the caller id info:
```
[2019-10-28 19:17:07] DEBUG[17518][C-0000000a]: pbx.c:2893 pbx_extension_helper: Launching 'ExecIf'
    -- Executing [s@macro-user-callerid:4] ExecIf("SCCP/25-00000043", "1?Set(REALCALLERIDNUM=25)") in new stack
[2019-10-28 19:17:07] DEBUG[17518][C-0000000a]: app_macro.c:472 _macro_exec: Executed application: ExecIf
[2019-10-28 19:17:07] DEBUG[17518][C-0000000a]: pbx.c:2893 pbx_extension_helper: Launching 'Set'
    -- Executing [s@macro-user-callerid:5] Set("SCCP/25-00000043", "AMPUSER=25") in new stack
[2019-10-28 19:17:07] DEBUG[17518][C-0000000a]: app_macro.c:472 _macro_exec: Executed application: Set
[2019-10-28 19:17:07] DEBUG[17518][C-0000000a]: pbx_variables.c:844 pbx_substitute_variables_helper_full: Expression result is '0'
[2019-10-28 19:17:07] DEBUG[17518][C-0000000a]: pbx.c:2893 pbx_extension_helper: Launching 'GotoIf'
    -- Executing [s@macro-user-callerid:6] GotoIf("SCCP/25-00000043", "0?limit") in new stack
[2019-10-28 19:17:07] DEBUG[17518][C-0000000a]: pbx_builtins.c:1180 pbx_builtin_gotoif: Not taking any branch
[2019-10-28 19:17:07] DEBUG[17518][C-0000000a]: app_macro.c:472 _macro_exec: Executed application: GotoIf
[2019-10-28 19:17:07] DEBUG[17518][C-0000000a]: pbx.c:2893 pbx_extension_helper: Launching 'Set'
    -- Executing [s@macro-user-callerid:7] Set("SCCP/25-00000043", "AMPUSERCIDNAME=Mike T") in new stack
```
Not saying they are doing anything wrong. It is just hard to deal with all these possible dialplan interactions from a channel perspective. Will come back to this in 1-2 hours and see if there is someting simple we can do (appart from adding a custom dialplan entry in freepbx :-))

dkgroot, 2019-10-28:
Can you have a look in freepbx to see if there is a way, not to set this cid information on this freepbx extension ? Maybe removing 'REALCALLERID` etc might fix things.

tabbertmj, 2019-10-28:
Ok, let me look to find out how to remove it.

dkgroot, 2019-10-28:
```
    -- Executing [s@macro-user-callerid:12] Set("SCCP/25-00000043", "CALLERID(all)="Mike T" <25>") in new stack
[2019-10-28 19:17:07] DEBUG[17518][C-0000000a]: app_macro.c:472 _macro_exec: Executed application: Set
[2019-10-28 19:17:07] DEBUG[17518][C-0000000a]: pbx_variables.c:844 pbx_substitute_variables_helper_full: Expression result is '1'
[2019-10-28 19:17:07] DEBUG[17518][C-0000000a]: pbx_variables.c:844 pbx_substitute_variables_helper_full: Expression result is '0'
[2019-10-28 19:17:07] DEBUG[17518][C-0000000a]: pbx.c:2893 pbx_extension_helper: Launching 'ExecIf'
    -- Executing [s@macro-user-callerid:13] ExecIf("SCCP/25-00000043", "1?Set(CALLERID(all)=31)") in new stack
[2019-10-28 19:17:07] DEBUG[17518][C-0000000a]: app_macro.c:472 _macro_exec: Executed application: ExecIf
[2019-10-28 19:17:07] DEBUG[17518][C-0000000a]: pbx_variables.c:844 pbx_substitute_variables_helper_full: Expression result is '0'
[2019-10-28 19:17:07] DEBUG[17518][C-0000000a]: pbx.c:2893 pbx_extension_helper: Launching 'GotoIf'
```
That's really confusing, first they set it to "Mike T <25>" and then in the next line they set it to "31", that looks just wrong to me.

tabbertmj, 2019-10-28:
It doesn't do that if I have text for the cid_num field.  The problem with that workaround is when you hit the messages button, the extension doesn't match and you have to enter it.

dkgroot, 2019-10-28:
Then that is not a good work around :-). We should be able to add some debugging on the chan-sccp side to see what callerid information we are receiving on an incoming call, maybe we can figure out what they are trying to tell us that way. In the meantime you might want to have a look at the freepbx manual to see if there is any way to suppress changing the callerid on incoming calls.

BTW: Updated the wiki, thanks for the hint.

tabbertmj, 2019-10-28:
I hear you.  It makes it easy to get started, but they manipulate things a bit too much. 
I think it is odd that they are defaulting to the called number if only numbers are present.  I'm not sure what they were trying to fix wit that.

dkgroot, 2019-10-28:
Yes they suck you in and then you are left to fend for your self, without knowing anything about how asterisk actually works. I thought that looked really confusing too.

tabbertmj, 2019-10-28:
I see that musicclass is also not being handled properly from the database.  Line 25 is set to Office.  If it is set to default, it handles it fine.  

-- Executing [s@macro-dial-one:40] ExecIf("SCCP/25-00000054", "0?Set(CHANNEL(musicclass)=)") in new stack

dkgroot, 2019-10-28:
@tabbetmj / @rofferman 

I added a little debug output during initial call setup, to find out what FreePBX has changed in the meantime. This has been put separate branch (see below)

You need to set `sccp debug core, channel, pbx, callinfo` during debug  + `core set debug 1` and `core set verbose 5`

***Git***
_Checkout this temporary branch_
```
git pull
git fetch -a
git branch
git checkout issues/487
./configure [....]
make -j4 && make install && make reload
```

_move back to develop afterwards:_
```
git checkout develop
```
give it a try and post the new result please.

Using my dialplan, this looks like:
```
 SCCP: (sccp_pbx_softswitch) exit
    -- Executing [98031@internal:1] NoOp("SCCP/98041-00000002", "98031") in new stack
    -- Executing [98031@internal:2] NoOp("SCCP/98041-00000002", "internal context called on Server B to 98031") in new stack
    -- Executing [98031@internal:3] Set("SCCP/98041-00000002", "CHANNEL(video)=auto") in new stack
  == SCCP/98041-00000002: (setVideoMode) Setting Video Mode to Auto
    -- Executing [98031@internal:4] Dial("SCCP/98041-00000002", "SCCP/98031,20") in new stack
[Oct 28 23:09:00] NOTICE[20578][C-00000000]: app_dial.c:2583 dial_exec_full: SCCP/98041-00000002: (app_dial) NativeAudioFormats:(g722)
    -- SCCP: Asterisk asked us to create a channel with type=SCCP, format=(g722), lineName=98031, options=
    -- 2: New channel number: 3 on line 98031
    -- SCCP: (pbx_channel_allocate) try to allocate Inbound channel on line: 98031
    -- SCCP:             cid_num: 98031
    -- SCCP:            cid_name: Diederik-Phone3
    -- SCCP:         accountcode: 98031
    -- SCCP:               exten: 
    -- SCCP:             context: internal
    -- SCCP:            amaflags: 0
    -- SCCP:           chan/call: SCCP/98031-00000003
    -- SCCP: combined audio caps: [slin16, g722/64k, ulaw/64k, alaw/64k, g722/56k, g722/48k, ilbc, g729b, g729ab, g729, g729a]
    -- SCCP: reduced audio prefs: [g722/64k, g722/56k, g722/48k, ulaw/64k, alaw/64k, g729, g729a, g729b, g729ab]
    -- SCCP: combined video caps: [h264]
    -- SCCP: reduced video prefs: [h264]
    -- SCCP: (allocPBXChannel) Create New Channel with name: SCCP/98031-00000003
    -- 2: Allocated asterisk channel SCCP/98031-00000003
SCCP/98031-00000003: (sccp_astwrap_request) Before Connected Line Update:
	- Requestor: 1572300540.1
	- CallingParty: PHONE4 <98041>
	- DialedParty : 98031
	- ConnectedParty : (N/A) <(N/A)>
	- RedirectingParty : (N/A) <(N/A)>
SCCP/98031-00000003: (sccp_astwrap_request) After Forced Connected Line Update:
	- Requestor: 1572300540.1
	- CallingParty: PHONE4 <98041>
	- DialedParty : 98031
	- ConnectedParty : (N/A) <(N/A)>
	- RedirectingParty : (N/A) <(N/A)>
 SCCP: (get_rtp_info) Asterisk requested EarlyRTP peer for channel SCCP/98031-00000003
 SCCP: (get_vrtp_info) Asterisk requested EarlyRTP peer for channel SCCP/98031-00000003
 SEP0023043403F9: (get_rtp_info) Asterisk requested EarlyRTP peer for channel SCCP/98041-00000002
 SEP0023043403F9: (get_vrtp_info) Asterisk requested EarlyRTP peer for channel SCCP/98041-00000002
    -- SCCP: Asterisk request to call SCCP/98031-00000003 (dest:98031, timeout: 0)
    -- 98031: Asterisk request to call SCCP/98031-00000003
    -- SCCP/98031: Incoming calls:0, Incoming calls limit: 3
    -- SCCP: (sccp_pbx_call) asterisk callerid='PHONE4 <98041>'
    -- SCCP/98031-00000003: SCCP/98031-00000003 : CallInfo:
        |- calledParty: Diederik-Phone3 <98031>, valid
        |- callingParty: PHONE4 <98041>, valid
        |- presentation: CallerId Presentation Allowed
```

The added output begins with:
`SCCP/98031-00000003: (sccp_astwrap_request) Before Connected Line Update:`
and
`SCCP/98031-00000003: (sccp_astwrap_request) After Forced Connected Line Update:`

dkgroot, 2019-10-28:
> I see that musicclass is also not being handled properly from the database. Line 25 is set to Office. If it is set to default, it handles it fine.
> 
> -- Executing [s@macro-dial-one:40] ExecIf("SCCP/25-00000054", "0?Set(CHANNEL(musicclass)=)") in new stack

Which database are your talking about, the asterisk realtime db (mysql/postgresql) or the astdb (CLI: `database show`) ? If the dialplan (FreePBX) pushes/presses a new musicclass we should follow that during the the current call. We will however use the musicclass defined in sccp.conf/realtime if nothing is set during dialplan execution or fallback to the asterisk default.

tabbertmj, 2019-10-29:
Here is the latest output.
https://pastebin.com/6eTZz7M9

Thanks!

dkgroot, 2019-10-29:
Thanks: That's just messed up:
```
    -- 31: Allocated asterisk channel SCCP/31-00000002
SCCP/31-00000002: (sccp_astwrap_request) Before Connected Line Update:
        - Requestor: 1572315757.0
        - CallingParty: (N/A) <31>
        - DialedParty : 31
        - ConnectedParty : Mike Test <31>
        - RedirectingParty : (N/A) <(N/A)>
SCCP/31-00000002: (sccp_astwrap_request) After Forced Connected Line Update:
        - Requestor: 1572315757.0
        - CallingParty: (N/A) <31>
        - DialedParty : 31
        - ConnectedParty : Mike Test <31>
        - RedirectingParty : (N/A) <(N/A)>
    -- SCCP/31-00000002: (sccp_callinfo_setter) : CallInfo:
        |- calledParty: Mike Test <Mike Test x31>, valid
        |- callingParty:  <31>, valid
        |- presentation: CallerId Presentation 
```
I am getting this information directly from asterisk not from chan_sccp. It/They broke their own calling party as far as i can tell. I have altered the patch (ie: `git pull` please on `issues/487`) to get a view even earlier in time, to see if we are the ones or freepbx.

dkgroot, 2019-10-29:
If this last update also shows the same state, then there is not much we can do (I would not know what). If it shows the correct data, then chan_sccp is messing up, and we can find out where.

tabbertmj, 2019-10-29:
I have an error when running the make command.

https://pastebin.com/5Yph7sx3

dkgroot, 2019-10-29:
Woops. i messed up. Pushed a fix. sorry

tabbertmj, 2019-10-29:
No worries.  Here is the latest output.

https://pastebin.com/XHaXMvg2

dkgroot, 2019-10-29:
Ok i have some sad news:
```
    -- Executing [s@macro-dial-one:54] Dial("SCCP/25-00000001", "SCCP/31,15,HhTtrIb(func-apply-sipheaders^s^1)") in new stack
    -- SCCP: Asterisk asked us to create a channel with type=SCCP, format=(ulaw), lineName=31, options=
SCCP: (sccp_astwrap_request) Before chan-sccp incoming channel is created:
        - Requestor: 1572358011.13
        - CallingParty: (N/A) <31>
        - DialedParty : 31
        - ConnectedParty : Mike Test <31>
        - RedirectingParty : (N/A) <(N/A)>
```
This is the information available in asterisk when we receive notification of an upcoming incoming call. This is even before the create the sccp channel plus it's callinfo structure. Before this moment we do 0 manipulation of the callerid information. So this is what is left after the sccp outgoing call and the dialplan 'mangling' by freepbx. Zero usefull information about the actual caller i am sorry to say. I guess they will correct that once they start ringing the channel or when it is connected. But on the initial incoming callside we are going to display the wrong information with what is provided.
```
 SCCP: callinfo constructor: 0x7f8cc8022380
 SCCP: Adding channel 2 to line 31
    -- 31: New channel number: 2 on line 31
    -- SCCP: (pbx_channel_allocate) try to allocate Inbound channel on line: 31
    -- SCCP/31-00000002: (sccp_callinfo_setter) : CallInfo:
        |- calledParty: Mike Test <Mike Test x31>, valid
        |- presentation: CallerId Presentation Allowed
```

With vanilla asterisk (ie with zero changing of any of the caller infomation) we do the right thing. I don't think there is much we can do in this case.

Maybe ask PhantomVl (developer of the sccp_manager plugin) on the [gitter.im](https://gitter.im/sccp_manager/community) chat if he is seeing the same thing, or not. If not he might have some different settings, that correct this behaviour. It might also be that there has been a regression in chan_sccp in recent time, that is having this side effect (ie: no hangling of later connected_line updates or something).

dkgroot, 2019-10-30:
@tabbertmj I see in the [sccp_manager chat](https://gitter.im/sccp_manager/community) that @PhantomVl is not able to reproduce the behaviour you are seeing. Maybe the two of you can figure out where the difference between your systems is.
Please close this issue when the problem is resolved. I will keep the changes made in the `issue/487` branch, but remark them out, so they can be used if ever necessary.

dkgroot, 2019-11-05:
See updates on #473

ErmakovDmitriy, 2019-11-14:
Hi,
I believe I have the same problem with my FreePBX distro installation (FreePBX 14).
When somebody calls affected phone incoming side sees only his/her phone number but not outgoing's number.

I use:
```
[root@kdast03 dev]# asterisk  -rx 'core show version'
Asterisk 13.27.1 built by mockbuild @ jenkins7 on a x86_64 running Linux on 2019-09-06 21:11:59 UTC
[root@kdast03 dev]# asterisk  -rx 'sccp show  version'
Skinny Client Control Protocol (SCCP). Release: 4.3.2 develop - 1c5cc89 (built by 'root' on '2019-11-05 11:52:58 UTC')

```
and static files configuration.
In FreePBX interface I just create "Extension => Other Extension" and type into dial string "SCCP/146".

I have phones which are the same model Cisco 7911 and have the same firmware version (<loadInformation>SCCP11.9-1-1SR1S</loadInformation>). One of them is affected by the problem (line 146) while another is not. I can provide more information if you need it.

Below are some logs and statuses:
[sccp.conf](https://pastebin.com/6mw3rR6Z)
[call log](https://pastebin.com/sj9ks2rL)
[affected device status](https://pastebin.com/cNmyEj5T)

Thank you for your help.

dkgroot, 2019-11-14:
@ErmakovDmitriy  Have a look at #473, it contains a possible solution to the issue toward the end

ErmakovDmitriy, 2019-11-14:
Thank you
I did as stoyanmarinov said [](https://github.com/chan-sccp/chan-sccp/issues/473#issuecomment-549908171) before sending the information.

I will try to fix the php code in FreePBX as mentioned here [](https://github.com/chan-sccp/chan-sccp/issues/473#issuecomment-549882633) tomorrow and reload the PBX (I can not reload it now).

dkgroot, 2019-11-14:
Ooh ok, I don't know if it is directly related or not, so we will see tomorrow, if more is needed. Thanks for checking it out !

ErmakovDmitriy, 2019-11-15:
Today I applied fix to macro-dial-one.php and reloaded FreePBX.
I also added Verbose output of CALLERID(all) and make a test call.

Here is the call full log [call-180-to-146.log](https://pastebin.com/GLzxC4df).
The call was made from line 180 to line 146 so I think CALLERID must be <username> 180.

Here is caller id output:
```
    -- Executing [s@macro-dial-one:55] Verbose("SCCP/180-00000024", "Caller ID: "" <146>") in new stack
Caller ID:  <146>
```
As I can see there is something strange with CALLERID value.
I will try to read dialplan to get some information about the problem.

Could you take a look at the logs?

ErmakovDmitriy, 2019-11-15:
I finally found the problem.
It was with FreePBX dialplan (sorry for disturbing you here).

There is a macro [macro-user-callerid] in extensions_additional.conf.
It contains a string:
`exten => s,n,ExecIf($["${DB(DEVICE/${ARG2}/tech)}"="custom"]?Set(CALLERID(all)=${IF($[${LEN(${DB(AMPUSER/${ARG2}/outboundcid)})}]?${DB(AMPUSER/${ARG2}/outboundcid)}:${ARG2})}))`

which sets CALLERID for all devices with "custom" tech in Asterisk DB to ${ARG2} (and I do not know why it is actually CALLED ID).

For my situation:
`
asterisk -rx 'database show' | grep custom | grep 146               
/DEVICE/146/tech                                  : custom     
`
The record in database with /DEVICE/<number>/tech == custom is added on creating a new extension or update of existing. Because I had performed update from FreePBX 12 to 14 two months before the problem I have some extensions with the record "tech==custom" set and some without it. It confused me because of strange behavior (some phones have the problem and others do not have).

The string is generated by line 44 in modules/core/functions.inc/macro-user-callerid.php of FreePBX.
Here is a [commit which added the line](https://git.freepbx.org/projects/FREEPBX/repos/core/diff/functions.inc/macro-user-callerid.php?until=60eec6b5e34b88459c2407e4c5093a08c6e16bbf).

Initially the line was added on 02 May 2019 by [commit](https://git.freepbx.org/projects/FREEPBX/repos/core/diff/functions.inc/macro-user-callerid.php?until=8f7a73c64c0ad3536567d6eeb1f19b6c096fff83).

I have to comment the affecting line in my `/var/www/html/admin/modules/core/functions.inc/macro-user-callerid.php`:
`// commented by demonihin to fix CallerID  $ext->add($context, $exten, '', new ext_execif('$["${DB(DEVICE/${ARG2}/tech)}"="custom"]', 'Set', 'CALLERID(all)=${IF($[${LEN(${DB(AMPUSER/${ARG2}/outboundcid)})}]?${DB(AMPUSER/${ARG2}/outboundcid)}:${ARG2})}'));`.

I hope the information might help somebody.

I will try to open a bug at FreePBX. Should I attach a URL to the bug here, if I open it?

dkgroot, 2019-11-15:
@ErmakovDmitriy 
Thank you so much for finding and solving the problem !
Please do report the issue to FreePBX.
I don't mind if you include a link to the issue here, but i am not sure if it will help.

dkgroot, 2019-11-15:
@tabbertmj Can you replicate the solution ?

tabbertmj, 2019-11-16:
> @tabbertmj Can you replicate the solution ?

I will try to test it tomorrow.

tabbertmj, 2019-11-16:
That resolved it!  It also resolves calling from one context to another with a custom device showing a CID assigned to a trunk.

dkgroot, 2019-11-16:
@tabbertmj Thanks for reporting back. 
@ErmakovDmitriy Thank you so much for figuring this out and helping to resolve the issue.

Can you make a small announcement to the sccp_mananger user community on gitter, about the issue and it's resolution ? Any ideas about what else to ? Should we close the issue, or leave it open for others to find ?

ErmakovDmitriy, 2019-11-19:
I do not mind closing the issue because the problem was solved for my installation. I commented closed bug at [FreePBX.org](https://issues.freepbx.org/browse/FREEPBX-19451) so I hope the FreePBX developers will fix it.

Thank you for your help!

dkgroot, 2019-11-19:
Ok let's close it for now, and i will try to remember that it was fixed (my memory is hopeless in this regard :-))

Note:
- chan_sccp issue is closed
- FreePBX issue: [19451](https://issues.freepbx.org/browse/FREEPBX-19451)

@ErmakovDmitriy In the FreePBX issue they asked you to open a new jira issue maybe look into that !

closed 2019-11-19 by dkgroot

ErmakovDmitriy, 2019-11-19:
I am sorry, I have not updated my response.
Here is a new issue: [FREEPBX-20805](https://issues.freepbx.org/browse/FREEPBX-20805) (just for information).

billsimon, 2020-06-30:
Hey, I think this is fixed on FreePBX with https://issues.freepbx.org/browse/FREEPBX-21582 - for FreePBX 15 at least.

dkgroot, 2020-07-01:
@billsimon Thanks for closing this open issue