chan-sccp/chan-sccp #488: error compiling develop branch

rofferman, 2019-11-04:
I get the following error when I compile the latest develop branch
Freepbx 15.0.16.22
Asterisk 16.6.0
CentOS 7

make -j4
Making all in src
make[1]: Entering directory `/usr/src/chan_sccp/src'
make  all-recursive
make[2]: Entering directory `/usr/src/chan_sccp/src'
Making all in pbx_impl
make[3]: Entering directory `/usr/src/chan_sccp/src/pbx_impl'
make[3]: Nothing to be done for `all'.
make[3]: Leaving directory `/usr/src/chan_sccp/src/pbx_impl'
Making all in pbx_impl/ast
make[3]: Entering directory `/usr/src/chan_sccp/src/pbx_impl/ast'
  CC       libast_la-ast.lo
In file included from ../../../src/common.h:34:0,
                 from ast.c:12:
ast.c: In function âget_sccp_channel_from_pbx_channelâ:
../../../src/define.h:180:2: error: â__auto_typeâ undeclared (first use in this function)
  __auto_type __attribute__((unused)) __dummy = (_T)(_ptr); \
  ^
../../../src/sccp_refcount.h:75:13: note: in expansion of macro âPTR_TYPE_CMPâ
  pbx_assert(PTR_TYPE_CMP(const _type *const, (_x) ) == 1);           \
             ^
../../../src/sccp_channel.h:17:34: note: in expansion of macro âsccp_refcount_retain_typeâ
 #define sccp_channel_retain(_x)  sccp_refcount_retain_type(sccp_channel_t, _x)
                                  ^
ast.c:579:11: note: in expansion of macro âsccp_channel_retainâ
    return sccp_channel_retain(c);
           ^
../../../src/define.h:180:2: note: each undeclared identifier is reported only once for each function it appears in
  __auto_type __attribute__((unused)) __dummy = (_T)(_ptr); \
  ^
../../../src/sccp_refcount.h:75:13: note: in expansion of macro âPTR_TYPE_CMPâ
  pbx_assert(PTR_TYPE_CMP(const _type *const, (_x) ) == 1);           \
             ^
../../../src/sccp_channel.h:17:34: note: in expansion of macro âsccp_refcount_retain_typeâ
 #define sccp_channel_retain(_x)  sccp_refcount_retain_type(sccp_channel_t, _x)
                                  ^
ast.c:579:11: note: in expansion of macro âsccp_channel_retainâ
    return sccp_channel_retain(c);
           ^
../../../src/define.h:180:14: error: expected â;â before â__attribute__â
  __auto_type __attribute__((unused)) __dummy = (_T)(_ptr); \
              ^
../../../src/sccp_refcount.h:75:13: note: in expansion of macro âPTR_TYPE_CMPâ
  pbx_assert(PTR_TYPE_CMP(const _type *const, (_x) ) == 1);           \
             ^
../../../src/sccp_channel.h:17:34: note: in expansion of macro âsccp_refcount_retain_typeâ
 #define sccp_channel_retain(_x)  sccp_refcount_retain_type(sccp_channel_t, _x)
                                  ^
ast.c:579:11: note: in expansion of macro âsccp_channel_retainâ
    return sccp_channel_retain(c);
           ^
In file included from ../../../src/common.h:49:0,
                 from ast.c:12:
ast.c: In function âsccp_astgenwrap_carefullHangupâ:
../../../src/define.h:180:2: error: â__auto_typeâ undeclared (first use in this function)
  __auto_type __attribute__((unused)) __dummy = (_T)(_ptr); \
  ^
../../../src/sccp_refcount.h:68:18: note: in definition of macro â__AUTO_RELEASE2â
  _type * _var = (_initial);                                                   \
                  ^
../../../src/sccp_refcount.h:71:78: note: in expansion of macro â__AUTO_RELEASE1â
 # define AUTO_RELEASE(_type, _var, _initial)                                 __AUTO_RELEASE1(_type, _var, _initial, __FILE__, __PRETTY_FUNCTION__, __LINE__, __COUNTER__)
                                                                              ^
ast.c:600:2: note: in expansion of macro âAUTO_RELEASEâ
  AUTO_RELEASE(sccp_channel_t, channel , sccp_channel_retain(c));
[... 101 lines cut ...]
                  ^
../../../src/sccp_refcount.h:71:78: note: in expansion of macro â__AUTO_RELEASE1â
 # define AUTO_RELEASE(_type, _var, _initial)                                 __AUTO_RELEASE1(_type, _var, _initial, __FILE__, __PRETTY_FUNCTION__, __LINE__, __COUNTER__)
                                                                              ^
ast.c:671:2: note: in expansion of macro âAUTO_RELEASEâ
  AUTO_RELEASE(sccp_channel_t, channel , sccp_channel_retain(c));
  ^
../../../src/sccp_refcount.h:75:13: note: in expansion of macro âPTR_TYPE_CMPâ
  pbx_assert(PTR_TYPE_CMP(const _type *const, (_x) ) == 1);           \
             ^
../../../src/sccp_channel.h:17:34: note: in expansion of macro âsccp_refcount_retain_typeâ
 #define sccp_channel_retain(_x)  sccp_refcount_retain_type(sccp_channel_t, _x)
                                  ^
ast.c:671:41: note: in expansion of macro âsccp_channel_retainâ
  AUTO_RELEASE(sccp_channel_t, channel , sccp_channel_retain(c));
                                         ^
ast.c: At top level:
cc1: warning: unrecognized command line option "-Wno-format-truncation" [enabled by default]
make[3]: *** [libast_la-ast.lo] Error 1
make[3]: Leaving directory `/usr/src/chan_sccp/src/pbx_impl/ast'
make[2]: *** [all-recursive] Error 1
make[2]: Leaving directory `/usr/src/chan_sccp/src'
make[1]: *** [all] Error 2
make[1]: Leaving directory `/usr/src/chan_sccp/src'
make: *** [all-recursive] Error 1

dkgroot, 2019-11-04:
@rofferman Aaah mann, is Centos using these lame old compilers. I will revert the change (for now). But centos should get the behind in motion and move toward the 21 century.

rofferman, 2019-11-04:
Ok, its acutally Sangoma OS lol. 
Can I install the correct compilers on Centos?

dkgroot, 2019-11-04:
Reversion is done... Latest version should compile again.
And yes you can. We have had issues with old compilers on Centos before: See [Update-Compiler---Devtools-on-CentOS](https://github.com/chan-sccp/chan-sccp/wiki/Update-Compiler---Devtools-on-CentOS). Same repo should still apply (i hope).

dkgroot, 2019-11-04:
Pretty happy with the quick turn around on this issue report though.. 10 minutes, not bad :-)

rofferman, 2019-11-04:
Yes that is awesome, thank you so much

I have a weird issue also and I can create another ticket if need be , but what Is registration rejected: crossover session mean. I have just encountered this on one phone so far. I changed the switch port and still get it?

Phones are 7975 and 7942

dkgroot, 2019-11-04:
@rofferman 
I normally stop looking at closed issues :-)
A crossover session should be resolved automatically. It happens when the connection between the server and phone gets seperated, but the device information has not yet been cleaned up. We start the cleanup process at that moment, and tell the phone to return 60 seconds later and try again.

It's existence is something that i am not to happy about, but because re-creating this situation is impossible (as far as i know), it happens very sporadically. Which makes it hard to clean this part of the code up or make it better.

dkgroot, 2019-11-04:
If this issue is persistent and not just a question, then yes we should open a new issue report for this, instead of continuing on an existing (and already closed) one.

rofferman, 2019-11-04:
Its not going away. I will try and restart asterisk in abit and see if that fixes it.. 
If not I will open a different ticket

I also updated my compiler on Centos

dkgroot, 2019-11-04:
Ok then a reset will 'fix' it for sure, the question is how we got in this state. What sccp debug level were you running before it happened ?

rofferman, 2019-11-04:
Debug was set to all I believe

dkgroot, 2019-11-04:
Ooh that's good (in this case) Can you find the stop in the debug log file just before it started happening and then send me an excerpt of what went on (let's say 15 before and 15 min after the first case)

rofferman, 2019-11-04:
Here is some of the debug log when the clean was not working on extension 3525
[new 1.txt](https://github.com/chan-sccp/chan-sccp/files/3806363/new.1.txt)

dkgroot, 2019-11-04:
Just mail me the whole debug log, and tell me the device SEP to look for and the line(s) attached to it, so i can search through it. (ddegroot@talon.nl)

dkgroot, 2019-11-04:
After that you can restart the asterisk server and you should be ok.

dkgroot, 2019-11-05:
Ok downloading now. You can compress a log file you know :-)
This is going to take me quite a while to sift through. Which line is attached to this device ? And what extension ?

rofferman, 2019-11-05:
I sent the log to you via wetransfer. It was 180 MB. 
The offending Sep addresss is SEP00083031F110

A asterisk restart seemed to of fixed the problem.

rofferman, 2019-11-05:
The line is 812-492-3525
Ext 3525

And I didn't even think about compressing it.. That was too obvious lol

dkgroot, 2019-11-05:
If you like you can activate rotate in /etc/asterisk/logger.conf so that log files are broken up every XX Mb's and then compressed automatically. Might save some diskspace and search time in the future.

dkgroot, 2019-11-05:
@davidded 
Excerpt from the full log.
[analysis_full_20191105.zip](https://github.com/chan-sccp/chan-sccp/files/3810784/analysis_full_20191105.zip)

We are looking for:
line: 3525
device: SEP00083031F110
linedevice: SEP00083031F110/3525

in the excerpt the refcount is going out of control at [2019-11-04 12:42:28]

dkgroot, 2019-11-05:
@rofferman Thanks to your full log with the high debug setting, we may have an inroads to fixing this looping/reset/cleaning cycle.

rofferman, 2019-11-05:
That's great. Thanks for the help

dkgroot, 2019-11-05:
It will take a while to go through it all, so please give me a couple of days to figure this out !

dkgroot, 2019-11-05:
@davidded 
might have found the first clue:
```
[2019-11-04 12:40:30] NOTICE[9230][C-0000002f] sccp_session.c:     -- SEP00083031F110: Connection reset by peer
[2019-11-04 12:40:30] NOTICE[9230][C-0000002f] sccp_session.c:     -- SEP00083031F110: Exiting sccp_socket device thread
```

Happens in the middle of a hangup (before it ends)
And then during the device cleanup (we rename the device) so a subsequent sccp_device_findbyid will not be able to find the 'old' reference).
```
[2019-11-04 12:40:30] NOTICE[9230][C-0000002f] sccp_channel.c:   == XXX00083031F110: Ending call SCCP/3525-000001F8 (state:CONGESTION)
[2019-11-04 12:40:30] NOTICE[9230][C-0000002f] sccp_channel.c:     -- XXX00083031F110: Sending hangupRequest to Call SCCP/3525-000001F8 (state: CONGESTION)
```
Pretty sure we do something bad after that, linking them back up again, were we should.
Yep this does not make much sense (yet), referring to the same device by two different name in two different threads. This will not end well
```
2019-11-04 12:40:30] VERBOSE[9230][C-0000002f] sccp_refcount.c:  ast.c          :0643 (sccp_astgenwrap_requestQueueHangup ) <----                  refcount decreased 04  <- 05 for    channel: SCCP/3525-000001F8 (0x7f02e000c260)
[2019-11-04 12:40:30] NOTICE[7661] sccp_device.c:     -- SEP0021A0103BFE: >> Send message SpeedDial State Dynamic Message
[2019-11-04 12:40:30] VERBOSE[9230][C-0000002f] sccp_refcount.c:  sccp_channel.c :1468 (sccp_channel_endcall               ) <------------          refcount decreased 12  <- 13 for     device: SEP00083031F110 (0x7f02e0002f00)
[2019-11-04 12:40:30] NOTICE[7661] sccp_session.c: SEP0021A0103BFE: Send Message: SpeedDial State Dynamic Message(0x0146) 140 bytes length
[2019-11-04 12:40:30] VERBOSE[9230][C-0000002f] sccp_refcount.c:  sccp_device.c  :2439 (_sccp_dev_clean                    ) <-----------           refcount decreased 11  <- 12 for     device: SEP00083031F110 (0x7f02e0002f00)
[2019-11-04 12:40:30] VERBOSE[9230][C-0000002f] sccp_refcount.c:  sccp_device.c  :2437 (_sccp_dev_clean                    ) <---                   refcount decreased 03  <- 04 for    channel: SCCP/3525-000001F8 (0x7f02e000c260)
[2019-11-04 12:40:30] VERBOSE[18391][C-00000004] pbx.c: Spawn extension (from-internal, 35, 7) exited non-zero on 'SCCP/3525-000001F8'
[2019-11-04 12:40:30] NOTICE[9230][C-0000002f] sccp_device.c:   == SCCP: Remove Line 3525 from device XXX00083031F110
[2019-11-04 12:40:30] NOTICE[9230][C-0000002f] sccp_linedevice.c:     -- XXX00083031F110: remove device from line 3525
```

dkgroot, 2019-11-06:
#483
#482
#424

rofferman, 2019-11-12:
This happened to me again this morning, I have other installations of chan sccp that I use and they do not have this issue. Is there a trigger that causes this to happen?

dkgroot, 2019-11-12:
@rofferman
I have not had time to fully analyze the log and come up with a possible fix yet. I have been down with the flu over the weekend. Also 'this' is not very clear on an issue report containing more than one issue, so i don't if you are talking about a compile issue or the last problem you reported, regarding looping/cleanup cycle.

rofferman, 2019-11-12:
The looping/cleanup cycle,
Came  in this morning to phones stuck in crossover session
I hope you are feeling better and don't mean to bother, just wanted to checkin on the issue

dkgroot, 2019-11-12:
No problem. Sorry you ran into the same issue again. I will report back as soon as i have had time to go over the log in detail and figure out the cause. In the meantime, you might want to compare the logs and see if you can see if there is some similar behaviour before running into the issue (for example the same device/line combination being involved or doing a callforward or transfer). Sorry i don't have a ready made solution for the issue right now.

dkgroot, 2019-11-26:
@rofferman I have rewritten some of the hangup handling. Note sure if this is going to help fix the issue, but from analyzing the debug log, i noticed that when the hangup failed at some time, all the subsequent hanging up caused new and more problems. I have added a debug log, to print the variables related to channels and hangup under the sccp log 'pbx' keyword.

***Git***
_Checkout this temporary branch_
```
git pull
git fetch -a
git branch
git checkout issues/488
./configure [....]
make -j4 && make install && make reload
```

_move back to develop afterwards:_
```
git checkout develop
```

If possible use the same debug settings you used previous (or at least sccp.conf `debug=core,pbx,channel,device,line,refcount,pbx`).

rofferman, 2019-11-26:
Great,, So you want me to pull that version. Run Debug for a while on it, Send you the Debug log from that version, then switch back to develop.
Is that correct?

dkgroot, 2019-11-26:
Please make sure you run a couple of tests before putting in production  (if in corporate environment). If it works without side effects, we should merge the changes, so don't switch back to develop just yet, only when encountering more issues

rofferman, 2019-11-26:
Ok do you want the debug log or part of if it when or is that not necessary if there are no issues

dkgroot, 2019-11-26:
If you have the space, yank the debug up nice and high :-) Especially in the beginning to see if there are negative side effects. Once it stabilizes we need to figure out if the old looping issues still occur.

rofferman, 2019-11-26:
Ok its implemented and I will keep you posted, Debug is set to all.
instead of make reload can you do a chan_sccp module reload.
I cant take the system down right now

dkgroot, 2019-11-26:
You can do a module unload / load but the sccp phones will have to be dropped and reregister. If that is not allowed at the moment, then you will have to wait.
In the meantime it might be a good idea to clone the machine and setup a test server (if you are running on a VM).

rofferman, 2019-11-26:
Yes i am doing a clone right now

dkgroot, 2019-11-27:
Pushed another little update, to get more information when something does go wrong.

rofferman, 2019-11-27:
great. I just pulled it
so far so good

dkgroot, 2019-11-27:
Ooh good news. This is on production or your test-clone ?
What was your previous 'uptime' compared to now ?

rofferman, 2019-11-27:
Production.. and Its not active yet, I have not had a chance to do a make reload. I will do it tonight

dkgroot, 2019-11-27:
You do know you can schedule a reload using `core restart when convenient` right. It will restart asterisk when there are zero active calls :-) But it would be recommended to have tested the module before doing something like that.

dkgroot, 2019-11-27:
If your uptime is this important, you might want to look into clustering your asterisk server and make two identical copies next to one another. With quick failover.

rofferman, 2019-11-27:
yeah I know, its that when it restarts all the phones go down any my users freak.

rofferman, 2019-11-27:
I do have a warm spare that is backed up to every night thru freepbx

dkgroot, 2019-11-27:
Push an `SCCP Message "server maintenance planned at 1:00AM"`  :-)

dkgroot, 2019-11-27:
Ooh nice, already a good start in that sense. Active/Active clustering is also possible of course, but a little more work.

rofferman, 2019-11-27:
LOL I should , tomorrow is a holiday though so not problem getting it done soon

dkgroot, 2019-11-27:
Ooh enjoy you 'partial' day off then :-)

rofferman, 2019-11-27:
I have only had one or two other instances with the crossover section error since I started this ticket

dkgroot, 2019-11-27:
But without the problem of it getting stuck ? So should we call that a slight improvement ?

dkgroot, 2019-11-29:
@rofferman:
I forced a rebase + push, if you want to get the latest version, you need to:
```
git checkout develop
git branch -D issues/488
git pull
git fetch -a
git checkout issues/488
```
it is not absolutely crucial, if you already got the latest from before, the functionality has not changed, i just improved the reporting a little.

Hope to hear if the latest version has improved the situation on your side, and you are not seeing the stuck devices any more.

rofferman, 2019-11-29:
Got both updates applied and have not had any issues yet
Is there anything I should look for specifically in the debug logs that would help?

dkgroot, 2019-11-29:
@rofferman
How many calls and how much uptime with the new version ? Should the issue have presented itself yet, if on the old version ?
Look for crossover, followed by a series of crossovers, re-register over and over (ie: looping & stuck) A single crossover and recover is not the issue, this fix was concentrating on. The objective was not to get stuck.

dkgroot, 2019-11-29:
You can do something like:
`grep -i -e "Connection reset by peer" -e "Sending hangupRequest" -e "is registering -e "Cleaning previous session, come back later" /var/log/asterisk/message`
And see if a device is continuing to have trouble, or it is just a one of re-register.

dkgroot, 2019-12-05:
@rofferman  Any updates ?

rofferman, 2019-12-05:
@dkgroot  So far so good. I have not had any problems as of yet..

dkgroot, 2019-12-05:
I think that means we should merge it back to develop as an improvement

dkgroot, 2019-12-05:
Merged and Pushed. Please switch back to develop branch:
```
git checkout develop
git pull
./configure [...]
make -j4 && make install && make reload
```

closed 2019-12-05 by dkgroot

rofferman, 2019-12-05:
Ok thanks, and I will keep an eye on it

commit 1c5cc89a (2019-11-04): Partial revert of a092c0f: Remove use of __auto_type_ to support older Centos Versions

Fixes: #488
commit 18ff82f0 (2019-12-05): Add dumpchan information when performing actual hangup is being evaded
Revised requestHangup/requestQueueHangup/requestCallfullHangup handling

Fixes: #488