chan-sccp/chan-sccp #489: Need some help installing and setting up sccp in asterisk

FoxMcCloud592, 2019-11-06:
Hello everyone, I am new to this and don't have much experience with working on command lines. I am trying to get a Cisco 7945 and a 8961 to register to a FreePBX server which is hosted on a VMware Virtual Machine, just so I can make snapshots and have access to the local disk easily.

I get the issue right at the start following these instructions https://github.com/Omid-Mohajerani/SCCP-Asterisk/wiki/Installing-CHAN-SCCP-in-Asterisk

when I cd /usr/src/ It just says that the directory does not exist

I am running Freepbx 13

I expect it to change directory to the issued one 

If someone could show me the commands for getting this to work, that would be fantastic. I am a dummy when it comes to linux and other OS's of the kind

(Screenshots which show reproduction steps and/or demonstrate the problem)
https://imgur.com/a/xup2yeT

dkgroot, 2019-11-06:
@FoxMcCloud592 
The github url you are using looks to be outdated, and would lead to problems for sure. The first issue is that he is still using the master branch, which has not been updated for a long time (And has been changed so that it generated a warning when you try to compile it).

When running on FreePBX you can follow our [official wiki](https://github.com/chan-sccp/chan-sccp/wiki/FreePBX).

Step one:
```
cd /usr/src
git clone https://github.com/chan-sccp/chan-sccp
cd chan-sccp
./configure
make -j4
make install
```

FoxMcCloud592, 2019-11-06:
Alright, looks like I got somewhere. I got until step 5.build a sccp.conf config file (look in the conf/ directory for examples). For example you could use the freepbx examples by doing this:

cp conf/sccp.conf.freepbx /etc/asterisk/sccp.conf

mv /etc/asterisk/sccp_extensions.conf /etc/asterisk/sccp_extensions.conf.bak   ; sccp_manager uses realtimedb for line entries
mv /etc/asterisk/sccp_hardware.conf /etc/asterisk/sccp_hardware.conf.bak       ; sccp_manager uses realtimedb for device entries

Here is what happens when I put these commands in https://imgur.com/a/ZvMT1Fc

dkgroot, 2019-11-06:
for that command to work, you need to be still in the /usr/src/chan-sccp/ directory
```
cd /usr/src/chan-sccp
cp conf/sccp.conf.freepbx /etc/asterisk/sccp.conf
```
The mv command is not required, as you never had chan-sccp installed before.

FoxMcCloud592, 2019-11-06:
We took a short class on Linux essentials when I was taking Cisco Academy my senior year. But I really didn't want to do anything with Linux when I was taking the class because they were also showing router and switch configurations. Honestly, that was far more interesting. Than learning Linux. I think I'm just going to suck it up and learn the command line interface. Thanks for helping and I'll play around and read some more about SCCP on here and I'll write back for anymore questions

FoxMcCloud592, 2019-11-06:
So with all this installed, how do I register the phone with the newly installed SCCP? It has an IP address from DHCP on my router.

dkgroot, 2019-11-06:
Ok good

Checklist:
- [ ] have you setup TFTP server to provision the phones (AFAIK sccp_manager does this for you)
- [ ] have you setup DHCP to point to your TFTP Server (Check that this is the case)
- [ ] check /var/log/messages or /var/log/syslog to see if a phone is retrieving it's SEPxxxxxx.cnf.xml fi le
- [ ] check the phone settings to see if the ip-address is in the range
- [ ] check the phone settings to make sure that the tftp is correct
- [ ] check the phone settings to check the CM (Call Manager) Address

FoxMcCloud592, 2019-11-06:
Ah, I see. I have DHCP on my router but no way to point it to my PBX. I
have a copy of windows server, so I can install tftp, DHCP clients on the
computer. I will need some help with all thoes things on the check list. I
know it's a lot to ask for wince you have been so helpful already, but
maybe a voice chat would be the easiest way to get help with this. Also,
I'll have to load a new SCCP firmware on the phone since it has passwords
on it.

dkgroot, 2019-11-07:
For now you can set the tftp-server ip-address manually in the phone-settings-network page. You switch to admin mode on the phone using '**#', then you switch 'alternate tftp server':on and go to the field above to enter the ip-address of the tftp server. 

Do have a look at our wiki for example https://github.com/chan-sccp/chan-sccp/wiki/Change-firmware-from-SIP-to-SCCP and https://github.com/chan-sccp/chan-sccp/wiki/Factory-Reset which covers the switching of firmware and forcefully resetting to factory setting, which will also cause a firmware reload. **Mind you before doing either of these means you have to have a working DHCP/TFTP combination to work, this will not work with 'manually set tftp server' !!**.

Please use the tftp-server of the PBX, because it is automatically setup/configured by sccp_manager. I don't know what router you have or what options you can set on this router (check it's manual).

We do have a chat site: https://gitter.im/chan-sccp/chan-sccp so visit us there to speed up the process a little.

FoxMcCloud592, 2019-11-07:
Yes, unfortunately I'll have to upload a new config to the phone, it has a
password on it. Or is there an easier way to get rid of the password?

dkgroot, 2019-11-07:
Nope factory reset is the way to go. If need be you can setup a local dhcp/tftp combo on your laptop and connect the phone directly to it, make sure it boots and get's firmware/config from it, and when you are sure it does, perform the factory-reset. You will have to have the firmware for your specific phone ready on your laptop in that case.

FoxMcCloud592, 2019-11-07:
Okay, ill write back when I get home. Thanks.

dkgroot, 2019-11-07:
You can run DHCP / TFTP Server on the PBX side by side if that is easier. But setting up two DHCP side by side next to a 'hardware' router is going to be difficult.

dkgroot, 2019-11-07:
@davidded / @ojaksch 
One of you might be able to help out better regarding windows DHCP/TFTP then i do. I have not used them Windows in 15-20 years :-)

FoxMcCloud592, 2019-11-07:
Ah then lets use something that we all understand. Would Tftpd64 be a
familiar thing?
So my gateway address is 10.0.161.223
My host is 10.0.161.134
And the phone address is 10.0.161.118
Im going off to download the SCCP firmware. How do I upload it to the phone
with tftpd64

FoxMcCloud592, 2019-11-07:
Or lets run the one on the PBX

dkgroot, 2019-11-07:
Skim through this page, and read the DHCP/TFTP ones. They explain in detail how the process works, and gives serveral examples of possible setups. LInux has multiple versions of DHCP and TFTP services so you need to pick the one that fits your needs/distro/preference/etc. Regarding DHCP it even includes an (older) example for Windows how to add 150/66 dhcp custom options to redirect the client to the correct tftp server.

> Or lets run the one on the PBX.

As you have not told me/us the Linux Distro you are using i am not able to help much, saying FreePBX 13 does not tell me anything.

FoxMcCloud592, 2019-11-07:
Here is what this release of FreePbx reports as its version.
https://imgur.com/a/ReHne1x . I understand that getting your assistance is
a privilege, therefore I'd completely understand If you no longer would
like to walk me through the steps. My inexperienced is probably a tad
annoying =)

dkgroot, 2019-11-07:
@FoxMcCloud592 Please join the [chat](https://gitter.im/chan-sccp/chan-sccp) to speed up the proces. You have not tired me out just yet :-) ! Getting a little help from some other people is not a shame.

You are lucky that /proc/version contains the OS tag in this case :-). So you are running an old/outdated version of Centos(6).

On the chat the best person to ask for a little help might be Alexander (@PhantomVl) who developed the sccp_manager plugin. I think he uses Centos as well and knows what dhcp/tftp services to use there.

FoxMcCloud592, 2019-11-08:
Sure, see you guys there!

FoxMcCloud592, 2019-11-11:
I'll come by, let me read this some more and come back to you guys. Thank
you very much

dkgroot, 2019-12-06:
Switched to chat

closed 2019-12-06 by dkgroot