chan-sccp/chan-sccp #490: Freepbx 15 update module issue

vregal, 2019-11-11:
Following Quick Installation guide, point 4b, no way to add modules chan_skinny.so and chan_sccp.so

**Reproduction Steps:**

1. Freepbx 15 web gui, Admin->Asterisk Modules, under Excluded Modules
2. Try to add chan_skinny.so
3. system responds with message: "Call to a member function noload() on null File:/var/www/html/admin/modules/core/Core.class.php:486

dkgroot, 2019-11-11:
@vregal 
If i interpret that response correctly, than chan_skinny.so is not available on that system, which is fine. Because we want to prevent it from being loaded. So not there would have the same effect.

nickvsh, 2019-11-13:
> chan_skinny.so is not available on that system

In fact, it is.
[image]

I suppose it's a bug of FreePBX 15 (on v14 was OK), because it's now working even on modules that was already there like `chan_alsa.so`, or anything like `app_test.so, chan_sip.so, app_dial `etc. You can't **Add** or **Delete** anything from **Excluded/Loaded/Preloaded** list...

I used a workaround: `rm /usr/lib64/asterisk/modules/chan_skinny.so` + restart asterisk.
Luckily, there is **autoload=yes** in  autogenerated /etc/asterisk/modules.conf, so no need to add **chan_sccp.so** to the list from GUI.

[image]

[image]

[image]

dkgroot, 2019-11-13:
Looks like some errors in processing the page are occuring, has anyone checked if these were reported to the FreePBX developers ?
Should we update the wiki, or just wait until it is fixed upstream ? Instead of deleting the skinny module, you can also just edit /etc/asterisk/module.conf of course ?

Idkwhattocallme, 2019-11-13:
is there a workaround to this?

dkgroot, 2019-11-13:
@Idkwhattocallme 
Just edit /etc/asterisk/modules.conf manually, that should work. Or delete the chan_skinny.so module as @nickvsh mentioned (See above)

nickvsh, 2019-11-25:
@Idkwhattocallme  @dkgroot 
I don't recommend editing **/etc/asterisk/modules.conf**, because it is auto-generated by FreePBX.
Any manual modifications will be erased after first "Apply" in FreePBX GUI!

Updated FreePBX "core" module was released to fix this issue (27/Sep/2019)
But it's in the so called "edge" truck.

More detailes here
https://issues.freepbx.org/browse/FREEPBX-20599?focusedCommentId=167779&page=com.atlassian.jira.plugin.system.issuetabpanels%3Acomment-tabpanel#comment-167779

> This module will be pushed to the Stable repo as soon as it meets the criteria for transition.

You need to enable "edge" repo to see this version from GUI (Module Admin).

Or simply run this command to upgrade core module from "edge" truck/repository.

`fwconsole ma --edge upgrade core`

[image]

[image]

Chan_skinny.so is excluded

[image]

[image]

> Should we update the wiki, or just wait until it is fixed upstream ?
 
I think we should update wiki with information, that "Asterisk  Modules" bug is relevant until **core module** version **at least** **15.0.9.47** will be in stable repo.

dkgroot, 2019-11-25:
@nickvsh As you have experience with this, would you mind editing out freepbx wiki pages ? Thanks
@vregal Does this solve your issue and can we close this issue report ?

nickvsh, 2019-11-26:
> would you mind editing out freepbx wiki pages ? Thanks

Done.

dkgroot, 2019-11-26:
@nickvsh Thank you !!

dkgroot, 2019-11-26:
@vregal Closing this issue report for now. If there are still problems, reopen the issue report.

closed 2019-11-26 by dkgroot