chan-sccp/chan-sccp #491: Incoming call reloads cisco phone 7912

bknv, 2019-11-16:
Hello. 
When an incoming call arrives on the second line of the phone cisco 7912, It suddenly reboots. I also noticed that the problem appears if a call arrives from chan_sip.

Log (core, channel, pbx, rtp, socket, action) in attach
[cisco7912.txt](https://github.com/chan-sccp/chan-sccp/files/3854054/cisco7912.txt)

dkgroot, 2019-11-16:
@bknv
From a first look it looks like we might be sending something to the phone which causes it to reset. 
Can you perform another call (with sccp.conf `debug=core,channel,pbx,action,rtp,indicate,message`), so we can see the messages being send. Can you retrieve the phone debug logs from the phones web interface during these last moments ?

@jozzus / @nickvsh Have you seen anything similar on your 7912 phones ?

dkgroot, 2019-11-16:
@bknv The 7912 has turned out to be quite hard to setup exactly right, they have quite a number of quirks: for example : [Troubleshooting-7912G-Keeps-Rebooting-event-after-Successfull-Registration](https://github.com/chan-sccp/chan-sccp/wiki/Troubleshooting-7912G-Keeps-Rebooting-event-after-Successfull-Registration)

dkgroot, 2019-11-18:
@davidded / @ojaksch Didn't you guys figure out how to resolve this one a while ago ?

dkgroot, 2019-11-20:
@bknv You have updated chan-sccp and now multiple incoming calls to 7912 phones work, correct. Can we close this issue report ?

bknv, 2019-11-22:
@dkgroot I update main server and problem saved...
I also noticed that the problem remained on the test server

dkgroot, 2019-11-22:
I guess you will have to do a little digging and find the difference between the two machines.

nickvsh, 2020-02-07:
I have no such problem with incoming calls on 7912 devices.

```
App Load ID | CP7912080004SCCP080108A
Boot Load ID | LD0100BOOT021112A
Software Version | 8.0.4(080108A)
Hardware Revision | 0x0101 0x0001
Product ID | CP-7912G-A
```

Maybe @bknv just need to upgrade to the latest available version of sccp_firmware for 7912 + asterisk version 16 + latest chan_sccp from develop branch.

dkgroot, 2020-02-07:
@nickvsh Thanks for reporting this ! I have not had any further feedback from @bknv so far, not sure what the status is. Will give it another month and close the issue.

bknv, 2020-02-11:
I think you can close.
I disconnected the second line on devices, now I have no problems

closed 2020-02-11 by bknv

dkgroot, 2020-02-11:
@bknv Not really a solution but nice to hear you got everything stabilized.
@nickvsh have you tried running multiple lines on one of these devices ?

BTW: sorry guys for not having direct access to one of these phone (i don't have a large pile of all the phone types out there).

nickvsh, 2020-02-13:
> have you tried running multiple lines on one of these devices ?

@dkgroot 
No, my 7912 devices don't support 2-nd line.

My config

```
[SEP...](7912)
description = NICK
button = line, 7317, default
button = line, 4455

```