chan-sccp/chan-sccp #492: SPA 524G phone resets after startup

mpaland, 2019-11-21:
I upgraded asterisk from 13.20.0 to 16.6.1 and the same time I updated the chan-sccp driver from an older (around begin of this year) version to the actual one of last weekend.

All phones (7970 and 7975) are working fine as expected except for the (older) SPA 524G phone.

Behavior:
After loading all necessary files via tftp it resets and starts again.
Loads all files again and restarts.

Here is the according log (3 loops):
```
    -- SCCP: Accepted Client Connection from 192.168.2.153:40761
    -- SCCP: Connected on server via 192.168.2.1:2000
[Nov 21 13:54:08] WARNING[24742]: sccp_session.c:1114 sccp_session_check_crossdevice: Session(0x764ad8003a60) and Device Session(0x764ad80036c0) are out of sync.
[Nov 21 13:54:08] NOTICE[24742]: sccp_actions.c:714 handle_SPCPTokenReq: SEP001DE5EA99B8: Cleaning previous session, come back later (tokenState:Progress)
  == 192.168.2.1:2000: Session 0x764ad80038f0 needs to be closed!
 SCCP: Clean Device SEP001DE5EA99B8, remove from global:No, restart_device:No
  == SCCP: Remove Line 13 from device SEP001DE5EA99B8
  == SCCP: Unregister Device SEP001DE5EA99B8
    -- SEP001DE5EA99B8: (sccp_devstate_deviceRegisterListener) device unregistered
 SEP001DE5EA99B8: LineDevice FREE 0x764a94003a70
    -- SCCP: Accepted Client Connection from 192.168.2.153:22853
    -- SCCP: Connected on server via 192.168.2.1:2000
    -- SCCP: Alarm Message: Severity: Informational (2), Name=SEP001DE5EA99B8  Load=8.1.17 Last=Halt+7372016c [0/-1727878976]
    -- SEP001DE5EA99B8: Accessory 'Headset' is 'On Hook' (0)
    -- SEP001DE5EA99B8: Device has 12 Capabilities
    -- SEP001DE5EA99B8: Accessory 'Headset' is 'On Hook' (0)
    -- SEP001DE5EA99B8: Sending ConfigStatMessage, lines 1, speeddials 0
    -- SEP001DE5EA99B8: Building button template Cisco SPA 524SG(80001), user config G524
    -- SCCP: Accepted Client Connection from 192.168.2.153:44107
    -- SCCP: Connected on server via 192.168.2.1:2000
[Nov 21 13:54:20] WARNING[24753]: sccp_session.c:1114 sccp_session_check_crossdevice: Session(0x764ad8003a60) and Device Session(0x764ad80036c0) are out of sync.
[Nov 21 13:54:20] NOTICE[24753]: sccp_actions.c:714 handle_SPCPTokenReq: SEP001DE5EA99B8: Cleaning previous session, come back later (tokenState:Progress)
  == 192.168.2.1:2000: Session 0x764ad80038f0 needs to be closed!
 SCCP: Clean Device SEP001DE5EA99B8, remove from global:No, restart_device:No
  == SCCP: Remove Line 13 from device SEP001DE5EA99B8
  == SCCP: Unregister Device SEP001DE5EA99B8
    -- SEP001DE5EA99B8: (sccp_devstate_deviceRegisterListener) device unregistered
 SEP001DE5EA99B8: LineDevice FREE 0x764aa801d4d0
    -- SCCP: Accepted Client Connection from 192.168.2.153:52145
    -- SCCP: Connected on server via 192.168.2.1:2000
    -- SCCP: Alarm Message: Severity: Informational (2), Name=SEP001DE5EA99B8  Load=8.1.17 Last=Halt+7372016c [0/-1727878976]
    -- SEP001DE5EA99B8: Accessory 'Headset' is 'On Hook' (0)
    -- SEP001DE5EA99B8: Device has 12 Capabilities
    -- SEP001DE5EA99B8: Accessory 'Headset' is 'On Hook' (0)
    -- SEP001DE5EA99B8: Sending ConfigStatMessage, lines 1, speeddials 0
    -- SEP001DE5EA99B8: Building button template Cisco SPA 524SG(80001), user config G524
    -- SCCP: Accepted Client Connection from 192.168.2.153:35064
    -- SCCP: Connected on server via 192.168.2.1:2000
[Nov 21 13:54:32] WARNING[24764]: sccp_session.c:1114 sccp_session_check_crossdevice: Session(0x764ad8003a60) and Device Session(0x764ad80036c0) are out of sync.
[Nov 21 13:54:32] NOTICE[24764]: sccp_actions.c:714 handle_SPCPTokenReq: SEP001DE5EA99B8: Cleaning previous session, come back later (tokenState:Progress)
  == 192.168.2.1:2000: Session 0x764ad80038f0 needs to be closed!
 SCCP: Clean Device SEP001DE5EA99B8, remove from global:No, restart_device:No
  == SCCP: Remove Line 13 from device SEP001DE5EA99B8
  == SCCP: Unregister Device SEP001DE5EA99B8
    -- SEP001DE5EA99B8: (sccp_devstate_deviceRegisterListener) device unregistered
 SEP001DE5EA99B8: LineDevice FREE 0x764ab000c360
    -- SCCP: Accepted Client Connection from 192.168.2.153:63827
    -- SCCP: Connected on server via 192.168.2.1:2000
    -- SCCP: Alarm Message: Severity: Informational (2), Name=SEP001DE5EA99B8  Load=8.1.17 Last=Halt+7372016c [0/-1727878976]
    -- SEP001DE5EA99B8: Accessory 'Headset' is 'On Hook' (0)
    -- SEP001DE5EA99B8: Device has 12 Capabilities
    -- SEP001DE5EA99B8: Accessory 'Headset' is 'On Hook' (0)
    -- SEP001DE5EA99B8: Sending ConfigStatMessage, lines 1, speeddials 0
    -- SEP001DE5EA99B8: Building button template Cisco SPA 524SG(80001), user config G524
```

Do you have any idea how to track this down?
Alternatively I can replace the phone, but I won't like that, cause it worked fine all the years.
I don't think that it has something to do with a wrong/corrupted SEPxxx.cnf.xml file, cause it hasn't changed and worked for years.
There must be something new after registration what the phone doesn't like at all...

dkgroot, 2019-11-21:
@mpaland 
My SPA524G works perfectly well with the latest version, so there is no reason to throw it out at all. I love the little spa's
I looks like you have a cross over session, and it's not clear why. Can you restart the chan-sccp module (unload/load) and see if that takes care of the issue. This does not resolve the reason why it got in the start, but should just let it connect again.
If that does not work, we need to add some debug flags like sccp.conf `debug=core,device,socket,action` to figure out why it is getting into this state.

mpaland, 2019-11-21:
@dkgroot, thanks a lot for giving such fast feedback!

I tried to unload/reload the module and restart asterisk several times, no success, result the same.
Here is a more detailed log:
```
Loaded chan_sccp.so => (Skinny Client Control Protocol (SCCP). Release: 4.3.2 TRUNK - unknown (built by 'root' on '2019-11-15 23:01:48 UTC')
 )
    -- SCCP: Accepted Client Connection from 192.168.2.153:27297
    -- SCCP: Connected on server via 192.168.2.1:2000
       > 192.168.2.1:2000: keepalive:62, keepaliveinterval:60
  == SEP001DE5EA99B8: is requesting a token, Instance: 16777216, Type: Cisco SPA 524SG (80001)
    -- SEP001DE5EA99B8: Allocating device to session (20) 192.168.2.153
    -- SEP001DE5EA99B8: checkACL returning TRUE
    -- SEP001DE5EA99B8: Registration State is 'Token'
       > SEP001DE5EA99B8: keepalive:62, keepaliveinterval:60
    -- SCCP: Alarm Message: Severity: Informational (2), Name=SEP001DE5EA99B8  Load=8.1.17 Last=Halt+7372016c [0/-1727878976]
 SEP001DE5EA99B8: is registering, Instance: 1, UserId: 0, Type: Cisco SPA 524SG (80001), Version: 8 (loadinfo '')
    -- SEP001DE5EA99B8: checkACL returning TRUE
    -- SEP001DE5EA99B8: Our Session IP4 Address 192.168.2.153:0
    -- SCCP: searching for our capability for device protocol version 8
    -- SCCP: searching for our capability for device protocol SPCP
    -- SPCP: found protocol version '8' at 8
    -- SEP001DE5EA99B8: (sccp_dev_set_registered) Setting Registered Status for Device from Token to Progress
    -- SEP001DE5EA99B8: Registration State is 'Progress'
    -- SEP001DE5EA99B8: Accessory 'Headset' is 'On Hook' (0)
    -- SEP001DE5EA99B8: Device has 12 Capabilities
    -- SEP001DE5EA99B8: Accessory 'Headset' is 'On Hook' (0)
    -- SEP001DE5EA99B8: Sending ConfigStatMessage, lines 1, speeddials 0
    -- SEP001DE5EA99B8: Building button template Cisco SPA 524SG(80001), user config G524
    -- SEP001DE5EA99B8: Allocated 4 Device buttons.
    -- SEP001DE5EA99B8: Allocated 52 Abbreviate Dial Buttons.
    -- SEP001DE5EA99B8: searching for softkeyset: softkeyset1!
    -- SEP001DE5EA99B8: ONHOOK          => |0 :Redial   |1 :Empty    |2 :DND      |
    -- SEP001DE5EA99B8: CONNECTED       => |0 :Hold     |1 :EndCall  |2 :Park     |3 :Select   |
    -- SEP001DE5EA99B8: ONHOLD          => |0 :Resume   |1 :NewCall  |2 :EndCall  |3 :Transfer |4 :Select   |
    -- SEP001DE5EA99B8: RINGIN          => |0 :Answer   |1 :EndCall  |
    -- SEP001DE5EA99B8: OFFHOOK         => |0 :Redial   |1 :EndCall  |3 :CFwdALL  |4 :CFwdBusy |8 :Barge    |
    -- SEP001DE5EA99B8: CONNTRANS       => |0 :Hold     |1 :EndCall  |2 :Transfer |3 :Confrn   |4 :Park     |5 :Select   |6 :DirTrfr  |8 :CFwdALL  |9 :CFwdBusy |
    -- SEP001DE5EA99B8: DIGITSFOLL      => |0 :Dial     |1 :<<       |2 :EndCall  |
    -- SEP001DE5EA99B8: CONNCONF        => |0 :ConfList |1 :EndCall  |2 :Join     |3 :Hold     |
    -- SEP001DE5EA99B8: RINGOUT         => |0 :EndCall  |
    -- SEP001DE5EA99B8: OFFHOOKFEAT     => |0 :Redial   |1 :EndCall  |
    -- SEP001DE5EA99B8: INUSEHINT       => |0 :Redial   |1 :NewCall  |4 :Barge    |
    -- SEP001DE5EA99B8: ONHOOKSTEALABLE => |0 :Redial   |1 :NewCall  |2 :CFwdALL  |5 :DND      |6 :Intercept|
    -- SEP001DE5EA99B8: Set softkeyset to ONHOOK(0) on line 0  and call 0
    -- SEP001DE5EA99B8: validKeyMask 4294967294
    -- SCCP: Accepted Client Connection from 192.168.2.153:6276
    -- SCCP: Connected on server via 192.168.2.1:2000
       > 192.168.2.1:2000: keepalive:62, keepaliveinterval:60
  == SEP001DE5EA99B8: is requesting a token, Instance: 16777216, Type: Cisco SPA 524SG (80001)
[Nov 21 18:56:01] WARNING[12377]: sccp_session.c:1114 sccp_session_check_crossdevice: Session(0x764b04001fa0) and Device Session(0x764b04000f80) are out of sync.
[Nov 21 18:56:01] NOTICE[12377]: sccp_actions.c:714 handle_SPCPTokenReq: SEP001DE5EA99B8: Cleaning previous session, come back later (tokenState:Progress)
  == 192.168.2.1:2000: Session 0x764b040011b0 needs to be closed!
    -- SEP001DE5EA99B8: cleanup session
    -- SEP001DE5EA99B8: Destroy Device Session 192.168.2.153
 SCCP: Clean Device SEP001DE5EA99B8, remove from global:No, restart_device:No
    -- SEP001DE5EA99B8: Registration State is 'Cleaning'
  == SEP001DE5EA99B8: checking buttonconfig index:0, type:Line (0) to see if there are any connected lines/channels
  == SCCP: Remove Line 13 from device SEP001DE5EA99B8
    -- 13: (updateCapabilitiesFromDevicesToLine) Could not retrieve capabilities from line or device.
Using Fallback Capabilities Alaw/Ulaw
  == SEP001DE5EA99B8: checking buttonconfig for pendingDelete (index:0, type:Line (0), pendingDelete:False, pendingUpdate:False)
  == SEP001DE5EA99B8: checking buttonconfig for pendingDelete (index:1, type:Empty (4), pendingDelete:False, pendingUpdate:False)
  == SEP001DE5EA99B8: checking buttonconfig for pendingDelete (index:2, type:Empty (4), pendingDelete:False, pendingUpdate:False)
  == SEP001DE5EA99B8: checking buttonconfig for pendingDelete (index:3, type:Empty (4), pendingDelete:False, pendingUpdate:False)
  == SCCP: Unregister Device SEP001DE5EA99B8
    -- SEP001DE5EA99B8: (sccp_devstate_deviceRegisterListener) device unregistered
 SEP001DE5EA99B8: LineDevice FREE 0x23c5f70
    -- SEP001DE5EA99B8: Registration State is 'None'
    -- SEP001DE5EA99B8: Registration State is 'None'
[... 33 lines cut ...]
    -- SEP001DE5EA99B8: Allocated 52 Abbreviate Dial Buttons.
    -- SEP001DE5EA99B8: searching for softkeyset: softkeyset1!
[... cut ...]
       > 192.168.2.1:2000: keepalive:62, keepaliveinterval:60
  == SEP001DE5EA99B8: is requesting a token, Instance: 16777216, Type: Cisco SPA 524SG (80001)
[Nov 21 18:56:13] WARNING[12388]: sccp_session.c:1114 sccp_session_check_crossdevice: Session(0x764b04002f90) and Device Session(0x764b04001fa0) are out of sync.
[Nov 21 18:56:13] NOTICE[12388]: sccp_actions.c:714 handle_SPCPTokenReq: SEP001DE5EA99B8: Cleaning previous session, come back later (tokenState:Progress)
```

I don't see an obvious reason.

dkgroot, 2019-11-21:
Could it be that you have two CUCM entries in your SPA config ? It would then send a second token to the second CUCM. if that is the same ip address as the first, you would get something like this. Can also happen if you have srst active.

dkgroot, 2019-11-21:
Ooh btw i have a SPA504 not a SPA524. So i can be that they are slightly different and that we accidently introduced a regression for the SPA524 without noticing.

mpaland, 2019-11-22:
I'm pretty sure, that I don't have two CUCM entries active.
What do you mean by "srst" active and the debug log times?
I'll make a complete trace on monday. Thanks a lot!

dkgroot, 2019-11-22:
> I'm pretty sure, that I don't have two CUCM entries active.

Ok

> What do you mean by "srst" active and the debug log times?

srst is a failover system from cisco, where remote routers can take over phone server tasks for central CUCM when the connection is lost. I can have the same effect as a second CUCM entry in the config. I am not 100% sure SPA's phones even support SRST to be honest.

> I'll make a complete trace on monday. Thanks a lot!

Ok looking forward to that. Maybe set sccp.conf 'debug=core,device,action,socket,message' so we can also see the messages being send to the device.
Please setup /etc/asterisk/logger.conf to create a 'full' log, so we can have a look at that. Please include the SPA phone config and sccp.conf when reporting back.

mpaland, 2019-11-28:
So far I'm not using any Cisco routers anymore, the asterisk and system setup is pretty plain.

I attached a full log file, but it's not very informative.
[log.txt](https://github.com/chan-sccp/chan-sccp/files/3902088/log.txt)
I guess we need a wireshark or tcpdump trace to see what's going on.

I had a test version (a27a98cc7785ea3e0d5d8cfe23fc9952776b3411) from May which I tried, but it shows the same results, so it may be asterisk 16 related or the regression is earlier than the test version.

Attaches:
[SEP001DE5EA99B8.cnf.xml.txt](https://github.com/chan-sccp/chan-sccp/files/3902128/SEP001DE5EA99B8.cnf.xml.txt)
[sccp.conf.txt](https://github.com/chan-sccp/chan-sccp/files/3902167/sccp.conf.txt)

dkgroot, 2019-11-28:
Hi Marco,

Ok the reset time between registering and restarting is pretty fast, my guess would be that we are sending the phone something that it does not like at all. Might be a faulty message.

The phone debug log might be able to tell us something about that. Best way to collect it would be via the phones ssh interface. You can try and grab the debug log from the webinterface, but that would be a little harder (trial and error).

A tcpdump might help, but that would not really show the reaction the phone is having.

It's  a little strange why my SPA504 is not having a similar reaction though.

dkgroot, 2019-11-28:
I am actually having difficulty finding the SPA524 model on the cisco website. Also could not find firmware for it :-)  Maybe you can provide a link or 2 ?

dkgroot, 2019-11-28:
Can you do a quick test using the same chan-sccp version but backed by asterisk-13, because i don't see how asterisk-16 could be the cause. I don't know if this is an easy switch for you or not (i have multiple asterisk versions installed, but that is not as easy for everyone). We use one source base for all asterisk version, so there should not be any difference in the messages we send the devices.

dkgroot, 2019-11-28:
I just found this site:
http://www.shanekillen.com/2012/08/cisco-spcp-register-token-rejected1.html
About firmware issues, also related 'Tokens' etc. Maybe you can downgrade/upgrade the firmware to see if that does anything. Might be hard to find the firmware though, so make sure you have everything in place.

mpaland, 2019-11-29:
Thanks a lot for the link, very interesting. From my past UC5xx times I still have several firmware versions for the phones. But the 524 has the latest one, cp524g-8-1-17.bin
Because it worked perfectly fine before the upgrade/update, the cause must be something else.
I already tried to enable ssh on the phone, but without success.

Switching the asterisk version back to 13 is possible, but takes a moment, I'm going to try this next.

dkgroot, 2019-11-29:
If you can switch firmwares back and forth that might already check of one thing.
(Can you send me a copy so i can put it in my firmware collection ?)

dkgroot, 2019-11-29:
Do you know the revision you were using, when it still worked. If so we can do a git bisect to figure out where it went wrong.

mpaland, 2019-11-29:
You get it by mail, no problem.
The phone was working the past 4 years with the same firmware, 1-17.bin, no changes, so I was really wondering about this reset.

dkgroot, 2019-12-17:
@mpaland Have you been able to make any progress in figuring out what might have happened ?

mpaland, 2020-02-28:
Sorry, time passed, with a disabled G524...

Okay, today I tried commit ad6cd243463957bb28649adca9833cce45170198 with asterisk 13.31.0.
Same result, phone crashes, so I switched back (forward) to 16.8.0, still same result.
Something in chan-sccp must have changed which causes this behavior.
Next I try is to take an much older version to narrow down the problem.

dkgroot, 2020-02-28:
Hi Marco, sorry for this ! We need to do a bit of backtracking to figure out where it went wrong. If you can provide a link to the first occurrence (ballpark date/time is fine) we should be able to `git bisect pretty quickly.

dkgroot, 2020-04-07:
@mpaland Have you had any luck finding an old revision that worked fine with the 524G for bisecting ?

mpaland, 2020-04-07:
Hi Diederik,
thanks for asking. No, not yet, this is not so easy, cause chan sccp is running on a secure router in my office and software changes require a distro recompilation.
This week I'll setup a second asterisk on a virtual box just with chan sccp to track down the problem finally.
Saying "not so easy" above means, that the mwi struct in asterisk (13 and 16) changed, so that I cannot take an old chan sccp version and compile it with the latest asterisk version.
I can fix this or take the according asterisk version, 16 is too young so I'll take 13 for the tests.
Keep you informed here shortly.

dkgroot, 2020-04-07:
I know what you mean, the simple fix i used for that, is cleaning out the MailboxSubscribe function in sccp_mwi.c (ie: make it non-functional) or checkout an old asterisk to compile against. But i have had the same issue as well.