chan-sccp/chan-sccp #493: What is "empty" softkey supposed to do?

billsimon, 2019-11-26:
On ring-out, a softkey labeled "Empty" appears in the first softkey slot. Pressing the key during ring-out causes the phone to display "No channel to perform empty on!" and the call fails. The phone is also rendered unusable for about 30 seconds (all buttons inoperative). What is the purpose of this softkey? I intend to adjust the template to remove it, but would like to understand its meaning first.

dkgroot, 2019-11-26:
@billsimon 
An empty softkey should normally not popup on a phone, and just leave an empty spot, used to line up softkeys nicely. It shoud not have any side effects when pressed either. I am wondering what type of phone you are using, that has this behaviour. You can create a specialize softkeyset if necessary (for now) until we figure this out.

dkgroot, 2019-12-17:
@darkponay helped clarify the problem :-)

@billsimon Thanks for reporting the issue. Just committed a fix for the problem
The empty button is not suppressed again but still leaving an open space.

closed 2019-12-17 by dkgroot

billsimon, 2019-12-17:
Confirmed, problem solved on my phones. Thanks!

dkgroot, 2019-12-17:
Sorry if took 3 weeks for me to finally understand the problem :-))

commit 350a8a76 (2019-12-17): Fix empty button issue (reported by Bill Simon)

Fixes: #493