chan-sccp/chan-sccp #494: cant setup realtime

Idkwhattocallme, 2019-11-26:
help i cant set up realtime im following the instructions here
 https://github.com/chan-sccp/chan-sccp/wiki/Realtime-Configuration

but none of the commands work 

i am using asterisk 16 with freepbx

dkgroot, 2019-11-26:
@Idkwhattocallme Can you give me some more details:
the platform/version/cpu-arch you are using
Which commands don't work
What error you get back

Helping with realtitime might be easier using [chat](gitter.im/chan-sccp/chan-sccp).

Idkwhattocallme, 2019-11-26:
i am using freepbx and following the guide but when i try to install sccp_manager i get a error cannot find sccpdevice table

dkgroot, 2019-11-26:
Which freepbx version. Have you seen the notice in FreePBX_Installation:4b ?
Please log in to the gitter chat make this process a little easier
For help with sccp_manager you might need some help from @PhantomVl

Idkwhattocallme, 2019-11-26:
what is gitter chat

dkgroot, 2019-11-26:
https://gitter.im/chan-sccp/chan-sccp

dkgroot, 2020-01-10:
Closing issue because of lack of reply

closed 2020-01-10 by dkgroot